raspberrypi/linux #4097: FKMS detects 4k60 but KMS does not

alanbork, 2021-01-26:
My pi4 can display 4k60 on my TV (sony xbr800D), when configured under FKMS. But when I switch to KMS it maxes out at 4K30. 

Linux raspberrypi 5.10.10-v7l+ raspberrypi/firmware#1397 SMP Mon Jan 25 18:42:27 GMT 2021 armv7l GNU/Linux 
rpi-updated to 8c2ba64b27a4b6a1dfcd114203c6309f766e2ef4
(same problem with 039088f7d929b0d1cc49e8d2012df0652f908518)

config.txt:
[pi4]
#dtoverlay=vc4-fkms-v3d
dtoverlay=vc4-kms-v3d
max_framebuffers=1
hdmi_force_hotplug=1
hdmi_group=1
hdmi_mode=4
hdmi_enable_4kp60=1

root@raspberrypi:~# edidparser  edid
Enabling fuzzy format match...
Parsing edid...
HDMI:EDID version 1.3, 1 extensions, screen size 95x54 cm
HDMI:EDID features - videodef 0x80 !standby !suspend !active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
HDMI:EDID found monitor name descriptor tag 0xfc
HDMI:EDID monitor name is SONY_TV___00
HDMI:EDID found monitor range descriptor tag 0xfd
HDMI:EDID monitor range offsets: V min=0, V max=0, H min=0, H max=0
HDMI:EDID monitor range: vertical is 23-62 Hz, horizontal is 14-136 kHz, max pixel clock is 600 MHz
HDMI:EDID monitor range does not support GTF
HDMI:EDID found preferred CEA detail timing format: 3840x2160p @ 60 Hz (97)
HDMI:EDID found CEA detail timing format: 1920x1080p @ 60 Hz (16)
HDMI:EDID established timing I/II bytes are 21 08 00
HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 9, 800x600p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
HDMI:EDID standard timings block x 8: 0x8180 A9C0 714F B300 0101 0101 0101 0101
HDMI:EDID found DMT format: code 35, 1280x1024p @ 60 Hz (5:4) in standard timing 0
HDMI:EDID found DMT format: code 83, 1600x900p @ 60 Hz (16:9) in standard timing 1
HDMI:EDID found DMT format: code 21, 1152x864p @ 75 Hz (4:3) in standard timing 2
HDMI:EDID found DMT format: code 58, 1680x1050p @ 60 Hz (16:10) in standard timing 3
HDMI:EDID parsing v3 CEA extension 0
HDMI:EDID monitor support - underscan IT formats:yes, basic audio:yes, yuv444:yes, yuv422:yes, #native DTD:0
HDMI:EDID found CEA detail timing format: 1280x720p @ 60 Hz (4)
HDMI:EDID found CEA format: code 97, 3840x2160p @ 60Hz
HDMI:EDID found CEA format: code 96, 3840x2160p @ 50Hz
HDMI:EDID found CEA format: code 93, 3840x2160p @ 24Hz
HDMI:EDID found CEA format: code 94, 3840x2160p @ 25Hz
HDMI:EDID found CEA format: code 95, 3840x2160p @ 30Hz
HDMI:EDID found CEA format: code 98, 4096x2160p @ 24Hz
HDMI:EDID found CEA format: code 31, 1920x1080p @ 50Hz
HDMI:EDID found CEA format: code 16, 1920x1080p @ 60Hz
HDMI:EDID found CEA format: code 20, 1920x1080i @ 50Hz
HDMI:EDID found CEA format: code 5, 1920x1080i @ 60Hz
HDMI:EDID found CEA format: code 19, 1280x720p @ 50Hz
HDMI:EDID found CEA format: code 4, 1280x720p @ 60Hz
HDMI:EDID found CEA format: code 32, 1920x1080p @ 24Hz
HDMI:EDID found CEA format: code 34, 1920x1080p @ 30Hz
HDMI:EDID found CEA format: code 60, 1280x720p @ 24Hz
HDMI:EDID found CEA format: code 62, 1280x720p @ 30Hz
HDMI:EDID found CEA format: code 18, 720x576p @ 50Hz
HDMI:EDID found CEA format: code 22, 1440x576i @ 50Hz
HDMI:EDID found CEA format: code 3, 720x480p @ 60Hz
HDMI:EDID found CEA format: code 7, 1440x480i @ 60Hz
HDMI:EDID found CEA format: code 17, 720x576p @ 50Hz
HDMI:EDID found CEA format: code 21, 1440x576i @ 50Hz
HDMI:EDID found CEA format: code 2, 720x480p @ 60Hz
HDMI:EDID found CEA format: code 6, 1440x480i @ 60Hz
HDMI:EDID found CEA format: code 1, 640x480p @ 60Hz
HDMI:EDID found CEA format: code 101, 4096x2160p @ 50Hz
HDMI:EDID found CEA format: code 102, 4096x2160p @ 60Hz
HDMI:EDID found audio format 6 channels PCM, sample rate: 32|44|48|88|96|176|192 kHz, sample size: 16|20|24 bits
HDMI:EDID found audio format 6 channels AC3, sample rate: 32|44|48 kHz, bitrate: 640 kbps
[... 124 lines cut ...]
  raspberrypi/firmware#18 1600x900 60.00 1600 1624 1704 1800 900 901 904 1000 108000 flags: phsync, pvsync; type: driver
  raspberrypi/firmware#19 1280x1024 60.02 1280 1328 1440 1688 1024 1025 1028 1066 108000 flags: phsync, pvsync; type: driver
  raspberrypi/firmware#20 1152x864 75.00 1152 1216 1344 1600 864 865 868 900 108000 flags: phsync, pvsync; type: driver
  raspberrypi/firmware#21 1280x720 60.00 1280 1390 1430 1650 720 725 730 750 74250 flags: phsync, pvsync; type: userdef, driver
  raspberrypi/firmware#22 1280x720 59.94 1280 1390 1430 1650 720 725 730 750 74176 flags: phsync, pvsync; type: driver
  raspberrypi/firmware#23 1280x720 50.00 1280 1720 1760 1980 720 725 730 750 74250 flags: phsync, pvsync; type: driver
  raspberrypi/firmware#24 1280x720 30.00 1280 3040 3080 3300 720 725 730 750 74250 flags: phsync, pvsync; type: driver
  raspberrypi/firmware#25 1280x720 29.97 1280 3040 3080 3300 720 725 730 750 74176 flags: phsync, pvsync; type: driver
  raspberrypi/firmware#26 1280x720 24.00 1280 3040 3080 3300 720 725 730 750 59400 flags: phsync, pvsync; type: driver
  raspberrypi/firmware#27 1280x720 23.98 1280 3040 3080 3300 720 725 730 750 59341 flags: phsync, pvsync; type: driver
  raspberrypi/firmware#28 1024x768 60.00 1024 1048 1184 1344 768 771 777 806 65000 flags: nhsync, nvsync; type: driver
  raspberrypi/firmware#29 800x600 60.32 800 840 968 1056 600 601 605 628 40000 flags: phsync, pvsync; type: driver
  raspberrypi/firmware#30 720x576 50.00 720 732 796 864 576 581 586 625 27000 flags: nhsync, nvsync; type: driver
  raspberrypi/firmware#31 720x480 60.00 720 736 798 858 480 489 495 525 27027 flags: nhsync, nvsync; type: driver
  raspberrypi/firmware#32 720x480 59.94 720 736 798 858 480 489 495 525 27000 flags: nhsync, nvsync; type: driver
  raspberrypi/firmware#33 640x480 60.00 640 656 752 800 480 490 492 525 25200 flags: nhsync, nvsync; type: driver
  raspberrypi/firmware#34 640x480 59.94 640 656 752 800 480 490 492 525 25175 flags: nhsync, nvsync; type: driver
  #0  nan 0 0 0 0 0 0 0 0 0 flags: ; type:
  #0  nan 0 0 0 0 0 0 0 0 0 flags: ; type:
  #0  nan 0 0 0 0 0 0 0 0 0 flags: ; type:
  #0 1280x720 60.00 1280 1390 1430 1650 720 725 730 750 74250 flags: phsync, pvsync; type: userdef, driver
  #0  nan 0 0 0 0 0 0 0 0 0 flags: ; type:
  #0  nan 0 0 0 0 0 0 0 0 0 flags: ; type:

and indeed if I force mode 97 via config.txt it doesn't work.

alanbork, 2021-01-26:
Kernel as in mainline kernel? or RPI specific?

It sure would be nice if that had been documented somewhere so I didn't have to waste my time debugging and eventually submitting a bug report here. I'm not allergic to reading;  but for all my browsing I've never seen a document listing the tradeoffs of using kms vs fkms on the Pi, just the constant push on the forums to switch from FKMS to KMS.

6by9, 2021-01-27:
Maxime Ripard is working on mainline kernel and getting the KMS driver working there. Follow dri-devel and you'd see the patches.

We're backporting his work frequently (often before being merged upstream), and also working on parts that are proving difficult to him as Pi Towers often have either additional information we can't release, or can try replicating things from the firmware more easily. Those patches often appear in our tree first, and then Maxime will pick them up, clean up where necessary, and then upstream them.
So the Pi specific 5.10 (or possibly 5.11 if you want really bleeding edge) kernel is generally the best bet.

Whilst you may read documentation, the vast majority of people seem not to. Particularly on something that is still under heavy development it generally isn't worthwhile trying to keep anything updated.

pelwell, 2021-01-27:
And now, thanks to google indexing, this issue becomes documentation. The title has been tweaked to help others find it.

alanbork, 2021-07-03:
problem has been fixed both kms and fkms can detect 4k60p now.

closed 2021-07-03 by alanbork