raspberrypi/linux #4098: unexpected tc358743 crash during the day

xtreegpt, 2021-01-26:
**Describe the bug**
The system is used as a IP based KVM server using PIKVM. The Pi 4 has a Lusya Standard HDMI To CSI-2 Adapter to provide video from HDMI. The continuous video is passed through the ustreamer sever (https://github.com/pikvm/ustreamer) and the video stream is viewed using TigerVNC Client. During the day there are process crash dumps.

**To reproduce**
Run PiKVM application and use the KVM output for an extended period of time. 

**Expected behaviour**
Run PiKVM without process and system crash. 

**Actual behaviour**
After running PiKVM for some time the process will crash and eventually the pi will reboot. 

**System**
uname -a
Linux pikvm 5.4.83-5-ARCH #1 SMP PREEMPT Sun Jan 3 16:11:27 UTC 2021 armv7l GNU/Linux

[root@pikvm ~]# pistat
# Raspberry Pi 4 Model B Rev 1.2

CPU temp: 46.25'C
GPU temp: 46.2'C

Throttled flags: 0x0

Throttled now:  no
Throttled past: no

Undervoltage now:  no
Undervoltage past: no

Frequency capped now:  no
Frequency capped past: no

**Logs**
[pastebin dmesg](https://pastebin.com/vgm8hAyq)

[pastbin top output](https://pastebin.com/4y38BzS3)

**Additional context**
None

Add any other relevant context for the problem.
None

pelwell, 2021-01-26:
Apart from the kernel messages (which are warnings, not crashes), did your Pi misbehave or malfunction in any way?

xtreegpt, 2021-01-26:
Yes after some time the system will crash and reboot. I consider the reboot a malfuction

pelwell, 2021-01-26:
Yes, rebooting counts as a malfunction, but whatever caused it doesn't appear in the log you posted.

The warnings are coming from `vb2_buffer_done`, which is complaining that it is being passed a buffer parameter is not marked as ACTIVE.

I notice that you are running a 5.4 kernel, while we are just about to switch over to rpi-5.10.y. A lot of work has been done since 5.4 was released, so it might be worth upgrading now to see if the problem has already been solved. Run `sudo rpi-update` to upgrade the kernel and firmware; it is still technically pre-release software so you would be wise to backup your SD card first, but I don't know of any major problems with the current snapshot.

xtreegpt, 2021-01-26:
Is there a way to capture the crash? Is there a debug option in the kernel or grub I should use? PiKVM is an appliance and does not have all the regular repi commands. I can ping the developer to find out if upgrading to 5.10 is advisable.

pelwell, 2021-01-26:
If you are lucky, some of it may already be recorded in /var/log/kern.log - it depends on how instantly fatal the crash was. If that doesn't show anything, the next step would be to set up the kdbg kernel debugger.

Don't forget about 5.10.

xtreegpt, 2021-01-26:
I'll chat with PiKVM dev Maxim about upgrading the kernel. I'll update when I hear back. Is there docs on setting up kdbg? BTW no /var/log/kern.log

mdevaev, 2021-01-27:
I'm here. This is Arch, the 5.10 kernel was added to the repository today. Try it and tell the results.

xtreegpt, 2021-01-27:
I updated the kernel this AM (EST). I will update if I see a crash today. If the system runs clean I will close the ticket. 

Linux pikvm 5.10.10-1-ARCH #1 SMP Mon Jan 25 21:57:16 UTC 2021 armv7l GNU/Linux

xtreegpt, 2021-01-27:
threw a core again. [Pastebin Dmesg output.](https://pastebin.com/Y4HqqbMH).  Let me know if you need anything else.

xtreegpt, 2021-01-29:
I went to a shorter USB-C cable to tidy things up. Between the kernel update and the cable switch I have gone 24hrs without a core. Closing..

closed 2021-01-29 by xtreegpt