raspberrypi/linux #4103: cec not working with vc4-kms-v3d

henkv1, 2021-01-29:
cec does not work when vc4-kms-v3d is used:
```
echo 'scan' | cec-client -s -d 1
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Unknown
osd string:    TV
CEC version:   unknown
power status:  unknown
language:      ???

device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
Segmentation fault (core dumped)
```
It works fine with vc4-fkms-v3d:
```
echo 'scan' | cec-client -s -d 1
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Samsung
osd string:    TV
CEC version:   1.3a
power status:  on
language:      dut

device #1: Recorder 1
address:       1.0.0.0
active source: yes
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: Recorder 1 (1)
`````

```
[root@harrie-boven ~]# uname -a   
Linux harrie-boven 5.10.11-2-ARCH #1 SMP Thu Jan 28 13:31:33 UTC 2021 armv7l GNU/Linux
```

/boot/config.txt:
```
gpu_mem=96
max_usb_current=1
overscan_left=16
overscan_bottom=16
hdmi_force_edid_audio=1
dtparam=audio=on
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off
hdmi_force_hotplug=1
dtoverlay=vc4-kms-v3d,cma-256
```
I've attached the kernel config.
[config.gz](https://github.com/raspberrypi/linux/files/5894487/config.gz)

6by9, 2021-01-29:
vc4-kms-v3d exposes CEC through the kernel CEC API (/dev/cec0 and /dev/cec1). vc4-fkms-v3d uses the Pi specific VCHIQ service.

Is your version of cec-client built with the appropriate support?

graysky2, 2021-01-29:
@6by9 - I believe @henkv1 is using the Arch ARM package for [libcec](https://github.com/archlinuxarm/PKGBUILDs/blob/master/alarm/libcec-rpi/PKGBUILD).  It is built like this:
```
 unset LDFLAGS

    mkdir build
    cd build
    cmake .. \
        -DCMAKE_BUILD_TYPE=Release \
        -DBUILD_SHARED_LIBS=1 \
        -DCMAKE_INSTALL_PREFIX=/usr \
        -DCMAKE_INSTALL_LIBDIR=/usr/lib \
        -DCMAKE_INSTALL_LIBDIR_NOARCH=/usr/lib \
        -DHAVE_EXYNOS_API=0 \
        -DHAVE_RPI_API=1 \
        -DRPI_INCLUDE_DIR=/opt/vc/include \
        -DRPI_LIB_DIR=/opt/vc/lib
    make
```

Is there something else that needs to be configured in the cmake stanza?

6by9, 2021-01-29:
```-DHAVE_LINUX_API=1``` to enable the standard Linux CEC API.
You may need to disable HAVE_RPI_API, or otherwise get it to choose to use the Linux API instead of RPI.

https://github.com/Pulse-Eight/libcec/blob/master/include/cectypes.h#L287 makes a little tricky to get CEC working on HDMI-1 too. No way to override it in the open call - https://github.com/Pulse-Eight/libcec/blob/master/src/libcec/adapter/Linux/LinuxCECAdapterCommunication.cpp#L72

popcornmix, 2021-01-29:
Note: you don't have to use libcec. You can use cec-ctl which should work out of the box. e.g.
```
cec-ctl -d0 --playback # Configure the PC as a CEC Playback device
cec-ctl -d1 -S # Show the CEC topology
```

graysky2, 2021-01-29:
@6by9 - I cannot speak for henkv1, but I specifically want Kodi to be able to use it.  I removed the following from the cmake stanza and it seems that kodi successfully initializes CEC now:
```
-DHAVE_EXYNOS_API=0 \
-DHAVE_RPI_API=1 \
```

Thank you for the reply.

henkv1, 2021-01-29:
@graysky2 - I want to be able to detect if my tv is switched on and use it to start kodi. I also want to control kodi with CEC.
Do you have Arch ARM packages of libcec and kodi which use the Kernel CEC API?

graysky2, 2021-01-29:
@6by9 - So I actually got a mixed result.  Compiling libcec like the following gave a functional CEC interface for Kodi Matrix RC1 to use on arm64 but not on 32-bit arm.  Both using rpi-5.10.y kernel (latest commit).  Thoughts?

```
    cmake .. \
        -DCMAKE_BUILD_TYPE=Release \
        -DBUILD_SHARED_LIBS=1 \
        -DCMAKE_INSTALL_PREFIX=/usr \
        -DCMAKE_INSTALL_LIBDIR=/usr/lib \
        -DCMAKE_INSTALL_LIBDIR_NOARCH=/usr/lib \
        -DRPI_INCLUDE_DIR=/opt/vc/include \
        -DRPI_LIB_DIR=/opt/vc/lib
```

graysky2, 2021-01-29:
@6by9 - I specifically want Kodi to be able to use it.  I removed the following from the cmake stanza but Kodi Matrix is not able to initializes CEC:
```
-DHAVE_EXYNOS_API=0 \
-DHAVE_RPI_API=1 \
```

I also tried building with:
```
-DHAVE_RPI_API=1 \
-DHAVE_LINUX_API=1 \
```

And simply omitting these lines entirely but same result: Kodi cannot initialize CEC.

graysky2, 2021-01-29:
> vc4-kms-v3d exposes CEC through the kernel CEC API (/dev/cec0 and /dev/cec1). vc4-fkms-v3d uses the Pi specific VCHIQ service.

I think that is the key!  If I build libcec without rpi support, I can use CEC when using kms BUT, if I boot using fkms, the very same package does not work.

```
    _args=(
        -DCMAKE_BUILD_TYPE=Release
        -DBUILD_SHARED_LIBS=1
        -DCMAKE_INSTALL_PREFIX=/usr
        -DCMAKE_INSTALL_LIBDIR=/usr/lib
        -DCMAKE_INSTALL_LIBDIR_NOARCH=/usr/lib
        -DHAVE_LINUX_API=1
      )
    cmake "${_args[@]}" ..
```

Example booted using kms
```
% echo 'volup' | cec-client -s -d 1
opening a connection to the CEC adapter...
volume up: 7F
```

Example rebooting using fkms:
```
% echo 'volup' | cec-client -s -d 1
autodetect FAILED
```

This is paralleled in Kodi.  Matrix (kms) works with CEC but Leia (fkms) does not.

So the big question: can libcec be compiled to work with both the fkms and kms driver?

popcornmix, 2021-01-29:
That's a question for libcec - not something we can affect in kernel.
You should try with -DHAVE_RPI_API=1 and -DHAVE_LINUX_API=1 and see if it works.
If not try here: https://github.com/Pulse-Eight/libcec/issues/

graysky2, 2021-01-29:
@popcornmix - Thanks.  There has been several open issues with them but no response has been offered (https://github.com/Pulse-Eight/libcec/issues/540 and https://github.com/Pulse-Eight/libcec/issues/543 for example).

I understand this is not a kernel issue and thank both you and @6by9 for the help.

In answer to your question, building with both options does not give a functional result under kms.

@henkv1 - I am thinking that once Matrix goes final, I will adjust Arch ARM's libcec package to work with kms.  Until then you can build one yourself now from https://github.com/graysky2/PKGBUILDs/tree/libcec-kms/alarm/libcec-rpi

Please test with kodi (I assume you're using the kodi-rpi-git package) and report back.

6by9, 2021-01-29:
I guess the firmware can detect the kms driver enabling the cec interrupt, and then disable the vchiq cec service.
Hopefully libcec behaves itself should the vchiq service not open.

posuch, 2021-01-30:
> @popcornmix - Thanks. There has been several open issues with them but no response has been offered ([Pulse-Eight/libcec#540](https://github.com/Pulse-Eight/libcec/issues/540) and [Pulse-Eight/libcec#543](https://github.com/Pulse-Eight/libcec/issues/543) for example).
> 
> I understand this is not a kernel issue and thank both you and @6by9 for the help.
> 
> In answer to your question, building with both options does not give a functional result under kms.
> 
> @henkv1 - I am thinking that once Matrix goes final, I will adjust Arch ARM's libcec-rpi to work with kms only since seemingly have to choose (kms or fkms) and since Matrix is best used with kms. Until then you can build one yourself now from https://github.com/graysky2/PKGBUILDs/tree/libcec-kms/alarm/libcec-rpi
> 
> Please test with kodi (I assume you're using the kodi-rpi-git package) and report back.

@graysky2 latest PKGBUILD works with kodi on 64bit arm rpi4b, all packages installed from official repos

henkv1, 2021-01-30:
@graysky2 - The PKGBUILD works fine on 32 bit arm with the standard kodi-rpi leia.
Thank you all for the support.

graysky2, 2021-01-30:
> @graysky2 - The PKGBUILD works fine on 32 bit arm with the standard kodi-rpi leia. Only, the kodi package does not work well with kms, but that has nothing to do with the CEC issue.

Your reply might be cryptic to folks not using Arch ARM... frankly, it's a bit cryptic to me :smile:

Currently, Arch ARM provides Kodi in these packages, all use forks from @popcornmix:
* **kodi-rpi-legacy** built from _newclock5_.
* **kodi-rpi** built from _leia_pi4_ and requires vc4-fkms-v3d.
* **kodi-rpi-git** built from _gbm_ and uses vc4-kms-v3d.

So I think you're saying two things:
1. **kodi-rpi** using vc4-fkms-v3d and my modified **libcec-rpi** package I linked above work together.
2. **kodi-rpi** does not work with vc4-kms-v3d.

Am I correct?

henkv1, 2021-01-30:
What I was trying to say was that CEC works fine in kodi-rpi with vc4-kms-v3d and the modified libcec, but there are issues playing videos.
The modified libcec does not work with vc4-fkms-v3d

graysky2, 2021-01-30:
OK.  That is consistent with my observations as well.

closed 2021-01-31 by henkv1

PaulWebster, 2021-02-01:
I have it working with kms - but needs extra kernel setting
CONFIG_MEDIA_CEC_RC

Also - can you enable things for RPI0 and 1 ?
Looks like default config has CONFIG_DRM_VC4_HDMI_CEC disable for old devices ... but would be good to enable even if with warnings (all of my testing was on an old B+)
Sorry for duplication between forum and github - but I started the forum thread 2 hours before this github issue was raised (perhaps the forum post is why the issue was raised).

See https://www.raspberrypi.org/forums/viewtopic.php?p=1809460#p1809460
which includes a link to a short video showing it working

graysky2, 2021-02-01:
> I have it working with kms and without libcec - but needs extra kernel setting
CONFIG_MEDIA_CEC_RC

Don't think adding that is going to help with the fkms vs kms thing.  Arch ARM's rpi-kernel has that option enabled and users still hit this issue.

```
% zgrep CEC /proc/config.gz
CONFIG_TABLET_USB_ACECAD=m
CONFIG_CEC_CORE=m
CONFIG_CEC_NOTIFIER=y
CONFIG_CEC_PIN=y
CONFIG_MEDIA_CEC_RC=y
# CONFIG_CEC_PIN_ERROR_INJ is not set
CONFIG_MEDIA_CEC_SUPPORT=y
CONFIG_CEC_CH7322=m
CONFIG_CEC_GPIO=m
CONFIG_USB_PULSE8_CEC=m
CONFIG_USB_RAINSHADOW_CEC=m
# CONFIG_VIDEO_VIVID_CEC is not set
# CONFIG_VIDEO_TC358743_CEC is not set
# CONFIG_DRM_DP_CEC is not set
CONFIG_DRM_VC4_HDMI_CEC=y
```

> Sorry for duplication between forum and github - but I started the forum thread 2 hours before this github issue was raised (perhaps the forum post is why the issue was raised).

No, [this](https://github.com/Pulse-Eight/libcec/issues/543#issuecomment-769727689) is why.  You already have two Raspberry Pi Engineers replying to you in your thread.  Not sure why you pinged this issue too.

PaulWebster, 2021-02-01:
I thought it was fairly clear why ... but just in case it wasn't, I explained.
Thanks for the link to the other github issue.

I had also seen that cec-client did not seem happy when HAV_RPI and HAVE_LINUX were enabled but there is no vchiq (because of using KMS rather than FKMS) ... which is why I switched to cec-ctl.
Just in case anyone else reading this in the future tries to make the switch and wants to use CEC via the built in RC mechanism ... do a
cec-cli --playback
first to set the profile otherwise it looks like it is not working.

Feel free to remove this if you think it is inappropriate in this issue.