raspberrypi/linux #4105: RPI 4 pending Kernel Upgrade does not install

sirg2, 2021-01-30:
**Describe the bug**
When I use `apt` to update/upgrade my Pi or install/update packages, after installation the following shows up in terminal
[image]
For those not speaking German:
> Pending kernel upgrade
> New kernel available
> The current kernel version is 5.4.79-v7l+, but version 5.4.83-v8+ is expected.
> The system does not automatically reboot to load the new kernel. A reboot should be performed

I thought I just need to reboot and then the new kernel will load. However, that is not the case. The old kernel is used again after every reboot.
In the file system I can find the files for both kernels. In some forums I have read that one should reinstall those files with different commands or wait until the next update releases. None of this has worked so far. The problem has been going on for about a quarter of a year.

So far it is not fatal, my Pi works as far as I can tell. However, I would still prefer to get the kernel issue resolved.
Do you know how to fix this or isn’t it a problem and I can ignore it?
I appreciate your help and support 😊 

Greetings Luca ✌🏻

* RPI B4 with 4 GB RAM
* Which OS and version (`cat /etc/rpi-issue`)?:
```
Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage5
```

* Which firmware version (`vcgencmd version`)?
```
Nov 30 2020 22:12:08
Copyright (c) 2012 Broadcom
Version ab1181cc0cb6df52bfae3b1d3fef0ce7c325166c (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
```
Linux raspberrypi 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux
```

XECDesign, 2021-02-01:
That's not a feature of the images we ship. I believe there's software called 'needrestart', which some people end up with for some reason. It makes some wrong assumptions when it tries to figure out when a restart is actually required. As far as I know, you should be able to just `apt purge` it.

sirg2, 2021-02-03:
> That's not a feature of the images we ship. I believe there's software called 'needrestart', which some people end up with for some reason. It makes some wrong assumptions when it tries to figure out when a restart is actually required. As far as I know, you should be able to just `apt purge` it.

Thanks a lot 👍🏻 Problem solved so far. I could deinstall `needrestart` and the window doesn't show up anymore.

But one more question if you don't mind. It said that the kernel version I am using is not the currently default one. By the infos I gave you, can you tell me if this is true? And if so does it, after I've purged `needrestart`, install the "default"/standard kernel currently in release with the next update automatically?

Thanks a lot @XECDesign for your help ￼👍🏻

XECDesign, 2021-02-04:
It does look like you're a little behind. After a reboot, uname should show `5.4.83-v7l+`. And there's another update coming a little later today that will take it to a 5.10 kernel.

sirg2, 2021-02-05:
Thank you very much. It is as you said ✌

closed 2021-02-05 by sirg2

puccaso, 2022-07-02:
The funny thing is, this is still the case today. Albeit, I am running ubuntu arm64, but I just wanted to share. 
```
disp:/boot$ ls -lsaht
total 214M
4.0K drwxr-xr-x  4 root root 4.0K Jul  2 13:20 .
   0 lrwxrwxrwx  1 root root   44 Jul  2 13:19 dtb -> dtbs/5.15.0-1005-raspi/./bcm2711-rpi-4-b.dtb
   0 lrwxrwxrwx  1 root root   44 Jul  2 13:19 dtb-5.15.0-1005-raspi -> dtbs/5.15.0-1005-raspi/./bcm2711-rpi-4-b.dtb
 40M -rw-r--r--  1 root root  40M Jul  2 13:19 initrd.img-5.15.0-1005-raspi
   0 lrwxrwxrwx  1 root root   47 Jul  2 13:10 dtb-5.15.0-30-lowlatency -> dtbs/5.15.0-30-lowlatency/./bcm2711-rpi-4-b.dtb
138M -rw-r--r--  1 root root 138M Jul  2 13:10 initrd.img-5.15.0-30-lowlatency
4.0K drwxr-xr-x  4 root root 4.0K Jul  2 13:06 dtbs
4.0K drwxr-xr-x 20 root root 4.0K Jul  2 12:52 ..
   0 lrwxrwxrwx  1 root root   31 Jul  2 12:30 initrd.img -> initrd.img-5.15.0-30-lowlatency
   0 lrwxrwxrwx  1 root root   28 Jul  2 12:30 vmlinuz -> vmlinuz-5.15.0-30-lowlatency
 15M -rw-------  1 root root  15M May  5 13:07 vmlinuz-5.15.0-30-lowlatency
6.7M -rw-------  1 root root 6.7M May  5 11:20 System.map-5.15.0-30-lowlatency
292K -rw-r--r--  1 root root 291K May  5 11:20 config-5.15.0-30-lowlatency
   0 lrwxrwxrwx  1 root root   28 Apr 19 11:09 initrd.img.old -> initrd.img-5.15.0-1005-raspi
   0 lrwxrwxrwx  1 root root   25 Apr 19 11:09 vmlinuz.old -> vmlinuz-5.15.0-1005-raspi
5.0M -rw-------  1 root root 5.0M Apr  4 12:33 System.map-5.15.0-1005-raspi
240K -rw-r--r--  1 root root 240K Apr  4 12:33 config-5.15.0-1005-raspi
9.9M -rw-------  1 root root 9.9M Apr  4 12:33 vmlinuz-5.15.0-1005-raspi
7.0K drwxr-xr-x  5 root root 7.0K Jan  1  1970 firmware
```
but still, 
```disp:/boot$ uname -a
Linux disp 5.15.0-1005-raspi #5-Ubuntu SMP PREEMPT Mon Apr 4 12:21:48 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux```

HinTak, 2022-07-02:
@puccaso Ubuntu uses a different mechanism to boot different kernels than Raspian, so your comment is unrelated  See my notes from a while ago:

https://github.com/HinTak/RaspberryPi-Dev/blob/master/Downgrading-Pi-Kernel.md