raspberrypi/linux #4106: isolcpus=0 not working on 5.10.x and 5.4.x

klslz, 2021-01-30:
I am wondering if there's an issue on RPi4 and RPi3  and kernel 5.10. and 5.4.x 
All run latest firmware. And RPiOS64. And show the same behavior.

Isolating CPU0 doesn't work anymore. 
I am pretty sure it used to work. Not sure when it stopped working.

isolcpus=0,3 

in cmdline.txt ends up with:

#cat /sys/devices/system/cpu/isolated 
3

#taskset -cp 1
pid 1's current affinity list: 0-2

CPU0 remains available.

Is there any explanation for it ? Am I doing anything wrong here.

Thx

pelwell, 2021-01-30:
Have you looked in the kernel log (`dmesg`) for signs of why it is rejecting isolcpus=0?

klslz, 2021-01-30:
Enclosed the dmesg.

There's this Housekeeping CPU0 message. One CPU must be present.

To me it looked OK.

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.10.5-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1392 SMP PREEMPT Sat Jan 9 18:56:30 GMT 2021
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003befffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Zeroed struct page in unavailable ranges: 256 pages
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0 totalpages: 1015552
[    0.000000]   DMA zone: 3836 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 245504 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 12032 pages used for memmap
[    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
[    0.000000] percpu: Embedded 32 pages/cpu s91224 r8192 d31656 u131072
[    0.000000] pcpu-alloc: s91224 r8192 d31656 u131072 alloc=32*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 999684
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:22:5D:30 vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  root=PARTUUID=98360210-02 rootfstype=ext4 elevator=deadline rootwait ipv6.disable=1 noswap isolcpus=0
[    0.000000] Kernel parameter elevator= does not have any effect anymore.
               Please use sysfs to set IO scheduler for individual devices.
[    0.000000] Housekeeping: must include one present CPU, using boot CPU:0
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000037f00000-0x000000003bf00000] (64MB)
[    0.000000] Memory: 3828552K/4062208K available (10880K kernel code, 1908K rwdata, 3908K rodata, 3648K init, 1242K bss, 168120K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 36054 entries in 141 pages
[    0.000000] ftrace: allocated 141 pages with 4 groups
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x3b8/0x578 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000003] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000122] Console: colour dummy device 80x25
[    0.000356] printk: console [tty0] enabled
[    0.000387] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000404] pid_max: default: 32768 minimum: 301
[    0.000474] LSM: Security Framework initializing
[    0.000593] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.000637] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[... 231 lines cut ...]
[    1.899130] EXT4-fs (sda2): re-mounted. Opts: (null)
[    1.933512] systemd-journald[116]: Received request to flush runtime journal from PID 1
[    2.123696] rpivid-mem feb00000.hevc-decoder: rpivid-hevcmem initialised: Registers at 0xfeb00000 length 0x00010000
[    2.124169] rpivid-mem feb10000.rpivid-local-intc: rpivid-intcmem initialised: Registers at 0xfeb10000 length 0x00001000
[    2.124389] rpivid-mem feb20000.h264-decoder: rpivid-h264mem initialised: Registers at 0xfeb20000 length 0x00010000
[... cut ...]
[    3.338281] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    3.338809] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[    3.338822] cfg80211: failed to load regulatory.db
[    3.408928] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    3.409208] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   31.711028] vcc-sd: disabling
[   40.306601] EXT4-fs (sda2): re-mounted. Opts: commit=300
```

pelwell, 2021-01-30:
Do you specifically want to isolate CPU 0? You have to keep at least one core online, so why not 0?

klslz, 2021-01-31:
Does your question imply ?

* You were able to reproduce the issue
* It's RPI specific
* Due to whatever reason - you might want to tell us -  something has changed
* It could be a bit tricky to touch it
* Is it worth it !?!?

Let's put it this way.

* it's not my problem specifically, it's a standard kernel feature
* that restriction might be a diversion from mainline
* afaik (correct me if I'm wrong), basically non of the documentation restricts from using CPU0
* because, usually as long as there is one (any) non isolated CPU, it'll work.

Now. 
The whole thing is not a huge issue. However. If in fact we'd be talking about a (rather new) RPI restriction, it should at least be documented and an error code/warning should be thrown out in dmesg, telling us about the restriction. 

Not nice. But at least a confirmed status quo.

pelwell, 2021-01-31:
Genuine show-stopping problems are more likely to get attention than unexpected behaviours with obvious workarounds.

pelwell, 2021-02-01:
The issue appears to be an artefact of the startup sequence. The `isolcpus` command line parameter is parsed and handled very early - notice the zero timestamp on the Housekeeping message in your log - whereas the secondary CPUs are started until quite a bit later.

The housekeeping code is trying to ensure that there is a housekeeping CPU available, and it does this by checking for an overlap between the present CPUs and the housekeeping CPUs. At that point, the only present CPU is CPU 0, and the housekeeping CPUs are all the possible CPUs (0-3) minus the isolated CPU 0, i.e. 1-3, guaranteeing no overlap.

Unless you want to rewrite the common start-up code to somehow avoid this problem then you are going to have to live with it, although you should note the words of the kernel parameter documentation for `isolcpus`:
```
	isolcpus=	[KNL,SMP,ISOL] Isolate a given set of CPUs from disturbance.
			[Deprecated - use cpusets instead]
			^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

closed 2021-02-01 by pelwell

klslz, 2021-02-01:
Thx. For the feedback. Highly appreciated.

I was simply wondering why it used to work for years and than suddenly stopped working. And there' hasn't been 
a CPU0 limitation after all.

I'll have a look at cpusets.

pelwell, 2021-02-01:
Taking a quick look at the 4.19 kernel shows that it used to check for a non-empty intersection between the housekeeping CPUs and the _possible_ CPUs (now _present_ CPUs), i.e. it assumed that the secondary CPUs would come online.