raspberrypi/linux #4107: environment variable hostname missing

jrbrearley, 2021-01-31:
Debian doesn't seem to have an environment variable $HOSTNAME.

When I list the variables available via "env", I see many variables, but no HOSTNAME.

When I list the environment variables visible to python with attached script, I see the same list of variables,  but no HOSTNAME.
[environment_vars.py.txt](https://github.com/raspberrypi/linux/files/5900461/environment_vars.py.txt)

What's a bit puzzling is default shell seems to know $HOSTNAME:
echo $HOSTNAME
mpptinv

In the short term, I am using a python script to get STDOUT from running external command: /bin/hostname

Any chance you could have an environment variable HOSTNAME that is visible to python and other coding frameworks?

I am using Raspberry Pi 3B+, build 5.4.79-v7+ 1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GNU/Linux

XECDesign, 2021-02-01:
That's something `bash` does, so it's not a real environment variable that you should be trying to use outside of shell scripts.

If you'd like to get the hostname from a python script, take a look at https://docs.python.org/3/library/socket.html#socket.gethostname

jrbrearley, 2021-02-01:
Yes, when using sockets, gethostname works.

Here is the stand-alone get_hostname script I wrote.
[get_hostname.py.txt](https://github.com/raspberrypi/linux/files/5904624/get_hostname.py.txt)

pelwell, 2021-02-11:
Closing as answered.

closed 2021-02-11 by pelwell