raspberrypi/linux #4108: Non-blocking ALSA audio is broken in kernel 5.10

timothyjward, 2021-02-01:
**Is this the right place for my bug report?**

I was trying out the 5.10 kernel and found that it has broken ALSA audio in conjunction with MPD. I did some debug with the lead for MPD and the issue looks to be related to the handling of non blocking io when using memory mapped buffers. 

It seems pretty certain that this is a kernel bug because playback worked fine on earlier kernels.

The original MPD bug is visible at https://github.com/MusicPlayerDaemon/MPD/issues/1056 - it was easily reproducible on 5.10.y but worked fine on 5.4.y (and 4.x).

**Describe the bug**
When attempting to play audio using a non-blocking source (e.g. MPD) the audio stream fails to make progress as no events are sent to the source indicating that the buffer is ready for more data. 

This only happens when there are one or more PCMs in the ALSA processing chain - direct hardware playback (with or without a plug converter) works correctly

**To reproduce**

This issue is simple to reproduce using the Pi headphone jack (seen on a Pi B, Pi 2B and Pi 3B)

Create a simple ALSA configuration in `/etc/asound.conf` that targets the Headphone Jack, such as:

```
pcm.mpdTest {
    type copy
    slave.pcm "hwOutput"
}

# There is always a plug before the hardware to be safe
pcm.output {
    type plug
    slave.pcm "hwOutput"
}

pcm.hwOutput {
    type hw
    card "Headphones"
}
```

Then play some CD quality audio (S16_LE, 44100Hz, 2 channels) to each of the PCMs using a non-blocking source (MPD is non blocking `aplay` isn't).

**Expected behaviour**

All of the PCMs should play correctly through the headphones

**Actual behaviour**

*If I write to the hardware through the pcm hw everything works
* If I write to the hardware through the pcm output everything works
* If I write to the hardware through the pcm mpdTest then audio hangs

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
  * Seen on Pi B, 2B and 3B
* Which OS and version (`cat /etc/rpi-issue`)?
  * I was using a Volumio beta image, but this was reproduced on a non volumio image
* Which firmware version (`vcgencmd version`)?
```
vcgencmd version
Dec 15 2020 18:34:30
Copyright (c) 2012 Broadcom
version 4b4aff21f72c5b9ba39d83c7b0f8fa910a6ef99b (clean) (release) (start)
```   
* Which kernel version (`uname -a`)?
```
uname -a
Linux pi2-test 5.10.3-v7+ #1386 SMP Sun Dec 27 14:01:57 GMT 2020 armv7l GNU/Linux
```
**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Additional information (including trace of the ALSA event calls) is present in the linked MPD bug

pelwell, 2021-02-01:
Recent Raspberry Pi OS releases have switched to using PulseAudio for audio control and mixing. Please confirm whether the regression with MPD is due to a kernel version bump or to the use of PulseAudio, e.g. by verifying that your current image pre-dates PulseAudio or by downgrading your kernel and showing that it works again.

timothyjward, 2021-02-01:
> Recent Raspberry Pi OS releases have switched to using PulseAudio for audio control and mixing. Please confirm whether the regression with MPD is due to a kernel version bump or to the use of [...]

The bug was visible to @MaxKellermann  when updating from `5.4.83-v7+` to `5.10.7-v7+` with no other OS changes. The Volumio image also started working correctly again after moving from `5.10.7-v7+` back to `5.4.83-v7+` with no other changes to the image. PulseAudio does not seem to be involved in the issue.

pelwell, 2021-02-01:
Thanks for confirming.

timothyjward, 2021-02-01:
> Thanks for confirming.

Not a problem. I won't pretend that I understand the underlying issue that well. According to Max 
_"...after the SNDRV_PCM_IOCTL_START, the file descriptor is never again reported "ready" by the kernel."_

It's also the case that:

* _with a "hw" PCM, libasound uses SNDRV_PCM_IOCTL_WRITEI_FRAMES_
* _with a "copy" PCM, libasound uses mmap_
* _on working kernels the "copy" pcm works (via mmap and SNDRV_PCM_IOCTL_HWSYNC)_
* _on failing kernels the "copy" pcm doesn't work, but it uses mmap and SNDRV_PCM_IOCTL_SYNC_PTR which is a fallback mode called mmap_status_fallbacked in libasound; this mode is used if mapping the status area fails._

I'm hoping that this will be of some help in working out what's wrong.

timothyjward, 2021-02-01:
Some further possibly helpful details for debugging this. I've been able to replicate the issue using the alsa-lib example code meaning it is even more unlikely to be an mpd error.
The [pcm.c test](https://github.com/alsa-project/alsa-lib/blob/d824b461ae807ea436e2df36da9c2212e485e3e6/test/pcm.c) allows you to create a sine wave test signal in a variety of ways.

The following (blocking call) runs fine:

    ./pcm -D mpdTest -r 44100 -c 2 -f 440 -b 100000 -p 10000 -m write -o S16_L£ -v -n -e

The non-blocking version **does not work**:

    ./pcm -D mpdTest -r 44100 -c 2 -f 440 -b 100000 -p 10000 -m write_and_poll -o S16_L£ -v -n -e

Unless you write directly to the hardware, which does work:

    ./pcm -D hwOutput -r 44100 -c 2 -f 440 -b 100000 -p 10000 -m write_and_poll -o S16_L£ -v -n -e

pelwell, 2021-02-01:
That's really helpful and timely - I just started looking for something similar.

pelwell, 2021-02-01:
What must I do to get mpdTest? I've installed the mpd apt package.

timothyjward, 2021-02-01:
MPD isn't necessary to show the bug (it was just how I found it). If you create the following file at `/etc/asound.conf` then it will add the PCMs that I've referred to in the bug.

```
pcm.mpdTest {
    type copy
    slave.pcm "hwOutput"
}

# There is always a plug before the hardware to be safe
pcm.output {
    type plug
    slave.pcm "hwOutput"
}

pcm.hwOutput {
    type hw
    card "Headphones"
}
```

pelwell, 2021-02-01:
Yes - that's not working.

DictumMortuum, 2021-02-09:
I have the same issue, but for me audio_format=44100:24:2 isn't playable, while other formats are.

Don't laugh, but I've got two buttons: one plays careless whisper, the other one just plays rain sound in a loop. The rain mp3 doesn't work, while careless whisper works :)

euri10, 2021-02-13:
I think I have the same issue without mpd, using plexamp :

1.  I installed yesterday a fresh image (https://downloads.raspberrypi.org/raspios_lite_armhf/release_notes.txt) that ships `Linux raspberrypi 5.4.83-v7+ #1379 SMP Mon Dec 14 13:08:57 GMT 2020 armv7l GNU/Linux` , 
2. I installed plexamp, everything works perfect, sound is as good as it gets :)
3. as soon as `raspberrypi-kernel` is upgraded from `20210108-1` to `20210201-1`, after a reboot, the sounds is like it is played in slow motion.

no pulseaudio here, and no change made to alsa.

Dunno if this adds to the understanding, only thing I can confirm is that the kernel upgrade is what caused audio to sound very slowed.

MaxKellermann, 2021-02-13:
> I think I have the same issue without mpd, using plexamp :
 
FYI: Plexamp is built on top of MPD, see https://www.musicpd.org/commercial.html
So it's not "without MPD". It is very much MPD.

timothyjward, 2021-09-26:
Has there been any progress with this? It would be nice to be able to move to a 5.10 kernel!

macmpi, 2021-10-16:
@popcornmix would you have any clue by chance?

macmpi, 2021-11-03:
@timothyjward has using `vc4-hdmi` ([`dtoverlay=vc4-kms-v3d`](https://forums.raspberrypi.com/viewtopic.php?f=29&t=269769&p=1636828#p1636828)) instead of `snd_bcm2835` (`dtparam=audio=on`) any influence?

timothyjward, 2021-11-13:
> @timothyjward has using `vc4-hdmi` ([`dtoverlay=vc4-kms-v3d`](https://forums.raspberrypi.com/viewtopic.php?f=29&t=269769&p=1636828#p1636828)) instead of `snd_bcm2835` (`dtparam=audio=on`) any influence?

We’ve tried this out in Volumio using a recent 5.10 and still don’t have working audio.