raspberrypi/linux #4110: How to get the specific kernel version?

jamwu, 2021-02-01:


popcornmix, 2021-02-01:
In general you can't automatically. You can rebase the downstream patches onto any stable point. e.g.
```
git rebase --onto=v5.10.3 stable/linux-5.10.y rpi-5.10.y
```
(you may need to fix up minor conflicts)
Assuming this repo is origin and stable is git://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git

It might help if you explained why you wanted this, and whether you want to pre-built kernel or to build it yourself.

HinTak, 2021-04-15:
@popcornmix apologies for jumping in - is there a collection of pre-built kernels I could try, without bisecting through commits that may or may not boot - I am thinking of #4279 . i.e. I am looking for a raspberry pi equivalent to the desktop linux's beta-build farm like https://koji.fedoraproject.org/ , or Ubuntu's mainline kernel build series  https://kernel.ubuntu.com/~kernel-ppa/mainline/ .

popcornmix, 2021-04-16:
Some are available, see:
https://github.com/Hexxeh/rpi-firmware/commits/master
https://github.com/Hexxeh/rpi-firmware/commits/next

But this only includes kernel versions we are actively testing, so there are many 5.4 builds, but no 5.5 ones.

HinTak, 2021-04-16:
@popcornmix thanks. I have looked at the history of rpi-5.4.y and rpi-5.5.y - they looks like they are fresh side branch of linux-stable so cannot be meaningfully bisected. It looks more like a selective reverse-patch to find what breaks...

popcornmix, 2021-04-16:
Yes. It may be possible to run the upstream kernel on the pi (depending on newness of pi and newness of upstream kernel),
which allows easier bisecting but it can be tricky to get running, especially if you rely on features like dtoverlays.
You can also rebase the downstream patches onto upstream at various points to bisect (but that will be slow).

If you can make a list of suspicious upstream commits then reverting them individually is probably the quickest option, but it depends on how good your guesses are.

I'm assuming the breakage was from an upstream commit, as it would have likely appeared on the 5.4 tree if it was due to a downstream commit as that continued to be updated long after 5.5 tree was abandoned.

HinTak, 2021-04-16:
@popcornmix thanks. I am pretty sure it is some subtle change in overlay processing (and what is valid) . Sound/soc has undergone substantial rewrite just before 5.5 (despite no-change in external API hence require no adjustments in out-of-tree driver to build against) . At the moment I am suspecting this series - https://lore.kernel.org/alsa-devel/87ftj23jph.wl-kuninori.morimoto.gx@renesas.com/ - buy unfortunately it is a large series with interlocking changes...