raspberrypi/linux #4111: Tag "raspberrypi-kernel_1.20210111.master-1" causes disk corruption and should be deleted from github

dflogeras, 2021-02-02:
Hi, I run Gentoo on my Pi's and thus compile my own custom kernel.  In the past I have always upgraded kernels to whatever the most recently released tag on Github was.  When I read that the 5.10.x kernel was ready for prime-time, I built a kernel based on the aforementioned tag but soon realized that it causes very bad corruption on any mounted volumes.

After chatting in IRC, apparently the 5.10.5 kernels were known to have a bug that does this.  Is is possible to please remove this tag from Github and hopefully spare others who build their own kernels?

popcornmix, 2021-02-02:
The tags are there for users who have a specific kernel build installed through apt to find the source code it was was built from.
As such we can't delete the tag. There are more recent tags that don't have the issue and I'd expect new builds to use those.

Also the corruption was with raspberrypi-kernel_1.20210108.master-1 and raspberrypi-kernel_1.20210111.master-1 contains the fix. Deleting tags would make it harder to investigate what code was in which kernel releases.

dflogeras, 2021-02-02:
Ah, yes I supposed tags should be immutable.  And I apologize, I did mean I ran raspberrypi-kernel_1.20210108.master-1 (the latest on GH last as of yesterday when I compiled it).  I was running the firmware blobs from 20210111 hence my typo.

I'll correct the title in case anyone else digs up this issue, then close.  Thanks for the clarification.

closed 2021-02-02 by dflogeras