raspberrypi/linux #4115: Can't sudo apt upgrade Kali Linux

MaVicho, 2021-02-03:
### Hello everyone! Today I'm going to tell you what I did to upgrade Kali Linux on Raspberry.
The problem is: when you do the sudo apt upgrade, at the moment that the progress bar appears (15% - 20% in my case), the screen goes black and if you try rebooting the system (or wait in the black screen, it's the same), it gets stuck. So after many attempts I decided to do some changes:
**(1) I was using Raspberry Pi Imager for the iso. Then I change it to balenaEtcher.**
**(2) When kali boot open a terminal and execute:**
$ cd /etc/apt
$ sudo nano sources.list
In my case I just have to delete '#' to leave it like this:
deb http://http.kali.org/kali kali-rolling main contrib non-free
deb-src http://http.kali.org/kali kali-rolling main contrib non-free
Then use CTRL+X; Y; ENTER to save it.
**(3) Go to Power Manager and change this settings:** _I thought the problem was the screen turning off after a while_
**SYSTEM:**
System sleep mode: Suspend
When inactive for: Never
**DISPLAY:**
Display power management: Off
Blank after; Put to sleep after; Switch off after: 60 minutes.
**Security:**
Automatically lock the sesion: Never
**(4) (Sudo apt update before this step) Instead of: sudo apt upgrade, I execute:** 
$ sudo apt dist-upgrade 
After a long time, the progress bar is completed. Now execute: 
$sudo apt upgrade 
It ends and you can do: 
$sudo apt autoremove 
to delete outdated packages.

**My specs:
Raspberry 4b 4gb
Kingston 32gb class 10
kali-linux-2020.4-rpi4-nexmon-64.img**

I'm leaving this open for any comments or contributions. I don't if all this thing solve it or just one, but I hope it helps someone.

Bye.

popcornmix, 2021-02-03:
This repo is for reporting bugs in the linux kernel.
Your issue seems to be with configuration options in Kali linux (something we here are not involved with producing).
It may be better to post this on a Kali linux forum.

pelwell, 2021-02-03:
https://forums.kali.org/

closed 2021-02-03 by pelwell