raspberrypi/linux #4117: LAN7430 driver needs patching - fix included

mangodan2003, 2021-02-04:
**Describe the bug**
lan743x module fails to map DMA here https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/net/ethernet/microchip/lan743x_main.c#L1986

**To reproduce**
Using the CM4IO board along with the LAN7430 evaluation board (PCIe) build and load the lan743x module.

**Expected behaviour**
Should work

**Actual behaviour**
Module init fails (note I added some debug printk's)

```
[   68.229990] lan743x 0000:01:00.0 eth1: using MSIX interrupts, number of vectors = 6
[   68.245017] lan743x 0000:01:00.0 eth1: Link is Down
[   68.246830] ------------[ cut here ]------------
[   68.246862] WARNING: CPU: 3 PID: 396 at kernel/dma/swiotlb.c:683 swiotlb_map+0x388/0x438
[   68.246883] lan743x 0000:01:00.0: swiotlb addr 0x0000000419400000+9236 overflow (mask ffffffff, bus limit 47fffffff).
[   68.246898] Modules linked in: sha256_generic cfg80211 rfkill 8021q garp stp llc panel_auo_h139bln01_2(O) v3d gpu_sched raspberrypi_hwmon i2c_brcmstb bcm2835_v4l2(C) dwc2 bcm2835_codec(C) roles bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common vc4 videodev cec mc vc_sm_cma(C) drm_kms_helper snd_bcm2835(C) drm drm_panel_orientation_quirks snd_soc_core lan743x(O) snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight rpivid_mem uio_pdrv_genirq uio ip_tables x_tables ipv6
[   68.247642] CPU: 3 PID: 396 Comm: dhcpcd Tainted: G         C O      5.10.11-v7l+ #1
[   68.247652] Hardware name: BCM2711
[   68.247662] Backtrace: 
[   68.247690] [<c020c37c>] (dump_backtrace) from [<c020c6c8>] (show_stack+0x20/0x24)
[   68.247705]  r7:ffffffff r6:00000000 r5:60000013 r4:c12e69bc
[   68.247724] [<c020c6a8>] (show_stack) from [<c0b62db4>] (dump_stack+0xcc/0xf8)
[   68.247742] [<c0b62ce8>] (dump_stack) from [<c02213a4>] (__warn+0x110/0x114)
[   68.247758]  r10:c1205048 r9:c02a7380 r8:000002ab r7:00000009 r6:00000000 r5:c0e22528
[   68.247768]  r4:c3ce7af4 r3:c1205094
[   68.247785] [<c0221294>] (__warn) from [<c022142c>] (warn_slowpath_fmt+0x84/0xc0)
[   68.247799]  r9:00000009 r8:c02a7380 r7:000002ab r6:c0e22528 r5:c0e224e4 r4:c1205048
[   68.247816] [<c02213ac>] (warn_slowpath_fmt) from [<c02a7380>] (swiotlb_map+0x388/0x438)
[   68.247830]  r9:00000000 r8:19400000 r7:00002414 r6:c1bcb070 r5:00000000 r4:ffffffff
[   68.247848] [<c02a6ff8>] (swiotlb_map) from [<c02a1fcc>] (dma_map_page_attrs+0x254/0x360)
[   68.247862]  r10:00000040 r9:00000000 r8:c1bcb1e0 r7:c1205048 r6:c1bcb070 r5:00000000
[   68.247872]  r4:ffffffff
[   68.247905] [<c02a1d78>] (dma_map_page_attrs) from [<bf137750>] (lan743x_rx_init_ring_element+0xa0/0x1d0 [lan743x])
[   68.247919]  r10:c1bcb000 r9:ded46000 r8:ded46000 r7:c270ed70 r6:00000000 r5:00000000
[   68.247928]  r4:c3c20000
[   68.247959] [<bf1376b0>] (lan743x_rx_init_ring_element [lan743x]) from [<bf138948>] (lan743x_netdev_open+0x978/0x1300 [lan743x])
[   68.247973]  r10:c1205048 r9:c270e580 r8:c270ed70 r7:00000000 r6:00000000 r5:c270ed70
[   68.247983]  r4:c270e580
[   68.248009] [<bf137fd0>] (lan743x_netdev_open [lan743x]) from [<c09f71fc>] (__dev_open+0x108/0x1a4)
[   68.248023]  r10:00000000 r9:00000000 r8:c270e028 r7:bf13c024 r6:c1205048 r5:00000000
[   68.248033]  r4:c270e000
[   68.248052] [<c09f70f4>] (__dev_open) from [<c09f762c>] (__dev_change_flags+0x178/0x1d0)
[   68.248065]  r8:00001002 r7:c1205048 r6:00001043 r5:00000001 r4:c270e000
[   68.248085] [<c09f74b4>] (__dev_change_flags) from [<c09f76ac>] (dev_change_flags+0x28/0x58)
[   68.248098]  r9:00000000 r8:00008914 r7:c270e144 r6:00000000 r5:00001002 r4:c270e000
[   68.248118] [<c09f7684>] (dev_change_flags) from [<c0ab7690>] (devinet_ioctl+0x5f0/0x768)
[   68.248132]  r9:00000000 r8:00008914 r7:c272e50c r6:c1205048 r5:00000000 r4:c3ce7e40
[   68.248149] [<c0ab70a0>] (devinet_ioctl) from [<c0abb2ac>] (inet_ioctl+0x1e4/0x320)
[   68.248162]  r10:00000009 r9:be9d5a24 r8:c131e700 r7:c131e700 r6:00008914 r5:be9d5a24
[   68.248172]  r4:c1205048
[   68.248188] [<c0abb0c8>] (inet_ioctl) from [<c09ca31c>] (sock_ioctl+0x340/0x510)
[   68.248201]  r8:c131e700 r7:be9d5a24 r6:00008914 r5:c1205048 r4:00008914
[   68.248219] [<c09c9fdc>] (sock_ioctl) from [<c0443e84>] (sys_ioctl+0x280/0x8c8)
[   68.248233]  r9:be9d5a24 r8:00000000 r7:c3349840 r6:00008914 r5:c3349840 r4:c1205048
[   68.248250] [<c0443c04>] (sys_ioctl) from [<c0200040>] (ret_fast_syscall+0x0/0x28)
[   68.248261] Exception stack(0xc3ce7fa8 to 0xc3ce7ff0)
[   68.248275] 7fa0:                   0152cb08 00001043 00000009 00008914 be9d5a24 be9d5b00
[   68.248290] 7fc0: 0152cb08 00001043 0005dca0 00000036 431bde83 000f423f 3b9aca00 0152f158
[   68.248301] 7fe0: 0005dea4 be9d5a1c 0001ba6c b6ea251c
[   68.248316]  r10:00000036 r9:c3ce6000 r8:c0200204 r7:00000036 r6:0005dca0 r5:00001043
[   68.248326]  r4:0152cb08
[   68.248338] ---[ end trace 5e4455b751f391fa ]---
[   68.248355] lan743x_rx_init_ring_element dma_ptr check failed to map 9236 bytes @ 0xC33F4040
[   68.248411] lan743x 0000:01:00.0 eth1: lan743x_rx_open failed : -12, index : 0
[   68.265349] lan743x 0000:01:00.0 eth1: Error opening LAN743x
```

[... 666 lines cut ...]
```

**Additional context**
Though it may not be the complete solution this allows the module to work. Note line  numbers are likely off as this is a subset of the patch with a load of printks removed.

```
diff --git a/drivers/net/ethernet/microchip/lan743x_main.c b/drivers/net/ethernet/microchip/lan743x_main.c
index 8947c3a62..0bf25e03c 100644
--- a/drivers/net/ethernet/microchip/lan743x_main.c
+++ b/drivers/net/ethernet/microchip/lan743x_main.c
@@ -2318,6 +2322,13 @@ static int lan743x_rx_ring_init(struct lan743x_rx *rx)
                ret = -EINVAL;
                goto cleanup;
        }
+       
+       if (dma_set_mask_and_coherent(&rx->adapter->pdev->dev, DMA_BIT_MASK(64))) {
+               dev_warn(&rx->adapter->pdev->dev, "lan743x_: No suitable DMA available\n");
+        ret = -ENOMEM;
+        goto cleanup;
+    }
+       
        ring_allocation_size = ALIGN(rx->ring_size *
                                     sizeof(struct lan743x_rx_descriptor),
                                     PAGE_SIZE);
```

martinum4, 2021-02-28:
can confirm this bug, EVB-LAN7430 is not recognized correctly and needs fixing.
Unfortunately i'm too dumb to compile the kernel correctly and thus can't confirm the fix.

6by9, 2021-02-28:
The lan743x driver is unmodified from mainline so any real bugs ought to be reported upstream and fixed there.

That said, the patch feels like the default mask is incorrect as I wouldn't have expected individual drivers to need patching like this. @pelwell may know what those defaults will be/how it is initialised.

LPAE extensions do confuse matters significantly between 32 and 64 bit systems though. Does it work on the 64bit kernel?

6by9, 2021-02-28:
Then again there appear to be a number of drivers that do manually set the dma mask
- https://elixir.bootlin.com/linux/latest/source/drivers/net/ethernet/realtek/r8169_main.c#L5337
- https://elixir.bootlin.com/linux/latest/source/drivers/net/ethernet/intel/e1000/e1000_main.c#L1001
- https://elixir.bootlin.com/linux/latest/source/drivers/net/ethernet/broadcom/genet/bcmgenet.c#L4003 (the onboard Gigabit PHY for BCM2711)

martinum4, 2021-02-28:
> LPAE extensions do confuse matters significantly between 32 and 64 bit systems though. Does it work on the 64bit kernel?

Tried with Ubuntu Server ARM (uname -a -> Linux ubuntu 5.4.0-1028-raspi #31-Ubuntu SMP PREEMPT Wed Jan 20 11:30:45 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux), not working either.

mangodan2003, 2021-03-05:
> The lan743x driver is unmodified from mainline so any real bugs ought to be reported upstream and fixed there.
>
> That said, the patch feels like the default mask is incorrect as I wouldn't have [...]

Last I tried to boot a 64Bit kernel it hung waiting for the root device and I haven't tried again since. I will  so so when I get a spare moment.

mangodan2003, 2021-03-30:
I've now tested 5.10.17 64bit. Doesn't work out of the box. Works with the above fix applied.

aep, 2021-04-13:
patch works for me with 5.10.27-v8+

gnijieb, 2021-07-30:
Any updates on this issue? Is the above fix deemed good enough to be used in production?

Backporting all the lan743x changes from mainline to the rpi-5.10.y branch did not fix the issue.

Can confirm the issue and the fix.

pelwell, 2021-07-30:
As @6by9 rightly said:

> The lan743x driver is unmodified from mainline so any real bugs ought to be reported upstream and fixed there.

user-redans, 2022-08-23:
Hello,

I ' ve used this driver on my RPY kernel version:

> root@raspberrypi:~# uname -a
> Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux

And it's works !

But, when I 've tried to another RPI, with another kernel version: 

> raspberrypi:/home/pi# uname -a
Linux raspberrypi 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux

It doesn't work! System can 't find my module and, of course.... I can't start it with modprobe: 

> raspberrypi:/home/pi# modprobe lan743x
> modprobe: FATAL: Module lan743x not found in directory /lib/modules/5.15.32-v7l+
> 
> raspberrypi:/lib/modules/5.15.32-v7l+/kernel/drivers/net/ethernet/microchip# modinfo lan743x
> modinfo: ERROR: Module lan743x not found.
> raspberrypi:/lib/modules/5.15.32-v7l+/kernel/drivers/net/ethernet/microchip# ll
> total 84
> -rw-r--r-- 1 root root 32308 Mar 31 22:40 enc28j60.ko
> -rw-r--r-- 1 root root 52828 Jul 30 15:20 lan743x.ko

NOTE: That lan743x.ko is the same for both kernel versions.... Do I have to re-compile this patch ( lan743x_main.c file) for every versions?

pelwell, 2022-08-23:
> Do I have to re-compile this patch ( lan743x_main.c file) for every versions?

Yes, you do.

user-redans, 2022-08-23:
Ok, but how can I re-compile for a new kernel version?

When I compile with make command, I' ve received this: 

> raspberrypi:/home/pi/lan743x# make
> make: *** No targets.  Stop.

This is my lan743x directory:
> raspberrypi:/home/pi/lan743x# ll
> total 96
> -rw-r--r-- 1 pi pi  1494 May 26  2021 Kconfig
> -rw-r--r-- 1 pi pi 86947 May 26  2021 lan743x_main.c
> -rw-r--r-- 1 pi pi   317 Aug 23 11:09 Makefile

pelwell, 2022-08-23:
Microchip have done the decent thing and upstreamed the patch - see https://github.com/raspberrypi/linux/commit/95a359c9553342d36d408d35331ff0bfce75272f. The patch is in the current 5.15 kernel, which means you no longer need special patched versions.

closed 2022-08-23 by pelwell

user-redans, 2022-08-29:
Thanks for your suggestion pelwell, but I need to re-compile follow dir. files: 
```
-rw-r--r--  1 pi pi  1586 Aug 29 13:50 Kconfig
-rw-r--r--  1 pi pi 21261 Aug 23 12:20 lan743x_ethtool.c
-rw-r--r--  1 pi pi   261 Aug 23 12:20 lan743x_ethtool.h
-rw-r--r--  1 pi pi 87219 Aug 23 12:20 lan743x_main.c
-rw-r--r--  1 pi pi 27547 Aug 23 12:20 lan743x_main.h
-rw-r--r--  1 pi pi 35604 Aug 23 12:20 lan743x_ptp.c
-rw-r--r--  1 pi pi  2683 Aug 23 12:20 lan743x_ptp.h
-rw-r--r--  1 pi pi   362 Aug 29 14:04 Makefile
```
I need another make file or how can I compile those files?

I did an upgrade to my RPI distribution: 

```
sudo apt-get update
sudo apt-get dist-upgrade
sudo rpi-update
sudo reboot
```

 And now I ' m using a new kernel version:

```
root@raspberrypi:/home/pi/lan743x_v3# uname -a
Linux raspberrypi 5.15.61-v7l+ #1579 SMP Fri Aug 26 11:13:03 BST 2022 armv7l GNU/Linux
```

I thought that doing this upgrade, I will find lan743x.ko module in it.... already compiled...

pelwell, 2022-08-30:
I can see why you might have thought that, but that's not what I meant. It's surprising that nobody has requested that the LAN7430 be included in our kernel builds, but up to now that has been the case - I suppose it's a bit niche, being a PCIe device.

Commit 010d2dbf92ca6 adds the necessary config setting to the BCM2711 defconfigs. It will be a while before that change appears in a kernel build, but until then you can build your own kernel - see https://www.raspberrypi.com/documentation/computers/linux_kernel.html#building for detailed instructions.

user-redans, 2022-08-30:
Thanks for extra explications! From the beginning I didn't want to build my own kernel, because it's a job that will take a long time to do.... I have at least 30 new RPI 's to re-build it... with different kernels versions on many of them.... that's why I'm asking about a solution to build just LAN7430 driver, not entire kernel distribution and result... somehow.... that loadable kernel file : "lan743x.ko"

user-redans, 2022-09-08:
Thank you for all your advice and suggestions!  I've finished successfully this job!