raspberrypi/linux #4120: Kernel panic when serial device resets

benlye, 2021-02-05:
**Describe the bug**
I maintain a plugin for the 3D printer host software OctoPrint which is used to update printer firmware. One of the types of printer is updated by resetting the serial device. I have observed that resetting this serial device will periodically cause a kernel panic.

I have reproduced this problem on two different Pi 3 Model B including one which was a clean install of Raspbian, fully updated this morning.

**To reproduce**
Running this shell script with the printer board plugged in will eventually crash the Pi (typically after 5-15 resets):
```
stty -F /dev/ttyACM0 115200 -echo > /dev/null
i=0; while true; do echo $i; echo M997 >> /dev/ttyACM0; ((i=i+1)); sleep 15; done
```

'M997' is the command that triggers the board to reset.  It resets almost immediately on receiving the command.  My last test took 20 resets before it crashed but it's been as few as 1.  I can reproduce the problem at will by running the code above for a few minutes

The board in this case is a BTT SKR v1.4 Turbo, which has an LPC1769 MCU.  The board presents it's on-board SD card as a USB storage device and a serial port to the host.  I've had the same problem with other boards of a similar configuration.

**Expected behaviour**
A kernel panic should not occur after resetting the serial device

**Actual behaviour**
The system crashes with a kernel panic.

Kernel panic output captured from the serial console:
https://pastebin.com/zSiAAZCg

**System**
* raspinfo output
https://pastebin.com/FyhjskHU

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi 3 model B

* Which OS and version (`cat /etc/rpi-issue`)?
```
pi@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2021-01-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 21090519d85bdaa1615d5d5057d37b09368ea5d2, stage2
```

* Which firmware version (`vcgencmd version`)?
```
pi@raspberrypi:~ $ vcgencmd version
Jan 27 2021 22:26:53
Copyright (c) 2012 Broadcom
version 99d9a48302e4553cff3688692bb7e9ac760a03fa (clean) (release) (start)
```

* Which kernel version (`uname -a`)?
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.11-v7+ #1399 SMP Thu Jan 28 12:06:05 GMT 2021 armv7l GNU/Linux
```

pelwell, 2021-02-05:
I don't see any Pi-specific software in that stack trace - have you tried in a different Linux system such as a PC?

benlye, 2021-02-05:
I can set up a PC to try it on.  Any suggestions for best (most approximate to Raspbian) distribution to use?  Just a vanilla Buster install?

benlye, 2021-02-05:
Ubuntu 20.04 - 50 consecutive resets with no panics

benlye, 2021-02-05:
If building a debug kernel (or anything similar) would be useful, I'm happy to give it a go.  I'm willing to do whatever is helpful to dig into this, but I need some pointers.

benlye, 2021-02-05:
Another data point - I can also make the crash occur on a Pi 4 Model B.

kantlivelong, 2021-03-15:
Seeing this as well.