raspberrypi/linux #4125: Not working scripts for native compiling kernel modules

pmelikov, 2021-02-06:
**Describe the bug**
The `fixdep` command as provided in the armhf header file is a x86-64 version which cannot run on armhf and arm64. This break native compilation of kernel modules. I am compiled my own kernel end headers with preemt-rt patch. Used 
```
root@raspberrypi:/home/pi/Desktop/LEDG0_timer# file /usr/src/$(uname -r)/scripts/basic/fixdep
/usr/src/5.10.13-rt26-preemt_rt-v7+/scripts/basic/fixdep: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=fc5b93fa068bb02ca6fa1f1e2eeb99fa74a19c84, for GNU/Linux 3.2.0, not stripped
```
**CROSS COMPILE HOST**: Linux VirtualBox 5.8.0-41-generic #46~20.04.1-Ubuntu SMP Mon Jan 18 17:52:23 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
**DEVICE**: Pi3B+

I tryed to compile end test 4 different options :
- kernel 5.4 and 5.10 with this parameters + rt_preemt patch:
```
export KERNEL=kernel8
export ARCH=arm64
export CROSS_COMPILE=aarch64-linux-gnu-
export BIT_32=false
make -j$((`nproc`*2)) bcmrpi3_defconfig
make -j$((`nproc`*2))
```

- kernel 5.4 and 5.10 with this parameters + rt_preemt patch:
```
export KERNEL=kernel7
export ARCH=arm
export CROSS_COMPILE=arm-linux-gnueabihf-
export BIT_32=true
make -j$((`nproc`*2)) bcm2709_defconfig
make -j$((`nproc`*2))
```
Two of 4 tested configs:
kernel 5.10.13-rt26-v7+ with rt-preemt patch 5.10.12-rt26
kernel 5.4.83-v8+ with rt-preemt patch 5.4.84-rt47

As base SD image i used this and just put my headers, modules, dts and kernel into it:

- 2021-01-11-raspios-buster-armhf.zip
- 2020-08-20-raspios-buster-arm64.zip

```
env PATH=$PATH make -j$((`nproc`*2)) INSTALL_MOD_PATH=../mnt/ext4 modules_install
env PATH=$PATH make -j$((`nproc`*2)) INSTALL_HDR_PATH=../mnt/ext4 headers_instal
#for armhf 
mv ../mnt/fat32/kernel.img ../mnt/fat32/kernel-backup.img
cp arch/arm/boot/zImage ../mnt/fat32/kernel.img
cp arch/arm/boot/dts/*.dtb ../mnt/fat32/
cp arch/arm/boot/dts/overlays/*.dtb* ../mnt/fat32/overlays/
cp arch/arm/boot/dts/overlays/README ../mnt/fat32/overlays/

release="$(cat ./include/config/kernel.release)" #in linux dir
cp -R linux ./mnt/ext4/usr/src/
mv -R ./mnt/ext4/usr/src/linux ./mnt/ext4/usr/src/"$release" #rename sources
cd ./mnt/ext4/
ln -sr ./usr/src/"$release" ./lib/modules/"$release"/build
```

pmelikov, 2021-02-06:
Tryed this: https://forum.loverpi.com/discussion/555/how-to-fix-dkms-error-bin-sh-1-scripts-basic-fixdep-exec-format-error
But it didnt help. Same error.

pyavitz, 2021-02-07:
Cross Compiling
Simply applying the byteshift patch and running make scripts doesn't always work, I recommend trying the following.

```sh
make M=scripts clean
patch -p1 < headers-byteshift.patch
echo 'y' | make -s scripts
echo 'y' | make -s M=scripts/mod/
```

pmelikov, 2021-02-07:
Where can i find working "headers-byteshift.patch" for rpi?

pmelikov, 2021-02-07:
> Cross Compiling
> Simply applying the byteshift patch and running make scripts doesn't always work, I recommend trying the following.
> 
> ```shell
> make M=scripts clean
> patch -p1 < headers-byteshift.patch
> echo 'y' | make -s scripts
> echo 'y' | make -s M=scripts/mod/
> ```

I did not find where to get headers-byteshift.patch for rassberry, but even without a patch, when you run these three commands from raspberry, everything starts working. Thank you!
```
sudo apt-get -y install git bc bison flex libssl-dev make
make -s scripts
make -s M = scripts / mod /
```
Can I somehow compile these scripts with a crosscompiler? Or is this patch necessary for this? If it need a patch for this, then please tell me where to get it ...

pyavitz, 2021-02-08:
I originally got the patch from Armbian, but I don't believe they created
it and there is no official patch for the Pi.
https://raw.githubusercontent.com/pyavitz/rpi-img-builder/master/patches/headers-byteshift.patch

I created packaging patches for kernel builds and included it in the build
process.
You can review the patches here:
https://github.com/pyavitz/rpi-img-builder/tree/master/patches/cc

pmelikov, 2021-02-10:
> Cross Compiling
> 
> Simply applying the byteshift patch and running make scripts doesn't always work, I recommend trying the following.
> 
> 
> 
> ```sh
> 
> make M=scripts clean
> 
> patch -p1 < headers-byteshift.patch
> 
> echo 'y' | make -s scripts
> 
> echo 'y' | make -s M=scripts/mod/
> 
> ```

Thanks a lot, my friend. Armbian patch and making of "scripts" and "scripts/mod" solved the problem. All the best to you🙋‍♂️. Ok, later I will research your repo🤓.

closed 2021-02-10 by pmelikov