raspberrypi/linux #4127: Upgrade to 5.10.11+ breaks wlan driver

gs4711, 2021-02-07:
Linux pi1ModBplus  5.10.11+ #1399 Thu Jan 28 12:02:28 GMT 2021 armv6l GNU/Linux

[    5.430438] usb 1-1.5: new high-speed USB device number 6 using dwc_otg
[    5.563077] usb 1-1.5: New USB device found, idVendor=7392, idProduct=7811, bcdDevice= 2.00
[    5.568183] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    5.573109] usb 1-1.5: Product: 802.11n WLAN Adapter
[    5.575617] usb 1-1.5: Manufacturer: Realtek
[    5.578006] usb 1-1.5: SerialNumber: 00e04c000001

no wlan0

pelwell, 2021-02-07:
What did you upgrade from? In other words, how far back might we have to trawl for the offending commit?

gs4711, 2021-02-07:
Coming from:

`Linux version 4.19.66+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1253 Thu Aug 15 11:37:30 BST 2019
`

having upgraded to:

`Linux version 5.10.11+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1399 Thu Jan 28 12:02:28 GMT 2021
`

The recent boot log has been:

```
Feb  6 13:17:13 home-pi02 kernel: [    0.000000] Booting Linux on physical CPU 0x0
Feb  6 13:17:13 home-pi02 kernel: [    0.000000] Linux version 4.19.66+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1253 Thu Aug 15 11:37:30 BST 2019
Feb  6 13:17:13 home-pi02 kernel: [    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
Feb  6 13:17:13 home-pi02 kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Feb  6 13:17:13 home-pi02 kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Model B Plus Rev 1.2
Feb  6 13:17:13 home-pi02 kernel: [    0.000000] Memory policy: Data cache writeback
Feb  6 13:17:13 home-pi02 kernel: [    0.000000] cma: Reserved 8 MiB at 0x1e400000
Feb  6 13:17:13 home-pi02 kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x49c with crng_init=0
Feb  6 13:17:13 home-pi02 kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 125860
Feb  6 13:17:13 home-pi02 kernel: [    0.000000] Kernel command line: coherent_pool=1M bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x1fa00000 v
c_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Feb  6 13:17:13 home-pi02 kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Feb  6 13:17:13 home-pi02 kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Feb  6 13:17:13 home-pi02 kernel: [    0.000000] Memory: 483144K/507904K available (6971K kernel code, 635K rwdata, 2080K rodata, 468K init, 797K bss, 16568K reserved, 8192K cma-reserved)
Feb  6 13:17:13 home-pi02 kernel: [    0.000000] Virtual kernel memory layout:
Feb  6 13:17:13 home-pi02 kernel: [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
Feb  6 13:17:13 home-pi02 kernel: [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
Feb  6 13:17:13 home-pi02 kernel: [    0.000000]     vmalloc : 0xdf800000 - 0xff800000   ( 512 MB)
Feb  6 13:17:13 home-pi02 kernel: [    0.000000]     lowmem  : 0xc0000000 - 0xdf000000   ( 496 MB)
Feb  6 13:17:13 home-pi02 kernel: [    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
Feb  6 13:17:13 home-pi02 kernel: [    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (6972 kB)
Feb  6 13:17:13 home-pi02 kernel: [    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   ( 468 kB)
Feb  6 13:17:13 home-pi02 kernel: [    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 636 kB)
Feb  6 13:17:13 home-pi02 kernel: [    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 798 kB)
Feb  6 13:17:13 home-pi02 kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Feb  6 13:17:13 home-pi02 kernel: [    0.000000] ftrace: allocating 25191 entries in 74 pages
Feb  6 13:17:13 home-pi02 kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Feb  6 13:17:13 home-pi02 kernel: [    0.000031] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
Feb  6 13:17:13 home-pi02 kernel: [    0.000095] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
Feb  6 13:17:13 home-pi02 kernel: [    0.000220] bcm2835: system timer (irq = 27)
Feb  6 13:17:13 home-pi02 kernel: [    0.000645] Console: colour dummy device 80x30
Feb  6 13:17:13 home-pi02 kernel: [    0.001275] console [tty1] enabled
Feb  6 13:17:13 home-pi02 kernel: [    0.001358] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
Feb  6 13:17:13 home-pi02 kernel: [    0.060396] pid_max: default: 32768 minimum: 301
Feb  6 13:17:13 home-pi02 kernel: [    0.060976] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Feb  6 13:17:13 home-pi02 kernel: [    0.061034] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Feb  6 13:17:13 home-pi02 kernel: [    0.062483] CPU: Testing write buffer coherency: ok
Feb  6 13:17:13 home-pi02 kernel: [    0.063884] Setting up static identity map for 0x8200 - 0x8238
Feb  6 13:17:13 home-pi02 kernel: [    0.065376] devtmpfs: initialized
Feb  6 13:17:13 home-pi02 kernel: [    0.076029] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
Feb  6 13:17:13 home-pi02 kernel: [    0.076520] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Feb  6 13:17:13 home-pi02 kernel: [    0.076598] futex hash table entries: 256 (order: -1, 3072 bytes)
Feb  6 13:17:13 home-pi02 kernel: [    0.077976] pinctrl core: initialized pinctrl subsystem
Feb  6 13:17:13 home-pi02 kernel: [    0.079576] NET: Registered protocol family 16
Feb  6 13:17:13 home-pi02 kernel: [    0.082804] DMA: preallocated 1024 KiB pool for atomic coherent allocations
Feb  6 13:17:13 home-pi02 kernel: [    0.090269] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[... cut ...]
Feb  7 11:29:29 home-pi02 kernel: [   23.372107] smsc95xx 1-1.1:1.0 eth0: Link is Up - 100Mbps/Full - flow control off
Feb  7 11:29:29 home-pi02 kernel: [   23.372207] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Feb  7 11:29:29 home-pi02 kernel: [   23.687833] Adding 4194300k swap on /var/swap.  Priority:-2 extents:6 across:5439488k SSFS
Feb  7 11:29:29 home-pi02 kernel: [   23.971127] random: crng init done
Feb  7 11:29:29 home-pi02 kernel: [   23.971162] random: 1 urandom warning(s) missed due to ratelimiting
```

clivem, 2021-02-07:
The (vendor) 8192cu.ko driver was removed from the Pi 5.10.y tree. Remove the blacklist of the rtl8192cu.ko module from whatever  file in /etc/modprobe.d has blacklisted it, reboot and see how you get on with that. (It's an inferior driver IMHO, but should at least get you back up and running.) I personally still have a bunch of Edimax EW-7811Un adapters in-use with Pi2's, so now build the vendor 8192cu.ko driver out-of-tree. YMMV.

gs4711, 2021-02-09:
Great. This has been the solution!

closed 2021-02-09 by gs4711