raspberrypi/linux #4128: TC358743 - crash (or NULL pointer dereference) on cable plugging

mdevaev, 2021-02-07:
**Describe the bug**
If you connect the cable while the ustreamer is running, the driver may crash. In rarer cases, with the same reproducing method - NULL pointer dereference. Perhaps the nature of both of these bugs is the same, so this is one issue. The bug is not related to image encoding, just getting a RAW stream is enough.

```
[  228.324204] ------------[ cut here ]------------
[  228.328944] WARNING: CPU: 3 PID: 612 at drivers/media/common/videobuf2/videobuf2-core.c:1945 __vb2_queue_cancel+0x1d0/0x24c [videobuf2_common]
[  228.341870] Modules linked in: usb_f_mass_storage usb_f_hid spidev btsdio bluetooth ecdh_generic ecc brcmfmac snd_soc_spdif_rx brcmutil snd_soc_simple_card snd_soc_simple_card_utils i2c_mux_pinctrl i2c_mux cfg80211 snd_soc_bcm2835_i2s raspberrypi_hwmon bcm2835_unicam snd_soc_core snd_compress snd_pcm_dmaengine i2c_bcm2835 spi_bcm2835 bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 iproc_rng200 videobuf2_common rng_core rpivid_mem nvmem_rmem uio_pdrv_genirq uio sch_fq_codel i2c_dev snd_bcm2835(C) snd_pcm snd_timer snd tc358743 v4l2_dv_timings v4l2_fwnode videodev mc cec libcomposite dwc2 udc_core roles drm fuse drm_panel_orientation_quirks ip_tables x_tables ipv6
[  228.410770] CPU: 3 PID: 612 Comm: stream Tainted: G         C        5.10.13-2-ARCH #1
[  228.418762] Hardware name: BCM2711
[  228.422221] [<c020f0a0>] (unwind_backtrace) from [<c020ae7c>] (show_stack+0x10/0x14)
[  228.430044] [<c020ae7c>] (show_stack) from [<c0dd60dc>] (dump_stack+0xa4/0xc4)
[  228.437342] [<c0dd60dc>] (dump_stack) from [<c022ae0c>] (__warn+0xec/0x148)
[  228.444372] [<c022ae0c>] (__warn) from [<c0dce37c>] (warn_slowpath_fmt+0x7c/0xac)
[  228.453323] [<c0dce37c>] (warn_slowpath_fmt) from [<bf278bf0>] (__vb2_queue_cancel+0x1d0/0x24c [videobuf2_common])
[  228.466613] [<bf278bf0>] (__vb2_queue_cancel [videobuf2_common]) from [<bf279548>] (vb2_core_streamoff+0x18/0xac [videobuf2_common])
[  228.481555] [<bf279548>] (vb2_core_streamoff [videobuf2_common]) from [<bf1d5474>] (__video_do_ioctl+0x238/0x490 [videodev])
[  228.495854] [<bf1d5474>] (__video_do_ioctl [videodev]) from [<bf1d6354>] (video_usercopy+0x170/0x5ac [videodev])
[  228.509073] [<bf1d6354>] (video_usercopy [videodev]) from [<c043ca78>] (sys_ioctl+0x34c/0x904)
[  228.520652] [<c043ca78>] (sys_ioctl) from [<c0200040>] (ret_fast_syscall+0x0/0x4c)
[  228.531168] Exception stack(0xc55e5fa8 to 0xc55e5ff0)
[  228.537691] 5fa0:                   b68de46c 00000004 00000008 40045613 b68dd968 01147638
[  228.548766] 5fc0: b68de46c 00000004 00000000 00000036 40045613 0054c35c b68dd968 00000008
[  228.559827] 5fe0: 0054c11c b68dd944 004f99bc b6c209fc
[  228.566357] ---[ end trace e70ee86b0b0a292c ]---
[  228.572403] videobuf2_common: driver bug: stop_streaming operation is leaving buf fed1a223 in active state
```

**To reproduce**
At the moment, it can't be reproduced on Raspbian, as it has not yet built the 5.10.13 kernel and cannot be installed with rpi-update. However the approximate method looks like this:
* Install [ustreamer](https://github.com/pikvm/ustreamer)
* Load some [EDID](https://github.com/pikvm/kvmd/blob/master/configs/kvmd/tc358743-edid.hex) to the tc358743 device.
* Run `ustreamer --dv-timings --device /dev/kvmd-video --persistent --format uyvy --encoder=noop`
* Plug HDMI
* Stop ustreamer
* Unplug HDMI
* Start ustreamer
* Plug HDMI
* ...
* Error

The error may occur after the second or twentieth iteration, it is non-periodic, but it occurs when the cable is connected. In rarer cases, some kind of NULL pointer dereference may occur, but I have not yet managed to catch it in the log, I only saw it briefly in the UART console three times, after which the Pi rebooted. In even rarer cases, something strange happens like double free or 'munmap_chunk(): invalid pointer'. I suppose it's all one problem.

On older 5.4 kernels, I didn't see this. Interestingly, if you now try to reproduce the problem on Raspbian with the kernel 5.10.11, the first connection of the cable will cause Segfault in ustreamer, this was also not observed before.

If you need any more information, I'm ready to help as usual.

**System**
* Raspberry Pi 4 Model B Rev 1.1
* Arch Linux ARM
* firmware: `version 99d9a48302e4553cff3688692bb7e9ac760a03fa (clean) (release) (start) / Jan 27 2021 22:19:57`
* `Linux pikvm 5.10.13-2-ARCH #1 SMP Fri Feb 5 15:47:26 UTC 2021 armv7l GNU/Linux`

**Logs**
Just dmesg: [dmesg1.txt](https://github.com/raspberrypi/linux/files/5939397/dmesg1.txt)
Another crash (UART log): 
[screenlog1.txt](https://github.com/raspberrypi/linux/files/5939401/screenlog1.txt)

mdevaev, 2021-02-07:
@6by9 In ancient times, ambassadors with bad news could be killed, but I hope you will spare me in this time :) I brought another nasty bug related to TC358743.

6by9, 2021-02-07:
__vb2_queue_cancel is a warning, not a crash. See the comment at https://elixir.bootlin.com/linux/latest/source/drivers/media/common/videobuf2/videobuf2-core.c#L1939 just before it.

I will look at it again as we should return all queued buffers, but it won't crash the system.

mdevaev, 2021-02-07:
Thanks. In this case, I will try to reproduce the NULL pointer dereference.

mdevaev, 2021-02-07:
Catched NULL pointer dereference:
```
[  113.910609] Unable to handle kernel NULL pointer dereference at virtual address 00000122
[  113.918751] pgd = c2ff1852
[  113.921470] [00000122] *pgd=80000000004003, *pmd=00000000
[  113.926905] Internal error: Oops: a06 [#1] SMP ARM

Entering kdb (current=0xc3fada00, pid 140) on processor 0 Oops: (null)
due to oops @ 0xbf2e06e0
CPU: 0 PID: 140 Comm: kworker/0:3 Tainted: G         C        5.10.13-2-ARCH #1
Hardware name: BCM2711
Workqueue: events dbs_work_handler
PC is at unicam_isr+0x2bc/0x360 [bcm2835_unicam]
LR is at 0x122
pc : [<bf2e06e0>]    lr : [<00000122>]    psr: 60000193
sp : c3fa1d10  ip : c25f5e80  fp : eece3c10
r10: eece3000  r9 : 000004fc  r8 : 00000100
r7 : 00000608  r6 : 00000000  r5 : eece34fc  r4 : eece3000
r3 : c25f6280  r2 : 00000122  r1 : 00000000  r0 : c25f6000
Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
Control: 30c5383d  Table: 02947280  DAC: fffffffd
CPU: 0 PID: 140 Comm: kworker/0:3 Tainted: G         C        5.10.13-2-ARCH #1
Hardware name: BCM2711
Workqueue: events dbs_work_handler
[<c020f0a0>] (unwind_backtrace) from [<c020ae7c>] (show_stack+0x10/0x14)
[<c020ae7c>] (show_stack) from [<c0dd60dc>] (dump_stack+0xa4/0xc4)
[<c0dd60dc>] (dump_stack) from [<c02ffd50>] (kdb_main_loop+0x4bc/0x9a0)
[<c02ffd50>] (kdb_main_loop) from [<c0302e04>] (kdb_stub+0x258/0x458)
[<c0302e04>] (kdb_stub) from [<c02f7f90>] (kgdb_cpu_enter+0x3e0/0x6fc)
```

And also this, similar as discribed in the first report but with another trace:
```
[   76.559135] ------------[ cut here ]------------
[   76.563844] WARNING: CPU: 3 PID: 541 at drivers/media/common/videobuf2/videobuf2-core.c:997 unicam_return_buffers+0x84/0xd0 [bcm2835_unicam]
[   76.576556] Modules linked in: usb_f_mass_storage usb_f_hid spidev btsdio bluetooth snd_soc_spdif_rx snd_soc_simple_card ecdh_generic snd_soc_simple_card_utils ecc brcmfmac i2c_mux_pinctrl brcmutil i2c_mux raspberrypi_hwmon bcm2835_unicam cfg80211 snd_soc_bcm2835_i2s snd_soc_core bcm2835_v4l2(C) snd_compress i2c_bcm2835 bcm2835_codec(C) snd_pcm_dmaengine spi_bcm2835 bcm2835_isp(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 iproc_rng200 videobuf2_common rng_core rpivid_mem uio_pdrv_genirq nvmem_rmem uio sch_fq_codel i2c_dev snd_bcm2835(C) snd_pcm snd_timer snd tc358743 v4l2_dv_timings v4l2_fwnode videodev mc cec libcomposite dwc2 udc_core roles drm fuse drm_panel_orientation_quirks ip_tables x_tables ipv6
[   76.645367] CPU: 3 PID: 541 Comm: stream Tainted: G         C        5.10.13-2-ARCH #1
[   76.653302] Hardware name: BCM2711
[   76.656722] [<c020f0a0>] (unwind_backtrace) from [<c020ae7c>] (show_stack+0x10/0x14)
[   76.664490] [<c020ae7c>] (show_stack) from [<c0dd60dc>] (dump_stack+0xa4/0xc4)
[   76.671734] [<c0dd60dc>] (dump_stack) from [<c022ae0c>] (__warn+0xec/0x148)
[   76.678716] [<c022ae0c>] (__warn) from [<c0dce37c>] (warn_slowpath_fmt+0x7c/0xac)
[   76.686848] [<c0dce37c>] (warn_slowpath_fmt) from [<bf30c72c>] (unicam_return_buffers+0x84/0xd0 [bcm2835_unicam])
[   76.698427] [<bf30c72c>] (unicam_return_buffers [bcm2835_unicam]) from [<bf214a48>] (__vb2_queue_cancel+0x28/0x24c [videobuf2_common])
[   76.711830] [<bf214a48>] (__vb2_queue_cancel [videobuf2_common]) from [<bf215548>] (vb2_core_streamoff+0x18/0xac [videobuf2_common])
[   76.725079] [<bf215548>] (vb2_core_streamoff [videobuf2_common]) from [<bf1d5474>] (__video_do_ioctl+0x238/0x490 [videodev])
[   76.737669] [<bf1d5474>] (__video_do_ioctl [videodev]) from [<bf1d6354>] (video_usercopy+0x170/0x5ac [videodev])
[   76.749188] [<bf1d6354>] (video_usercopy [videodev]) from [<c043ca78>] (sys_ioctl+0x34c/0x904)
[   76.759122] [<c043ca78>] (sys_ioctl) from [<c0200040>] (ret_fast_syscall+0x0/0x4c)
[   76.768026] Exception stack(0xc5599fa8 to 0xc5599ff0)
[   76.773747] 9fa0:                   b16f9474 00000004 0000000c 40045613 b16f8960 009ad660
[   76.783251] 9fc0: b16f9474 00000004 00000000 00000036 40045613 0048e06c b16f8960 0000000c
[   76.792767] 9fe0: 0048de18 b16f893c 00439c08 b6bcd9fc
[   76.798490] ---[ end trace 599a6883e85eaf43 ]---
```

I use a slightly different command: here, jpeg encoding is enabled for omx and h264 encoding for mmal using zerocopy from the video device. But it seems to me that this is not related, because I added oops and stacktrace as above (not warnings) and without these options.

`ustreamer --dv-timings --device /dev/video0 --persistent --format=uyvy --encoder=omx --h264-sink test`

mdevaev, 2021-02-08:
@6by9 I continued testing attempts and realized that NULL dereference (described in the previous comment) happens stably for about 10-20 attempts to connect the cable. Next, the kernel runs kdbg, but I can't do anything there since the kernel is rebooted after that (probably Arch settings). The command causing the crash is also in the previous comment.

mdevaev, 2021-02-13:
Another catch on zero. Crash and warning:
```
Feb 13 08:07:26 pikvm kernel: ------------[ cut here ]------------
Feb 13 08:07:26 pikvm kernel: WARNING: CPU: 0 PID: 128 at drivers/media/common/videobuf2/videobuf2-core.c:997 unicam_isr+0xdc/0x36c [bcm2835_unicam]
Feb 13 08:07:26 pikvm kernel: Modules linked in: usb_f_mass_storage usb_f_hid brcmfmac brcmutil cfg80211 raspberrypi_hwmon i2c_mux_pinctrl i2c_mux bcm2835_unicam bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(>
Feb 13 08:07:26 pikvm kernel: CPU: 0 PID: 128 Comm: kworker/u2:2 Tainted: G         C        5.10.14-2-ARCH #1
Feb 13 08:07:26 pikvm kernel: Hardware name: BCM2835
Feb 13 08:07:26 pikvm kernel: Workqueue:  0x0 (brcmf_wq/mmc1:0001:1)
Feb 13 08:07:26 pikvm kernel: [<8010ef8c>] (unwind_backtrace) from [<8010ae04>] (show_stack+0x10/0x14)
Feb 13 08:07:26 pikvm kernel: [<8010ae04>] (show_stack) from [<80d03c88>] (dump_stack+0xd0/0xf8)
Feb 13 08:07:26 pikvm kernel: [<80d03c88>] (dump_stack) from [<8012bd1c>] (__warn+0xd4/0x140)
Feb 13 08:07:26 pikvm kernel: [<8012bd1c>] (__warn) from [<80cfb694>] (warn_slowpath_fmt+0x74/0xa8)
Feb 13 08:07:26 pikvm kernel: [<80cfb694>] (warn_slowpath_fmt) from [<7f2bae4c>] (unicam_isr+0xdc/0x36c [bcm2835_unicam])
Feb 13 08:07:26 pikvm kernel: [<7f2bae4c>] (unicam_isr [bcm2835_unicam]) from [<8018eb18>] (__handle_irq_event_percpu+0x48/0x1ec)
Feb 13 08:07:26 pikvm kernel: [<8018eb18>] (__handle_irq_event_percpu) from [<8018eda8>] (handle_irq_event+0x64/0xd8)
Feb 13 08:07:26 pikvm kernel: [<8018eda8>] (handle_irq_event) from [<8019334c>] (handle_level_irq+0xb0/0x1a0)
Feb 13 08:07:26 pikvm kernel: [<8019334c>] (handle_level_irq) from [<8018e430>] (__handle_domain_irq+0x74/0xd4)
Feb 13 08:07:26 pikvm kernel: [<8018e430>] (__handle_domain_irq) from [<8089e644>] (bcm2835_handle_irq+0x2c/0x34)
Feb 13 08:07:26 pikvm kernel: [<8089e644>] (bcm2835_handle_irq) from [<80100af8>] (__irq_svc+0x58/0x74)
Feb 13 08:07:26 pikvm kernel: Exception stack(0x84367e68 to 0x84367eb0)
Feb 13 08:07:26 pikvm kernel: 7e60:                   8306cb00 40000700 0000fde3 00000000 81266b80 00000000
Feb 13 08:07:26 pikvm kernel: 7e80: 8306cb00 9b6d2b80 1a46c000 84220f00 00000000 84367ef4 846e8040 84367eb8
Feb 13 08:07:26 pikvm kernel: 7ea0: 80d0b0d8 80157940 60000013 ffffffff
Feb 13 08:07:26 pikvm kernel: [<80100af8>] (__irq_svc) from [<80157940>] (finish_task_switch+0x7c/0x218)
Feb 13 08:07:26 pikvm kernel: [<80157940>] (finish_task_switch) from [<80d0b0d8>] (__schedule+0x2e4/0xa54)
Feb 13 08:07:26 pikvm kernel: [<80d0b0d8>] (__schedule) from [<80d0b8a8>] (schedule+0x60/0xdc)
Feb 13 08:07:26 pikvm kernel: [<80d0b8a8>] (schedule) from [<80147e74>] (worker_thread+0xcc/0x594)
Feb 13 08:07:26 pikvm kernel: [<80147e74>] (worker_thread) from [<8014dca8>] (kthread+0x130/0x144)
Feb 13 08:07:26 pikvm kernel: [<8014dca8>] (kthread) from [<80100138>] (ret_from_fork+0x14/0x3c)
Feb 13 08:07:26 pikvm kernel: Exception stack(0x84367fb0 to 0x84367ff8)
Feb 13 08:07:26 pikvm kernel: 7fa0:                                     00000000 00000000 00000000 00000000
Feb 13 08:07:26 pikvm kernel: 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Feb 13 08:07:26 pikvm kernel: 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
Feb 13 08:07:26 pikvm kernel: ---[ end trace 1e400d33e83b8123 ]---
...
Feb 13 08:08:06 pikvm kernel: ------------[ cut here ]------------
Feb 13 08:08:06 pikvm kernel: WARNING: CPU: 0 PID: 459 at drivers/media/common/videobuf2/videobuf2-core.c:1945 __vb2_queue_cancel+0x1cc/0x250 [videobuf2_common]
Feb 13 08:08:06 pikvm kernel: Modules linked in: usb_f_mass_storage usb_f_hid brcmfmac brcmutil cfg80211 raspberrypi_hwmon i2c_mux_pinctrl i2c_mux bcm2835_unicam bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(>
Feb 13 08:08:06 pikvm kernel: CPU: 0 PID: 459 Comm: stream Tainted: G        WC        5.10.14-2-ARCH #1
Feb 13 08:08:06 pikvm kernel: Hardware name: BCM2835
Feb 13 08:08:06 pikvm kernel: [<8010ef8c>] (unwind_backtrace) from [<8010ae04>] (show_stack+0x10/0x14)
Feb 13 08:08:06 pikvm kernel: [<8010ae04>] (show_stack) from [<80d03c88>] (dump_stack+0xd0/0xf8)
Feb 13 08:08:06 pikvm kernel: [<80d03c88>] (dump_stack) from [<8012bd1c>] (__warn+0xd4/0x140)
Feb 13 08:08:06 pikvm kernel: [<8012bd1c>] (__warn) from [<80cfb694>] (warn_slowpath_fmt+0x74/0xa8)
Feb 13 08:08:06 pikvm kernel: [<80cfb694>] (warn_slowpath_fmt) from [<7f259a94>] (__vb2_queue_cancel+0x1cc/0x250 [videobuf2_common])
Feb 13 08:08:06 pikvm kernel: [<7f259a94>] (__vb2_queue_cancel [videobuf2_common]) from [<7f25a3f4>] (vb2_core_streamoff+0x18/0xac [videobuf2_common])
Feb 13 08:08:06 pikvm kernel: [<7f25a3f4>] (vb2_core_streamoff [videobuf2_common]) from [<7f1a0570>] (__video_do_ioctl+0x228/0x474 [videodev])
Feb 13 08:08:06 pikvm kernel: [<7f1a0570>] (__video_do_ioctl [videodev]) from [<7f1a14b8>] (video_usercopy+0x148/0x598 [videodev])
Feb 13 08:08:06 pikvm kernel: [<7f1a14b8>] (video_usercopy [videodev]) from [<8033aa68>] (sys_ioctl+0x32c/0x92c)
Feb 13 08:08:06 pikvm kernel: [<8033aa68>] (sys_ioctl) from [<80100060>] (ret_fast_syscall+0x0/0x4c)
Feb 13 08:08:06 pikvm kernel: Exception stack(0x833c7fa8 to 0x833c7ff0)
Feb 13 08:08:06 pikvm kernel: 7fa0:                   72600b80 00000004 00000009 40045613 737f8960 02087798
Feb 13 08:08:06 pikvm kernel: 7fc0: 72600b80 00000004 00000000 00000036 40045613 004df06c 737f8960 00000009
Feb 13 08:08:06 pikvm kernel: 7fe0: 004dee18 737f893c 00489bf8 76c9a92c
Feb 13 08:08:06 pikvm kernel: ---[ end trace 1e400d33e83b8124 ]---
Feb 13 08:08:06 pikvm kernel: videobuf2_common: driver bug: stop_streaming operation is leaving buf cbe27ef1 in active state
```

mdevaev, 2021-02-15:
@6by9 Is there anything I can do to help solve this bug? I get pecked at the head because of the NULL dereference.

mdevaev, 2021-03-02:
Sup?

6by9, 2021-03-02:
Sorry, still a way down the priority list.

@naushir It looks like there is a path where the unicam driver can return a buffer that isn't queued to it. That's what the videobuf2-core.c:997 warning is about.
TC358743 has no metadata pad, so it should only be the image pad active. It's as if there is a race between unicam_isr and unicam_stop_streaming.

Is it just that we've forgotten to clear UNICAM_ICTL, so the interrupts are still active?

mdevaev, 2021-03-02:
@6by9 In this situation, I'm more concerned about the NULL dereference rather than the warning that occurs. Apparently you're right and it's a race, given the diverse range of errors. Most of all, I would like to get rid of crash. Warning is a much lesser evil.

naushir, 2021-03-03:
The only way I can see this problem occurring is if `unicam_isr` were to be preempted by `unicam_stop_streaming` but can that really happen??  If so, here is a possible (and entirely untested) fix:

```
diff --git a/drivers/media/platform/bcm2835/bcm2835-unicam.c b/drivers/media/platform/bcm2835/bcm2835-unicam.c
index a8b3f5433f04..5496f93aa43c 100644
--- a/drivers/media/platform/bcm2835/bcm2835-unicam.c
+++ b/drivers/media/platform/bcm2835/bcm2835-unicam.c
@@ -831,10 +831,12 @@ static irqreturn_t unicam_isr(int irq, void *dev)
                        if (!unicam->node[i].streaming)
                                continue;
 
+                       spin_lock(&unicam->node[i].dma_queue_lock);
                        if (unicam->node[i].cur_frm)
                                unicam_process_buffer_complete(&unicam->node[i],
                                                               sequence);
                        unicam->node[i].cur_frm = unicam->node[i].next_frm;
+                       spin_unlock(&unicam->node[i].dma_queue_lock);
                }
                unicam->sequence++;
        }
@@ -849,6 +851,7 @@ static irqreturn_t unicam_isr(int irq, void *dev)
                        if (!unicam->node[i].streaming)
                                continue;
 
+                       spin_lock(&unicam->node[i].dma_queue_lock);
                        if (unicam->node[i].cur_frm)
                                unicam->node[i].cur_frm->vb.vb2_buf.timestamp =
                                                                ts;
@@ -858,6 +861,7 @@ static irqreturn_t unicam_isr(int irq, void *dev)
                         * from the queue.
                         */
                        unicam_schedule_dummy_buffer(&unicam->node[i]);
+                       spin_unlock(&unicam->node[i].dma_queue_lock);
                }
 
                unicam_queue_event_sof(unicam);
```

mdevaev, 2021-03-03:
@naushir Well, yes, I have reproduced it stably, as you can see :) Can your patch fix NULL dereference, do you think?

naushir, 2021-03-03:
The patch above should fix *a potential* NULL deference, but it may or may not be the one that you have seen triggered, it's difficult to tell.  Please do give it a try and let me know.

mdevaev, 2021-03-03:
@naushir A potential fix is much better than no fix :) Thank you! I'll take care of it immediately.

mdevaev, 2021-03-03:
@naushir Got this:
```
[  151.372449] ------------[ cut here ]------------
[  151.377155] WARNING: CPU: 0 PID: 652 at drivers/media/common/videobuf2/videobuf2-core.c:997 unicam_return_buffers+0x58/0xd0 [bcm2835_unicam]
[  151.389868] Modules linked in: usb_f_mass_storage usb_f_hid spidev btsdio snd_soc_spdif_rx bluetooth ecdh_generic ecc snd_soc_simple_card brcmfmac
snd_soc_simple_card_utils brcmutil i2c_mux_pinctrl i2c_mux bcm2835_unicam snd_soc_bcm2835_i2s cfg80211 raspberrypi_hwmon snd_soc_core i2c_bcm2835 snd_
compress spi_bcm2835 snd_pcm_dmaengine bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_dma_contig videobu
f2_vmalloc videobuf2_memops iproc_rng200 videobuf2_v4l2 rng_core videobuf2_common rpivid_mem nvmem_rmem uio_pdrv_genirq uio sch_fq_codel i2c_dev snd_b
cm2835(C) snd_pcm snd_timer snd tc358743 v4l2_dv_timings v4l2_fwnode videodev mc cec libcomposite dwc2 udc_core roles drm fuse drm_panel_orientation_q
uirks ip_tables x_tables ipv6
[  151.458693] CPU: 0 PID: 652 Comm: stream Tainted: G         C        5.10.17-2-ARCH #1
[  151.466677] Hardware name: BCM2711
[  151.470132] [<c020f10c>] (unwind_backtrace) from [<c020aed0>] (show_stack+0x10/0x14)
[  151.477954] [<c020aed0>] (show_stack) from [<c0de7864>] (dump_stack+0xa4/0xc4)
[  151.485250] [<c0de7864>] (dump_stack) from [<c022ae9c>] (__warn+0xec/0x148)
[  151.492278] [<c022ae9c>] (__warn) from [<c0ddfb00>] (warn_slowpath_fmt+0x7c/0xac)
[  151.501226] [<c0ddfb00>] (warn_slowpath_fmt) from [<bf54e700>] (unicam_return_buffers+0x58/0xd0 [bcm2835_unicam])
[  151.514436] [<bf54e700>] (unicam_return_buffers [bcm2835_unicam]) from [<bf20fa48>] (__vb2_queue_cancel+0x28/0x24c [videobuf2_common])
[  151.529507] [<bf20fa48>] (__vb2_queue_cancel [videobuf2_common]) from [<bf210548>] (vb2_core_streamoff+0x18/0xac [videobuf2_common])
[  151.544455] [<bf210548>] (vb2_core_streamoff [videobuf2_common]) from [<bf1d5474>] (__video_do_ioctl+0x238/0x490 [videodev])
[  151.558797] [<bf1d5474>] (__video_do_ioctl [videodev]) from [<bf1d6354>] (video_usercopy+0x170/0x5ac [videodev])
[  151.572027] [<bf1d6354>] (video_usercopy [videodev]) from [<c043df50>] (sys_ioctl+0x34c/0x904)
[  151.583625] [<c043df50>] (sys_ioctl) from [<c0200040>] (ret_fast_syscall+0x0/0x4c)
[  151.594146] Exception stack(0xc530ffa8 to 0xc530fff0)
[  151.600672] ffa0:                   b0505000 00000004 0000000c 40045613 b16f8960 01d67648
[  151.611754] ffc0: b0505000 00000004 00000000 00000036 40045613 0048f06c b16f8960 0000000c
[  151.622822] ffe0: 0048ee1c b16f893c 00439c08 b6ba59fc
[  151.629321] ---[ end trace 90a2d269f970e9d2 ]---
[  151.635469] ------------[ cut here ]------------
[  151.641586] WARNING: CPU: 0 PID: 652 at drivers/media/common/videobuf2/videobuf2-core.c:1945 __vb2_queue_cancel+0x1d0/0x24c [videobuf2_common]
[  151.656335] Modules linked in: usb_f_mass_storage usb_f_hid spidev btsdio snd_soc_spdif_rx bluetooth ecdh_generic ecc snd_soc_simple_card brcmfmac snd_soc_simple_card_utils brcmutil i2c_mux_pinctrl i2c_mux bcm2835_unicam snd_soc_bcm2835_i2s cfg80211 raspberrypi_hwmon snd_soc_core i2c_bcm2835 snd_compress spi_bcm2835 snd_pcm_dmaengine bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops iproc_rng200 videobuf2_v4l2 rng_core videobuf2_common rpivid_mem nvmem_rmem uio_pdrv_genirq uio sch_fq_codel i2c_dev snd_bcm2835(C) snd_pcm snd_timer snd tc358743 v4l2_dv_timings v4l2_fwnode videodev mc cec libcomposite dwc2 udc_core roles drm fuse drm_panel_orientation_quirks ip_tables x_tables ipv6
[  151.731803] CPU: 0 PID: 652 Comm: stream Tainted: G        WC        5.10.17-2-ARCH #1
[  151.741047] Hardware name: BCM2711
[  151.745110] [<c020f10c>] (unwind_backtrace) from [<c020aed0>] (show_stack+0x10/0x14)
[  151.754156] [<c020aed0>] (show_stack) from [<c0de7864>] (dump_stack+0xa4/0xc4)
[  151.762679] [<c0de7864>] (dump_stack) from [<c022ae9c>] (__warn+0xec/0x148)
[  151.770303] [<c022ae9c>] (__warn) from [<c0ddfb00>] (warn_slowpath_fmt+0x7c/0xac)
[  151.779094] [<c0ddfb00>] (warn_slowpath_fmt) from [<bf20fbf0>] (__vb2_queue_cancel+0x1d0/0x24c [videobuf2_common])
[  151.790760] [<bf20fbf0>] (__vb2_queue_cancel [videobuf2_common]) from [<bf210548>] (vb2_core_streamoff+0x18/0xac [videobuf2_common])
[  151.803997] [<bf210548>] (vb2_core_streamoff [videobuf2_common]) from [<bf1d5474>] (__video_do_ioctl+0x238/0x490 [videodev])
[  151.816558] [<bf1d5474>] (__video_do_ioctl [videodev]) from [<bf1d6354>] (video_usercopy+0x170/0x5ac [videodev])
[  151.828079] [<bf1d6354>] (video_usercopy [videodev]) from [<c043df50>] (sys_ioctl+0x34c/0x904)
[  151.838018] [<c043df50>] (sys_ioctl) from [<c0200040>] (ret_fast_syscall+0x0/0x4c)
[  151.846950] Exception stack(0xc530ffa8 to 0xc530fff0)
[  151.852662] ffa0:                   b0505000 00000004 0000000c 40045613 b16f8960 01d67648
[  151.862185] ffc0: b0505000 00000004 00000000 00000036 40045613 0048f06c b16f8960 0000000c
[  151.871715] ffe0: 0048ee1c b16f893c 00439c08 b6ba59fc
[  151.877495] ---[ end trace 90a2d269f970e9d3 ]---
[  151.882820] videobuf2_common: driver bug: stop_streaming operation is leaving buf dca779ef in active state
```

And NULL dereference after ~30 retries:
```
[  372.258026] 8<--- cut here ---
[  372.261116] Unable to handle kernel NULL pointer dereference at virtual address 00000122
[  372.269256] pgd = a13d13c3
[  372.271977] [00000122] *pgd=80000000004003, *pmd=00000000
[... cut ...]
CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        5.10.17-2-ARCH #1
Hardware name: BCM2711
[<c020f10c>] (unwind_backtrace) from [<c020aed0>] (show_stack+0x10/0x14)
[<c020aed0>] (show_stack) from [<c0de7864>] (dump_stack+0xa4/0xc4)
[<c0de7864>] (dump_stack) from [<c03010a8>] (kdb_main_loop+0x4bc/0x9a0)
[<c03010a8>] (kdb_main_loop) from [<c030415c>] (kdb_stub+0x258/0x458)
[<c030415c>] (kdb_stub) from [<c02f92e8>] (kgdb_cpu_enter+0x3e0/0x6fc)
[<c02f92e8>] (kgdb_cpu_enter) from [<c02f9ca8>] (kgdb_handle_exception+0xe0/0x13c)
more>
(reboot)
```

naushir, 2021-03-03:
Ok, so that potential fix did nothing, which sort of reassures me that `unicam_isr` is not preempted by `unicam_stop_streaming`.

Need to continue looking into this one in more detail...

mdevaev, 2021-03-03:
If necessary, I can give you access to the device via the serial console.

mdevaev, 2021-03-03:
@naushir I'm not one hundred percent sure, but could this problem have something to do with using DMA? At least, NULL dereference was played exactly with H264 encoding enabled. It works like this: [ustreamer exports v4l2 buffers, then feeds them to the VC](https://github.com/pikvm/ustreamer/blob/master/src/ustreamer/device.c#L224). At least so far I haven't been able to reproduce this without h264. But maybe I'm just unlucky.

Hypothesis based on ustreamer behavior:
* I connect the cable
* ustreamer exports b4l2 buffers
* there is some race in the driver
* inside the exported buffer turns out to be NULL (???)

mdevaev, 2021-03-03:
It may be generally unrelated. I'm just informing you of what I'm observing. Perhaps exporting buffers somehow negatively affects the warning that we observe, and with some probability exacerbates the effect.

6by9, 2021-03-03:
I was thinking of ```unicam_stop_streaming``` being preempted by ```unicam_isr```.

Supposedly ```unicam_disable``` should have shut down the analogue side, but nothing actually clears the interrupt mask register in unicam. ```unicam_isr``` can therefore still fire.

```
[  151.377155] WARNING: CPU: 0 PID: 652 at drivers/media/common/videobuf2/videobuf2-core.c:997 unicam_return_buffers+0x58/0xd0 [bcm2835_unicam]
...
[  151.501226] [<c0ddfb00>] (warn_slowpath_fmt) from [<bf54e700>] (unicam_return_buffers+0x58/0xd0 [bcm2835_unicam])
[  151.514436] [<bf54e700>] (unicam_return_buffers [bcm2835_unicam]) from [<bf20fa48>] (__vb2_queue_cancel+0x28/0x24c [videobuf2_common])
[  151.529507] [<bf20fa48>] (__vb2_queue_cancel [videobuf2_common]) from [<bf210548>] (vb2_core_streamoff+0x18/0xac [videobuf2_common])
[  151.544455] [<bf210548>] (vb2_core_streamoff [videobuf2_common]) from [<bf1d5474>] (__video_do_ioctl+0x238/0x490 [videodev])
[  151.558797] [<bf1d5474>] (__video_do_ioctl [videodev]) from [<bf1d6354>] (video_usercopy+0x170/0x5ac [videodev])
```
That warn is that the buffer isn't in the ACTIVE state, ie it's not with the driver such that it should be returned.
That sort of implies that something (presumably ```unicam_isr```) has returned a buffer that ```vb2_core_streamoff``` calling ```unicam_return_buffers``` then tries to return again.

naushir, 2021-03-03:
> @naushir I'm not one hundred percent sure, but could this problem have something to do with using DMA? At least, NULL dereference was played exactly with H264 encoding enabled. It works like this: [...]

It's possible that we might be looking at two separate and unrelated problem here, the NULL dereference during encode, and the vb2_buf asserts.

naushir, 2021-03-03:
> That warn is that the buffer isn't in the ACTIVE state, ie it's not with the driver such that it should be returned.
> That sort of implies that something (presumably `unicam_isr`) has returned a buffer that `vb2_core_streamoff` calling `unicam_return_buffers` then tries to return again.

Yes, that was my thinking, and the patch above should have ensured this cannot happen by testing `cur_frame` within the lock, but I must have missed something.

mdevaev, 2021-03-03:
But if it's related, then maybe it's underground knocks, and if you fix the problem with active buffers, it can also disappear. In any case, I am at your disposal for testing.

naushir, 2021-03-04:
@mdevaev can you try the below patch (only slightly different from the earlier one) and send me a log if/when you hit the NULL ptr exception?  Could you also enable debug logging on the unicam driver with the below command:

```
sudo su
echo 5 > /sys/module/bcm2835_unicam/parameters/debug
```

```
diff --git a/drivers/media/platform/bcm2835/bcm2835-unicam.c b/drivers/media/platform/bcm2835/bcm2835-unicam.c
index a8b3f5433f04..03779d656e5b 100644
--- a/drivers/media/platform/bcm2835/bcm2835-unicam.c
+++ b/drivers/media/platform/bcm2835/bcm2835-unicam.c
@@ -794,7 +794,7 @@ static void unicam_queue_event_sof(struct unicam_device *unicam)
 static irqreturn_t unicam_isr(int irq, void *dev)
 {
        struct unicam_device *unicam = dev;
-       unsigned int lines_done = unicam_get_lines_done(dev);
+       unsigned int lines_done = 0;
        unsigned int sequence = unicam->sequence;
        unsigned int i;
        u32 ista, sta;
@@ -831,10 +831,12 @@ static irqreturn_t unicam_isr(int irq, void *dev)
                        if (!unicam->node[i].streaming)
                                continue;
 
+                       spin_lock(&unicam->node[i].dma_queue_lock);
                        if (unicam->node[i].cur_frm)
                                unicam_process_buffer_complete(&unicam->node[i],
                                                               sequence);
                        unicam->node[i].cur_frm = unicam->node[i].next_frm;
+                       spin_unlock(&unicam->node[i].dma_queue_lock);
                }
                unicam->sequence++;
        }
@@ -849,6 +851,7 @@ static irqreturn_t unicam_isr(int irq, void *dev)
                        if (!unicam->node[i].streaming)
                                continue;
 
+                       spin_lock(&unicam->node[i].dma_queue_lock);
                        if (unicam->node[i].cur_frm)
                                unicam->node[i].cur_frm->vb.vb2_buf.timestamp =
                                                                ts;
@@ -858,6 +861,7 @@ static irqreturn_t unicam_isr(int irq, void *dev)
                         * from the queue.
                         */
                        unicam_schedule_dummy_buffer(&unicam->node[i]);
+                       spin_unlock(&unicam->node[i].dma_queue_lock);
                }
 
                unicam_queue_event_sof(unicam);
@@ -886,6 +890,9 @@ static irqreturn_t unicam_isr(int irq, void *dev)
                reg_write_field(unicam, UNICAM_ICTL, 1, UNICAM_TFC);
                reg_write_field(unicam, UNICAM_ICTL, 0, UNICAM_FCM);
        }
+
+       unicam_dbg(3, unicam, "ISR: End");
+
        return IRQ_HANDLED;
 }
 
@@ -1750,6 +1757,7 @@ static void unicam_stop_streaming(struct vb2_queue *vq)
        struct unicam_node *node = vb2_get_drv_priv(vq);
        struct unicam_device *dev = node->dev;
 
+       unicam_dbg(3, dev, "Stop streaming: Begin");
        node->streaming = false;
 
        if (node->pad_id == IMAGE_PAD) {
@@ -1785,6 +1793,8 @@ static void unicam_stop_streaming(struct vb2_queue *vq)
 
        /* Clear all queued buffers for the node */
        unicam_return_buffers(node, VB2_BUF_STATE_ERROR);
+
+       unicam_dbg(3, dev, "Stop streaming: End");
 }
 
 static int unicam_enum_input(struct file *file, void *priv,
```
```

mdevaev, 2021-03-04:
@naushir Sure. Can I just ask you to attach it as a file because github messed up the indentation. In the last patch, I had to arrange them manually :)

naushir, 2021-03-04:
Sure thing.  Here is a zip of the patch file you can apply with `git am`.  I had to zip as github does not allow plaintext attachments.

[0001-Unicam-testing.zip](https://github.com/raspberrypi/linux/files/6082301/0001-Unicam-testing.zip)

mdevaev, 2021-03-04:
Thanks! I'll start assembling it now. The result will be available in the next hour.

mdevaev, 2021-03-04:
@naushir Bingo! Got it via UART.
[screenlog.txt](https://github.com/raspberrypi/linux/files/6082789/screenlog.txt)

naushir, 2021-03-04:
Thanks!  Would be useful to see the videobuf2_common warning messages as well if you can capture them.

naushir, 2021-03-04:
@mdevaev , can I just confirm, you only see the NULL ptr exception if you are doing a H.264 encode?  If so, any chance you could try reproducing without encode?

mdevaev, 2021-03-04:
@naushir Here without trying of H264. Just a warning. I will try to reproduce NULL dereference without H264.
[screenlog.txt](https://github.com/raspberrypi/linux/files/6082941/screenlog.txt)

mdevaev, 2021-03-04:
UPD: The second part of the log is when I disconnected the cable after the previous failure.
[screenlog.txt](https://github.com/raspberrypi/linux/files/6082951/screenlog.txt)

naushir, 2021-03-04:
> @naushir Here without trying of H264. Just a warning. I will try to reproduce NULL dereference without H264.
> [screenlog.txt](https://github.com/raspberrypi/linux/files/6082941/screenlog.txt)

Interestingly this log does show the warning in videobuf2-core, but it happens in the middle of a streaming situation.  There is no stop stream call happening anywhere in the vicinity of the warning stack dump.  Is something in userland manipulating the buffers while they are queued in unicam?  Doing that is strictly not allowed.

naushir, 2021-03-04:
> @naushir Bingo! Got it via UART.
> [screenlog.txt](https://github.com/raspberrypi/linux/files/6082789/screenlog.txt)

Again, the NULL ptr dereference happens mid-streaming.  At least from this log snippet here, there is not stop streaming call either.  I struggle to see how the unicam driver could cause this in those cases.

@6by9 any thoughts?

mdevaev, 2021-03-04:
> Is something in userland manipulating the buffers while they are queued in unicam? Doing that is strictly not allowed.

I don't think I do anything like that. I only work with the buffer that I explicitly get after VIDIOC_DQBUF. Otherwise, I would have noticed this problem earlier. I have several thousand users here and they would have reported it :)

PS: Weird behavior. If I turn off H264, then on enabled kernel debug logging ustreamer freezes on select(dev) and does not accept SIGINT. This does not happen on a regular kernel build. The kernel seems to be too busy with log IO.

JakubVanek, 2021-03-04:
I tried to reproduce this on Raspbian on RPi4B1.4 with the latest commit ec967eb45f8d4ed59bebafb5748da38118383be7 (32bit build) and I wasn't able to trigger the crash nor the warning. I did change some kconfig values, but these were (I suppose) related to kernel debugging. I was using the Auvidea B101 rev 4 board for this.

I will try to test this with the latest official kernel to see if I can reproduce the segfault.

I tried multiple ways of reproducing this:
* run `ustreamer --dv-timings --device=/dev/video0 --persistent --format=uyvy --encoder=noop` and repeatedly plug and unplug hdmi
* run `ustreamer --dv-timings --device /dev/video0 --persistent --format=uyvy --encoder=omx --h264-sink test` and repeatedly plug and unplug hdmi
* run `ustreamer --dv-timings --device=/dev/video0 --persistent --format=uyvy --encoder=noop` and repeatedly upload new edid (this caused the bridge to be redetected on the other side)
* run `ustreamer --dv-timings --device=/dev/video0 --persistent --format=uyvy --encoder=noop` and repeatedly upload new edid (this caused the bridge to be redetected on the other side)

JakubVanek, 2021-03-04:
Taking back: I reproduced both the warning and NULL dereference this successfully on the official 5.10.17 build, but I had to use another RPi as the signal source.

JakubVanek, 2021-03-04:
This is the NULL dereference from a local debug build without the zipped patch applied. Here is a fuller stack trace of it:
```
[  165.377940] 8<--- cut here ---
[  165.381007] Unable to handle kernel NULL pointer dereference at virtual address 00000104
[  165.389102] pgd = 543cd94e
[  165.391811] [00000104] *pgd=0316a003, *pmd=7ff7d003
[  165.396712] Internal error: Oops: a07 [#1] SMP ARM

Entering kdb (current=0xc3181f00, pid 635) on processor 0 Oops: (null)
due to oops @ 0xbf258ba8
CPU: 0 PID: 635 Comm: stream Tainted: G        WC        5.10.17-v7l+ #1
Hardware name: BCM2711
PC is at unicam_isr+0x2f8/0x3b4 [bcm2835_unicam]
LR is at 0x100
pc : [<bf258ba8>]    lr : [<00000100>]    psr: 60000193
sp : c4073890  ip : 00000100  fp : c40738d4
r10: 00000122  r9 : c18fd000  r8 : 00000608
r7 : 000004fc  r6 : 00000000  r5 : c18fd4fc  r4 : c18fd000
r3 : c31f3280  r2 : 00000122  r1 : 00000000  r0 : c31f3000
Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
Control: 30c5383d  Table: 03449500  DAC: fffffffd
CPU: 0 PID: 635 Comm: stream Tainted: G        WC        5.10.17-v7l+ #1
Hardware name: BCM2711
Backtrace:
[<c020c3f0>] (dump_backtrace) from [<c020c73c>] (show_stack+0x20/0x24)
 r7:ffffffff r6:00000000 r5:60000193 r4:c12e69fc
[<c020c71c>] (show_stack) from [<c0b7a99c>] (dump_stack+0xcc/0xf8)
[<c0b7a8d0>] (dump_stack) from [<c0208d6c>] (show_regs+0x1c/0x20)
 r10:c120518c r9:00000005 r8:c13842c8 r7:0000000f r6:c1211ee4 r5:c1383e0c
 r4:c1383e10 r3:49c86701
[<c0208d50>] (show_regs) from [<c0305460>] (kdb_main_loop+0x3bc/0x7d8)
[<c03050a4>] (kdb_main_loop) from [<c0308570>] (kdb_stub+0x268/0x474)
 r10:c120518c r9:00000005 r8:c13842c8 r7:00000000 r6:c13845c4 r5:c137f654
 r4:c4073694
[<c0308308>] (kdb_stub) from [<c02fd848>] (kgdb_cpu_enter+0x450/0x7d4)
 r10:00000001 r9:ffffcfc7 r8:ffffe000 r7:00000000 r6:c127d368 r5:c4073694
 r4:c137f640
[<c02fd3f8>] (kgdb_cpu_enter) from [<c02fe378>] (kgdb_handle_exception+0x11c/0x140)
 r10:00000a07 r9:00000001 r8:00000001 r7:0000000b r6:c4073840 r5:00000000
 r4:c1205048
[<c02fe25c>] (kgdb_handle_exception) from [<c0210f9c>] (kgdb_notify+0x40/0x94)
 r9:ffffe000 r8:c0210f5c r7:00000001 r6:c4073738 r5:00000080 r4:20000193
[<c0210f5c>] (kgdb_notify) from [<c02485f4>] (atomic_notifier_call_chain+0x68/0x8c)
 r7:c4073738 r6:00000001 r5:fffffffc r4:00000000
[<c024858c>] (atomic_notifier_call_chain) from [<c0248668>] (notify_die+0x50/0x6c)
 r8:c0e1c778 r7:c4073840 r6:c3181f00 r5:49c86701 r4:c1205048
[<c0248618>] (notify_die) from [<c020c864>] (die+0x124/0x374)
 r5:c120a444 r4:c135631c
[<c020c740>] (die) from [<c0b6ad68>] (__do_kernel_fault.part.0+0x88/0x98)
 r10:00000122 r9:c4072000 r8:ffffe000 r7:c28d3a00 r6:00000a07 r5:c4073840
 r4:00000104
[<c0b6ace0>] (__do_kernel_fault.part.0) from [<c0b85200>] (do_translation_fault+0x0/0x108)
 r7:c28d3a00 r4:c4073840
[<c0b84ef0>] (do_page_fault) from [<c02154d4>] (do_DataAbort+0x40/0xc4)
 r10:00000122 r9:c4072000 r8:c4073840 r7:00000104 r6:c0b84ef0 r5:00000a07
 r4:c120af08
[<c0215494>] (do_DataAbort) from [<c0200a34>] (__dabt_svc+0x54/0x80)
Exception stack(0xc4073840 to 0xc4073888)
3840: c31f3000 00000000 00000122 c31f3280 c18fd000 c18fd4fc 00000000 000004fc
3860: 00000608 c18fd000 00000122 c40738d4 00000100 c4073890 00000100 bf258ba8
3880: 60000193 ffffffff
 r8:00000608 r7:c4073874 r6:ffffffff r5:60000193 r4:bf258ba8
[<bf2588b0>] (unicam_isr [bcm2835_unicam]) from [<c0289c70>] (__handle_irq_event_percpu+0xac/0x258)
 r10:c4073920 r9:c120509c r8:ffffe000 r7:c1abd900 r6:00000000 r5:c1abd96c
 r4:c335bb40
[<c0289bc4>] (__handle_irq_event_percpu) from [<c0289e58>] (handle_irq_event_percpu+0x3c/0x8c)
 r10:00000100 r9:c4072000 r8:c1810800 r7:00000001 r6:c1abd900 r5:c1abd96c
 r4:c1205048
[<c0289e1c>] (handle_irq_event_percpu) from [<c0289efc>] (handle_irq_event+0x54/0x78)
 r6:00000000 r5:c1abd96c r4:c1abd900
[... 45 lines cut ...]
(vb2_fop_poll [videobuf2_v4l2]) from v4l2_poll (drivers/media/v4l2-core/v4l2-dev.c:348) videodev
(v4l2_poll [videodev]) from do_select (./include/linux/file.h:45) 
(do_select) from core_sys_select (fs/select.c:679) 
(core_sys_select) from kern_select (fs/select.c:719) 
(kern_select) from sys_select (fs/select.c:722) 
(sys_select) from ret_fast_syscall (arch/arm/kernel/entry-common.S:82) 
```
KGDB stacktrace from the ISR:
```
#0  0xbf258ba8 in __list_del (next=0x100, prev=0x122)
#1  __list_del_entry (entry=0xc31f3280) at ./include/linux/list.h:135
#2  list_del (entry=0xc31f3280) at ./include/linux/list.h:146
#3  unicam_schedule_next_buffer (node=<optimized out>) at drivers/media/platform/bcm2835/bcm2835-unicam.c:745
#4  unicam_isr (irq=<optimized out>, dev=0xc18fd000) at drivers/media/platform/bcm2835/bcm2835-unicam.c:879
#5  0xc0289c70 in __handle_irq_event_percpu (desc=desc@entry=0xc1abd900, flags=flags@entry=0xc4073920) at kernel/irq/handle.c:156
#6  0xc0289e58 in handle_irq_event_percpu (desc=desc@entry=0xc1abd900) at kernel/irq/handle.c:196
#7  0xc0289efc in handle_irq_event (desc=desc@entry=0xc1abd900) at kernel/irq/handle.c:213
#8  0xc028ea34 in handle_fasteoi_irq (desc=0xc1abd900) at kernel/irq/chip.c:714
#9  0xc0288c18 in generic_handle_irq_desc (desc=<optimized out>) at ./include/linux/irqdesc.h:152
#10 generic_handle_irq (irq=<optimized out>) at kernel/irq/irqdesc.c:650
#11 0xc0289368 in __handle_domain_irq (domain=0xc1810800, hwirq=<optimized out>, lookup=lookup@entry=true, regs=regs@entry=0xc40739e0) at kernel/irq/irqdesc.c:687
#12 0xc020135c in handle_domain_irq (regs=0xc40739e0, hwirq=<optimized out>, domain=<optimized out>) at ./include/linux/irqdesc.h:170
#13 gic_handle_irq (regs=0xc40739e0) at drivers/irqchip/irq-gic.c:370
#14 0xc0200abc in __irq_svc () at arch/arm/kernel/entry-armv.S:205
```

JakubVanek, 2021-03-04:
(this was triggered by the original reproducer - `ustreamer --dv-timings --device=/dev/video0 --persistent --format=uyvy --encoder=noop`)

JakubVanek, 2021-03-04:
<details>
<summary>Contents of the affected unicam_device struct</summary>

```
$13 = {
  kref = {
    refcount = {
      refs = {
        counter = 2
      }
    }
  }, 
  asd = {
    match_type = V4L2_ASYNC_MATCH_FWNODE, 
    match = {
      fwnode = 0xeff68b90, 
      device_name = 0xeff68b90 "", 
      i2c = {
        adapter_id = -269055088, 
        address = 0
      }, 
      custom = {
        match = 0xeff68b90, 
        priv = 0x0
      }
    }, 
    list = {
      next = 0x100, 
      prev = 0x122
    }, 
    asd_list = {
      next = 0xc18fd24c, 
      prev = 0xc18fd24c
    }
  }, 
  base = 0xf0821000, 
  clk_gate_base = 0xf0825004, 
  clock = 0xc335ba40, 
  vpu_clock = 0xc335bac0, 
  clocks_enabled = true, 
  v4l2_dev = {
    dev = 0xc1a4b610, 
    mdev = 0xc18fd090, 
    subdevs = {
      next = 0xc340b0e0, 
      prev = 0xc340b0e0
    }, 
    lock = {
      {
        rlock = {
          raw_lock = {
            {
              slock = 65537, 
              tickets = {
                owner = 1, 
                next = 1
              }
            }
          }
        }
      }
    }, 
    name = "unicam fe801000.csi", '\000' <repeats 16 times>, 
    notify = 0xbf258c64 <unicam_notify>, 
    ctrl_handler = 0x0, 
    prio = {
      prios = {{
          counter = 0
        }, {
          counter = 0
[... 1548 lines cut ...]
      prev = 0xc18fdeec
    }, 
    req_obj = {
      ops = 0x0, 
      priv = 0x0, 
      req = 0x0, 
      list = {
        next = 0xc18fdf00, 
        prev = 0xc18fdf00
      }, 
      kref = {
        refcount = {
          refs = {
            counter = 1
          }
        }
      }, 
      completed = false
    }
  }
}
```
</details>

I'd be happy to send a complete memory dump, but I couldn't find anything like that apart from kdump, which is likely not working due to its reliance on kexec.

JakubVanek, 2021-03-04:
After rebuilding the kernel with list debugging enabled, I received a BUG caused by list corruption here:
```
(gdb) bt
#0  0xc0730ad8 in __list_add_valid (new=new@entry=0xc340e680, prev=prev@entry=0xc340e680, next=next@entry=0xc2c084fc) at lib/list_debug.c:29
#1  0xbf1f4748 in __list_add (next=0xc2c084fc, prev=0xc340e680, new=0xc340e680)
#2  list_add_tail (head=0xc2c084fc, new=0xc340e680) at ./include/linux/list.h:100
#3  unicam_buffer_queue (vb=0xc340e400) at drivers/media/platform/bcm2835/bcm2835-unicam.c:1312
#4  0xbf0eac50 in __enqueue_in_driver (vb=vb@entry=0xc340e400)
#5  0xbf0ed398 in vb2_core_qbuf (q=q@entry=0xc2c083a8, index=<optimized out>, pb=pb@entry=0xc367be3c, req=<optimized out>)
#6  0xbf264e8c in vb2_qbuf (q=0xc2c083a8, mdev=<optimized out>, b=0xc367be3c)
#7  0xbf264f20 in vb2_ioctl_qbuf (file=0xc31a9f00, priv=<optimized out>, p=0xc367be3c)
#8  0xbf164ee4 in v4l_qbuf (ops=0xbf1f8364 <unicam_ioctl_ops>, file=file@entry=0xc31a9f00, fh=fh@entry=0xc32d1c00, arg=arg@entry=0xc367be3c) at drivers/media/v4l2-core/v4l2-ioctl.c:2072
#9  0xbf169954 in __video_do_ioctl (file=file@entry=0xc31a9f00, cmd=cmd@entry=3226490383, arg=arg@entry=0xc367be3c) at drivers/media/v4l2-core/v4l2-ioctl.c:2996
#10 0xbf16a408 in video_usercopy (file=0xc31a9f00, orig_cmd=3225703951, arg=3044017664, func=func@entry=0xbf169788 <__video_do_ioctl>) at drivers/media/v4l2-core/v4l2-ioctl.c:3312
#11 0xbf16a888 in video_ioctl2 (file=<optimized out>, cmd=<optimized out>, arg=<optimized out>) at drivers/media/v4l2-core/v4l2-ioctl.c:3349
#12 0xbf161094 in v4l2_ioctl (filp=<optimized out>, cmd=<optimized out>, arg=<optimized out>)
#13 0xc0448dc0 in vfs_ioctl (arg=<optimized out>, cmd=<optimized out>, filp=<optimized out>) at fs/ioctl.c:48
#14 __do_sys_ioctl (arg=3044017664, cmd=3225703951, fd=<optimized out>) at fs/ioctl.c:753
#15 __se_sys_ioctl (fd=<optimized out>, cmd=-1069263345, arg=-1250949632) at fs/ioctl.c:739
#16 0xc0200040 in __idmap_text_end ()
```

naushir, 2021-03-05:
@mdevaev and @JakubVanek thank you for the logs and traces you have provided!

I'm afraid I still cannot see where this may be going wrong from the code.  Would you be able to capture some more logs for me with the attached patch please?  It adds a few more log points to help trace buffer usage.  I am weary of adding to much logging, as it could hide the problem.  Thanks again for helping to debug this!

With the below patch, there is no need to do `echo 5 > /sys/module/bcm2835_unicam/parameters/debug` as I bump up the log level in the driver directly.

[0001-Unicam-testing.zip](https://github.com/raspberrypi/linux/files/6089242/0001-Unicam-testing.zip)

mdevaev, 2021-03-05:
Time to build the kernel!

mdevaev, 2021-03-05:
Warning without null dereference (first on plug, second on unplug): [no-crash.txt](https://github.com/raspberrypi/linux/files/6089599/no-crash.txt)

naushir, 2021-03-05:
Thank you.  Ok, I have started to see something strange with the way interrupts occur.  It seems this device does not return Unicam Frame End interrupts?  Instead we rely on the packet capture PI0 to indicate end of frame?  @6by9 can you confirm that?

Additionally, from sequence 2 onward, we only get one isr call per frame, with FS + LC + PI0 which is really not great.  After that, our videobuf2 warning occurs.  Our buffer handling code is probably not robust enough to handle this sequence, particularly when FE is missing and we rely and PI0.

```
[  705.473604] unicam fe801000.csi: ISR: ISTA: 0x1, STA: 0xC002, sequence 0, lines done 0       <<< FS + PI0
[  705.483659] unicam fe801000.csi: ISR: FE: cur c53c9800, next c53c9800
[  705.508863] unicam fe801000.csi: ISR: End
[  707.203854] unicam fe801000.csi: ISR: ISTA: 0x1, STA: 0x4000, sequence 1, lines done 0       <<< FS
[  707.229328] unicam fe801000.csi: ISR: End
[  707.239425] unicam fe801000.csi: ISR: ISTA: 0x4, STA: 0x4003, sequence 1, lines done 0       <<< LC
[  707.244531] unicam fe801000.csi: ISR: End
[  707.253918] unicam fe801000.csi: ISR: ISTA: 0x1, STA: 0xC001, sequence 1, lines done 0       <<< FS + PI0
[  707.268064] unicam fe801000.csi: ISR: FE: cur c53c9800, next c53c9000
[  707.292300] unicam fe801000.csi: ISR: End
[  707.302030] unicam fe801000.csi: ISR: ISTA: 0x5, STA: 0xC003, sequence 2, lines done 0       <<< FS + LC + PI0
[  707.316801] unicam fe801000.csi: ISR: FE: cur c53c9000, next c53c8c00
[  707.342066] unicam fe801000.csi: ISR: End
[  707.351564] unicam fe801000.csi: ISR: ISTA: 0x5, STA: 0xC001, sequence 3, lines done 0       <<< FS + LC + PI0
[  707.366263] unicam fe801000.csi: ISR: FE: cur c53c8c00, next c53c8800
[  707.390834] unicam fe801000.csi: ISR: End
[  707.400312] unicam fe801000.csi: ISR: ISTA: 0x5, STA: 0xC003, sequence 4, lines done 0       <<< FS + LC + PI0
[  707.414654] unicam fe801000.csi: ISR: FE: cur c53c8800, next c53c9800
[  707.438182] unicam fe801000.csi: ISR: End
[  707.438196] unicam fe801000.csi: ISR: ISTA: 0x5, STA: 0xC001, sequence 5, lines done 0       <<< FS + LC + PI0
[  707.459265] unicam fe801000.csi: ISR: FE: cur c53c9800, next 00000000
[  707.483773] unicam fe801000.csi: ISR: End
[  707.492739] unicam fe801000.csi: ISR: ISTA: 0x5, STA: 0xC002, sequence 6, lines done 0       <<< FS + LC + PI0
[  707.497458] unicam fe801000.csi: ISR: FE: cur 00000000, next c53c9800
[  707.526747] unicam fe801000.csi: ISR: End
[  707.526757] unicam fe801000.csi: ISR: ISTA: 0x5, STA: 0xC001, sequence 7, lines done 0       <<< FS + LC + PI0
[  707.854294] unicam fe801000.csi: ISR: FE: cur c53c9800, next 00000000
[  707.868427] unicam fe801000.csi: ISR: End

```

6by9, 2021-03-05:
Based on the original firwmare driver FE detection is sometimes dubious, so CMP0/PI0 is set up to also look for FE (VC = 0 and DT/short packet ID = 1 (FE))

mdevaev, 2021-03-05:
I can't catch null dereference right now on the same kernel with the new patch, it seems the new debug logs take up kernel time and reduce the chance of a race.

6by9, 2021-03-05:
The ISR is taking FAR to long to run
```
[  707.302030] unicam fe801000.csi: ISR: ISTA: 0x5, STA: 0xC003, sequence 2, lines done 0       <<< FS + LC + PI0
[  707.316801] unicam fe801000.csi: ISR: FE: cur c53c9000, next c53c8c00
[  707.342066] unicam fe801000.csi: ISR: End
```
That looks to be 40ms, which explains why FS + LC + FE/PI0 all occur at the same time. FS+FE/PI0 simultaneously is a condition we can't handle well to determine whether the buffer is actually swapped or not.

naushir, 2021-03-05:
> That looks to be 40ms, which explains why FS + LC + FE/PI0 all occur at the same time. FS+FE/PI0 simultaneously is a condition we can't handle well to determine whether the buffer is actually swapped or not.

Good catch, I missed that.  This is likely due to my patch that adds a spinlock wait in that bit of code.

I think the problem starts on the first interrupt where we get FS + PI0.  Since LC has not had a chance to swap buffers, cur_buf and next_buf are the same values.  We could not do the FE handling if we have a FS + FE simultaneously, but I'm not sure that is the right answer.

6by9, 2021-03-05:
It is Unicam that sometimes fails to trigger on FE, not the CSI2 source failing to produce it.
You'll find the same thing in ```unicam_lisr``` with STA & CAM_PI0 -> UNICAM_STAT_FE, and ISTA & CAM_FEI -> UNICAM_STAT_FE.

If you want one of these then I have about 6 here. I need to head past the office and can detour slightly to your place.
(They're generally the Chinese boards from Ebay that come up under a search for "TC358743 Pi", but they work the same as the Auvidea ones for video).

naushir, 2021-03-05:
I think I have a grasp on the buffer warning now.  Hopefully the attached patch should get rid of it.  However, I am not convinced that this would also get rid of the NULL deference exception.

@mdevaev can you give this a try whenever you are able to please?  I've removed the extra spinlocks so that interrupts should also behave better with this patch.

[0001-Unicam-testing.zip](https://github.com/raspberrypi/linux/files/6090039/0001-Unicam-testing.zip)

mdevaev, 2021-03-05:
Building...

mdevaev, 2021-03-05:
I had to set debug level to 3 to be able to use ctrl+c.

Also found some interesting thing about i2c (maybe not related):
```
[  582.917872] unicam fe801000.csi: ISR: ISTA: 0x5, STA: 0xC003, sequence 84, lines done 0
[  582.922439] unicam fe801000.csi: ISR: FE: cur 00000000, next 00000000
[  582.931550] unicam fe801000.csi: Scheduling dummy buffer for node 0
[  582.945364] unicam fe801000.csi: ISR: End
[  582.945371] unicam fe801000.csi: ISR: ISTA: 0x4, STA: 0x4001, sequence 85, lines done 0
[  582.949932] unicam fe801000.csi: schedule_next_buffer: next buf c52fec00, vb c52fec00
[  582.959057] unicam fe801000.csi: ISR: End
[  582.968014] unicam fe801000.csi: ISR: ISTA: 0x5, STA: 0xC003, sequence 85, lines done 0
[  582.972592] unicam fe801000.csi: ISR: FE: cur 00000000, next c52fec00
[  582.981739] unicam fe801000.csi: Scheduling dummy buffer for node 0
[  582.981848] i2c-bcm2835 fe205000.i2c: i2c transfer timed out
[  582.988810] unicam fe801000.csi: ISR: End
[  582.988820] unicam fe801000.csi: ISR: ISTA: 0x4, STA: 0x4001, sequence 86, lines done 0
[  582.995720] tc358743 10-000f: i2c_rd: reading register 0x14 from 0xf failed
[  583.001917] unicam fe801000.csi: schedule_next_buffer: next buf c52fe400, vb c52fe400
[  583.001920] unicam fe801000.csi: ISR: End
```
Warning and null dereference not yet reproduced

JakubVanek, 2021-03-05:
Neither can I reproduce the warning (neither from videobuf2 nor from list corruption) nor the NULL dereference.

Sample dmesg logs with ustreamer using JPEG and H264 encoding: [jpeg.log](https://github.com/raspberrypi/linux/files/6090351/jpeg.log), [h264.log](https://github.com/raspberrypi/linux/files/6090352/h264.log).

I think I saw a similar I2C warning earlier, but I cannot get it to appear now.

mdevaev, 2021-03-05:
It seems the problem is fixed. At least I didn't manage to reproduce it for an hour and a half. I suggest to clean the patch and apply it. It doesn't seem to get any worse :)

naushir, 2021-03-05:
Good news! I'll prepare a clean commit for a PR soon.

mdevaev, 2021-03-05:
@naushir If it's not difficult, plz attach the new patch, I will test the final version.

naushir, 2021-03-05:
No problem. I will cc you on the PR. Once you confirm it working we will submit it.

However I may not be able to get it ready until later on in the day.

mdevaev, 2021-03-05:
It's okay, I'll wait. Solving this problem is a top priority for me.

mdevaev, 2021-03-07:
Given the presence of the fix, I think that this issue can be closed.

closed 2021-03-07 by mdevaev