raspberrypi/linux #4130: Realtek RTL9210 chipset (NVMe to USB) initialized but seems doesn't work correctly on PRI 4

hayorov, 2021-02-08:
**Describe the bug**
NVMe to USB encloser with RTL9210 chipset detected as USB2 (high-speed) device initialized but seems doesn't work. The device is present but cannot be used for partitioning 

**To reproduce**
List the steps required to reproduce the issue.

**Expected behavior**
Discovered as 5000M (USB3 Bus) device and allows to manage NVMe disk.

**Actual behaviour**
```

[   18.936785] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   18.945449] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar 23 2020 02:19:54 version 7.45.206 (r725000 CY) FWID 01-88ee44ea
[   19.711692] usb 1-1.3: new high-speed USB device number 3 using xhci_hcd
[   19.818618] usb 1-1.3: New USB device found, idVendor=0bda, idProduct=9210, bcdDevice=20.01
[   19.818628] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   19.818633] usb 1-1.3: Product: RTL9210--VB
[   19.818638] usb 1-1.3: Manufacturer: Realtek
[   19.818643] usb 1-1.3: SerialNumber: 012345689321
[   19.996580] usb-storage 1-1.3:1.0: USB Mass Storage device detected
[   19.998010] scsi host0: usb-storage 1-1.3:1.0
[   19.998617] usbcore: registered new interface driver usb-storage
[   20.004271] usbcore: registered new interface driver uas
[   20.211855] alua: device handler registered
[   20.214924] emc: device handler registered
[   20.218323] rdac: device handler registered
[   21.029378] scsi 0:0:0:0: Direct-Access     Realtek  RTL9210 NVME     1.00 PQ: 0 ANSI: 6
[   21.030078] sd 0:0:0:0: Attached scsi generic sg0 type 0
[   21.035682] sd 0:0:0:0: [sda] Read Capacity(10) failed: Result: hostbyte=DID_OK driverbyte=DRIVER_SENSE
[   21.035693] sd 0:0:0:0: [sda] Sense Key : Illegal Request [current] 
```

**System**

* Pi4B
* Ubuntu 20.04.2 LTS \n \l
* version 194a85abd768c7334bbadc3f1911c10a7d18ed14 (clean) (release) (start)
* Linux raspberrypi 5.4.0-1028-raspi #31-Ubuntu SMP PREEMPT Wed Jan 20 11:30:45 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux

**Logs**
dmesg
```
[   18.936785] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   18.945449] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar 23 2020 02:19:54 version 7.45.206 (r725000 CY) FWID 01-88ee44ea
[   19.711692] usb 1-1.3: new high-speed USB device number 3 using xhci_hcd
[   19.818618] usb 1-1.3: New USB device found, idVendor=0bda, idProduct=9210, bcdDevice=20.01
[   19.818628] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   19.818633] usb 1-1.3: Product: RTL9210--VB
[   19.818638] usb 1-1.3: Manufacturer: Realtek
[   19.818643] usb 1-1.3: SerialNumber: 012345689321
[   19.996580] usb-storage 1-1.3:1.0: USB Mass Storage device detected
[   19.998010] scsi host0: usb-storage 1-1.3:1.0
[   19.998617] usbcore: registered new interface driver usb-storage
[   20.004271] usbcore: registered new interface driver uas
[   20.211855] alua: device handler registered
[   20.214924] emc: device handler registered
[   20.218323] rdac: device handler registered
[   21.029378] scsi 0:0:0:0: Direct-Access     Realtek  RTL9210 NVME     1.00 PQ: 0 ANSI: 6
[   21.030078] sd 0:0:0:0: Attached scsi generic sg0 type 0
[   21.035682] sd 0:0:0:0: [sda] Read Capacity(10) failed: Result: hostbyte=DID_OK driverbyte=DRIVER_SENSE
[   21.035693] sd 0:0:0:0: [sda] Sense Key : Illegal Request [current] 
[   21.035701] sd 0:0:0:0: [sda] Add. Sense: Invalid command operation code
[   21.035711] sd 0:0:0:0: [sda] 0 512-byte logical blocks: (0 B/0 B)
[   21.035716] sd 0:0:0:0: [sda] 0-byte physical blocks
[   21.037426] sd 0:0:0:0: [sda] Test WP failed, assume Write Enabled
[   21.039130] sd 0:0:0:0: [sda] Asking for cache data failed
[   21.039147] sd 0:0:0:0: [sda] Assuming drive cache: write through
[   21.062525] sd 0:0:0:0: [sda] Read Capacity(10) failed: Result: hostbyte=DID_OK driverbyte=DRIVER_SENSE
[   21.062543] sd 0:0:0:0: [sda] Sense Key : Illegal Request [current] 
[   21.062556] sd 0:0:0:0: [sda] Add. Sense: Invalid command operation code
[   21.066197] sd 0:0:0:0: [sda] Attached SCSI disk
```

```
 ❯ sudo lsblk -l   
NAME      MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
loop0       7:0    0 48.8M  1 loop /snap/core18/1936
loop1       7:1    0 48.9M  1 loop /snap/core18/1949
loop2       7:2    0   27M  1 loop /snap/snapd/10494
loop3       7:3    0 62.9M  1 loop /snap/lxd/19168
loop4       7:4    0   27M  1 loop /snap/snapd/10709
loop5       7:5    0 62.9M  1 loop /snap/lxd/19020
mmcblk0   179:0    0 14.9G  0 disk 
mmcblk0p1 179:1    0  256M  0 part /boot/firmware
mmcblk0p2 179:2    0 14.6G  0 part /
```

```
 ❯ lsusb -t                      
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 3: Dev 3, If 0, Class=Mass Storage, Driver=usb-storage, 480M
```

**Additional context**
Latest smartctl with chipset support
```
smartctl -a /dev/sda
smartctl 7.2 2020-12-30 r5155 [aarch64-linux-5.4.0-1028-raspi] (local build)
Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org

Read NVMe Identify Controller failed: scsi error unsupported scsi opcode

 ❯ sudo smartctl -x -d scsi /dev/sda
smartctl 7.2 2020-12-30 r5155 [aarch64-linux-5.4.0-1028-raspi] (local build)
Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Vendor:               Realtek
Product:              RTL9210 NVME
Revision:             1.00
Compliance:           SPC-4
LU is fully provisioned
Logical Unit id:      0x3001237923792379
Serial number:        0000000000000000
Device type:          disk
Local Time is:        Sat Feb  6 13:01:11 2021 +08
SMART support is:     Unavailable - device lacks SMART capability.
Read Cache is:        Unavailable
Writeback Cache is:   Unavailable

=== START OF READ SMART DATA SECTION ===
Current Drive Temperature:     0 C
Drive Trip Temperature:        0 C

Error Counter logging not supported

Device does not support Self Test logging
Device does not support Background scan results logging
```

Looking for some help and ideas.

P33M, 2021-02-08:
What happens if you connect the drive after power-on (and Linux has booted) - is the behaviour any different?

hayorov, 2021-02-09:
@P33M I've tried to connect the enclosure after the RPI4 boot, the behavior is absolutely the same, I've tried other cables and even different power adapters. 

I still confused and can't get why it's a USB2 device, but the main point - why I cannot interact with the disk... is that a wrong controller detection or RPI <-> Controller <-> NVMe

hayorov, 2021-02-09:
One more unsuccessful attempt.

Installed Raspbian GNU/Linux 10 (latest), booted, attached enclosure

dmesg
```
[  378.509022] usb 1-1.1: new high-speed USB device number 3 using xhci_hcd
[  378.645936] usb 1-1.1: New USB device found, idVendor=0bda, idProduct=9210, bcdDevice=20.01
[  378.645955] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  378.645972] usb 1-1.1: Product: RTL9210--VB
[  378.645987] usb 1-1.1: Manufacturer: Realtek
[  378.646003] usb 1-1.1: SerialNumber: 012345689321
[  378.658234] usb-storage 1-1.1:1.0: USB Mass Storage device detected
[  378.659600] usb-storage 1-1.1:1.0: Quirks match for vid 0bda pid 9210: 800000
[  378.659807] scsi host0: usb-storage 1-1.1:1.0
[  379.680950] scsi 0:0:0:0: Direct-Access     Realtek  RTL9210 NVME     1.00 PQ: 0 ANSI: 6
[  379.689818] sd 0:0:0:0: [sda] Read Capacity(10) failed: Result: hostbyte=0x00 driverbyte=0x08
[  379.689839] sd 0:0:0:0: [sda] Sense Key : 0x5 [current] 
[  379.689858] sd 0:0:0:0: [sda] ASC=0x20 ASCQ=0x0 
[  379.689883] sd 0:0:0:0: [sda] 0 512-byte logical blocks: (0 B/0 B)
[  379.689900] sd 0:0:0:0: [sda] 0-byte physical blocks
[  379.691690] sd 0:0:0:0: [sda] Test WP failed, assume Write Enabled
[  379.695031] sd 0:0:0:0: [sda] Asking for cache data failed
[  379.695047] sd 0:0:0:0: [sda] Assuming drive cache: write through
[  379.698882] sd 0:0:0:0: Attached scsi generic sg0 type 0
[  379.756380] sd 0:0:0:0: [sda] Read Capacity(10) failed: Result: hostbyte=0x00 driverbyte=0x08
[  379.756399] sd 0:0:0:0: [sda] Sense Key : 0x5 [current] 
[  379.756415] sd 0:0:0:0: [sda] ASC=0x20 ASCQ=0x0 
[  379.760119] sd 0:0:0:0: [sda] Attached SCSI disk
```

lsusb
```
root@raspberrypi:/home/pi# lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Mass Storage, Driver=usb-storage, 480M
```

`Linux raspberrypi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux`

Tried Quirks also, no effect.

Any thoughts?

hayorov, 2021-02-09:
Some additional information about the enclosure and NVMe from CrystalDiskInfo (Windows)
[image]

P33M, 2021-02-09:
From what I see, there are at least two things going wrong when plugging into the Pi. The first is that USB3.0 negotiation fails and the device falls back to USB2.0. The second is that the disk isn't reporting back any capacity figures - the command is timing out. The fact that switching to USB mass storage rather than UAS has the same result points to a basic initialisation-style problem rather than UAS protocol breakage.

Does the adapter work at all if you plug it into one of the physical USB2.0 ports (or use a USB2.0 cable)?

timg236, 2021-02-09:
Please could you also post links to the enclosure and NVMe device. I'm using a Realtek NVMe adapter with what appears to be the same chipset on a Pi4.

hayorov, 2021-02-10:
@timg236 happy that I'm not alone here.

My hardware
- Enclosure - ORICO M2PV-C3 <https://www.orico.cc/us/product/detail/7214.html>, Realtek RTL9210
- NVMe device - Reletech P400, <https://www.newegg.com/reletech-1tb-p400/p/0D9-004F-00031>, Phison E12S

hayorov, 2021-02-10:
@P33M USB2 physical port gives the same result

```
pi@raspberrypi:~ $ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 3: Dev 4, If 0, Class=Mass Storage, Driver=usb-storage, 480M
```

the same dmesg
```
[  378.509022] usb 1-1.1: new high-speed USB device number 3 using xhci_hcd
[  378.645936] usb 1-1.1: New USB device found, idVendor=0bda, idProduct=9210, bcdDevice=20.01
[  378.645955] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  378.645972] usb 1-1.1: Product: RTL9210--VB
[  378.645987] usb 1-1.1: Manufacturer: Realtek
[  378.646003] usb 1-1.1: SerialNumber: 012345689321
[  378.658234] usb-storage 1-1.1:1.0: USB Mass Storage device detected
[  378.659600] usb-storage 1-1.1:1.0: Quirks match for vid 0bda pid 9210: 800000
[  378.659807] scsi host0: usb-storage 1-1.1:1.0
[  379.680950] scsi 0:0:0:0: Direct-Access     Realtek  RTL9210 NVME     1.00 PQ: 0 ANSI: 6
[  379.689818] sd 0:0:0:0: [sda] Read Capacity(10) failed: Result: hostbyte=0x00 driverbyte=0x08
[  379.689839] sd 0:0:0:0: [sda] Sense Key : 0x5 [current] 
[  379.689858] sd 0:0:0:0: [sda] ASC=0x20 ASCQ=0x0 
[  379.689883] sd 0:0:0:0: [sda] 0 512-byte logical blocks: (0 B/0 B)
[  379.689900] sd 0:0:0:0: [sda] 0-byte physical blocks
[  379.691690] sd 0:0:0:0: [sda] Test WP failed, assume Write Enabled
[  379.695031] sd 0:0:0:0: [sda] Asking for cache data failed
[  379.695047] sd 0:0:0:0: [sda] Assuming drive cache: write through
[  379.698882] sd 0:0:0:0: Attached scsi generic sg0 type 0
[  379.756380] sd 0:0:0:0: [sda] Read Capacity(10) failed: Result: hostbyte=0x00 driverbyte=0x08
[  379.756399] sd 0:0:0:0: [sda] Sense Key : 0x5 [current] 
[  379.756415] sd 0:0:0:0: [sda] ASC=0x20 ASCQ=0x0 
[  379.760119] sd 0:0:0:0: [sda] Attached SCSI disk
[ 1529.563004] usb 1-1.1: USB disconnect, device number 3
[82719.801100] usb 1-1.3: new high-speed USB device number 4 using xhci_hcd
[82719.938052] usb 1-1.3: New USB device found, idVendor=0bda, idProduct=9210, bcdDevice=20.01
[82719.938071] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[82719.938088] usb 1-1.3: Product: RTL9210--VB
[82719.938103] usb 1-1.3: Manufacturer: Realtek
[82719.938118] usb 1-1.3: SerialNumber: 012345689321
[82719.949972] usb-storage 1-1.3:1.0: USB Mass Storage device detected
[82719.950390] usb-storage 1-1.3:1.0: Quirks match for vid 0bda pid 9210: 800000
[82719.950597] scsi host0: usb-storage 1-1.3:1.0
[82720.992715] scsi 0:0:0:0: Direct-Access     Realtek  RTL9210 NVME     1.00 PQ: 0 ANSI: 6
[82720.993388] sd 0:0:0:0: Attached scsi generic sg0 type 0
[82720.999310] sd 0:0:0:0: [sda] Read Capacity(10) failed: Result: hostbyte=0x00 driverbyte=0x08
[82720.999332] sd 0:0:0:0: [sda] Sense Key : 0x5 [current] 
[82720.999358] sd 0:0:0:0: [sda] ASC=0x20 ASCQ=0x0 
[82720.999383] sd 0:0:0:0: [sda] 0 512-byte logical blocks: (0 B/0 B)
[82720.999400] sd 0:0:0:0: [sda] 0-byte physical blocks
[82721.018583] sd 0:0:0:0: [sda] Test WP failed, assume Write Enabled
[82721.020382] sd 0:0:0:0: [sda] Asking for cache data failed
[82721.020400] sd 0:0:0:0: [sda] Assuming drive cache: write through
[82721.088512] sd 0:0:0:0: [sda] Read Capacity(10) failed: Result: hostbyte=0x00 driverbyte=0x08
[82721.088534] sd 0:0:0:0: [sda] Sense Key : 0x5 [current] 
[82721.088552] sd 0:0:0:0: [sda] ASC=0x20 ASCQ=0x0 
[82721.097176] sd 0:0:0:0: [sda] Attached SCSI disk
```

hayorov, 2021-02-21:
Ok, I'm still fighting and decided to check one more case: lack of power.

I use [ZMI 20000 Power Bank](https://www.zmi.com/products/zmi-powerpack-20k-pro-usb-c-power-bank) with a built-in USB hub `Genesys Logic, Inc. USB-2.0 4-Port HUB`. 

```
root@raspberrypi:/home/pi# lsusb -d 05e3:0608 -vvv

Bus 001 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x05e3 Genesys Logic, Inc.
  idProduct          0x0608 Hub
  bcdDevice           85.37
  iManufacturer           0 
  iProduct                1 USB2.0 Hub
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0019
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             4
  wHubCharacteristic 0x00e4
    Ganged power switching
    Compound device
    Ganged overcurrent protection
    TT think time 32 FS bits
    Port indicators
  bPwrOn2PwrGood       50 * 2 milli seconds
  bHubContrCurrent    100 milli Ampere
  DeviceRemovable    0x18
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0100 power
[... 114 lines cut ...]
Temperature:                        27 Celsius
Available Spare:                    100%
Available Spare Threshold:          5%
Percentage Used:                    2%
Data Units Read:                    454,278 [232 GB]
Data Units Written:                 863,865 [442 GB]
Host Read Commands:                 22,541,193
Host Write Commands:                40,123,635
Controller Busy Time:               48
Power Cycles:                       454
Power On Hours:                     93
Unsafe Shutdowns:                   344
Media and Data Integrity Errors:    0
Error Information Log Entries:      2
Warning  Comp. Temperature Time:    0
Critical Comp. Temperature Time:    0

Warning: NVMe Get Log truncated to 0x200 bytes, 0x200 bytes zero filled
Error Information (NVMe Log 0x01, 16 of 63 entries)
No Errors Logged
```

I've ordered USB3 Hub based on [VL812](https://www.via-labs.com/product_show.php?id=41) with an external power supply.

I'll report the final results but as for now looks like it's a power shortage.

hayorov, 2021-03-01:
USB3 Gen1  hub with complementary power delivery arrived. So, I can fully confirm that **the root cause of the problem is a lack of pi's power ** that leads to incorrect enclose detection and a problem with NVMe drive recognition. 

My new setup with external power via USB3 Gen 1 hub works as expected:

```
root@raspberrypi:/home/pi# lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 3: Dev 3, If 0, Class=Mass Storage, Driver=uas, 5000M
```
```
[    1.843319] usb 2-1: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[    1.875454] usb 2-1: New USB device found, idVendor=2109, idProduct=0817, bcdDevice= 2.14
[    1.875492] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    1.875524] usb 2-1: Product: USB3.0 Hub             
[    1.875552] usb 2-1: Manufacturer: VIA Labs, Inc.         
[    1.879015] hub 2-1:1.0: USB hub found
[    1.879343] hub 2-1:1.0: 4 ports detected
[   10.503583] usb 2-1.3: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[   10.540650] usb 2-1.3: New USB device found, idVendor=0bda, idProduct=9210, bcdDevice=20.01
[   10.540696] usb 2-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   10.540731] usb 2-1.3: Product: RTL9210--VB
[   10.540760] usb 2-1.3: Manufacturer: Realtek
[   10.540788] usb 2-1.3: SerialNumber: 012345689321
[   10.545129] usb 2-1.3: Enable of device-initiated U1 failed.
[   10.557009] usb 2-1.3: Enable of device-initiated U2 failed.
[   10.578631] usb 2-1.3: Enable of device-initiated U1 failed.
[   10.590361] usb 2-1.3: Enable of device-initiated U2 failed.
[   10.622839] scsi host0: uas
[   10.629115] scsi 0:0:0:0: Direct-Access     Realtek  RTL9210 NVME     1.00 PQ: 0 ANSI: 6
[   10.648708] sd 0:0:0:0: [sda] 2000409264 512-byte logical blocks: (1.02 TB/954 GiB)
[   10.650292] sd 0:0:0:0: [sda] Write Protect is off
[   10.650326] sd 0:0:0:0: [sda] Mode Sense: 37 00 00 08
[   10.653486] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[   10.656318] sd 0:0:0:0: [sda] Optimal transfer size 33553920 bytes
[   10.752387]  sda: sda1 sda2
[   10.769966] sd 0:0:0:0: [sda] Attached SCSI disk
```

Hub <https://linux-hardware.org/index.php?id=usb:2109-0817>

closed 2021-03-01 by hayorov

maxromanovsky, 2021-05-14:
@hayorov could you please share a link to the USB3 Gen 1 hub you're using? Cheers!

hayorov, 2021-07-03:
@maxromanovsky I have this https://www.orico.cc/usmobile/product/detail/id/3238 It is based on the VL817 VIA chip.

mmokrejs, 2021-09-05:
I have same Enclosure - ORICO M2PVdevce and I already found that I had to prevent uas driver (linux-5.4.143) to pick it up by adding the following line:

```
# cat /etc/modprobe.d/ignore_uas.conf 
options usb-storage quirks=0bda:9210:u
#
```

For basic use it worked "flawlessly" but if I want to re-write the SSD drive using `dd_rescue` or add it to an `mdraid`, it chokes and disconnects from USB.

```
# smartctl -d sntrealtek  -x /dev/sda
smartctl 7.2 2020-12-30 r5155 [x86_64-linux-5.4.143] (local build)
Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       Force MP510
Serial Number:                      19378239000128951104
Firmware Version:                   ECFM22.5
PCI Vendor/Subsystem ID:            0x1987
IEEE OUI Identifier:                0x6479a7
Total NVM Capacity:                 1,920,383,410,176 [1.92 TB]
Unallocated NVM Capacity:           0
Controller ID:                      1
NVMe Version:                       1.3
Number of Namespaces:               1
Namespace 1 Size/Capacity:          1,920,383,410,176 [1.92 TB]
Namespace 1 Formatted LBA Size:     512
Namespace 1 IEEE EUI-64:            6479a7 26b1863034
Local Time is:                      Sun Sep  5 16:33:45 2021 CEST
Firmware Updates (0x12):            1 Slot, no Reset required
Optional Admin Commands (0x0017):   Security Format Frmw_DL Self_Test
Optional NVM Commands (0x005d):     Comp DS_Mngmt Wr_Zero Sav/Sel_Feat Timestmp
Log Page Attributes (0x08):         Telmtry_Lg
Maximum Data Transfer Size:         512 Pages
Warning  Comp. Temp. Threshold:     75 Celsius
Critical Comp. Temp. Threshold:     80 Celsius

Supported Power States
St Op     Max   Active     Idle   RL RT WL WT  Ent_Lat  Ex_Lat
 0 +    11.52W       -        -    0  0  0  0        0       0
 1 +     8.09W       -        -    1  1  1  1        0       0
 2 +     6.37W       -        -    2  2  2  2        0       0
 3 -   0.0490W       -        -    3  3  3  3     2000    2000
 4 -   0.0018W       -        -    4  4  4  4    25000   25000

Supported LBA Sizes (NSID 0x1)
Id Fmt  Data  Metadt  Rel_Perf
 0 +     512       0         2
 1 -    4096       0         1

=== START OF SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

SMART/Health Information (NVMe Log 0x02)
Critical Warning:                   0x00
Temperature:                        49 Celsius
Available Spare:                    100%
Available Spare Threshold:          5%
Percentage Used:                    5%
Data Units Read:                    34,042,467 [17.4 TB]
Data Units Written:                 29,298,920 [15.0 TB]
Host Read Commands:                 343,811,672
Host Write Commands:                436,547,290
Controller Busy Time:               2,474
Power Cycles:                       566
Power On Hours:                     8,449
Unsafe Shutdowns:                   82
Media and Data Integrity Errors:    0
Error Information Log Entries:      287
Warning  Comp. Temperature Time:    0
Critical Comp. Temperature Time:    0
Thermal Temp. 1 Transition Count:   4
Thermal Temp. 2 Transition Count:   2
Thermal Temp. 1 Total Time:         3441
Thermal Temp. 2 Total Time:         19

Warning: NVMe Get Log truncated to 0x200 bytes, 0x200 bytes zero filled
Error Information (NVMe Log 0x01, 16 of 63 entries)
No Errors Logged
```

Did anyone manage to get a firmware update? See https://forums.anandtech.com/threads/stable-nvme-usb-adapter.2572973/page-29

hayorov, 2021-09-25:
@mmokrejs maybe I'm wrong but the controller (Realtek) has a setting something like idle-timeout and if no IO activity for a while it switches off the power and it looks like a USB disconnect, I believe that this device was designed to work as a USB storage stick rather them permanent drive. Probably you can disable idle timeout but make sure that the chip (both memory and the controller) is cold enough.

partment, 2022-04-13:
@mmokrejs You can head over orico's china website for firmware that doesn't sleep after 10 minutes.
https://www.orico.com.cn/download.html
[image]
[image]
Change congifure
Lite_0 means never sleep
Lite_10 means sleep after 10 minutes

mmokrejs, 2022-06-05:
@partment Thank you for your help. Finally I got to try the new firmware. The 1.27.xx series update went to 100% and I used the `_0.cfg` file to get rid of the sleeping issue. Unplugged, reconnected, was recognized again. I did not check the NVMe data contents ... yet.

Sadly I then decided that while at it I shall try the 1.29.xx series image using the same 64bit EXE distributed with the 1.27.xx firmware and the same `_0.cfg`. But, something went wrong, the update process stoped at 97%. The device is dead, does nothing after plugging in. USB does ot report a new device (Linux kernel too, not only Win10).

I did check before that the 1.29.xx image is selected and the other part was in some subdirectory. So either using he older EXE was the culprit, or the old config was not compatible with the newer firmware files ...

partment, 2022-06-05:
@mmokrejs 
Sorry to hear that.
I'm using `UTNVME_B_v1.29.12.011122.bin` as firmware BIN and `UTGDFW_B_v4.22.27.081220.bin` as firmware GD.
These two are paired with each other, maybe you flushed wrong firmware GD?

partment, 2022-06-05:
@mmokrejs I remembered that there's is a method to reset with shorting certain pins under or near the chip.
I will post the method here once I confirm that.

mmokrejs, 2022-07-31:
Right, that saved me. Short-circuiting the ping before USB power is attached was ineed the way to reset it. I had to flash and reset several times, I used several USB cables but mostly the flashing process interrupted. I am just guessing but maybe limiting the USB to USB2-speeds would eliminate the issue with cable or I/O error handling, whatever. I had 2 successfull attempt out of say 20! The flashing stopped anywhere between 35% to 98%, it felt like a random occassion than a rule. Yeah, I also enabled the serial programing option in a few atempts but with not difference.

avelytchko, 2022-10-08:
I have RTL9210B-CG with exactly same issue. However, advise to use externally powered usb hub didn't help. I updated firmware from 1.23 to 1.27 with _0.cfg - didn't help either.

S474N, 2022-11-16:
> I have RTL9210B-CG with exactly same issue. However, advise to use externally powered usb hub didn't help. I updated firmware from 1.23 to 1.27 with _0.cfg - didn't help either.

Tested also v1.29.12 and nothing.

Lajcisvk, 2022-11-19:
For me, it does not work with the latest kernel. After downgrading to 20220830 it started using UAS driver again.

mmokrejs, 2022-11-19:
> For me, it does not work with the latest kernel. After downgrading to 20220830 it started using UAS driver again.

That probably has to do with the device+firmware being reported in the past not behaving properly under UAS driver. Look for these in dmesg:

```
[360801.653060] usb 2-4.1.1: new SuperSpeed USB device number 15 using xhci_hcd
[360801.706193] usb 2-4.1.1: New USB device found, idVendor=0bda, idProduct=9210, bcdDevice=20.01
[360801.706197] usb 2-4.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[360801.706199] usb 2-4.1.1: Product: RTL9210
[360801.706200] usb 2-4.1.1: Manufacturer: Realtek
[360801.706201] usb 2-4.1.1: SerialNumber: 012345678908
[360801.737745] usb 2-4.1.1: UAS is ignored for this device, using usb-storage instead
[360801.737749] usb-storage 2-4.1.1:1.0: USB Mass Storage device detected
[360801.738080] usb-storage 2-4.1.1:1.0: Quirks match for vid 0bda pid 9210: 800000
```

You can disable the blacklisting in some config file.

Lajcisvk, 2022-11-19:
You are right `US_FL_IGNORE_UAS` flag was added in this commit https://github.com/raspberrypi/linux/commit/528aba78ee0151162bcabc85c02b297801d5eefe

But I don't think there is a way to remove/disable it without rebuilding the kernel.

dionorgua, 2022-11-20:
This should override quirk:
```
echo "0bda:9210:" | sudo tee /sys/module/usb_storage/parameters/quirks
```

partment, 2022-11-20:
I add this to /boot/cmdline.txt and uas is back:
```
usb-storage.quirks=0bda:9210:
```

PANKAJ3312, 2022-12-25:
@partment 
I used below config.
`;Configure Start
;=================FW Configure=================
U2PHY = 02 f4 9b e0 e1 
U3PHY = 02 d4 09 00 d5 00 80 

VID = 0xbda
PID = 0x9210

MANUFACTURE = "Realtek"
SCSI_VENDOR = "Realtek"
PRODUCT = "RTL9210"
SCSI_PRODUCT = "RTL9210 NVME"
SERIAL = "012345678903"

CFEXPRESS = 0x0
LED = 0x1
PINMUX1 = 0x70000
PINMUX2 = 0x0
U2_MAXPWR = 0xfa
U3_MAXPWR = 0x70
ASPMDIS = 0x0

;10 minutes of idle time will disable pcie power
;PCIE_PWRCUT_THRES = 0xa

;EN_UPS = 0x0
;PD = 0x0

;CUSTOMIZED_LED is 15 bytes
;CUSTOMIZED_LED = 00
;SUSPEND_LED_OFF = 0x0
;FAN = 0x0

;DIS_SHOW_EMPTY_DISK = 0x1
;UART_DBG_PIN = 5

;=================Tool Configure=================
;DISCONN_FORCE_USB2 = n
;FORMAT_DISK = n
;INC_SERIAL = y

;PARTITION: GPT, MBR
PARTITION = "MBR"
;FORMAT: NTFS, exFAT
FORMAT = "NTFS"
;LABEL max allowed - characters
LABEL = "My USB"
`
Does it support 'sleep after 10 mins' or not.
I looked the files RTL9210_Lite_10.cfg and RTL9210_Lite_0.cfg both are same.

PANKAJ3312, 2022-12-25:
@partment 
I used below config.
Does it support **'sleep after 10 mins'** or not.
I looked the files **RTL9210_Lite_10.cfg** and **RTL9210_Lite_0.cfg** both are same.

```
;Configure Start
;=================FW Configure=================
U2PHY = 02 f4 9b e0 e1 
U3PHY = 02 d4 09 00 d5 00 80 

VID = 0xbda
PID = 0x9210

MANUFACTURE = "Realtek"
SCSI_VENDOR = "Realtek"
PRODUCT = "RTL9210"
SCSI_PRODUCT = "RTL9210 NVME"
SERIAL = "012345678903"

CFEXPRESS = 0x0
LED = 0x1
PINMUX1 = 0x70000
PINMUX2 = 0x0
U2_MAXPWR = 0xfa
U3_MAXPWR = 0x70
ASPMDIS = 0x0

;10 minutes of idle time will disable pcie power
;PCIE_PWRCUT_THRES = 0xa

;EN_UPS = 0x0
;PD = 0x0

;CUSTOMIZED_LED is 15 bytes
;CUSTOMIZED_LED = 00
;SUSPEND_LED_OFF = 0x0
;FAN = 0x0

;DIS_SHOW_EMPTY_DISK = 0x1
;UART_DBG_PIN = 5

;=================Tool Configure=================
;DISCONN_FORCE_USB2 = n
;FORMAT_DISK = n
;INC_SERIAL = y

;PARTITION: GPT, MBR
PARTITION = "MBR"
;FORMAT: NTFS, exFAT
FORMAT = "NTFS"
;LABEL max allowed - characters
LABEL = "My USB"
```

partment, 2022-12-26:
You just missed the commented PCIE_PWRCUT_THRES
0xa means 10
if you want 20 minutes then change it 0xf4, it's hexadecimal.

jglathe, 2022-12-27:
> ```
> usb-storage.quirks=0bda:9210:
> ```

Works also on other kernels. Here on Pop!OS

quantuumsnot, 2023-01-21:
3 days ago bought Orico MM2C3-G2-GY 10Gbps Space Gray version (despite that their site says Dark Green is the 10Gbps)
It has RTL9210 chip, cable is capable of 10Gbps
Enclosure works stable on USB 3.1 5Gbps ports, on USB 3.1 Gen 2 10Gbps it's not detected correctly under Win 8.1/10
Updated the firmware to realtek_rtl9210_1.29.12.011122 version (found it on station-drivers.com)
Now it is found on 10Gbps Type A ports but not stable, this chip model is so bad that even crashed the Gen2 motherboard controller (ASMedia, reviews say it's the best for the bucks used in most of the Asus or other brands models) so it went in BSOD state very fast, during the crash flight it managed to disconnect and WD_Black 4TB CMR external 2.5" HDD :D

I waited 4-5 years so they can bring a stable NVMe 10Gbps enclosures to the market, most of (if not all) JMS583/RTL9210 based  are pure shit, ASM2362 based are probably the most stable. But pinch that with a salt and use the regular USB3 5Gbps ports (btw USB3 is the worst protocol for ports in modern era, max 900mA is a joke, and it's age of 15 YEARS with problems is hilarious). If you have the opportunity just put a PCIe USB extender on your PC to at least offer more Amps for the USB devices

fanoush, 2023-02-02:
> Now it is found on 10Gbps Type A ports but not stable, this chip model is so bad that even crashed the Gen2 motherboard controller

I have good experience with this enclosure https://www.aliexpress.com/item/1005002776016696.html
works fine for me with 10Gbps with good usb-c to usb-c cable (I use this one https://www.aliexpress.com/item/1005003984553956.html), I don't have device with usb 3.1 gen 2 type A socket so cannot try original cable with 10Gbps speed.

I made several full ssd backup and restore / OS cloning operations with Macrium Reflect Free with 3 different laptops, transferred terabytes of data and it worked fine so far.

quantuumsnot, 2023-02-03:
> I have good experience with this enclosure https://www.aliexpress.com/item/1005002776016696.html works fine for me with 10Gbps with good usb-c to usb-c cable (I use this one [...]

Can you provide screenshots of the actual speeds with CrystalDiskMark 8?

fanoush, 2023-02-03:
I also got this one https://www.aliexpress.com/item/1005004792305540.html but it turns off very quickly and vid/pid is non standard so for smartctl I need to force `-d sntrealtek` to work. But it does 10Gbps too. Maybe some info from this thread about reflashing firmware could help? Can I use the files and config in such generic adapter? Parts like 
```
CFEXPRESS = 0x0
LED = 0x1
PINMUX1 = 0x70000
PINMUX2 = 0x0
U2_MAXPWR = 0xfa
U3_MAXPWR = 0x70
ASPMDIS = 0x1
```
look like specific only to some device so may break mine?

quantuumsnot, 2023-02-03:
> Also I have another two cheaper ones without any box and some turn off every few seconds and some are not readable via `smartctl -a` or `-x` but with this one I can't complain about anything, it just works. Maybe I got lucky piece.

This also shows and SMART data which is not reported on my enclosure ... Definitely going to return it, so shame on you ORICO!

mvadu, 2023-02-08:
@fanoush  do you know if the AliExpress enclosure you have uses same RTL9210 chip? does it reports as UAS device and supports smartctl without any config changes?

fanoush, 2023-02-09:
> @fanoush do you know if the AliExpress enclosure you have uses same RTL9210 chip? does it reports as UAS device and supports smartctl without any config changes?

yes and yes - the first one I mentioned  https://www.aliexpress.com/item/1005002776016696.html , it looks like this
[image]

the second one has the chip too but the vid/pid is changed so needs `-d sntrealtek`

mvadu, 2023-02-10:
@fanoush thank you for sharing the pic.. I am getting one too now.

quantuumsnot, 2023-02-11:
Whoa, I think the problems with RTL9210 based enclosures is in the chip's version - most problematic has RTL9210A, the newer ones are RTL9210B and they are somewhat OK. Anyone knows if there's somewhere on the Internet a list with the used chips in enclosures?

1kazekage, 2023-03-11:
I am facing the same issue.
It worked earlier but now it doesn't detect at all.
Can anyone help?

semaf, 2023-04-02:
> Whoa, I think the problems with RTL9210 based enclosures is in the chip's version - most problematic has RTL9210A, the newer ones are RTL9210B and they are somewhat OK. Anyone knows if there's somewhere on the Internet a list with the used chips in enclosures?

I have couple with RTL9210B and after a while it can't read the ssd.

quantuumsnot, 2023-04-03:
We definitely should create a repo or site with list of these sh*tty products - pictures of the internals, housing ... aaaand we can try mailing the brands to replace them even if warranty ended. FFS I have almost a bag full of crap like these last 5-6 years, backstabbed me with ~$250

bijavix, 2023-04-18:
> I have good experience with this enclosure https://www.aliexpress.com/item/1005002776016696.html
works fine for me with 10Gbps with good usb-c to usb-c cable (I use this one https://www.aliexpress.com/item/1005003984553956.html), I don't have device with usb 3.1 gen 2 type A socket so cannot try original cable with 10Gbps speed.

Just I case, I want to warn everyone here, the seller of the good NVMe enclosure is now sending lower quality boards, NOT Realtek.

[image]

Jafalex, 2023-07-23:
Did anyone fix their issue by hooking up the drive through a powered usb 3.0 hub?

I'm facing the issue of a degraded experience with a RTL9210b ([Ugreen NVME enclosure](https://www.aliexpress.com/item/1005004622441562.html?spm=a2g0o.order_list.order_list_main.17.75051802g3VoPn#nav-specification)) whereby my Raspberry Pi 4B handles the drive OK but `lsusb -t` lists the drive as 480M (i.e. USB 2.0), instead of 5000M (i.e. USB 3.0). Therefore I get dreadful wr speeds.

magnets110, 2023-08-09:
Does the RTL9210b support UAS?

actraiser30, 2024-01-18:
Hello i have a usb 3 case with chip rtl9201R it has firmware 1.32.7 the annoying thing is that every 1 minute the hdd goes into sleep,is there a firmware somewhere even the stock that i have now (1.32.7) that i can disable the sleep timer of the case,or can i extract the firmware from the hdd case and re-apply it ?

mmokrejs, 2024-01-18:
For RTL9210B chipset there is `1.32.87` available. But you nly need to update the configuration. Use the `MP Tool` to upload a new configuration file. Here is how you can disable the time-based sleep:

```
diff -u -w RTL9210AB_v1.27.24/RTL9210_Lite_10.cfg /mnt/ntfs/Users/dell_e5580/RTL9210B/RTL9210AB_v1.27.24/RTL9210_Lite_0.cfg
--- RTL9210AB_v1.27.24/RTL9210_Lite_10.cfg	2022-06-28 14:47:39.438314800 +0200
+++ RTL9210AB_v1.27.24/RTL9210_Lite_0.cfg	2022-06-28 15:17:33.399543900 +0200
@@ -21,7 +21,7 @@
 ASPMDIS = 0x0
 
 ;10 minutes of idle time will disable pcie power
-;PCIE_PWRCUT_THRES = 0xa
+PCIE_PWRCUT_THRES = 0x0
 
 ;EN_UPS = 0x0
 ;PD = 0x0
```

It was already described above in https://github.com/raspberrypi/linux/issues/4130#issuecomment-1364766371 

I fetched from
https://www.mediafire.com/file/t4vnjscwbw3g6ad/Realtek_RTL9210_1.32.87.zip/file
https://station-drivers.com/index.php/en/component/remository/Drivers/Realtek/NVMe-USB-3.1/Realtek-RTL92xx(B)-NVMe-USB-3.1-Controller-firmware-Version-1.32.87.082923/lang,en-gb/

See https://forum.gamer.com.tw/C.php?bsn=60030&snA=626346 anf https://station-drivers.com/index.php/en/component/remository/Drivers/Realtek/NVMe-USB-3.1/Realtek-RTL92xx(B)-NVMe-USB-3.1-Controller-firmware-Version-1.32.87.082923/lang,en-gb/

actraiser30, 2024-01-18:
> It was already described above in #4130 (comment)

Hi thanks for your help

My chipset is RTL9201R not RTL9210B,and it's a 2.5" case that takes
mechanical hard drive or ssd drive not nvme case,Will this work on my
chipset??

Zibri, 2024-01-28:
Can someone please LOG usb traffic (using wireshark for example) while updating the firmware and send me the capture?
I really need it for a project...

actraiser30, 2024-01-28:
Hi thanks

But will it work on mine? since I have RTL9201R ??? How can I upload the
configuration only on my chipset???

nstrelow, 2024-01-28:
Hi folks, thanks for all the insights.

I am having the same problem: 9210 (non-B version) throws an error upon connecting, doesn't recognize pratitions.

Is recognized and accessible under Windows AND on my Home Assistant Yellow (board for RPI4- compute module).

I went and updated the firmware, and now it is not even recognized in dmesg. When I connected it through a non-powered USB-Hub it got recognized again, with the same errors.

This enclosure used to work with a different NVME 2 years ago, I used it as a boot drive for the RPi4. I assume something in the RPi4 bootloader/kernel changed or my new drive which now has DRAM, might draw more power than the older one.

I used this firmware here: https://station-drivers.com/index.php/en/component/remository/Drivers/Realtek/NVMe-USB-3.1/Realtek-RTL92xx(A-B)-NVMe-USB-3.1-Controller-firmware-Version-1.29.12.011122/lang,en-gb/

actraiser30, 2024-01-29:
> For RTL9210B chipset there is `1.32.87` available. But you nly need to update the configuration. Use the `MP Tool` to upload a new configuration file. Here is how you can disable the time-based sleep:
> 
> ```
> diff -u -w RTL9210AB_v1.27.24/RTL9210_Lite_10.cfg RTL9210B/RTL9210AB_v1.27.24/RTL9210_Lite_0.cfg
> --- RTL9210AB_v1.27.24/RTL9210_Lite_10.cfg	2022-06-28 14:47:39.438314800 +0200
> +++ RTL9210AB_v1.27.24/RTL9210_Lite_0.cfg	2022-06-28 15:17:33.399543900 +0200
> @@ -21,7 +21,7 @@
>  ASPMDIS = 0x0
>  
>  ;10 minutes of idle time will disable pcie power
> -;PCIE_PWRCUT_THRES = 0xa
> +PCIE_PWRCUT_THRES = 0x0
>  
>  ;EN_UPS = 0x0
>  ;PD = 0x0
> ```
> 
> It was already described above in [#4130 (comment)](https://github.com/raspberrypi/linux/issues/4130#issuecomment-1364766371)
> 
> I fetched from https://www.mediafire.com/file/t4vnjscwbw3g6ad/Realtek_RTL9210_1.32.87.zip/file https://station-drivers.com/index.php/en/component/remository/Drivers/Realtek/NVMe-USB-3.1/Realtek-RTL92xx(B)-NVMe-USB-3.1-Controller-firmware-Version-1.32.87.082923/lang,en-gb/
> 
> See https://forum.gamer.com.tw/C.php?bsn=60030&snA=626346 and https://station-drivers.com/index.php/en/component/remository/Drivers/Realtek/NVMe-USB-3.1/Realtek-RTL92xx(B)-NVMe-USB-3.1-Controller-firmware-Version-1.32.87.082923/lang,en-gb/

But will it work on mine? since I have RTL9201R ??? How can I upload the
configuration only on my chipset???

C0rn3j, 2024-02-03:
>  Does the RTL9210b support UAS?

Not very well, there's a suggested patch to disable it for the device by default - https://patchwork.kernel.org/project/linux-usb/patch/20240117141132.1023457-1-bruno@clisp.org/

---

I have the RPi5 8GB with the official 27W power supply, I have a RTL9210B Sabre enclosure from AliExpress connected by a [PremiumCord kabel USB-C - USB 3.0 A (USB 3.1 generation 2, 3A, 10Gbit/s) 2m](https://www.czc.cz/premiumcord-kabel-usb-c-usb-3-0-a-usb-3-1-generation-2-3a-10gbit-s-2m/275798/produkt) with a `ADATA XPG SX8200 Pro 2TB` SSD. I have the UAS quirk in cmdline.txt, it works when it works.

**It CAN power it well if it manages to negotiate USB 3.2** which gives it 896mA, but in 99% of the cases it does 2.10 and fails to boot because it's limited to 500mA, as can be seen in `sudo lsusb -v`.

Bad Pi connect

```
Bus 001 Device 002: ID 0bda:9210 Realtek Semiconductor Corp. RTL9210 M.2 NVME Adapter
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x9210 RTL9210 M.2 NVME Adapter
  bcdDevice           f0.01
  iManufacturer           1 Sabrent
  iProduct                2 Sabrent
  iSerial                 3 830700000024
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Binary Object Store Descriptor:
[... 187 lines cut ...]
    bDescriptorType        16
    bDevCapabilityType     10
    bmAttributes         0x00000001
      Sublink Speed Attribute count 1
      Sublink Speed ID count 0
    wFunctionalitySupport   0x1100
    bmSublinkSpeedAttr[0]   0x000a4030
      Speed Attribute ID: 0 10Gb/s Symmetric RX SuperSpeedPlus
    bmSublinkSpeedAttr[1]   0x000a40b0
      Speed Attribute ID: 0 10Gb/s Symmetric TX SuperSpeedPlus
Device Status:     0x0000
  (Bus Powered)
```

The firmware was updated on the enclosure from:

[image]

To:

[image]

There is some stupid issue that happens with the USB negotiation most of the time.

My friend got it working well with [EEM2-SG2 SUPERSPEED+ USB-C - M.2 NVME & SATA SSD RAW BOX](https://www.axagon.eu/en/produkty/eem2-sg2) and its included cable and a `Samsung 980 PRO 1TB` when the quirk is enabled. Also Pi5 with official power supply updated to identical firmware.

Zibri, 2024-02-04:
> Can someone please LOG usb traffic (using wireshark for example) while updating the firmware and send me the capture? I really need it for a project...

Please guys, it takes a few minutes to do that. Can you help?

C0rn3j, 2024-02-04:
> > Can someone please LOG usb traffic (using wireshark for example) while updating the firmware and send me the capture? I really need it for a project...
> 
> Please guys, it takes a few minutes to do that. Can you help?

An enclosure with the chip is 12 EUR or less, takes a few minutes to order.

C0rn3j, 2024-02-12:
@actraiser30 pretty sure you need to remove `DISK_IPS_THRES`

actraiser30, 2024-02-12:
thanks i did that it sleeps again after 10-15 minutes,anything else?

actraiser30, 2024-02-18:
thanks i did that i works,the only thing i need to do is when i restart the Raspberry pi the disk power off and power on again
is it possible to not having that??

something to change in the configuration file about the on-off on restart???

muelli01, 2024-02-23:
This problem is not only happening on the Pi, also on various other Linux systems. It seems there is a general issue with RTL9210* and recent kernels.

actraiser30, 2024-02-25:
Anyone knows what is the setting in the config file that disables the power off at reboot or shutdown??  So the disk stays on/working???

Config file contents:

VID = 0xbda
PID = 0x9201
MANUFACTURE = "Realtek"
SCSI_VENDOR = "n/a"
PRODUCT = "RTL9201"
SCSI_PRODUCT = "n/a"
SERIAL = "012345678999"
DISK_IPS_THRES = 10
LED = 0x3
CUSTOMIZED_LED = 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00
SUSPEND_DISK_OFF = 0x7
SCSI_WP_PIN = false
UART_DBG_PIN = 0x64
PERIPH_API = false
UART_BAUD_RATE = false

[15 more replies]