raspberrypi/linux #4133: Degraded network performance after recent update

iandk, 2021-02-08:
**Describe the bug**
I'm using a Pi 4 as a basic Samba share for several months now. 
Recently after performing a system update & upgrade I noticed a degraded read and write performance.

It must be related to a recent update since otherwise, nothing has changed.
The write performance has dropped from a constant 113MB/s to around 75MB/s.

Since nothing else has changed in my setup I flashed an older build of Raspberry Pi OS ( 64bit was the only image I could find) to the Pi. 
With this version, it's performing very well, just like the last few months. 

But after running a system update && upgrade performance is significantly worse again.
The Samba version is exactly the same in both situations. 

I guess it's related to the newer kernel and maybe some changes to the underlying network stack?

   

          

          

          

**To reproduce**
Configure a basic samba share on some external storage with an older version of Raspberry Pi OS. 
[I've used this build
](https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2020-08-24/)

After running a system update && upgrade the write performance is much lower than before.

          

          

          
 
**Expected behaviour**
More or less stable transfer at 100MB/s+
[image]

          

          

          

**Actual behaviour**
After a few seconds, the write speed to the Pi drops to around 75MB/s and stays there. 
[image]

**System**

Raspberry Pi 4 2GB with Crucial MX500 attached via UASP USB3 adapter.
The disk is formatted as XFS and has trim support enabled. 

##### WORKING
```
Linux raspberrypi 5.4.51-v8+ #1333 SMP PREEMPT Mon Aug 10 16:58:35 BST 2020 aarch64 GNU/Linux
```
```
Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7252c154838ec5b4576f29c996ac8fe3750cae12, stage4

```

```
Aug 19 2020 17:38:16
Copyright (c) 2012 Broadcom
version e90cba19a98a0d1f2ef086b9cafcbca00778f094 (clean) (release) (start)

```

##### NOT WORKING
```
Linux raspberrypi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
```

```
Raspberry Pi reference 2021-01-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 21090519d85bdaa1615d5d5057d37b09368ea5d2, stage2
```

```
Jan 27 2021 22:19:57
Copyright (c) 2012 Broadcom
version 99d9a48302e4553cff3688692bb7e9ac760a03fa (clean) (release) (start)
```

**Logs**
No relevant output in dmesg or syslog 

**Additional context**
I've tested it with three different drives, and two different Pis.
I also reinstalled Raspberry Pi OS several times, with nothing other than samba installed. 
Therefore I'm 100% sure it's related to one of the updates. 

The disk itself is performing well in both cases.
```
root@raspberrypi/home # dd if=/dev/zero of=test.data bs=4M count=1K
1024+0 records in
1024+0 records out
4294967296 bytes (4.3 GB, 4.0 GiB) copied, 13.0589 s, 329 MB/s
```

mi-hol, 2021-02-10:
Could you please update your issue details for the used UASP USB3 adapter from dmesg?
Background:
certain chipset & firmware versions seem to have bugs that impact performance negatively.
example for detail I'm looking for
[    3.013976] usb 2-2: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[    3.045420] usb 2-2: New USB device found, idVendor=174c, idProduct=55aa, bcdDevice= 1.00
[    3.053773] usb 2-2: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[    3.061033] usb 2-2: Product: IB-RD3662-C31
[    3.065324] usb 2-2: Manufacturer: ICY BOX
[    3.069496] usb 2-2: SerialNumber: 123456789100

iandk, 2021-02-10:
I've attached the output below. 
[At least one guy is experiencing the same issue.
](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=288234&p=1816823#p1816633)

```
Bus 002 Device 002: ID 174c:55aa ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
[    1.793617] usb 1-1: new high-speed USB device number 2 using xhci_hcd
[    1.976374] usb 1-1: New USB device found, idVendor=2109, idProduct=3431, bcdDevice= 4.21
[    1.976424] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    1.976458] usb 1-1: Product: USB2.0 Hub
[    1.979228] hub 1-1:1.0: USB hub found
[    1.979571] hub 1-1:1.0: 4 ports detected
[    2.124045] usb 2-1: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[    2.137014] systemd[1]: System time before build time, advancing clock.
[    2.155349] usb 2-1: New USB device found, idVendor=174c, idProduct=55aa, bcdDevice= 1.00
[    2.155393] usb 2-1: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[    2.155427] usb 2-1: Product: Ugreen Storage Device
[    2.155458] usb 2-1: Manufacturer: Ugreen
[    2.155487] usb 2-1: SerialNumber: 091912070139
[    2.182292] scsi host0: uas
[    2.184443] scsi 0:0:0:0: Direct-Access     CT250MX5 00SSD1           0    PQ: 0 ANSI: 6
[    2.187262] sd 0:0:0:0: [sda] 488397168 512-byte logical blocks: (250 GB/233 GiB)
[    2.187500] sd 0:0:0:0: [sda] Write Protect is off
[    2.187537] sd 0:0:0:0: [sda] Mode Sense: 43 00 00 00
[    2.187922] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    2.188686] sd 0:0:0:0: [sda] Optimal transfer size 33553920 bytes
[    2.235616]  sda: sda1
[    2.238716] sd 0:0:0:0: [sda] Attached SCSI disk
```

mi-hol, 2021-02-10:
I've collected performance metrics using CrystalDiskMark previously and can confirm a significant drop after upgrade to kernel 5.10.11 (i.e. read of large 64MB file dropped from 93 MB/s  to  28.9 MB/s) on the same hardware & setup ([details here](https://forum.openmediavault.org/index.php?thread/37285-rpi4-read-write-performance-for-smb-shares/&postID=260232#post266012))

Metrics on 5.4.

Measurement | MB/s | IOPS
-- | -- | --
SEQ 1MiB (Q= 8, T= 1) | 93.0 | 88.7
SEQ 1MiB (Q= 1, T= 1) | 95.4 | 91.0
RND 4KiB (Q= 32, T= 1) | 43.2 | 10557.4
RND 4KiB (Q= 1, T= 1) | 8.4 | 2054.4

Metrics on 5.10.11
------------------------------------------------------------------------------
CrystalDiskMark 8.0.0 x86 (C) 2007-2020 hiyohiyo
                                  Crystal Dew World: https://crystalmark.info/
------------------------------------------------------------------------------
* MB/s = 1,000,000 bytes/s [SATA/600 = 600,000,000 bytes/s]
* KB = 1000 bytes, KiB = 1024 bytes

[Read]
  SEQ    1MiB (Q=  8, T= 1):    28.937 MB/s [     27.6 IOPS] <280809.89 us>
  SEQ    1MiB (Q=  1, T= 1):    28.101 MB/s [     26.8 IOPS] < 37209.92 us>
  RND    4KiB (Q= 32, T= 1):    16.972 MB/s [   4143.6 IOPS] <  7706.54 us>
  RND    4KiB (Q=  1, T= 1):     1.699 MB/s [    414.8 IOPS] <  2403.04 us>

[Write]
  SEQ    1MiB (Q=  8, T= 1):    43.206 MB/s [     41.2 IOPS] <188604.44 us>
  SEQ    1MiB (Q=  1, T= 1):    35.021 MB/s [     33.4 IOPS] < 29890.23 us>
  RND    4KiB (Q= 32, T= 1):    21.026 MB/s [   5133.3 IOPS] <  6210.67 us>
  RND    4KiB (Q=  1, T= 1):     1.458 MB/s [    356.0 IOPS] <  2254.01 us>

Profile: Default
   Test: 64 MiB (x5) [Z: 23% (1253/5544GiB)]
   Mode:
   Time: Measure 5 sec / Interval 5 sec 
   Date: 2021/02/10 16:28:37
     OS: Windows 10 Enterprise [10.0 Build 18363] (x64)

iandk, 2021-02-10:
64MB file? 
You're not testing the throughput to your disk, you're testing your cache.

Try again with a larger file and several read/ write operations.

iandk, 2021-02-10:
I just tested it again. 

Flashed the newest Raspberry Pi OS 32bit version (still running the old 5.4.83-v7l+ Kernel ), configured a basic samba share and ran a few tests with around 10-20GB of data. 
No issues at all, the transfer speed was at 110MB/s all the time. 

After that I updated to the kernel version 5.10.11-v7l+ and updated the preinstalled packages as well.
I then ran exactly the same test again, nothing else changed. 
And the transfer speed dropped after around 3-4GB from 110MB/s to 75MB/s.

popcornmix, 2021-02-11:
Can you test if the issue is samba related or network related?
E.g. measure network speed with iperf between the two machines?

iandk, 2021-02-11:
It doesn't seem to be network related. 
iPerf shows a constant 950mbit/s in both directions.

dq-git, 2021-02-11:
I am having a similar issue with my NFS shares that I am using for my kodi box. Everything was working just fine, but at the moment the video playback is impossible. Before the update everything was working fine. I even turned on my old rpi3 board with :5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l and just copied over my /etc/exports. Suddenly kodi went back to life. 
Strangely enough, when accessing the share from my windows laptop, on both the video playback is smooth as butter. 

This is strange as for example iperf3 shows stable 900 Mb/s+.  
For a test I have played the same movie from both machines, while it played just fine from the RPI3 board, the RPI4 is just full of stutter and after a while it crushes. 

RPI4 nfsstat after 2min:
`Server rpc stats:
calls      badcalls   badfmt     badauth    badclnt
61554      0          0          0          0

Server nfs v3:
null             getattr          setattr          lookup           access
7         0%     436       0%     0         0%     28117    45%     194       0%
readlink         read             write            create           mkdir
14        0%     32737    53%     0         0%     0         0%     0         0%
symlink          mknod            remove           rmdir            rename
0         0%     0         0%     0         0%     0         0%     0         0%
link             readdir          readdirplus      fsstat           fsinfo
0         0%     0         0%     39        0%     1         0%     8         0%
pathconf         commit
`

RPI3 after 2 min playback:

```
Server rpc stats:
calls      badcalls   badfmt     badauth    badclnt
1698       0          0          0          0

Server nfs v3:
null             getattr          setattr          lookup           access
1         0%     5         0%     0         0%     97        5%     3         0%
readlink         read             write            create           mkdir
0         0%     1589     93%     0         0%     0         0%     0         0%
symlink          mknod            remove           rmdir            rename
0         0%     0         0%     0         0%     0         0%     0         0%
link             readdir          readdirplus      fsstat           fsinfo
0         0%     0         0%     2         0%     0         0%     1         0%
pathconf         commit
0         0%     0         0%

```

popcornmix, 2021-02-11:
@dq-git your issue may be https://github.com/xbmc/xbmc/issues/19147

d33pd1ve, 2021-02-11:
I have the same problem with my Pi 4 4GB!

I use it as a NAS with an 8TB WD Elements attached.
Performance is much lower after the recent updates

iandk, 2021-02-12:
This is also the second time I'm noticing IO errors after the  kernel update.
I've never seen this issue in the past 6 months with the old kernel. 

```
[59761.546841] sd 1:0:0:0: [sdb] Attached SCSI disk
[152744.813849] XFS (sda1): log I/O error -5
[152744.813945] XFS (sda1): xfs_do_force_shutdown(0x2) called from line 1196 of file fs/xfs/xfs_log.c. Return address = 78401cd3
[152744.813963] XFS (sda1): Log I/O Error Detected. Shutting down filesystem
[152744.813996] XFS (sda1): Please unmount the filesystem and rectify the problem(s)
```

dq-git, 2021-02-12:
@popcornmix Thanks, it looks like this is just the case here. Meanwhile I reverted kernel to the same that was working on rpi3 and magic happened, everything works now.

jorensanbar, 2021-02-15:
​Same bug.

Upgrated to kernel 5.10.11 speed: 70 MB/s.
​
Returning to 5.4.79-v71+ #1373, speed: 115 MB/s

Pi 4B 2GB + WD Elements 5TB USB 3.0

jorensanbar, 2021-02-15:
Same bug.

Upgrated to kernel 5.10.11 speed: 70 MB/s.
​
Returning to 5.4.79-v71+ #1373, speed: 115 MB/s

Pi 4B 2GB + WD Elements 5TB USB 3.0

jorensanbar, 2021-02-15:
> Upgrated to kernel 5.10.11 speed: 70 MB/s.

New test: It is not the kernel, it is another update.

Continuous speed problem, after updating everything except the Kernel.
[image]

My OMV, is working with an old kernel, and after updating everything else, the problem starts.
[image]

iandk, 2021-02-15:
Oh.. I was quite sure it’s related to the kernel. 
I’ll have a look at the packages that will get updated

popcornmix, 2021-02-15:
Confirming exactly which package update causes the reduced performance would be helpful.

iandk, 2021-02-15:
I'll try to see which package reduces the performance. 
After a clean install, the following packages have an upgrade available, so one of them must be the problem.

```
base-files bluez-firmware ca-certificates device-tree-compiler file firmware-atheros firmware-brcm80211
firmware-libertas firmware-misc-nonfree firmware-realtek iproute2 libgnutls30 libldap-2.4-2 libldap-common
libmagic-mgc libmagic1 libpam-systemd libraspberrypi-bin libraspberrypi-dev libraspberrypi-doc libraspberrypi0
libsystemd0 libudev1 libzstd1 python-rpi.gpio raspberrypi-bootloader raspberrypi-kernel raspberrypi-sys-mods
raspi-config rpi-eeprom rpi.gpio-common sudo systemd systemd-sysv tzdata udev unzip
```

jorensanbar, 2021-02-16:
This combo has the bug.

In a Fresh Install, if you update this:

- bootloader
- kernel
- sys-mods
- raspi-config
- rpi-eeprom

When the server is restarted and changes are applied, then the speed is reduced.

We have to do the test by omitting the SYS-Mod, which is the only one with a description that mentions the performance of the system.

iandk, 2021-02-16:
I don't think it's one of those packages. 
I've just updated them one by one and rebootet but the performance is fine. 

So it leaves those packages:

```
  base-files bluez-firmware ca-certificates device-tree-compiler file firmware-atheros firmware-brcm80211
  firmware-libertas firmware-misc-nonfree firmware-realtek iproute2 libldap-2.4-2 libldap-common libmagic-mgc
  libmagic1 libpam-systemd libsystemd0 libudev1 libzstd1 python-rpi.gpio rpi.gpio-common sudo systemd systemd-sysv
  tzdata udev unzip
```

pelwell, 2021-02-16:
I think you should be able to rule out the following by upgrading them all together:
```
bluez-firmware device-tree-compiler firmware-atheros firmware-brcm80211
  firmware-libertas firmware-misc-nonfree firmware-realtek python-rpi.gpio rpi.gpio-common tzdata unzip
```

jorensanbar, 2021-02-16:
Test 3. Test performed in a completely new environment.

[image]

The following additional packages will be installed:
- libraspberrypi-bin
- libraspberrypi-dev
- libraspberrypi-doc
- libraspberrypi0

The following packages will be upgraded:
- libraspberrypi-bin
- libraspberrypi-dev
- libraspberrypi-doc
- libraspberrypi0
- raspberrypi-bootloader
- raspberrypi-kernel
- raspberrypi-sys-mods
- raspi-config
- rpi-eeprom

Just by updating those packages, it is already automatically reflected in performance. The update was done with the 5 packages simultaneously, not one by one.
[image]

iandk, 2021-02-16:
Well, this is super strange.

I've updated all packages one by one but the issue didn't occur. 
Then I've overclocked the Pi to 2GHz with an overvoltage of 6 and I noticed the random drops. 

The Pi didn't throttle, whatsoever.

@jorensanbar Is your Pi overclocked? 
If yes, try running at stock speed

jorensanbar, 2021-02-16:
@iandk , negative, my Pi is Stock

I'm going to reinstall the Pi Os Stock image again, and do the test skipping the SYS-Mod update. I don't know what other option I have.

iandk, 2021-02-16:
Are you seeing those issues with both write and read speed?
At least at the moment the write speed to the Pi seems fine, but read speed is much lower and quite unstable.

I'll also reinstall the Pi again and test if one of the packages you mentioned might be the problem.
Well it must be, since before those recent updates everything was working just fine.

iandk, 2021-02-16:
@popcornmix @pelwell 

Does anyone of you know if any of the recent updates changed some behavior that could explain this performance drop?

pelwell, 2021-02-16:
The xbmc issue @popcornmix linked to ended with this patch appearing in the upstream kernel: 
    https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v5.10.15&id=8e081627f3a7f733a4955ee40b385c972f010f05
which has only today appeared in rpi-5.10.y. Other than that I'm not aware of anything, so I've been watching the bisecting with interest.

pelwell, 2021-02-16:
P.S. The latest rpi-update kernel is now on 5.10.16, so might be worth a try.

iandk, 2021-02-16:
Thank you, how can I update to 5.10.16? With the rpi-update command?
My kernel version is  5.10.11-v7l+

At least the read performance issues seem to be somehow related to the kernel, I've done a clean reinstall and tested the read/write performance. Write speed seems to be fine so far (at least without OC) even after updating/ upgrading all packages and the Kernel. 

Read performance started to drop with the Kernel upgrade.

pelwell, 2021-02-16:
`sudo rpi-update` should do the trick. We normally warn at this point to back up your data first...

iandk, 2021-02-16:
Yep, I've just updated to the latest kernel.
Write @stock speed is fine so far, but the read performance still varies and is lower than with the old 5.4 kernel.

jorensanbar, 2021-02-16:
Test 4: Again I did another clean install. It is definitely a kernel problem.

**To test all packages Updated except: _Kernel + SYS-Mod_.**
[image]

Speed with Kernel 5.4.79-v71+ #1373: (110-115 MB/s stable)
[image]

**Now: Update Kernel except SYS-Mod.**
Speed with Kernel 5.10.11-v71+ #1399: Speed frop to 70-90 MB/s (Irregular)
[image]

**Now: Update Kernel except SYS-Mod**
Speed with Kernel 5.10.16-v71+ #1402: Worst (65 MB/s)
[image]

**Now: Update SYS-Mod (All packages are updated now).**
Same Speed of Kernel 5.10.16 (65MB/s Write / 90 MB/s Read)

iandk, 2021-02-16:
Yea I also think it's related to the kernel.

And somehow overclocking the Pi results in much lower and unstable performance.
This wasn't the case a few weeks ago. 

So it must be related to the issue as well. 

@pelwell @popcornmix  Do you guys need any more information?
**To wrap things up:** 
One of the recent updates introduced performance issues with samba.
The read and write performance is much lower/ dropping quite often and is very unstable in general. 

It seems to be the kernel update that's causing the issues, or maybe one of the supporting packages (sys-mods, eerprom..)
Overclocking the Pi intensifies the problem, at least in my case. 

I verified it by reinstalling it several times. 
When using the old image with the 5.4 kernel both the read and write speeds are fine. 

If you need any help to investigate or reproduce the issue I'm very happy to help.

pelwell, 2021-02-17:
I can reproduce the slowdown, but haven't found the reason yet.

pelwell, 2021-02-18:
I've run out of time on this today, but so far the only obvious difference is that the number of interrupts generated by the Ethernet controller is 20-25% higher on 5.10; in fact it's higher on all kernels since 5.5, 5.4 being the last "good" one.

pelwell, 2021-02-18:
If you want to experiment yourself, `grep eth /proc/interrupts` shows the cumulative interrupt count.

iandk, 2021-02-18:
If I remember correctly I noticed a higher interrupt count as well. 

Thank you for your investigation, I hope you'll be able to find the cause eventually.

iandk, 2021-02-27:
Jeff Gerling mentioned that he didn't notice this issue on his CM4 with a PCIe SATA board. 
So either it might be related to the USB/ UASP stack or he didn't transfer enough data to experience this bug.

geerlingguy, 2021-02-28:
@iandk - To add a little, I tested up to 5 GB file copies (thought I did 10 GB but realized I only did 5 GB for this round of testing), and it was rock solid over both NFS and Samba through a Broadcom MegaRAID card on the PCIe bus (connected directly to the CM4 IO Board). That card uses a Broadcom driver that was tweaked to work in that situation, too, but otherwise the drives are just normal SCSI volumes to the system.

I did the tests through the built-in network jack on the CM4 IO Board, running kernels 5.10.11 and 5.10.14, in both 32-bit and 64-bit Pi OS compiled from source.

iandk, 2021-03-02:
@pelwell Did you manage to find something?

pelwell, 2021-03-02:
No I didn't - it's hard to spot the reason for a slight slow down in a tidal wave of packets, and it's slipped down the priority list.

iandk, 2021-03-02:
I See
Wish I could help somehow.

mi-hol, 2021-03-04:
Today kernel 5.10.17 got deployed, does this change anything related to this issue?

iandk, 2021-03-04:
The write speed seems to be back to normal! 

But somehow the read speed still affected and is mostly sitting at around 80-90MB/s and even struggling with that. 
It was at a stable 112MB/s in the past.

jorensanbar, 2021-03-05:
Raspberry Pi4 B 2GB

Test in 5.10.17-v71+ #1403

Read: 92 MB/s 
[image]

Write: 110 MB/s
[image]

iandk, 2021-03-08:
So has this been caused by the upstream kernel? 
Unfortunately the upload speed (Pi -> Client) is still affected

iandk, 2021-03-10:
I just updated to 5.10.20 but I'm still seeing the same upload speed issue :(

`Linux raspberrypi 5.10.20-v7l+ #1404`

iandk, 2021-03-19:
Here's a screenshot of atop during the download of a file from the pi to a client. 
Pi -> SMB Client
No OC, latest 5.10.23 kernel

[image]

jorensanbar, 2021-03-19:
Now the problem is random.

Suddenly, the copy speed drops to 70MB/s.

It is necessary to restart the server for the speed to return to the maximum, but this does not guarantee, there are occasions that up to 3 continuous restarts are required.

iandk, 2021-03-19:
Didn’t have any issues copying TO the pi so far. 
at least after one of the recent updates. 

However I didn’t copy that much data so far. 

But I’m still seeing issues when the Pi is overclocked (speed is slower and unstable) or when downloading FROM the Pi to a client.

all of this worked a few months ago, no idea what has changed with the 5.10 kernel. 

@jorensanbar your external drive is an SMR drive, that might be reinforce the speed issues for you

trumpetsven, 2021-03-20:
Just a shot in the dark:
Could you try to revert this patch:
https://github.com/raspberrypi/linux/commit/27af8e2c90fba242460b01fa020e6e19ed68c495
and see if this changes something?

iandk, 2021-03-20:
> Just a shot in the dark:
> Could you try to revert this patch:
> [27af8e2](https://github.com/raspberrypi/linux/commit/27af8e2c90fba242460b01fa020e6e19ed68c495)
> and see if this changes something?

How can I revert a patch?

popcornmix, 2021-03-20:
> How can I revert a patch?

Can you build a kernel? See: https://www.raspberrypi.org/documentation/linux/kernel/building.md

Type
```
git revert 27af8e2
```
before building to best the suggestion.

iandk, 2021-03-20:
I've never build a kernel so far. 

Reverting the commit doesn't seem to work, even tho I cloned the kernel with the --depth=1 parameter
```
root@raspberrypi:~/linux# git revert 27af8e2
fatal: bad revision '27af8e2'
```

popcornmix, 2021-03-20:
Ah the commit linked was from 5.11.
Try
```
git revert 7369248
```

iandk, 2021-03-20:
I just recompiled the kernel and rebooted. 
And assuming that I did everything right, it seems to not be related to the issue. 

The read speed is still much lower than the write speed to the smb share hosted on the pi

popcornmix, 2021-03-22:
> And assuming that I did everything right

You should see a difference from `uname -a` (the date you compiled kernel, and the build number after the '#' will be small) if you are running the kernel you built.

iandk, 2021-03-22:
Looks good to me
`Linux raspberrypi 5.10.23-v7l+ #1 SMP Sat Mar 20 16:55:43 GMT 2021 armv7l GNU/Linux`

popcornmix, 2021-03-22:
Thanks for testing. It's ruled out one possibility.

jorensanbar, 2021-03-22:
> @jorensanbar your external drive is an SMR drive, that might be reinforce the speed issues for you

I have been my server for a year, and the speed has always been at maximum. It is from the updates since December that this is affected.

Although the disks are SMR, a speed of 100 MB / s does not have a considerable effect.

If I go back to Kernel 5.4, everything is corrected automatically.

[64 more replies]