raspberrypi/linux #4134: Wifi connection issues after upgrade to Linux version 5.10.11-v8+

mgrouch, 2021-02-09:
Much more difficult to connect to my home wifi.
I had to do 30 attempts or so.
I see error in dmesg log

Raspberry Pi4 8Gb Model

ERROR:

[  200.785149] v3d fec00000.v3d: MMU error from client L2T (0) at 0xea1000, pte invalid
[  217.256816] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[  259.634313] w1_master_driver w1_bus_master1: Attaching one wire slave 00.200000000000 crc 23
[  259.638797] w1_master_driver w1_bus_master1: Family 0 for 00.200000000000.23 is not registered.
[  275.368427] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[  275.942871] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[  296.273631] ieee80211 phy0: brcmf_update_bss_info: wl dtim_assoc failed (-52)
[  296.273973] ------------[ cut here ]------------
[  296.274097] WARNING: CPU: 2 PID: 912 at net/wireless/sme.c:757 __cfg80211_connect_result+0x3ac/0x440 [cfg80211]
[  296.274100] Modules linked in: fuse rfcomm aes_neon_blk crypto_simd cryptd bnep hci_uart btbcm bluetooth ecdh_generic ecc s
quashfs spidev joydev sg brcmfmac brcmutil sha256_generic hid_multitouch cfg80211 rfkill vc4 v3d cec gpu_sched drm_kms_helper 
raspberrypi_hwmon spi_bcm2835 drm drm_panel_orientation_quirks i2c_bcm2835 bcm2835_v4l2(C) videobuf2_vmalloc snd_soc_core bcm2
835_isp(C) bcm2835_codec(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem vc_sm_cma(C) videobuf2_dma_contig videobuf2_memops videobuf2_v4
l2 videobuf2_common videodev mc snd_compress snd_bcm2835(C) snd_pcm_dmaengine snd_pcm rpivid_mem snd_timer snd w1_gpio syscopy
area wire sysfillrect sysimgblt cn fb_sys_fops backlight nvmem_rmem uio_pdrv_genirq uio uinput ip_tables x_tables ipv6
[  296.274255] CPU: 2 PID: 912 Comm: kworker/u8:3 Tainted: G         C        5.10.11-v8+ #1399
[  296.274258] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[  296.274305] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[  296.274312] pstate: 60000005 (nZCv daif -PAN -UAO -TCO BTYPE=--)
[  296.274349] pc : __cfg80211_connect_result+0x3ac/0x440 [cfg80211]
[  296.274383] lr : __cfg80211_connect_result+0x21c/0x440 [cfg80211]
[  296.274386] sp : ffffffc011d63cc0
[  296.274389] x29: ffffffc011d63cc0 x28: 0000000000000000 
[  296.274396] x27: ffffff81039d56c8 x26: ffffff810ede1000 
[  296.274403] x25: dead000000000100 x24: dead000000000122 
[  296.274409] x23: ffffffc011d63d08 x22: ffffff8103d18000 
[  296.274415] x21: ffffffc011228948 x20: ffffff8103fe8e18 
[  296.274421] x19: ffffff8103bec008 x18: 0000000000000000 
[  296.274428] x17: 0000000000000000 x16: 0000000000000000 
[  296.274434] x15: 0000000000000000 x14: 0000087f00000000 
[  296.274440] x13: 0000000000000000 x12: 0000000000000028 
[  296.274446] x11: 0101010101010101 x10: 7f7f7f7f7f7f7f7f 
[  296.274452] x9 : ffffffc010085730 x8 : ffffff81332d2044 
[  296.274458] x7 : 0000000000000001 x6 : ffffffc011d63ba0 
[  296.274464] x5 : 0000000000000000 x4 : ffffff810eea9e40 
[  296.274471] x3 : ffffffc011229000 x2 : 0000000000000000 
[  296.274477] x1 : 0000000100000000 x0 : 0000000000000000 
[  296.274483] Call trace:
[  296.274521]  __cfg80211_connect_result+0x3ac/0x440 [cfg80211]
[  296.274556]  cfg80211_process_wdev_events+0x124/0x188 [cfg80211]
[  296.274591]  cfg80211_process_rdev_events+0x40/0x98 [cfg80211]
[  296.274625]  cfg80211_event_work+0x28/0x40 [cfg80211]
[  296.274632]  process_one_work+0x1f4/0x4c8
[  296.274636]  worker_thread+0x50/0x480
[  296.274641]  kthread+0x148/0x158
[  296.274645]  ret_from_fork+0x10/0x30
[  296.274649] ---[ end trace 4ff4410865608369 ]---
[  296.276210] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  318.252399] w1_master_driver w1_bus_master1: Attaching one wire slave 00.a00000000000 crc af
[  318.257037] w1_master_driver w1_bus_master1: Family 0 for 00.a00000000000.af is not registered.

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.10.11-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld 
(GNU Binutils for Ubuntu) 2.34) #1399 SMP PREEMPT Thu Jan 28 12:14:03 GMT 2021
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] efi: UEFI not found.

mgrouch, 2021-02-09:
[  296.273973] ------------[ cut here ]------------
[  296.274097] WARNING: CPU: 2 PID: 912 at net/wireless/sme.c:757 __cfg80211_connect_result+0x3ac/0x440 [cfg80211]
[  296.274100] Modules linked in: fuse rfcomm aes_neon_blk crypto_simd cryptd bnep hci_uart btbcm bluetooth ecdh_generic ecc s
quashfs spidev joydev sg brcmfmac brcmutil sha256_generic hid_multitouch cfg80211 rfkill vc4 v3d cec gpu_sched drm_kms_helper 
raspberrypi_hwmon spi_bcm2835 drm drm_panel_orientation_quirks i2c_bcm2835 bcm2835_v4l2(C) videobuf2_vmalloc snd_soc_core bcm2
835_isp(C) bcm2835_codec(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem vc_sm_cma(C) videobuf2_dma_contig videobuf2_memops videobuf2_v4
l2 videobuf2_common videodev mc snd_compress snd_bcm2835(C) snd_pcm_dmaengine snd_pcm rpivid_mem snd_timer snd w1_gpio syscopy
area wire sysfillrect sysimgblt cn fb_sys_fops backlight nvmem_rmem uio_pdrv_genirq uio uinput ip_tables x_tables ipv6
[  296.274255] CPU: 2 PID: 912 Comm: kworker/u8:3 Tainted: G         C        5.10.11-v8+ #1399
[  296.274258] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[  296.274305] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[  296.274312] pstate: 60000005 (nZCv daif -PAN -UAO -TCO BTYPE=--)
[  296.274349] pc : __cfg80211_connect_result+0x3ac/0x440 [cfg80211]
[  296.274383] lr : __cfg80211_connect_result+0x21c/0x440 [cfg80211]
[  296.274386] sp : ffffffc011d63cc0
[  296.274389] x29: ffffffc011d63cc0 x28: 0000000000000000 
[  296.274396] x27: ffffff81039d56c8 x26: ffffff810ede1000 
[  296.274403] x25: dead000000000100 x24: dead000000000122 
[  296.274409] x23: ffffffc011d63d08 x22: ffffff8103d18000 
[  296.274415] x21: ffffffc011228948 x20: ffffff8103fe8e18 
[  296.274421] x19: ffffff8103bec008 x18: 0000000000000000 
[  296.274428] x17: 0000000000000000 x16: 0000000000000000 
[  296.274434] x15: 0000000000000000 x14: 0000087f00000000 
[  296.274440] x13: 0000000000000000 x12: 0000000000000028 
[  296.274446] x11: 0101010101010101 x10: 7f7f7f7f7f7f7f7f 
[  296.274452] x9 : ffffffc010085730 x8 : ffffff81332d2044 
[  296.274458] x7 : 0000000000000001 x6 : ffffffc011d63ba0 
[  296.274464] x5 : 0000000000000000 x4 : ffffff810eea9e40 
[  296.274471] x3 : ffffffc011229000 x2 : 0000000000000000 
[  296.274477] x1 : 0000000100000000 x0 : 0000000000000000 
[  296.274483] Call trace:
[  296.274521]  __cfg80211_connect_result+0x3ac/0x440 [cfg80211]
[  296.274556]  cfg80211_process_wdev_events+0x124/0x188 [cfg80211]
[  296.274591]  cfg80211_process_rdev_events+0x40/0x98 [cfg80211]
[  296.274625]  cfg80211_event_work+0x28/0x40 [cfg80211]
[  296.274632]  process_one_work+0x1f4/0x4c8
[  296.274636]  worker_thread+0x50/0x480
[  296.274641]  kthread+0x148/0x158
[  296.274645]  ret_from_fork+0x10/0x30
[  296.274649] ---[ end trace 4ff4410865608369 ]---

solsticedhiver, 2021-02-11:
On a rpi3b, using 5.10.14-1-ARCH

```
ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
------------[ cut here ]------------
WARNING: CPU: 2 PID: 7 at kernel/workqueue.c:1477 __queue_work+0x4ac/0x628
Modules linked in: vc4 brcmfmac cec brcmutil evdev joydev drm_kms_helper cfg80211 snd_soc_core bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_codec(C) snd_compress bcm2835_mmal_vchiq(C) v4l2_mem2mem snd_pcm_dmaengi>
CPU: 2 PID: 7 Comm: kworker/u8:0 Tainted: G         C        5.10.14-1-ARCH #1
Hardware name: BCM2835
Workqueue: brcmf_wq/mmc1:0001:1 brcmf_sdio_dataworker [brcmfmac]
[<8010ef8c>] (unwind_backtrace) from [<8010ae04>] (show_stack+0x10/0x14)
[<8010ae04>] (show_stack) from [<80d03c88>] (dump_stack+0xd0/0xf8)
[<80d03c88>] (dump_stack) from [<8012bd1c>] (__warn+0xd4/0x140)
[<8012bd1c>] (__warn) from [<80cfb694>] (warn_slowpath_fmt+0x74/0xa8)
[<80cfb694>] (warn_slowpath_fmt) from [<80147154>] (__queue_work+0x4ac/0x628)
[<80147154>] (__queue_work) from [<80147328>] (queue_work_on+0x58/0x60)
[<80147328>] (queue_work_on) from [<7f4011b0>] (brcmf_sdio_dataworker+0x630/0x2148 [brcmfmac])
[<7f4011b0>] (brcmf_sdio_dataworker [brcmfmac]) from [<80147a70>] (process_one_work+0x204/0x53c)
[<80147a70>] (process_one_work) from [<80147df0>] (worker_thread+0x48/0x594)
[<80147df0>] (worker_thread) from [<8014dca8>] (kthread+0x130/0x144)
[<8014dca8>] (kthread) from [<80100138>] (ret_from_fork+0x14/0x3c)
Exception stack(0x818f5fb0 to 0x818f5ff8)
5fa0:                                     00000000 00000000 00000000 00000000
5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
5fe0: 00000000 00000000 00000000 00000000 00000013 00000000
---[ end trace e9640d3be52731ef ]---

```

mdevaev, 2021-02-12:
I'm also getting this on ZeroW:
```
[   21.668195] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   22.486482] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   22.543932] usbcore: registered new interface driver brcmfmac
[   22.645834] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.txt failed with error -2
[   22.861081] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   22.998763] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
[   23.008901] ------------[ cut here ]------------
[   23.016355] WARNING: CPU: 0 PID: 7 at kernel/workqueue.c:1477 __queue_work+0x4ac/0x628
[   23.029902] Modules linked in: brcmfmac brcmutil i2c_mux_pinctrl i2c_mux cfg80211 raspberrypi_hwmon bcm2835_unicam bcm2835_codec(C) bcm2835_v4l2(C)
 bcm2835_isp(C) i2c_bcm2835 bcm2835_mmal_vchiq(C) bcm2835_rng rng_core v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_
v4l2 videobuf2_common uio_pdrv_genirq fixed uio sch_fq_codel snd_bcm2835(C) snd_pcm snd_timer snd tc358743 v4l2_dv_timings v4l2_fwnode videodev mc cec libcomposite dwc2 udc_core roles fuse drm drm_panel_orientation_quirks ip_tables x_tables ipv6
[   23.097859] CPU: 0 PID: 7 Comm: kworker/u2:0 Tainted: G         C        5.10.14-2-ARCH #1
[   23.111946] Hardware name: BCM2835
[   23.118507] Workqueue: brcmf_wq/mmc1:0001:1 brcmf_sdio_dataworker [brcmfmac]
[   23.128624] [<8010ef8c>] (unwind_backtrace) from [<8010ae04>] (show_stack+0x10/0x14)
[   23.142290] [<8010ae04>] (show_stack) from [<80d03c88>] (dump_stack+0xd0/0xf8)
[   23.155447] [<80d03c88>] (dump_stack) from [<8012bd1c>] (__warn+0xd4/0x140)
[   23.165498] [<8012bd1c>] (__warn) from [<80cfb694>] (warn_slowpath_fmt+0x74/0xa8)
[   23.178925] [<80cfb694>] (warn_slowpath_fmt) from [<80147154>] (__queue_work+0x4ac/0x628)
[   23.193056] [<80147154>] (__queue_work) from [<80147328>] (queue_work_on+0x58/0x60)
[   23.206868] [<80147328>] (queue_work_on) from [<7f3b31b0>] (brcmf_sdio_dataworker+0x630/0x2148 [brcmfmac])
[   23.222764] [<7f3b31b0>] (brcmf_sdio_dataworker [brcmfmac]) from [<80147a70>] (process_one_work+0x204/0x53c)
[   23.238646] [<80147a70>] (process_one_work) from [<80147df0>] (worker_thread+0x48/0x594)
[   23.252722] [<80147df0>] (worker_thread) from [<8014dca8>] (kthread+0x130/0x144)
[   23.266109] [<8014dca8>] (kthread) from [<80100138>] (ret_from_fork+0x14/0x3c)
[   23.279321] Exception stack(0x818d3fb0 to 0x818d3ff8)
[   23.287392] 3fa0:                                     00000000 00000000 00000000 00000000
[   23.301583] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   23.315783] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[   23.325456] ---[ end trace f77024c6129753fd ]---
```

pavel-cernohorsky, 2021-06-09:
Happening here as well. WiFi connection is extremely unstable, still re-connects and kernel log full of these:
```
[Wed Jun  9 18:52:54 2021] ieee80211 phy0: brcmf_update_bss_info: wl dtim_assoc failed (-52)
[Wed Jun  9 18:52:54 2021] ------------[ cut here ]------------
[Wed Jun  9 18:52:54 2021] WARNING: CPU: 3 PID: 4121 at net/wireless/sme.c:757 __cfg80211_connect_result+0x3ac/0x440 [cfg80211]
[Wed Jun  9 18:52:54 2021] Modules linked in: algif_hash algif_skcipher af_alg bnep hci_uart btbcm bluetooth ecdh_generic ecc xt_conntrack xt_MASQUERADE nf_conntrack_netlink nft_counter xt_addrtype nft_compat nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables nfnetlink br_netfilter bridge overlay 8021q garp stp llc vc4 cec drm_kms_helper brcmfmac snd_soc_core brcmutil snd_compress snd_pcm_dmaengine snd_bcm2835(C) snd_pcm bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) v3d snd_timer bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_vmalloc cfg80211 snd videobuf2_dma_contig gpu_sched syscopyarea videobuf2_memops videobuf2_v4l2 sysfillrect rfkill sysimgblt videobuf2_common fb_sys_fops drm vc_sm_cma(C) videodev mc raspberrypi_hwmon rpivid_mem drm_panel_orientation_quirks backlight nvmem_rmem uio_pdrv_genirq uio i2c_dev aes_neon_bs sha256_generic aes_neon_blk crypto_simd cryptd ip_tables x_tables ipv6
[Wed Jun  9 18:52:54 2021] CPU: 3 PID: 4121 Comm: kworker/u8:3 Tainted: G        WC        5.10.17-v8+ #1421
[Wed Jun  9 18:52:54 2021] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[Wed Jun  9 18:52:54 2021] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[Wed Jun  9 18:52:54 2021] pstate: 60000005 (nZCv daif -PAN -UAO -TCO BTYPE=--)
[Wed Jun  9 18:52:54 2021] pc : __cfg80211_connect_result+0x3ac/0x440 [cfg80211]
[Wed Jun  9 18:52:54 2021] lr : __cfg80211_connect_result+0x21c/0x440 [cfg80211]
[Wed Jun  9 18:52:54 2021] sp : ffffffc01241bcc0
[Wed Jun  9 18:52:54 2021] x29: ffffffc01241bcc0 x28: 0000000000000000 
[Wed Jun  9 18:52:54 2021] x27: ffffff8126a1a448 x26: ffffff8105a17100 
[Wed Jun  9 18:52:54 2021] x25: dead000000000100 x24: dead000000000122 
[Wed Jun  9 18:52:54 2021] x23: ffffffc01241bd08 x22: ffffff8103877000 
[Wed Jun  9 18:52:54 2021] x21: ffffffc011238948 x20: ffffff811764e818 
[Wed Jun  9 18:52:54 2021] x19: ffffff8106bc8008 x18: 0000000000000000 
[Wed Jun  9 18:52:54 2021] x17: 0000000000000000 x16: 0000000000000000 
[Wed Jun  9 18:52:54 2021] x15: 0000000000000000 x14: 02080004087f0000 
[Wed Jun  9 18:52:54 2021] x13: 0000000000000000 x12: 0000000000000028 
[Wed Jun  9 18:52:54 2021] x11: 0101010101010101 x10: 7f7f7f7f7f7f7f7f 
[Wed Jun  9 18:52:54 2021] x9 : ffffffc010084fa8 x8 : ffffff8106ac8044 
[Wed Jun  9 18:52:54 2021] x7 : 0000000000000001 x6 : ffffffc01241bba0 
[Wed Jun  9 18:52:54 2021] x5 : 0000000000000000 x4 : ffffff810bb4bc80 
[Wed Jun  9 18:52:54 2021] x3 : ffffffc011239000 x2 : 0000000000000000 
[Wed Jun  9 18:52:54 2021] x1 : 0000000100000000 x0 : 0000000000000000 
[Wed Jun  9 18:52:54 2021] Call trace:
[Wed Jun  9 18:52:54 2021]  __cfg80211_connect_result+0x3ac/0x440 [cfg80211]
[Wed Jun  9 18:52:54 2021]  cfg80211_process_wdev_events+0x124/0x188 [cfg80211]
[Wed Jun  9 18:52:54 2021]  cfg80211_process_rdev_events+0x40/0x98 [cfg80211]
[Wed Jun  9 18:52:54 2021]  cfg80211_event_work+0x28/0x40 [cfg80211]
[Wed Jun  9 18:52:54 2021]  process_one_work+0x1f4/0x4d8
[Wed Jun  9 18:52:54 2021]  worker_thread+0x50/0x480
[Wed Jun  9 18:52:54 2021]  kthread+0x148/0x158
[Wed Jun  9 18:52:54 2021]  ret_from_fork+0x10/0x30
[Wed Jun  9 18:52:54 2021] ---[ end trace f76d162904a53f6a ]---
```
Using Raspberry Pi 4 8G, 64bit OS with `Linux 5.10.17-v8+ #1421 SMP PREEMPT Thu May 27 14:01:37 BST 2021 aarch64 GNU/Linux`.
Any help would be appreciated, this makes RPi practically unusable.

JamesH65, 2021-06-09:
Hmm, just saw a very similar issue on 4.19, when connecting to the Pi towers  guest network. Will report back to the network gurus.

JamesH65, 2021-06-10:
[image]

@pelwell

mgrouch, 2021-09-24:
It was due to not setting Wi-Fi Country

closed 2021-09-24 by mgrouch