raspberrypi/linux #4136: Kernel 5.10 - RPI1 + Kodi no longer usable because the connection to the screen always breaks down

mkreisl, 2021-02-09:
For days I've been trying in vain to get the kernel 5.10 (currently .14) and Kodi to work.

The connection to the screen (HDMI 1080p 50 or 60Hz) keeps breaking down and the TV finally reports 'No signal'.
Unfortunately there is nothing in the logs (neither dmesg nor syslog reports anything).

Using kernel 5.4 everything works fine of course. The 5.10 kernel was built with bcmrpi_defconfig, only BTRFS is builtin because BTRFS is used as root filesystem.

Trying with cma=xxx did not change anything

Did I miss something or is the kernel 5.10 currently really almost unusable?
(see also https://github.com/raspberrypi/linux/issues/4112)

micha53, 2021-02-09:
I also made an apt-get update/upgrade at my RPI4 on Feb.. 6th, 2021 with the result that kodi does not start anymore at all on this device. There comes a black screen, the trial to stop kodi  "sudo killall kodi" also makes no change, the black screen remains and I am forced to reboot the raspi via ssh.
No kodi crashlog to find.

popcornmix, 2021-02-09:
@mkreisl just to confirm you are talking about Pi1?
Is this kodi 18? Using legacy driver (no kms or fkms)?
Does screen break down when playing video or in gui?
Does 1080p24 or 720p50/60 work better?

mkreisl, 2021-02-09:
Yes, Pi1
It is Kodi v18
It seems that screen starts to flicker after a hwaccel video has been played. Doing this before  Kodi starts
Screen breaks while Kodi is idle. Playing vid inside of Kodi not necessary

720p/50 and 1080p/24 seems to work and the screen flickers as soon as I switch back to 1080p/50. Maybe something is wrong with the frequencies

And, before you're asking:

```
vcgencmd version
Jan  7 2021 18:34:00 
Copyright (c) 2012 Broadcom
version fb345a0c2d5544957f4ba1a2b9e968970e3312c4 (clean) (release) (start_x)
```

And already tried different HDMI ports and HDMI cables

popcornmix, 2021-02-09:
The kernel really has very little to do with HDMI when using the firmware/legacy driver.
Are you sure it was the kernel update rather than the firmware that caused this?

mkreisl, 2021-02-09:
> Are you sure it was the kernel update rather than the firmware that caused this?

Absolute sure. And it happens on 2 different PI1

As soon as I reinstall the 5.4 kernel the problems are gone immediately. I have done this several times now