raspberrypi/linux #4139: Kernel update 5.10.11 breaks Bluetooth LE binding / paring

JohnRucker, 2021-02-11:
After my Raspberry Pi 4B and Raspberry Pi Zero devices received the 5.10.11-v7l+ #1399 kernel update dated Thu Jan 28, 2021 they stoped pairing with my iPhone.

I have an app that sets up the Raspberry Pi 4B as a bluetooth peripheral. It has worked pretty much flawlessly for years on the Pi Zero and more recently on the Pi 4b. After the auto update to the 5.10.11 kernel I can no longer bind with my iPhone. If I roll back to the 5.4.83 #1379 kernel dated Mon Dec 14, 2020 everything works just fine. The only thing that doesn't seem to work is the binding process. If the phone is already bound before the update everything continues to work just fine. I just happened to be updating my documentation on the bind process and found the error. 

Btmon shows an error of:
Code: Select all

`= bluetoothd: No agent available for request type 2`                 
`= bluetoothd: device_confirm_passkey: Operation not permitted`

I have always used the default agent to bind the devices and now it doesn't handle this request. I'm not sure if the issue is with the kernel or Bluetoothd as they were both updated in this release. 

Any help would be appreciated.

pelwell, 2021-02-11:
How are you rolling back the kernel?

You can try using `rpi-update` to replace just the kernel and firmware:
```
# To install the Dec 14th 5.4.83 kernel & FW
$ sudo rpi-update 453e49bdd873

# To install the Jan 28th 5.10.11 kernel & FW
$ sudo rpi-update 43998c82a7e8
```
Compare old bluetoothd and new kernel with new bluetoothd and old kernel. My money is on the bluetoothd change, but I could be wrong.

JohnRucker, 2021-02-11:
I rolled back my Pi 4b by reinstalling the Dec 14 OS from SD card and then my app on top of that.  I'm able to bind with my iPhone and iPad as expected.

I then updated only the kernel as you suggested with the `sudo rpi-update 43998c82a7e8` command and rebooted.  Now when I trie to pair the iPhone or iPad I still get the same error:

`= bluetoothd: No agent available for request type 2`                                                                                                                                                            
`= bluetoothd: device_confirm_passkey: Operation not permitted `

I did a `bluetoothd -v` just to check and the version is 5.50

I then rolled back to Dec 14th 5.4.83 kernel & FW using the `sudo rpi-update 453e49bdd873` command, rebooted and now I can pair and bind my phone and iPad as before. 

So it looks like it is a kernel thing??

I will be happy to help an any way I can.  This is going to shut down my project so it has my attention.

JohnRucker, 2021-02-19:
Looks like there is a known bug [(Bug 210684)](https://bugzilla.kernel.org/show_bug.cgi?id=210681) in the kernel that is causing issues.  However my exact issue is not the same.  I'm starting to hear from other people that they are seeing this same thing I'm seeing on other forums.  I wonder if the bug is related to my issue.

JohnRucker, 2021-02-20:
Just confirmed this problem still exist in 5.10.16 kernel.

pelwell, 2021-02-22:
It looks like something got broken around 5.8. If the device is paired already it seems to reconnect successfully, but new pairing fail.

pelwell, 2021-03-01:
A whole lotta bisecting identified the commit that (in my testing) causes BLE pairing to fail - https://github.com/raspberrypi/linux/commit/92516cd97fd4d8ad5b1421a0d51771044f453a5f. It prompts a call to the userspace Bluetooth daemon for "Just Works" pairing, where previously the pairing would have been automatically approved. After reverting the commit (and a close relative - https://github.com/raspberrypi/linux/commit/ffee202a78c2980688bc5d2f7d56480e69a5e0c9), pairing work again, but the long-term solution is likely to be a userspace change (either BlueZ or the Bluetooth GUI elements).

chewitt, 2021-03-01:
@pelwell Since moving from LE 9.2.6 (4.19 kernel, older BlueZ) to LE master (5.10 kernel, latest BlueZ) I'd felt the behaviour of my BLE remote control device was very sluggish at reconnecting compared to before. Setting JustWorksRepairing=always in main.conf feels like it restores the previous behaviour. This has been bugging me during LE10 testing (more than all the other bugs) so thanks for doing the bisect work to flag the issue! :)

pelwell, 2021-03-01:
> Setting JustWorksRepairing=always in main.conf feels like it restores the previous behaviour.

Thanks - that's a useful pointer.

JohnRucker, 2021-03-01:
I will give the JustWorksRepairing=always a try.  But based on the reading I have been doing that fixes a repairing issue not my initial pairing issue.  Give me a few minutes to set everything up reporting back in a few....

HiassofT, 2021-03-01:
FYI: you'll need a quite recent bluez version. Just-Works policies were added in 5.54 and 5.55 included a fx for auto-accept

JohnRucker, 2021-03-01:
Oh crap.  I didn't see it was in the 5.54 version.  I'm testing against version 5.10.16-v7l of the kernel and its Bluetoothd -v reports 5.50 so 5.54 is way off in the future. 

So if I understand things correctly @pelwell (Phil) has submitted a commit that changes this behavior for 5.50 and reverts it back to the way it was before the problem appeard (pre 5.8)?

popcornmix, 2021-03-01:
Yes, a temporary fix (reverting undesired upstream changes) has been pushed to rpi-5.10.y tree.
That will appear in next rpi-update kernel (sometime this week).

pelwell, 2021-03-01:
We're looking at the GUI authentication agent - there's the suspicion that a bit of XML went astray.

closed 2021-03-09 by JohnRucker

pelwell, 2021-03-09:
Note that a pending (? it might have appeared somewhere by now) update to the Bluetooth plugin will prompt the user correctly, and a future BlueZ change should automatically accept the pairing when it has been instigated by the user, which is a triple fix.

JohnRucker, 2021-03-09:
Good to know!  Thanks @pelwell for taking the time to fix this.  This was a huge deal for my project.

Ionwerks, 2021-05-23:
Appreciate posting to a closed issue may not be the proper process but adding to the existing problem seems like a good first gambit.

Found myself in a similar situation to @JohnRucker - with a simple BT application using default agents which had been working fine on a Pi Zero W but, since moving to OS-Lite image 2021-03-04 (kernel 5.10.17), fails to complete a pairing with a device which had previously paired with my implementation and then removed it (presume an auto-generated PIN/PassKey has been cached). As with the original issue above, BTMon shows;

= bluetoothd: No agent available for request type 2
= bluetoothd: device_confirm_passkey: Operation not permitted 

After reading this thread I'd hoped an _rpi-update_ to 5.10.36+ would resolve the problem but the issue persisted, after manually removing the paired device MAC, pairing again, forgetting and then attempting to re-pair.

So I built a new image based on 2021-01-11 (kernel 5.4.83) and this behaviour is gone, I can pair, forget and re-pair with my device at will. This leads me to believe the issue has not yet been resolved or there has been a regression since 5.10.20.

eric1tran, 2021-10-02:
I'm attempting to connect to my Raspberry Pi from my iOS app/LightBlue and whenever the iOS Bluetooth Pairing Request pop up appears and I click "Pair", I see the following error in the bluetooth logs:

`bluetoothd[502]: No agent available for request type 2`
`bluetoothd[502]: device_confirm_passkey: Operation not permitted`
`kernel: [58168.506508] Bluetooth: hci0: unexpected SMP command 0x0d from 14:60:cb:00:ae:cb`

On kernel 5.10.60-v7l+ # 1449

The connection seems to work ok most of the time but seeing I'm some instability. I

Should I be still be seeing these errors after this fix? Does anyone have insight into the "unexpected SMP command 0x0d" error?