raspberrypi/linux #4142: hifiberry digi not recognized when using vc4-kms-v3d

amverweij, 2021-02-12:
**Describe the bug**
My hifiberry digi card is not recognized when I use the vc4-kms-v3d overlay. 

**To reproduce**
Using this config.txt:
`disable_overscan=1

dtoverlay=rpivid-v4l2
gpu_mem=128

dtparam=krnbt=on
dtoverlay=disable-wifi

#dtoverlay=vc4-fkms-v3d,cma-512
#dtparam=audio=on
dtoverlay=vc4-kms-v3d,cma-512

arm_64bit=1

force_eeprom_read=0
dtoverlay=hifiberry-digi
`

only yields vc4-hdmi audio. aplay -l gives:

`**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM vc4-hdmi-hifi-0 [MAI PCM vc4-hdmi-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM vc4-hdmi-hifi-0 [MAI PCM vc4-hdmi-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
`

Strangely enough, when using the fkms overlay with the following config.txt:
`disable_overscan=1

dtoverlay=rpivid-v4l2
gpu_mem=128

dtparam=krnbt=on
dtoverlay=disable-wifi

dtoverlay=vc4-fkms-v3d,cma-512
dtparam=audio=on
#dtoverlay=vc4-kms-v3d,cma-512

arm_64bit=1

force_eeprom_read=0
dtoverlay=hifiberry-digi
`

the hifiberry digi is recognized as expected:

`**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 3/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 2: sndrpihifiberry [snd_rpi_hifiberry_digi], device 0: HifiBerry Digi HiFi wm8804-spdif-0 [HifiBerry Digi HiFi wm8804-spdif-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
`

**Expected behaviour**
I expect to be able to use the hifiberry digi card in combination with the vc4-kms-v3d overlay.

**Actual behaviour**
When using the vc4-kms-v3d overlay, my hifi berry card is not recognized by the OS.

**System**

This is on a Pi4B.

Gentoo Linux, 64bit

uname -a:
Linux herbie 5.10.14-v8+ #1 SMP PREEMPT Fri Feb 12 17:29:02 CET 2021 aarch64 GNU/Linux

vcgencmd version:
Feb  8 2021 14:32:22 
Copyright (c) 2012 Broadcom
version 73b3cad64181954e67f6e9fe6d275378d3eefa10 (clean) (release) (start)

**Logs**
On the fkms overlay, dmesg | grep 8804 gives
[   12.678525] wm8804 1-003b: revision E

On the kms overlay, there is nothing related to the hifiberry digi in dmesg.

**Additional context**
I use the pi as a multimedia PC with Kodi, and the Kodi guys seem to prefer the kms overlay. I'm experiencing video tearing so I'd like to follow suit.

Libreelec nightly build (LibreELEC-RPi4.arm-9.80-nightly-20210209-06bafb5.img) does recognize the hifiberry digi when using the kms overlay;
Raspberry Pi OS (64 bit) does recognize the hifiberry digi when using the kms overlay as well.

I'm building the kernel with "make ARCH=arm64 CROSS_COMPILE=aarch64-unknown-linux-gnu- bcm2711_defconfig".

6by9, 2021-02-12:
#4132
An upstream change resulted in i2c0 and i2c1 not being able to claim their IRQ. It's always best to provide a full dmesg log as that would have been flagged in there.

I've created #4143 to revert those changes, and the same is happening upstream.

amverweij, 2021-02-12:
Thank you for the quick feedback. I've applied your change to my kernel build. I can confirm that with this change, my hifiberry-digi sound card is fully function in combination with the vc4-kms-v3d overlay.

closed 2021-02-12 by amverweij