raspberrypi/linux #4144: Several ipv6_ modules are missing from the RPI kernel in the distro

owendelong, 2021-02-13:
**Is this the right place for my bug report?**
Putting it here based on feedback from XECDesign in https://github.com/RPi-Distro/repo/issues/229

**Describe the bug**
The following modules are missing from the RPI kernel in the Raspberry PI OS distribution, but are present in every other Linux distribution I examined (Fedora, Debian, Ubuntu, VyOS).
/lib/modules/<version>/kernel/net/

From .../ipv6:
esp6_offload.ko
fou6.ko
ila/ila.ko
ip6_gre.ko
ip6_vti.ko

And from .../ipv6/netfilter:
ip6table_security.ko
ip6t_srh.ko
ip6t_SYNPROXY.ko

**To reproduce**
cd /lib/modules/<kernel_version>/kernel/net/ipv6
ls

**Expected behaviour**
Expect the modules to be present. They are necessary to support various IPv6 functionality, such as GRE tunnels using IPv6 transport.

**Actual behaviour**
Some modules are present, but the ones above are missing.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW -- I noticed this on a Pi4, but all models are equally affected.
* Which OS and version (`cat /etc/rpi-issue`)?  -- Included below, but all are likely affected

> Raspberry Pi reference 2020-05-27
> Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ee22f, stage5

* Which firmware version (`vcgencmd version`)? -- Included below, but all are affected identically (kernel module choices)

> Sep 2 2020 21:13;13
> Copyright (c) 2012 Broadcom
> version 4439d2aaa6c376a2d1ef4402f142e1cf4de37c43 (clean) (release) (start)

* Which kernel version (`uname -a`)?

> Linux <hostname> 5.4.51-v7l+ #2 SMP

**Logs**
Not relevant... This is about kernel modules which were omitted by choice.

**Additional context**
I spent quite a bit of time chasing configuration issues where I thought it was a failed dependency tree trying to bring up an ip6gre tunnel on the system. systemd-networkd silently reports that it is loading the Ip6_gre module without reporting any failure to load it and all of the available debugging information makes it look like everything is fine and hints that the problem is that the GRE interface in question is coming up before the ethernet interface it depends on. Turns out that this is false and that the module not present to load is the real issue. Upon compiling a new kernel with the modules included (admittedly based on the 5.10.13 sources), everything immediately came up as expected after a reboot.

pelwell, 2021-02-13:
Why not list the actual config changes required, or submit a Pull Request?

owendelong, 2021-02-16:
I don't know what they are... I know how to run menuconfig and build a new kernel with the correct changes, but have no familiarity with how the default configuration is constructed or where changes need to be applied in order to achieve that.

As such, I figured I would report the issue and name the modules in question in hopes that those far better versed in what is required to do this correctly could easily accomplish it.

Given that every other distro ships with these modules I'm figuring someone had to deliberately disable their inclusion at some point.

pelwell, 2021-02-16:
Using `diff` on the `.config` before and after making changes with the menu will show the relevant config settings - it removes a source of potential errors and puts some of the burden on the requester.

> Given that every other distro ships with these modules I'm figuring someone had to deliberately disable their inclusion at some point.

That's not the case - we started out with relatively minimal configurations (nearly 10 years ago now) and have gradually expanded them as necessary in order to keep the footprint modest.

pelwell, 2021-02-16:
The following config options seem to be required:
```
CONFIG_INET6_ESP_OFFLOAD=m
CONFIG_NET_FOU=m
CONFIG_IPV6_ILA=m
CONFIG_IPV6_GRE=m
CONFIG_IPV6_VTI=m
CONFIG_IP6_NF_SECURITY=m
CONFIG_IP6_NF_MATCH_SRH=m
CONFIG_IP6_NF_TARGET_SYNPROXY=m
```
which results in the following extra modules (with sizes):
```
 10372 esp6_offload.ko
  3396 fou6.ko
 40636 ip6_gre.ko
 25920 ip6_vti.ko
 30248 ila/ila.ko
  6260 netfilter/ip6table_security.ko
  8688 netfilter/ip6t_srh.ko
  7068 netfilter/ip6t_SYNPROXY.ko
```
Are you OK with an extra 130KB of modules, @popcornmix?

pelwell, 2021-02-17:
Updated defconfigs have been pushed to rpi-5.10.y and rpi-5.11.y.