raspberrypi/linux #4147: Kodi has no screen output at rpi4b

micha53, 2021-02-15:
starting kodi as normal user at rpi4 only a black screen appears instead of the kodi start screen.
starting kodi with command **sudo kodi** the normal screen appears.

After downgrading to kernel 5.4.50 everything works again, kodi starts with his normal screen.

**System**
rpi4b, Raspbian OS buster, error came with kernel 5.10.xx update

no kodi crashlog appears in this context, but in syslog I can see 

> Feb 14 12:25:42 Medienserver tvheadend[699]: htsp: Got connection from 192.168.178.43
> Feb 14 12:25:42 Medienserver tvheadend[699]: htsp: 192.168.178.43: Welcomed client software: Kodi Media Center (HTSPv34)
> Feb 14 12:25:42 Medienserver tvheadend[699]: htsp: 192.168.178.43 [ Kodi Media Center ]: Identified as user 'micha'
> Feb 14 12:25:42 Medienserver tvheadend[699]: htsp: 192.168.178.43 [ micha | Kodi Media Center ]: Privileges updated

so kodi seems to work, but without sudo it has no right to access the graphics output...?

popcornmix, 2021-02-15:
Can you post link to ~/.kodi/temp/kodi.log (e.g. on pastebin type site). Also dmesg output.

micha53, 2021-02-15:
dmsg is from after downgrade to 5.4. The logs probably too, not sure.
[dmesg.txt](https://github.com/raspberrypi/linux/files/5983535/dmesg.txt)
[kodi.log](https://github.com/raspberrypi/linux/files/5983537/kodi.log)
[kodi.old.log](https://github.com/raspberrypi/linux/files/5983538/kodi.old.log)

popcornmix, 2021-02-15:
Logs are only useful if they show the issue (i.e. on 5.10 kernel).

micha53, 2021-02-15:
I agree, but that is all I find, 'cause I turned back my system today to verify that the error came with 5.10.
But at least kodi.old.log is from 5.10. 
I made the downgrade between 3:00 and 4:00 p.m.

zito, 2021-02-15:
I can confirm problems too with kodi playback of some TV streams with vc4-fkms-v3d. Downgrade back to 5.4.83 solves the problem. I run kodi standalone as service:

> root@pi4:/home/kodi/.kodi/temp# cat /etc/systemd/system/kodi.service
> [Unit]
> Description = Kodi Media Center
> After = remote-fs.target network-online.target
> Wants = network-online.target
> 
> [Service]
> User = kodi
> Group = kodi
> Type = simple
> ExecStart = /usr/bin/kodi-standalone
> Restart = on-abort
> RestartSec = 5
> 
> [Install]
> WantedBy = multi-user.target
> 

stopping the service takes long time. Starting again and trying to switch channels leads to distorted graphics
and the whole system hangs finally. I can see disk activity.
I cannot find anything usable in logs :-(

Hmm, now I tried to run kodi with debug:

> root@pi4:~# sudo -u kodi /usr/lib/arm-linux-gnueabihf/kodi/kodi.bin_v8 --standalone --debug
> Cannot connect to server socket err = No such file or directory
> Cannot connect to server request channel
> jack server is not running or cannot be started
> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
> 
> 
> rpi_ctrl_ffmpeg_init
>  id=0xb0a5e770
> setup_io: 00000000 -> 0xa5247000 (fd:34)
> setup_io: 00000000 -> 0xa5237000 (fd:39)
> ***** gpu_malloc_uncached_internal, 67108864
> A 100000000 apb:0xa5247000 axi.arm:0x8a6fb000 axi.vc:20700000
> 
>

zito, 2021-02-15:
Hmm, I have done downgrade to 5.4.83 and kodi debug output is the same:

> root@pi4:~# uname -a
> Linux pi4 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux
> root@pi4:~# sudo -u kodi /usr/lib/arm-linux-gnueabihf/kodi/kodi.bin_v8 --standalone --debug
> Cannot connect to server socket err = No such file or directory
> Cannot connect to server request channel
> jack server is not running or cannot be started
> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
> rpi_ctrl_ffmpeg_init
>  id=0x42fe0c0
> setup_io: 00000000 -> 0x96902000 (fd:27)
> setup_io: 00000000 -> 0x94406000 (fd:44)
> ***** gpu_malloc_uncached_internal, 67108864
> A 100000000 apb:0x96902000 axi.arm:0x8b6fe000 axi.vc:3a2c2000

So nothing interesting there. :-(
Kodi works ok above kernel 5.4.83...

There is some ugly problem with 5.10.x

zito, 2021-02-16:
I recorded a short sample of "safe" source channel and a sample of "problem" source channel.

- [Udalosti v regionech.ts](https://drive.google.com/file/d/12bvwrXsRFS0IIKjoHWoWEFbNrQlDmv_U/view?usp=sharing) (safe)
> Input #0, mpegts, from 'Udalosti v regionech.ts':
>   Duration: 00:00:05.23, start: 66887.062511, bitrate: 3468 kb/s
>   Program 1 
>     Metadata:
>       service_name    : CT 1 HD T2
>       service_provider: CESKA TELEVIZE
>     Stream #0:1[0x83e]: Video: hevc (Main) ([36][0][0][0] / 0x0024), yuv420p(tv, bt709), 1920x1080 [SAR 1:1 DAR 16:9], 50 fps, 50 tbr, 90k tbn, 50 tbc
>     Stream #0:2[0x848](cze): Audio: aac_latm (LC) ([17][0][0][0] / 0x0011), 48000 Hz, stereo, fltp
>     Stream #0:3[0x849](qaa): Audio: aac_latm (LC) ([17][0][0][0] / 0x0011), 48000 Hz, stereo, fltp
>     Stream #0:4[0x84a](cze): Audio: ac3 (AC-3 / 0x332D4341), 48000 Hz, stereo, fltp, 448 kb/s
>     Stream #0:5[0x84b](cze): Audio: aac_latm (LC) ([17][0][0][0] / 0x0011), 48000 Hz, mono, fltp (visual impaired) (descriptions) (dependent)
>     Stream #0:6[0x852](cze): Subtitle: dvb_teletext ([6][0][0][0] / 0x0006)
>     Stream #0:7[0x866](cze): Subtitle: dvb_subtitle ([6][0][0][0] / 0x0006) (hearing impaired)
>   No Program
>     Stream #0:0[0x12]: Data: epg
> 

- [Slunce, seno, erotika.ts](https://drive.google.com/file/d/1Y1jrRNT4lx8KvcIUeSfZ1bC40LDK2pqP/view?usp=sharing) (problem)

> Input #0, mpegts, from 'Slunce, seno, erotika.ts':
>   Duration: 00:00:15.03, start: 40957.720178, bitrate: 1934 kb/s
>   Program 1 
>     Metadata:
>       service_name    : Nova Cinema
>       service_provider: CDG
>     Stream #0:1[0xc94]: Video: hevc (Main) ([36][0][0][0] / 0x0024), yuv420p(tv), 960x540 [SAR 1:1 DAR 16:9], 50 fps, 50 tbr, 90k tbn, 50 tbc
>     Stream #0:2[0xc95](cze): Audio: aac_latm (LC) ([17][0][0][0] / 0x0011), 48000 Hz, stereo, fltp
>     Stream #0:3[0xc9c](cze): Subtitle: dvb_subtitle ([6][0][0][0] / 0x0006)
>   No Program
>     Stream #0:0[0x12]: Data: epg
> 

A try to play "problem" sample causes kodi hang (with the fkms driver an the 5.10.11 kernel).

micha53, 2021-02-17:
issue update
today I made an installation of smb with the 5.4. kernel.
Starting kodi later kodi again does not appear at the screen, but sure it works. Somehow occasionally I started a TV program, and the TV screen appeared. Stopping TV again brought a black screen.
From a putty terminal I cold killall kodi, but not kodi.bin.v8. Even kill -9 

> pi@Medienserver:~ $ sudo killall kodi.bin_v8
> pi@Medienserver:~ $ ps -A | grep kodi
>  2819 tty1     00:03:41 kodi.bin_v8
> pi@Medienserver:~ $ ps -A | grep kodi
>  2819 tty1     00:03:43 kodi.bin_v8
> pi@Medienserver:~ $ sudo kill -9 2819
> pi@Medienserver:~ $ ps -A | grep kodi
>  2819 tty1     00:03:49 kodi.bin_v8 <defunct>

has only the result above.
Once started I can terminate kodi - blind - pressing "up" and twice "ret", like I would do it seeing the screen.
kodi.logs and dmesg are from the actual situation:
[dmesg.txt](https://github.com/raspberrypi/linux/files/5998152/dmesg.txt)
[kodi.log](https://github.com/raspberrypi/linux/files/5998153/kodi.log)
[kodi_crashlog-20210217_202739.log](https://github.com/raspberrypi/linux/files/5998155/kodi_crashlog-20210217_202739.log)

micha53, 2021-02-23:
I set up my system fresh from scratch and since I have done this, I cannot reproduce the error I described above. My new set up system works well - concerning this issue - and nobody else reported the same behavior, so I'll close this issue.

closed 2021-02-23 by micha53