raspberrypi/linux #4148: Kernel 5.10. Kodi CEC doesn't work anymore

micha53, 2021-02-15:
Since I made an Raspberry OS buster update, which installed the 5.10 kernel version, my TV remote control does not work with kodi anymore.
The kodi CEC adapter, Verion libCEC 4.0.4 - firmware v1 settings are the default. Physical address is 0, the TV has 1.
Same behavior on rpi3b+ and rpi 4b, acting the remote control only the TV reacts.
After downgrading the kernel to 5.4.50 (actually done on rpi4) CEC works good again.

[dmesg.txt](https://github.com/raspberrypi/linux/files/5983911/dmesg.txt)
[raspinfo.txt](https://github.com/raspberrypi/linux/files/5983932/raspinfo.txt)
[kodi.log](https://github.com/raspberrypi/linux/files/5983933/kodi.log)
[kodi.old.log](https://github.com/raspberrypi/linux/files/5983934/kodi.old.log)

pi@Raspi3:~ $ vcgencmd version
Jan 27 2021 22:26:53
Copyright (c) 2012 Broadcom
version 99d9a48302e4553cff3688692bb7e9ac760a03fa (clean) (release) (start)
pi@Raspi3:~ $ uname -a
Linux Raspi3 5.10.11-v7+ #1399 SMP Thu Jan 28 12:06:05 GMT 2021 armv7l GNU/Linux

micha53, 2021-02-23:
I made a new setup of my system, could reproduce the behavior, making the following steps:

**installation of Raspbian buster**

**Installation of kodi**
sudo apt-get update/upgrade
sudo apt-get install kodi

**raspi-config:**
	"Performance Options" -> "GPU Memory" -> 320 MB.
	„ Advanced Options“ → „ GL Driver“ → G2 GL (Fake KMS) OpenGL desktop driver with fake KMS
	start into CLI

**start kodi**
	settings/skin settings/main menue items-→ _selected my interests (music, TV, pictures, videos, favourites)_
	settings/System/play GUI sounds → never

**kodi autostart:** /lib/systemd/system/kodi.service:
```
[Unit]
Description = Kodi Media Center
After = remote-fs.target network-online.target
Wants = network-online.target

[Service]
User = pi
Group = pi
Type = simple
ExecStart = /usr/bin/kodi-standalone
Restart = on-abort
RestartSec = 5

[Install]
WantedBy = multi-user.target
```
sudo systemctl enable kodi.service

**Included Harddisk:**
sudo apt-get -y install ntfs-3g hfsutils hfsprogs exfat-fuse
sudo mkdir /media/pi
sudo mount -t ntfs-3g -o utf8,uid=pi,gid=pi,noatime /dev/sda1 /media/pi

sudo nano -w /etc/fstab
appended
	UUID=<my HD's uuid> /media/pi/ ntfs-3g utf8,uid=pi,gid=pi,noatime 0

Tested Kodi with the harddisk.

**Until here CEC within kodi was working like expected.**

**Then TvHeadend installed:**
sudo apt-get install tvheadend

Then I shut down my system and made a backup of my SD-card.

Since the next start of my system CEC does not work anymore with kodi. Behavior is like described above.

popcornmix, 2021-02-23:
Are you saying it was installing tvheadend that caused the issue?
Or rebooting? Or backing up?

micha53, 2021-02-23:
None of them. In my old system I got the problem with an update to kernel 5.10, but I could not reproduce exactly what I've done before. That is why I documented now what I've done with a new setup till the error occured again. Rebooting or backup I can't believe that it is the reason. Tvheadend, the server... ??? I am going to ask somebody who has the [same Issue ](https://forum.kodi.tv/showthread.php?tid=360500)if he has installed tvheadend.

wagnerch, 2021-02-24:
It's not related to tvheadend. It's fairly reproducible on 5.10.11 kernel by:
1. turn off TV
2. shutdown kodi (runs via service: sudo systemctl stop kodi)
3. start up kodi (runs via service: sudo systemctl start kodi)
4. TV will turn on, and CEC doesn't work.
5. If you restart Kodi, with TV on, CEC works.

This is on a Pi4B, 5.10.11 kernel, Kodi 18.9.

wagnerch, 2021-02-24:
Another thing is if you start "cec-client -m" that totally hoses up CEC with Kodi.  Not really sure if that is supposed to work not and no idea if that was an issue under 5.4, never used cec-client before or tried, was just seeing if there was anything more useful I could include here.

popcornmix, 2021-02-24:
> Another thing is if you start "cec-client -m" that totally hoses up CEC with Kodi. Not really sure if that is supposed to work or not
You can't run cec-client when kodi is running (which also uses kodi).
e.g. https://github.com/Pulse-Eight/libcec/issues/123

popcornmix, 2021-02-24:
> It's not related to tvheadend. It's fairly reproducible on 5.10.11 kernel by:

It's pretty unlikely this is related to kernel. Without kms driver, cec is purely implemented by userspace library (libcec) and firmware code (which acts as a dumb serial interface for cec commands).

micha53, 2021-02-24:
> It's pretty unlikely this is related to kernel. 

I agree. Could reproduce the behavior outlined by wagnerch. 
Does anybody know, where to direct this issue for an improvement?
I'll wait a day for an answer, but close this issue here tomorrow evening.

wagnerch, 2021-02-24:
> Does anybody know, where to direct this issue for an improvement?
> I'll wait a day for an answer, but close this issue here tomorrow evening.

Unless I missed something here, didn't you indicate above that there is no issue under 5.4, and upgrading to 5.10 that CEC no longer works?  I am not sure how this isn't kernel related, I recall seeing a post that indicated CEC under RPI was implemented in vchiq.  Isn't that in the kernel and a rewrite as a part of 5.10?

wagnerch, 2021-02-24:
> 
> 
> > Another thing is if you start "cec-client -m" that totally hoses up CEC with Kodi. Not really sure if that is supposed to work or not
> 
> You can't run cec-client when kodi is running (which also uses kodi).
> e.g. [Pulse-Eight/libcec#123](https://github.com/Pulse-Eight/libcec/issues/123)

I figured it was something along those lines, thanks.

wagnerch, 2021-02-24:
> It's pretty unlikely this is related to kernel. Without kms driver, cec is purely implemented by userspace library (libcec) and firmware code (which acts as a dumb serial interface for cec commands).

Unfortunately, I am not willing to go back to 5.4 to verify the issue. :)  It doesn't bother me enough, so I can live with it.  I use a HID device for Kodi anyways.

At least for me the issue is only if Kodi is restarted while the TV is off, but it sounds like the OP issue is it just doesn't work at all on 5.10.

micha53, 2021-02-25:
> didn't you indicate above that there is no issue under 5.4, and upgrading to 5.10 that CEC no longer works? I am not sure how this isn't kernel related, 

it seemed to be so and one [post #2 from Nachteule here](https://forum.kodi.tv/showthread.php?tid=360500) caused me to open this issue here.
But your post directed me to a subject I had not considered before and I **could reproduce** your description. CEC on kodi works fine if I give attention to your recommendation. So far your experience **was helpful** for me.

If I think about it also becomes clear. The TV must redirect the remote control to kodi. How shall it do that if it is still switched off - or not fully booted!
May be the libcec should poll the TV if it is ready to communicate before trying to send commands. I do not know how libcec communicates with the TV.

I close this issue now and turn to the [Pulse-Eight/libcec](https://github.com/Pulse-Eight/libcec) Discussion.

Thanks to all for taking the time for this discussion. Nevertheless it was helful for me.

closed 2021-02-25 by micha53

mriscoc, 2021-11-21:
I think than maybe there is a problem with CEC and fake KMS, I got the same problem when trying to configurate a VNC service using the script: https://raw.githubusercontent.com/MarkusLange/VNC-Server-install-script-for-OSMC/master/osmc_vnc_install_cli.bash

popcornmix, 2021-11-22:
Probably best asked on osmc forum. It doesn't sound like your issue is related to this (closed) issue.