raspberrypi/linux #4149: v3d / Qt applications: X graphics becoming non-functional when using large QT_SCALE_FACTOR

maxnet, 2021-02-15:
Starting Qt applications (that use OpenGL acceleration) with a larger QT_SCALE_FACTOR environment variable value set results in X becoming unusable.

If for testing purposes I install rpi-imager (`sudo apt install rpi-imager`) and do:

```
QT_SCALE_FACTOR=15 rpi-imager
```

The application exits, and errors out with GPU memory allocation error:

```
qt5ct: using qt5ct plugin
DRM_IOCTL_MODE_CREATE_DUMB failed: Cannot allocate memory
Failed to create scanout resource
QOpenGLFramebufferObject: Framebuffer incomplete attachment.
QOpenGLFramebufferObject: Framebuffer incomplete, missing attachment.
QOpenGLFramebufferObject: Framebuffer incomplete attachment.
QOpenGLFramebufferObject: Framebuffer incomplete, missing attachment.
Segmentation fault
```

and dmesg:

```
[  183.058879] [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
[  183.058903] vc4-drm gpu: [drm]                           dumb:  12168kb BOs (3)
[  204.120057] v3d fec00000.v3d: MMU error from client L2T (0) at 0x2de1000, pte invalid
```

Ok, that's to be expected, and reasonable enough.
However if one sets an even larger QT_SCALE_FACTOR like 20 it does NOT error out.
Just graphics/X becomes non-functional and screen looks like this:

[image]

The system itself did not hang completely. Mouse cursor still moves.
But X is useless. Cannot close the application with alt+f4, and cannot switch to another application.
I do can ssh to the box, and if send sigterm to rpi-imager, rpi-imager does exit properly, but screen will stay the same.
No memory allocation errors or anything out of the ordinary in dmesg.
Output from the program (if redirected to file) does show:

```
qt5ct: using qt5ct plugin
QOpenGLFramebufferObject: Framebuffer incomplete attachment.
QOpenGLFramebufferObject: Framebuffer incomplete, missing attachment.
QOpenGLFramebufferObject: Framebuffer incomplete attachment.
QOpenGLFramebufferObject: Framebuffer incomplete, missing attachment.
```

But no "Cannot allocate memory" line which one does sees with scale factor 15.

If no QT_SCALE_FACTOR environment variable is provided manually, Qt will calculate a reasonable value automatically based on the DPI of the display. It looks at the physical display dimensions reported by EDID for that, and the screen resolution.

As a result users will sometimes get high scale factors without them having set the QT_SCALE_FACTOR variable intentionally, e.g. if there display has incorrect EDID.
There is a user report here about xrandr reporting physical screen dimensions of only 16 mm x 9 mm for the user: https://github.com/raspberrypi/rpi-imager/issues/153
In which case graphics also become non-functional.

**System**

<details><summary>CLICK TO EXPAND RASPINFO</summary>
System Information
------------------

Raspberry Pi 400 Rev 1.0
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2021-01-11
[... 572 lines cut ...]
[    7.821250] random: 7 urandom warning(s) missed due to ratelimiting
[    8.327253] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[    8.333771] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    8.717119] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    8.717682] bcmgenet fd580000.ethernet eth0: Link is Down
[   10.363498] broken atomic modeset userspace detected, disabling atomic
[   12.783798] fuse: init (API version 7.32)
[   13.745104] Bluetooth: Core ver 2.22
[   13.745162] NET: Registered protocol family 31
[   13.745168] Bluetooth: HCI device and connection manager initialized
[   13.745180] Bluetooth: HCI socket layer initialized
[   13.745188] Bluetooth: L2CAP socket layer initialized
[   13.745201] Bluetooth: SCO socket layer initialized
[   13.757838] Bluetooth: HCI UART driver ver 2.3
[   13.757847] Bluetooth: HCI UART protocol H4 registered
[   13.757897] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   13.758028] Bluetooth: HCI UART protocol Broadcom registered
[   13.926335] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   13.985460] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   13.985469] Bluetooth: BNEP filters: protocol multicast
[   13.985482] Bluetooth: BNEP socket layer initialized
[   14.013943] Bluetooth: RFCOMM TTY layer initialized
[   14.013963] Bluetooth: RFCOMM socket layer initialized
[   14.013987] Bluetooth: RFCOMM ver 1.11
</details>

popcornmix, 2021-02-16:
This feels like a QT bug rather than a kernel bug. QT is asking the kernel to allocate a framebuffer of say, 16000x9000 which understandably fails and then it doesn't handle it well.

QT_AUTO_SCREEN_SCALE_FACTOR=0 is a workaround for the edid with incorrect dimensions in.
Some discussion here: https://www.raspberrypi.org/forums/viewtopic.php?t=259270

maxnet, 2021-02-16:
>This feels like a QT bug rather than a kernel bug.

Even after I terminate the application (sending SIGTERM with `killall rpi-imager` through ssh works correctly, the process terminates) the display/X is still messed up though.
I consider that part a kernel (or Mesa/X) bug.