raspberrypi/linux #4158: writeq() on 64bit 5.10.XX branch does issue PCIe cycle, switching to two writel() does work.

joshsinykin, 2021-02-19:
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
The megaraid_sas driver utilizes the writeq() to perform the PCIe cycle to its adapter. On the 64bit Pi OS kernel 5.10.11 , when loading the driver, the load hangs. The hang was debugged down to a cycle sent to the adapter, but the adapter never receiving the PCIe command. Switching from writeq() to 2 writel() allows the PCIe cycle to be received by the adapter. This driver has been tested to work with many other ARM 64 bit configurations and it would not make sense to impose the performance penalty into the driver for only the RPi.

**To reproduce**
List the steps required to reproduce the issue.
./compile.sh 
sudo insmod megaraid_sas.ko
driver hangs.

if we apply a patch to the driver to change from writeq to two writel., same steps above work.

Driver code in question:
/**
 * megasas_write_64bit_req_desc -	PCI writes 64bit request descriptor 
 * @instance:						Adapter soft state
 * @req_desc:						64bit Request descriptor
 */
static void
megasas_write_64bit_req_desc(struct megasas_instance *instance,
		union MEGASAS_REQUEST_DESCRIPTOR_UNION *req_desc)
{
#if defined(writeq) && defined(CONFIG_64BIT)
		u64 req_data = (((u64)le32_to_cpu(req_desc->u.high) << 32) |
			le32_to_cpu(req_desc->u.low));
		writeq(req_data, &instance->reg_set->inbound_low_queue_port);
#else
		unsigned long flags;
		spin_lock_irqsave(&instance->hba_lock, flags);
		writel(le32_to_cpu(req_desc->u.low),
			&instance->reg_set->inbound_low_queue_port);
		writel(le32_to_cpu(req_desc->u.high),
			&instance->reg_set->inbound_high_queue_port);
#if !((defined(RHEL_MAJOR) && (RHEL_MAJOR == 8) && (RHEL_MINOR >= 2)) || \
      (LINUX_VERSION_CODE >= KERNEL_VERSION(5,2,0)))
		mmiowb();
#endif
		spin_unlock_irqrestore(&instance->hba_lock, flags);
#endif

**Expected behaviour**
Broadcom Adapter is expecting a kernel modification to correctly handle the writeq(), so that it does not need specific platform workarounds in their driver.

**Actual behaviour**
writeq() is not performing the 64 bit cycle on PCIe.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW

Compute Module 4 Lite with 4 GB RAM, WiFi/Bluetooth

* Which OS and version (`cat /etc/rpi-issue`)?
* 
Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7252c154838ec5b4576f29c996ac8fe3750cae12, stage2

5.10.11 and 5.10.13 , with KConfig change to allow CONFIG_IRQ_POLL=y . necessary to compile the megaraid_sas driver for ARM.

* Which firmware version (`vcgencmd version`)?
Aug 19 2020 17:38:16
Copyright (c) 2012 Broadcom
version e90cba19a98a0d1f2ef086b9cafcbca00778f094 (clean) (release) (start)

* Which kernel version (`uname -a`)?

Linux raspberrypi 5.10.14-v8+ #3 SMP PREEMPT Fri Feb 12 16:47:24 GMT 2021 aarch64 GNU/Linux
[... 59 lines cut ...]
[  139.747904] scsi host0: Broadcom SAS based MegaRAID driver
[  139.769816] scsi 0:0:97:0: Enclosure         LSI      VirtualSES       03   PQ: 0 ANSI: 7
[  139.778076] scsi 0:0:97:0: Attached scsi generic sg0 type 13
[  139.789480] scsi 0:2:0:0: Direct-Access     AVAGO    MR9460-16i       5.16 PQ: 0 ANSI: 5
[  139.790170] sd 0:2:0:0: [sda] 700121088 512-byte logical blocks: (358 GB/334 GiB)
[  139.790216] sd 0:2:0:0: [sda] Write Protect is off
[  139.790223] sd 0:2:0:0: [sda] Mode Sense: 1f 00 00 08
[  139.790295] sd 0:2:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[  139.790410] sd 0:2:0:0: Attached scsi generic sg1 type 0
[  139.790606] sd 0:2:0:0: [sda] Optimal transfer size 196608 bytes
[  139.791735] scsi 0:2:1:0: Direct-Access     AVAGO    MR9460-16i       5.16 PQ: 0 ANSI: 5
[  139.792089] sd 0:2:0:0: [sda] Attached SCSI disk
[  139.792341] sd 0:2:1:0: Attached scsi generic sg2 type 0
[  139.792558] sd 0:2:1:0: [sdb] 1754529792 512-byte logical blocks: (898 GB/837 GiB)
[  139.793181] sd 0:2:1:0: [sdb] Write Protect is off
[  139.793196] sd 0:2:1:0: [sdb] Mode Sense: 1f 00 00 08
[  139.793281] sd 0:2:1:0: [sdb] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[  139.793594] sd 0:2:1:0: [sdb] Optimal transfer size 786432 bytes
[  139.799745] sd 0:2:1:0: [sdb] Attached SCSI disk

**Additional context**
Add any other relevant context for the problem.

This data came from efforts from work with Jeff Geerling:
https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/72

geerlingguy, 2021-02-19:
Just wanted to chime in and say it's possible (though I haven't looked too deeply) that this could be related to some of the issues Google's had getting their Coral TPU driver working with the CM4 as well: https://github.com/google-coral/edgetpu/issues/280

timg236, 2021-02-19:
The PCIe controller hardware only supports 32bit MMIO reads/writes. IIIRC there were similar bug reports for XHCI on BC2712

joshsinykin, 2021-02-19:
Do you see a KConfig change to disable writeq support? or reimplementing writeq in the 64 bit kernel to perform two writel()?

timg236, 2021-02-20:
Disabling writeq might be reasonable but it's not really my area of expertise

Coreforge, 2021-02-20:
You could try changing arch/arm64/include/asm/io.h and replace the 64bit accesses with two 32bit ones, but I've found it locks up somewhere else, so probably better to replace them where they are used so other things don't get messed up.

joshsinykin, 2021-02-22:
Broadcom Storage driver engineer is recommending disabling the writeq define in CONFIG.

pelwell, 2021-02-22:
Give it a go - kernel builds don't take so long on a Pi 4, and there are instructions here: https://www.raspberrypi.org/documentation/linux/kernel/building.md

joshsinykin, 2021-02-22:
Searching the kernel, I dont see a KConfig option available. Some kernels out there have an option "HAVE_WRITEQ" , but the pi kernel does not.

pelwell, 2021-02-22:
You could try the following (untested) patch:
```diff
diff --git a/arch/arm64/include/asm/io.h b/arch/arm64/include/asm/io.h
index fd172c41df90..3731da41c680 100644
--- a/arch/arm64/include/asm/io.h
+++ b/arch/arm64/include/asm/io.h
@@ -42,7 +42,9 @@ static __always_inline void __raw_writel(u32 val, volatile void __iomem *addr)
 #define __raw_writeq __raw_writeq
 static inline void __raw_writeq(u64 val, volatile void __iomem *addr)
 {
-       asm volatile("str %x0, [%1]" : : "rZ" (val), "r" (addr));
+       //asm volatile("str %x0, [%1]" : : "rZ" (val), "r" (addr));
+       asm volatile("str %w0, [%1]" : : "rZ" ((u32)val), "r" (addr));
+       asm volatile("str %w0, [%1]" : : "rZ" ((u32)(val>>32)), "r" (addr + 4));
 }
 
 #define __raw_readb __raw_readb
```

vianpl, 2021-02-22:
@pelwell I wonder if the order in which the 32bit accesses happen is relevant. I see XHCI does it in low/high order, as you did, but is it something we can generalize? Atomicity might be an issue too.

pelwell, 2021-02-22:
Order might indeed be an issue, but I just took an educated guess. There's nothing we can do about the lack of atomicity - if it doesn't work for some drivers then tough luck.

Interestingly, writeq doesn't exist in arm32 and none of the kernel code used by our defconfigs makes use of it.

joshsinykin, 2021-02-22:
Broadcom also does it low, then high order (using writel)

static void
megasas_write_64bit_req_desc(struct megasas_instance *instance,
		union MEGASAS_REQUEST_DESCRIPTOR_UNION *req_desc)
{
#if defined(writeq) && defined(CONFIG_64BIT)
		u64 req_data = (((u64)le32_to_cpu(req_desc->u.high) << 32) |
			le32_to_cpu(req_desc->u.low));
		writeq(req_data, &instance->reg_set->inbound_low_queue_port);
#else
		unsigned long flags;
		spin_lock_irqsave(&instance->hba_lock, flags);
		writel(le32_to_cpu(req_desc->u.low),
			&instance->reg_set->inbound_low_queue_port);
		writel(le32_to_cpu(req_desc->u.high),
			&instance->reg_set->inbound_high_queue_port);
#if !((defined(RHEL_MAJOR) && (RHEL_MAJOR == 8) && (RHEL_MINOR >= 2)) || \
      (LINUX_VERSION_CODE >= KERNEL_VERSION(5,2,0)))
		mmiowb();
#endif
		spin_unlock_irqrestore(&instance->hba_lock, flags);
#endif
}

geerlingguy, 2021-05-14:
Just wanted to link a relevant kernel discussion here: https://lore.kernel.org/linux-arm-kernel/c188698ca0de3ed6c56a0cf7880e1578aa753077.camel@suse.de/T/#u — I discovered that through a semi-related discussion about bringing up an AMD GPU on the CM4 in this thread: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/4#issuecomment-841388694

vianpl, 2021-05-15:
@geerlingguy Are you aware of more PCIe boards that depend on this? If we are to introduce infrastructure in the kernel to deal with limitations in MMIO buses it'd be nice to have as many examples as possible.

geerlingguy, 2021-05-15:
So far there's a small bit of evidence it _could_ be an issue with the Google Coral TPU, and very little evidence it is related to the problems we're having with Nvidia/AMD GPUs.

The only card I can confirm this issue directly affects currently is the MegaRAID.

geerlingguy, 2021-09-12:
Not `writeq()`, but it seems 64-bit writes could be one of the blockers to getting `amdgpu` (and `radeon`) drivers working as well—see https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/222