raspberrypi/linux #4161: RPI4: mmc1: Timeout when WiFi interface is under heavy load

Drachenkaetzchen, 2021-02-21:
**Short Description**

When running [ustreamer](https://github.com/pikvm/ustreamer) to stream the RPI camera over the network with 10 megabytes/s per client, and 3 clients connected, the system becomes non-responsive with the following messages appearing in `dmesg` (excerpt)

```
[  479.849229] mmc1: Timeout waiting for hardware interrupt.
[  479.849637] brcmfmac: mmc_submit_one: CMD53 sg block write failed -110
[  479.849656] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  490.089453] mmc1: Timeout waiting for hardware interrupt.
[  490.089845] brcmfmac: mmc_submit_one: CMD53 sg block write failed -110
[  490.089868] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  500.329673] mmc1: Timeout waiting for hardware interrupt.
[  500.330040] brcmfmac: mmc_submit_one: CMD53 sg block write failed -110
[  500.330062] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  510.569894] mmc1: Timeout waiting for hardware interrupt.
[  510.569974] brcmfmac: mmc_submit_one: CMD53 sg block write failed -110
[  510.569994] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  520.810106] mmc1: Timeout waiting for hardware interrupt.
[  520.810477] brcmfmac: mmc_submit_one: CMD53 sg block write failed -110
[  520.810499] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  520.810959] brcmfmac: mmc_submit_one: CMD53 sg block write failed -84
[  520.810976] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  520.816668] brcmfmac: brcmf_sdio_hdparse: HW header checksum error
[  520.816685] brcmfmac: brcmf_sdio_rxfail: terminate frame
[  520.816918] brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -84
```

**Verbose Description**

I run [ustreamer](https://github.com/pikvm/ustreamer) to stream video over the network. On my Raspberry Pi 4, this seems to trigger an issue where either the WiFi hardware or the driver crashes if I connect to the video stream from multiple machines. For a single client, this is about 10 megabytes per second. The WiFi interface is on a 5 GHz 325Mbit/s connection.

When about 3 clients are connected and the system tries to transfer about 30 megabytes per second over the wifi, the system often becomes non-responsive after a random amount of time. It is noteworthy that as soon as the non-responsive state is reached, the attached serial console via USB shows the kernel log extremely slowly, suggesting that the system is under heavy load by that issue.

However, writing the output of `dmesg` to a text file still seems to work. As such, I can rule out power supply issues as the system is using the original USB-C power supply for the Raspberry Pi 4, since an undervoltage condition would appear in the log. The captured log when the problem occurs can be found below.

It is noteworthy that the issue does not seem to occur if using an USB WiFi stick.

**To reproduce**
Since the issue only appears after a random amount of time, it's hard to reproduce. Any clues on how to make this reproducible are appreciated.

**Expected behaviour**
The system stays responsive even if the WiFi interface is under heavy load.

**Actual behaviour**
The systembecomes unresponsive under heavy WiFi load.

**System**

Raspberry Pi 4 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

[Full raspinfo output pastebin](http://paste.debian.net/1186324/)

**Logs**

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.11-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1399 SMP Thu Jan 28 12:09:48 GMT 2021
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000fc00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000001fffffff]
[... 458 lines cut ...]
[  608.252741] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  609.292298] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  614.608406] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  615.612456] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  616.652479] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  619.613743] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  620.652536] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  621.692593] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  625.499879] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  626.412636] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  626.572603] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  626.991248] brcmfmac: mmc_submit_one: CMD53 sg block write failed -84
[  626.991270] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  627.452721] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  627.612668] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  628.492714] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  630.625434] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  630.892808] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  631.692770] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  631.932735] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  632.732819] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  632.972790] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  635.630928] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  635.752621] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
```

Drachenkaetzchen, 2021-02-21:
I just noticed that the WiFi interface seems to be attached to `mmc1` which is initialized as high speed SDHC:

```
[    3.099665] mmc1: new high speed SDIO card at address 0001
```

According to [Wikipedia](https://en.wikipedia.org/wiki/SD_card#Speed), the high speed mode supports 25 MB/s, which would explain why the issue occurs only with 3 clients and 10 MB/s transfer rate each.

pelwell, 2021-02-21:
I don't think you can rule out power supply - the built-in detector only picks up gross undervoltage events on the main supply line, and doesn't necessarily notice when isolated areas of the device don't quite get the voltage they need to function correctly. The official power supply is only really intended to power the Pi itself, a mouse and a keyboard. I've successfully powered an SSD in a USB carrier at the same time, but the WiFi and Bluetooth subsystems were idle at the time.

I don't think you will get much more than about 110Mb/s through the WiFi, once you've factored in the multiple protocol overheads, i.e. about 1.5 of your 10MB/s streams.

Drachenkaetzchen, 2021-02-21:
> I don't think you can rule out power supply - the built-in detector only picks up gross undervoltage events on the main supply line, and doesn't necessarily notice when isolated areas of the device [...]

The issue also occurred when I didn't have an SSD via USB attached.

> I don't think you will get much more than about 110Mb/s through the WiFi, once you've factored in the multiple protocol overheads, i.e. about 1.5 of your 10MB/s streams.

The issue is not that I'm getting too little bandwidth - the issue is that the attempt to transfer more data than the link supports triggers the problem.

pelwell, 2021-02-21:
If what you are saying is correct then an iperf/iperf3 to a PC with a high-speed 5G WiFi interface should be enough to crash the WiFi stack, but I've never seen that happen. Also, firmware crashes are trapped and reported to the brcmfmac driver, so the symptoms don't quite match.

I suggest you try slightly increasing the voltage supplied to the board (the exact power requirements vary from unit to unit) - put `over_voltage=2` in config.txt and either remove the USB peripherals or put them on a powered hub.

Drachenkaetzchen, 2021-02-21:
> I suggest you try slightly increasing the voltage supplied to the board (the exact power requirements vary from unit to unit) - put `over_voltage=2` in config.txt and either remove the USB peripherals or put them on a powered hub.

Citing myself:

`The issue also occurred when I didn't have an SSD via USB attached (= booted from an SD card with no other USB devices attached).`

What makes you think it's an undervoltage problem?

pelwell, 2021-02-21:
Years of experience.

Drachenkaetzchen, 2021-02-21:
> Years of experience.

Years of experience comes with ignorance of facts, I guess.

Drachenkaetzchen, 2021-02-21:
Closing the issue as there doesn't seem to be any interest in fixing kernel bugs.

closed 2021-02-21 by Drachenkaetzchen