raspberrypi/linux #4163: Should be able to load rootfs from an nvme device

peterharperuk, 2021-02-22:
**Describe the bug**
With a 2711 cm4 device you should be able to load rootfs from an nvme device but as nvme is a module it can't be found

**To reproduce**
Change the kernel command line to use a partition from an nvme disk for root.

**Expected behaviour**
Should load roofs from the NVME device

**Actual behaviour**
On boot it keeps waiting for the partition to appear

pelwell, 2021-02-22:
I think that's just a case of changing `CONFIG_BLK_DEV_NVME=m` to `CONFIG_BLK_DEV_NVME=y` in the 2711 defconfigs. Have you verified that doing so results in a kernel that boots as expected?

peterharperuk, 2021-02-22:
Hi Phil. Yes I've verified that it resolves the problem and checked rpi4 still boots ok.

pelwell, 2021-02-22:
Closing, now that #4164 is merged.

closed 2021-02-22 by pelwell