raspberrypi/linux #4166: kworker hogs CPU only when using WIFI

ssackett, 2021-02-22:
**Describe the bug**
kworker hogs the CPU only when using WIFI, wired network connections are not affected.

**To reproduce**
This configuration allows a USB peripheral connected to the Pi4 to be controlled by a client running on a remote computer by emulating a USB connection over an IP network.

Install the VirtualHere service named "vhusbdarmpi4", connect SDRPlaydx to USB on the Pi4 and start the client software named "vhui64.exe" on a Windows 10 connected by wire to the same subnet as the Pi4.  Note the Pi4 communicates to a router using WIFI and the Win10 computer is connected to the same router but with a wired connection.  Using the client software connect remotely to the SDRPlaydx. Note the VirtualHere software uses a USBIP protocol to make it appear that the SDRPlaydx is connected directly to the client via wired USB.  On the client start the application "SDRuno".  SDRuno is an application that controlls the SDRPlaydx.

**Expected behaviour**
I expect that the configuration to work as well when the Pi4 communicates over WIFI as it does when using a wired connection.

**Actual behaviour**
When the Pi4 uses a WIFI connection a kworker hogs the CPU and lowers the performance of the communications.

**System**
 System Information
------------------

Raspberry Pi 4 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2021-01-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, ddba443b028e3673570b52d2fbf471da0341afd4, stage3

Linux playground 5.10.16-v7l+ #1402 SMP Tue Feb 16 14:12:18 GMT 2021 armv7l GNU/Linux
Revision	: c03111
Serial		: 10000000d9860c20
Model		: Raspberry Pi 4 Model B Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Feb 16 2021 14:17:31 
Copyright (c) 2012 Broadcom
version d6d82cf99bcb3e9a166a34cfde53130957a36bd3 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       30456660 2676036  26504584  10% /
devtmpfs         1827468       0   1827468   0% /dev
tmpfs            1959564      16   1959548   1% /dev/shm
tmpfs            1959564    8708   1950856   1% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs            1959564       0   1959564   0% /sys/fs/cgroup
tmpfs              65536       0     65536   0% /tmp/openwebrx
/dev/mmcblk0p1    258095   49864    208232  20% /boot
tmpfs             391912       0    391912   0% /run/user/1000

Filename				Type		Size		Used		Priority
/var/swap                               file		102396		0		-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: (none)
raspberrypi-sys-mods:
  Installed: 20210208
openbox:
  Installed: (none)
lxpanel:
[... 581 lines cut ...]
[ 1768.846843] usb 1-1.1: reset high-speed USB device number 6 using xhci_hcd
[ 1769.226872] usb 1-1.1: reset high-speed USB device number 6 using xhci_hcd
[ 1780.777056] usb 1-1.1: reset high-speed USB device number 6 using xhci_hcd
[ 1781.057073] usb 1-1.1: reset high-speed USB device number 6 using xhci_hcd
[ 2948.045653] logitech-hidpp-device 0003:046D:404D.0004: HID++ 4.1 device connected.
root@playground:/home/pi# 

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
I have a PI 4B and am experiencing a kworker that hogs the cpu only when using WIFI.  I have updated the kernel to 5.10.16-v7l+ and added "dtparam=sd_poll_once=on" to config.txt.  I have verified that "sd_poll_once" is active using "dtc -I fs /proc/device-tree" which shows "sd_poll_once = "\0\0\0?non-removable?";".  None of these changes fixed the problem.

Using top the the first two lines are:

1387 root      20   0       0      0      0 I  21.6   0.0   0:16.66 kworker/u8:2-brcmf_wq/mmc1:0001:1                                                                             
  839 root      20   0    6964   2380      4 S  14.0   0.1   0:18.22 vhusbdarmpi4                           

The offending kworker always shows as "kworker/u8:2-brcmf_wq/mmc1:0001:1" with the number after "/u8:" changing between 2, 3, and 4.  Also the number is followed alternately by "+"  and "-" (ie: "u8:2+brcmf" and "u8:2-brcmf").  

The process "vhusbdarmpi4" is running as a service started by systemd,  it is communicating via WIFI and is running a USBIP protocol.   If I use a wired connection instead of WIFI the kworker disappears and the service runs as expected.

Thanks,

Steve

cat007mine, 2021-04-12:
Any progress on this?
My pi just locked up and I had to hard power cycle it and upon reboot I have the below eating anything from 2-30% CPU:

1476 root      20   0       0      0      0 I  10.6   0.0   0:24.80 kworker/u8:2-brcmf_wq/mmc1:0001:1

I've done the usual apt-get update && upgrade and am on kernel 5.10.23-v7l+

ssackett, 2021-04-12:
Not as far as I know.

ssackett, 2021-04-12:
> @cat007mine <https://github.com/cat007mine>:
>
> What is your software configuration? The closer to a stock RPiOS, the more likely this issue will get looked at.
> How consistently do you see this CPU spin?

I have forgotten some of the details but they might be in my bug report.  To the best of my memory, I was using a fully updated  RPiOS that had started life as a packaged solution from a third party.   I was running a service from a different third party and when I connected over WIFI a kernel worker thread would hog 25% of the CPU, I found this would not happen if I connected to the network via wire.  I applied the fix for a similar problem where a kernel worker would hog the CPU by repeatedly checking if the SD card was connected, but this didn't help.  I also updated to the beta kernel but this didn't help either.  My solution was to stop connecting over WIFI.

Steve

pelwell, 2021-04-20:
`dtparam=sd_poll_once` only affects the SD card controller (emmc2 on a Pi 4), but both upstream and downstream DTBs already set the `non-removable` flag for the WiFi SD interface (EMMC/SD1) so poll-once is already enabled.

wagnerch, 2021-05-22:
I have this same issue on LibreELEC 9.95.2 using commit https://github.com/LibreELEC/LibreELEC.tv/tree/0f2dcbea9038df060f47913974111fbc8909f643

Appears to cause NFS read errors (error # -4), and NFS stream stalls.  There is also a brief lockup or responsiveness issue when logged in via SSH.

mengxp, 2023-01-02:
I remove cap-sdio-irq in dts and fix
cpu usage
from 20% to 0.5%

pelwell, 2023-01-09:
> I remove cap-sdio-irq in dts

Can you explain what you did? I don't see `cap-sdio-irq` in the Raspberry Pi DTS files.

georgehuang2, 2023-03-09:
@pelwell you can also check another post, I fixed the issue by removing WL_HOST_WAKE interrupt in dts.

https://github.com/grate-driver/linux/issues/37

pelwell, 2023-03-09:
That's in interesting suggestion, but the Pi boards and their DTS files don't use the host-wake feature, so there's nothing to delete.

georgehuang2, 2023-03-10:
At leaset, you can check if there are lots of brcmf_oob_intr interrupts recorded in /proc/interrupts in the Pi boards.
If yes, then it may be a clue for further investigation on the Pi boards.

fableman73, 2024-02-07:
Hi.
    This problem is still around at least for RPi Zero W2.
    RPi Zero W2
     All packages are up to date.
     
     I'am transferring files from a windows computer to the RPi Zero W2 over SMB using WIFI on the RPi Zero W2
     Transfering files over SMB to a connected SSD drive. ( I got a zero W2 to SSD boards connected)

     CPU 47% kworker/u9:0+brcmf_wq/mmc1:0001:1