raspberrypi/linux #4167: broadcom BCM_VC_SM removal from >=5.10.y kernels

nkichukov, 2021-02-22:
Hi folks,

drivers/staging/vc04_services/vc-sm-cma/vc_sm.c as configured by kernel configuration option: BCM_VC_SM has been removed from 5.10.y kernels and onwards.

It seems like this allowed kodi-rbpi to perform optimal hevc decoding through mmal, thus kodi-rbpi works smooth on kernels:
5.4.44, 5.4.83, 5.4.y and previous

but does not work on anything after it, such as 5.10.x and 5.11.0.

Can we have this broadcom char device re-added to the more recent kernels, same way it exists on 5.4.x:

```
./drivers/char/broadcom/vc_sm/Makefile:obj-$(CONFIG_BCM_VC_SM) := vc-sm.o
./drivers/char/broadcom/Kconfig:config BCM_VC_SM
./drivers/char/broadcom/Makefile:obj-$(CONFIG_BCM_VC_SM)         += vc_sm/
```
This will allow raspberrypi 3 devices to run more recent kernels with kodi-rbpi binaries which decode hevc pretty recently.
Thanks

pelwell, 2021-03-01:
Any comments, @6by9?

6by9, 2021-03-01:
It wasn't a specific decision by me to drop it.

vcsm-cma only allocates as uncached on the ARM, which does hit performance of HEVC decode on Pi3. It needs (another) rework to use dma-heaps for the allocs, so vcsm-cma becomes an importer only. If we can fiddle an ARM caching CMA dma-heap then that's the better solution.
(It'll upset John in Chromium again as we'll then need 3 file descriptors, and Chromium sandboxes these things).
It feels like that shouldn't be a huge task, so I might rejuggle priorities and see if it falls out easily.

pelwell, 2021-03-01:
Thanks - I was asking you as the person most familiar with the current state of VCSM and its replacements.

popcornmix, 2021-03-01:
Also https://github.com/raspberrypi/linux/issues/4112

nkichukov, 2021-03-05:
Seems like this got removed from 5.9.y and later. So for now I will be running 5.8.18 kernels on rpi3.

I have tried to port the changes into 5.11.y restoring /dev/vcsm however this has proven to be much harder than I initially thought, so I give up.

Does your idea to implement dma-heaps for allocs via vcsm-cma work for kodi-rbpi without any changes to xbmc(the rpi platform code)? Or is this the required step towards allowing raspberrypi 3 and lower to use the GBM platform with optimized HEVC decode, as described here:

https://forum.libreelec.tv/thread/23463-rpi3-hangs-while-trying-to-play-hevc-10bit-on-libreelec-10-test-builds/

which implies we have to wait for the GBM platform to catchup on the changes you will implement in the kernel API?

Thanks

6by9, 2021-03-05:
> Does your idea to implement dma-heaps for allocs via vcsm-cma work for kodi-rbpi without any changes to xbmc(the rpi platform code)? Or is this the required step towards allowing raspberrypi 3 and lower to use the GBM platform with optimized HEVC decode, as described here:

The [userland library](https://github.com/raspberrypi/userland/tree/master/host_applications/linux/libs/sm) will be updated to support dma heaps. As long as that library is updated along with the kernel then it should work.

nkichukov, 2021-09-04:
any progress with the kernel API and the library?