raspberrypi/linux #4169: I2C bus rate change on CSI Interface with update to 5.10

TIS-Christopher, 2021-02-23:
*** General overview: ***

We have a self designed device connected to the Pi 4B board CSI interface. This board has several chips on it that connect to the I2C
bus (EEPROM + tca6408 + ...). When using the Raspberry Pi OS with kernel 5.4 and older we don't have issues with this setup.
The last update to 5.10 (or a firmware update) introduced a new problem that the tca6408 sends NACKs on the I2C bus in certain circumstances.

We reduced this to an issue with the I2C bus, that it now scales from a baudrate of 400kHz to 1MHz on cpu usage.

Previously the I2C bus was at a (fixed) baudrate of 100kHz.

Special Settings in our setup:

- sudo raspi-config to start tool.
  - Select Interfacing Options -> Enable I2C
  - Select Interfacing Options -> Enable Camera
- Add `dtparam=i2c_vc=on` to /boot/config.txt

**Describe the bug**

The I2C bus rate of the CSI interface on the Pi4 changed from a steady 100kHz to a frequency of 400kHr - MHz
scaling with the cpu frequency.

**To reproduce**

- Measure the I2C bus rate on the CSI interface with the latest packages installed.
- Especially contrast this to earlier Raspberry Pi OS releases.

**Expected behaviour**

A fixed I2C bus rate of ~100kHz.

**Actual behaviour**

I2C bus rate 400kHz - 1MHz.

**System**

```
System Information
------------------

Raspberry Pi 4 Model B Rev 1.4 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage4

Linux chris-office-pi-upd 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
Revision	: d03114
Serial		: 10000000d842038d
Model		: Raspberry Pi 4 Model B Rev 1.4
Throttled flag  : throttled=0x0
Camera          : supported=1 detected=0

Videocore information
---------------------

Jan 27 2021 22:20:57 
Copyright (c) 2012 Broadcom
version 99d9a48302e4553cff3688692bb7e9ac760a03fa (clean) (release) (start_x)

alloc failures:     0
compactions:        0
legacy block fails: 0

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: 1.20200611
raspberrypi-sys-mods:
  Installed: 20210208
openbox:
  Installed: 3.6.1-8+rpt5
lxpanel:
  Installed: 0.10.0-2+rpt14
pcmanfm:
  Installed: 1.3.1-1+rpt26
rpd-plym-splash:
  Installed: 0.26

config.txt
----------

arm_freq=1500
audio_pwm_mode=514
config_hdmi_boost=5
core_freq=500
core_freq_min=200
disable_commandline_tags=2
disable_l2cache=1
disable_overscan=1
display_hdmi_rotate=-1
display_lcd_rotate=-1
dvfs=2
enable_gic=1
force_eeprom_read=1
force_pwm_open=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
gpu_freq=500
gpu_freq_min=250
init_uart_clock=0x2dc6c00
lcd_framerate=60
mask_gpu_interrupt0=1024
mask_gpu_interrupt1=0x10000
max_framebuffers=2
over_voltage_avs=-40000
pause_burst_frames=1
program_serial_random=1
total_mem=8192
hdmi_force_cec_address:0=65535
hdmi_force_cec_address:1=65535
hdmi_pixel_freq_limit:0=0x11e1a300
hdmi_pixel_freq_limit:1=0x11e1a300
device_tree=-
overlay_prefix=overlays/
hdmi_cvt:0=
hdmi_cvt:1=
hdmi_edid_filename:0=
hdmi_edid_filename:1=
hdmi_timings:0=
hdmi_timings:1=

cmdline.txt
-----------
coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1280x1024M@60 smsc95xx.macaddr=DC:A6:32:B6:13:B8 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=650104c1-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles

```

Thanks

[Full dump of raspinfo attached here](https://github.com/raspberrypi/linux/files/6028721/t.txt)

6by9, 2021-03-02:
I've just had a logic analyser on the SDA/SCL from CAM1/DISP1 on a CM4. That shows 100kHz on SCL as expected.

What does ```xxd /proc/device-tree/soc/i2c@7e205000/clock-frequency``` report? That should be ```0x0001 86a0``` for 100kHz.
Likewise ```sudo cat /sys/kernel/debug/clk/clk_summary``` to give a summary of all the clocks.

Have you got anything else strange about your system?
I see no additional dtoverlay entries being parsed, so I'm assuming your I2C devices are just poked from userspace.

TIS-Christopher, 2021-03-03:
I've just tried a CM4 board with our device attached.

Using the CM4 board:
* The frequency is 40kHz on an idle system and 100kHz under load.

Using the Pi4B:
* The frequency is 400kHz on an idle system and 1MHz under load.

Notes:
* Load is generated by executing `cat /dev/urandom > /dev/null`
* The CM4 board has a fresh image (SD Image '2021-01-11-raspios-buster-armhf'), used init wizard (and so updated to 5.10.11) and previously mentioned changes (Camera enabled, I2C enabled, `dtparam=i2c_vc=on`), enabled usb bus and enabled ssh.
* The Pi4B is a bit more 'used', but I have seen this issue with 'fresh' images

> Have you got anything else strange about your system?

I don't think so. I have seen the results of this issue on 'fresh' systems

> I see no additional dtoverlay entries being parsed, so I'm assuming your I2C devices are just poked from userspace.

Yes. We can see this just poking via `i2cget -y 10 0x20 0` with our I2C device at address 0x20.

Reports are [here: raspbian-i2c-bus-issue-2.txt](https://github.com/raspberrypi/linux/files/6074458/raspbian-i2c-bus-issue-2.txt)

6by9, 2021-03-03:
Have you looked with a logic analyser?

Hooking up a Pi4 (instead of CM4) I'm seeing non-I2C signals on 44&45 (I've hooked on a Pi display to make it easier to connect to). I see the line normally high, but 9 1usec pulses low / 45usec high on BOTH SDA and SCL at 2 to 11ms intervals. This also happens whilst the I2C controller is trying to send data, which therefore corrupts messages. Nothing else should be messing with those GPIOs so I'm a little confused.
I2C clock frequency is still the normal 40/100kHz based on load though.

I must admit to having abused this Pi slightly so I will try another one.

6by9, 2021-03-03:
Ahem, it helps to notice that the ground wire has become disconnected from the logic analyser. I'm surprised it was picking up as much as it was!

I2C clock frequency is now a nice clean 40kHz when idle, and 100kHz when the CPU is busy.

TIS-Christopher, 2021-03-08:
We just reproduced the 'error' with probes connected to the SDL/SDA on a ribbon cable on the CSI port of a PI4.
So we are not sure what we can do to help you reproduce it.

Just to list what I have done, here a recap:

```
PI4B
* Image from 'https://downloads.raspberrypi.org/raspios_armhf/images/raspios_armhf-2021-01-12/2021-01-11-raspios-buster-armhf.zip'
* Use wizard to init Language etc., but do not updated software.
* 'uname -a' => 'Linux raspberrypi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux'
* Reboot
* sudo raspi-config to start tool.
* Select Interfacing Options -> Enable I2C
* Select Interfacing Options -> Enable Camera
* Reboot
* Add dtparam=i2c_vc=on to /boot/config.txt
* Reboot

* Now doing i2cdetect -y 10
	- Shows 100 kHz on our scope
* Doing 'cat /dev/urandomg > /dev &' and then retrying 'i2cdetect' does not change the bus frequency

---

Updated image (apt update; upgrade; reboot)
* ' uname -a' => 'Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux'

* Now doing i2cdetect -y 10
	- Shows 400 kHz
* Doing 'cat /dev/urandomg > /dev &' and then retrying 'i2cdetect' changes the frequency to 1 MHz
````

I can send you an image of our setup with a logic analyzer/scope connected to the ribbon cable and showing the high frequencies if that helps ...

6by9, 2021-03-08:
I'm assuming doing ```'cat /dev/urandomg > /dev &``` is a typo, as /dev is a directory not a file, and /dev/urandomg doesn't exist on a normal system.

I've just tried again with my Pi4 running 5.10.17-v7l+, and it's again fine. Same revision code as yours (```d03114```).

Does /dev/i2c-1 on GPIOs 2&3 (pins 3&5) also end up running at 400kHz/1MHz? I'm assuming yes as they're derived from the same core clock. That'd isolate parent clock from peripheral setup.

...

Ah, enabling the camera seems to be the critical thing. I'd missed that step. 
 
The loading of bcm2835_v4l2 (triggered as a platform driver from vchiq) will be probing for the camera modules using BSC0 on GPIOs 44&45. I suspect that is changing the clock divider setup up the I2C peripheral that the kernel driver thinks it has already set up as it wants. It's then left in a half-and-half configuration.
This is one of the reasons we say that i2c_vc is reserved for VideoCore (the GPU/firmware).

Is there a specific reason that you are enabling the camera when you don't have one connected?
Either drop back to the standard firmware image, or blacklist bcm2835_v4l2 by creating a suitable file in /etc/modprobe.d.

A CM doesn't have the camera config defined by default, so won't attempt to probe anything.

6by9, 2021-03-08:
Correct - rawcam is only in the full variant.

V4L2 is recommended over rawcam, which would also then avoid having the firmware looking for a camera as the firmware would know that Unicam was assigned to Linux and therefore out-of-bounds to it.

TIS-Christopher, 2021-03-09:
>  blacklist bcm2835_v4l2 by creating a suitable file in /etc/modprobe.d

This works

Do you intend to fix this issue with the bcm2835_v4l2 driver, or should I just keep this blacklisted?

> V4L2 is recommended over rawcam

That might be the case, but keeping a kernel patch set for the V4L2 for our cameras seems to be more work for us then to rely on a stable user mode interface. Especially because we did not have any know how when we started this project.

I am not sure if basing our work on MMAL was the best idea, but it works (even if it was a bit light on documentation, and may get deprecated in the future)

6by9, 2021-03-09:
bcm2835_v4l2 is only triggering the firmware to probe for cameras, exactly the same as running raspistill or raspivid would.

I will have a quick look to see if there is anything funny going on in the firmware regarding I2C frequency, but I wouldn't be anticipating doing any significant fixes.

I have had a back burner project for a little while to sort a dummy V4L2 sensor driver that will just accept whatever format and resolution userspace requests in order to configure the CSI2 receiver. That can still leave the I2C setup to be done by userspace, but uses the preferred receiver driver.

TIS-Christopher, 2021-03-15:
> I have had a back burner project for a little while to sort a dummy V4L2 sensor driver that will just accept whatever format and resolution userspace requests in order to configure the CSI2 receiver. That can still leave the I2C setup to be done by userspace, but uses the preferred receiver driver.

That would be interesting.

We can live we the current solution of blacklisting the bcm2835_v4l2 driver.

Thank you very much for your help.

closed 2021-03-22 by TIS-Christopher

6by9, 2021-03-22:
PR for the dummy driver - https://github.com/raspberrypi/linux/pull/4226