raspberrypi/linux #4172: Message “Warning: The unit file, source configuration file or drop-ins of plexmediaserver.service changed on disk” After Every Reboot

SJMarty, 2021-02-24:
**Describe the bug**
I’ve installed the Plex Media Server software on a 4B running Buster with all updates.  After every reboot, the plexmediaserver service is logging the message "Warning: The unit file, source configuration file or drop-ins of plexmediaserver.service changed on disk. Run 'systemctl daemon-reload' to reload units."

**To reproduce**
Simply boot and run 'sudo systemctl status plexmediaserver.service' and the warning message above is reported.  'sudo systemctl daemon-reload' clears the warning message for that session but after a reboot, the warning message above returns.

I have an override.conf present in the /etc/systemd/system/plexmediaserver.service.d folder but other than that, no changes to the default plexmediaserver package install.  Just adding an _empty_ override.conf file causes the warning message.

**Expected behaviour**
1.  Install the default service file in '/lib/systemd/system/plexmediaserver.service' (dpkg deposits the file there per systemd spec).
2.  systemd looks at /etc/systemd/system/<servicename>.service.d/override.conf for service overrides.
3.  Specifying the following in override.conf...

[Service]
User=pi
Group=pi

4.  Execute 'sudo systemctl daemon-reload'
5.  After a reboot, systemd should not log message "Warning: The unit file, source configuration file or drop-ins of plexmediaserver.service changed on disk. Run 'systemctl daemon-reload' to reload units."

**Actual behaviour**
1.  Install the default service file in '/lib/systemd/system/plexmediaserver.service' (dpkg deposits the file there per systemd spec).
2.  systemd looks at /etc/systemd/system/<servicename>.service.d/override.conf for service overrides.
3.  Specifying the following in override.conf...

[Service]
User=pi
Group=pi

4.  Execute 'sudo systemctl daemon-reload'
5.  After a reboot, systemd raises message "Warning: The unit file, source configuration file or drop-ins of plexmediaserver.service changed on disk. Run 'systemctl daemon-reload' to reload units."

**System**

* Which model of Raspberry Pi?
4B 8 GB

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2021-01-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 21090519d85bdaa1615d5d5057d37b09368ea5d2, stage5

* Which firmware version (`vcgencmd version`)?
Jan 27 2021 22:19:57 
Copyright (c) 2012 Broadcom
version 99d9a48302e4553cff3688692bb7e9ac760a03fa (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux raspberry-pi-06 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux

**Additional context**
Suspecting this was a Plex problem, I contacted Plex via their support forums and the Plex team member that responded (ChuckPa) indicated that this is an OS issue and not an issue with the plexmediaserver service.

popcornmix, 2021-02-24:
Sorry this repo is for reporting bugs with the linux kernel and this doesn't sound like a kernel issue.
You may have more luck in the [raspberrypi forums](https://www.raspberrypi.org/forums).

> Warning: The unit file, source configuration file or drop-ins of plexmediaserver.service changed on disk. Run 'systemctl daemon-reload' to reload units."

Possibly a timestamp issue with configuration files? But this isn't the place to discuss this.

closed 2021-02-24 by popcornmix

XECDesign, 2021-02-24:
I looked into this a while back. I don't remember the specifics, but it's a systemd issue which is fairly harmless. Since we don't have an RTC or initramfs, systemd gets confused about the time.

SJMarty, 2021-02-24:
Thanks for the feedback.

The issue I'm having is that on my Plex server, I am trying to use a program that stops the plexmediaserver service, runs a backup, and then restarts the service.  Once 'systemctl daemon-reload' is executed, this whole process works fine until the Pi is rebooted.  After a reboot, the service cannot be stopped (or otherwise managed) due to the warning message and the backup fails.  Once 'systemctl daemon-reload' is executed again, all is fine until the next reboot.

What is the proper procedure to officially report it and hopefully, get it fixed?

XECDesign, 2021-02-25:
I don't quite follow. The warning is just a warning and shouldn't actually alter the behaviour in any way.

You could try reporting the issue here https://github.com/systemd/systemd, but my experience with reporting such issues upstream hasn't been productive.

SP3NGL3R, 2021-04-23:
I have the same issue and can't figure it out. Did you ever get it?

SJMarty, 2021-04-24:
I added the following command to /etc/rc.local...

`sleep 300 && sudo systemctl daemon-reload`

I consider this a workaround and not a fix but it has eliminated the issue.

SP3NGL3R, 2021-04-24:
> 
> 
> I added the following command to /etc/rc.local...
> 
> `sleep 300 && sudo systemctl daemon-reload`
> 
> I consider this a workaround and not a fix but it has eliminated the issue.

quite the bodge. but i'll take it. cheers. rebooting to test now .... and works for me.

mbhangui, 2021-07-08:
I have the same issue for a totally different service. It started ocurring when I created an override.conf for that service. If I remove the override.conf, the problem doesn't occur. Seems like a systemd bug where it is not recognizing that a reload has been done after the override.conf was created. Did a strace of systemctl daemon-reload. It goes to /run/systemd directory, communicates with systemd. Looks like it exchanges timestamps.

After reboot, systemd starts up before the raspberry pi sets the correct time. At this time the override.conf is in the future. But why this doesn't happen with the main systemd unit file of the service beats me

```
1156  fstatat64(AT_FDCWD, "/run/systemd/system/", {st_mode=S_IFDIR|0755, st_size=40, ...}, AT_SYMLINK_NOFOLLOW) = 0
1156  geteuid32()                       = 0
1156  getpid()                          = 1156
1156  socket(AF_UNIX, SOCK_STREAM|SOCK_CLOEXEC|SOCK_NONBLOCK, 0) = 3
1156  getsockopt(3, SOL_SOCKET, SO_RCVBUF, [212992], [4]) = 0
1156  setsockopt(3, SOL_SOCKET, SO_RCVBUF, [8388608], 4) = 0
1156  getsockopt(3, SOL_SOCKET, SO_SNDBUF, [212992], [4]) = 0
1156  setsockopt(3, SOL_SOCKET, SO_SNDBUF, [8388608], 4) = 0
1156  connect(3, {sa_family=AF_UNIX, sun_path="/run/systemd/private"}, 23) = 0
1156  getsockopt(3, SOL_SOCKET, SO_PEERCRED, {pid=1, uid=0, gid=0}, [12]) = 0
1156  getsockopt(3, SOL_SOCKET, SO_PEERSEC, 0x5f8d10, [64]) = -1 ENOPROTOOPT (Protocol not available)
1156  getsockopt(3, SOL_SOCKET, SO_PEERGROUPS, 0x5f8d58, [256->0]) = 0
1156  clock_gettime(CLOCK_MONOTONIC, {tv_sec=73, tv_nsec=537747621}) = 0
1156  fstat64(3, {st_mode=S_IFSOCK|0777, st_size=0, ...}) = 0
1156  getsockopt(3, SOL_SOCKET, SO_ACCEPTCONN, [0], [4]) = 0
1156  getsockname(3, {sa_family=AF_UNIX}, [128->2]) = 0
1156  geteuid32()                       = 0
1156  sendmsg(3, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\0AUTH EXTERNAL ", iov_len=15}, {iov_base="30", iov_len=2}, {iov_base="\r\nNEGOTIATE_UNI
1156  getsockopt(3, SOL_SOCKET, SO_PEERCRED, {pid=1, uid=0, gid=0}, [12]) = 0
1156  geteuid32()                       = 0
1156  clock_gettime(CLOCK_MONOTONIC, {tv_sec=73, tv_nsec=539287307}) = 0
1156  recvmsg(3, {msg_namelen=0}, MSG_DONTWAIT|MSG_CMSG_CLOEXEC) = -1 EAGAIN (Resource temporarily unavailable)
1156  clock_gettime(CLOCK_MONOTONIC, {tv_sec=73, tv_nsec=540214714}) = 0
1156  ppoll([{fd=3, events=POLLIN}], 1, {tv_sec=89, tv_nsec=997533000}, NULL, 8) = 1 ([{fd=3, revents=POLLIN}], left {tv_sec=89, tv_nsec=994347814})
1156  clock_gettime(CLOCK_MONOTONIC, {tv_sec=73, tv_nsec=543647640}) = 0
1156  recvmsg(3, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="OK b845306a10de4f6995365f5270fc2"..., iov_len=256}], msg_iovlen=1, msg_controllen=0, msg
1156  clock_gettime(CLOCK_MONOTONIC, {tv_sec=73, tv_nsec=544613881}) = 0
1156  sendmsg(3, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="l\1\4\1\0\0\0\0\1\0\0\0\211\0\0\0\1\1o\0\31\0\0\0/org/fre"..., iov_len=160}], msg_iovlen
1156  clock_gettime(CLOCK_MONOTONIC, {tv_sec=73, tv_nsec=545619195}) = 0
1156  recvmsg(3, {msg_namelen=0}, MSG_DONTWAIT|MSG_CMSG_CLOEXEC) = -1 EAGAIN (Resource temporarily unavailable)
1156  clock_gettime(CLOCK_MONOTONIC, {tv_sec=73, tv_nsec=545817621}) = 0
```

noah505050, 2021-12-11:
Same issue here, once i finish my website config I'll post whether or not it has any negative implications on the website.