raspberrypi/linux #4174: Philips TouCam

tony1tf, 2021-02-24:
Latest linux version on Raspberry Pi 400, running VLC and other players, this USB webcam fails:

libv4l2: error turning on stream: cannot allocate memory

The camera is detected in lsusb:
Bus 001 Device 005: ID 0471:0311 Philips (or NXP) PCVC740K ToUcam Pro [pwc]

Version:
Linux raspberrypi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux

Also fails if you try to use pygame cam.start, can't allocate memory.

Previous version was OK:
Linux raspberrypi 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux

P33M, 2021-02-25:
In each case please post full logs. At a minimum, `dmesg` and `lsusb -v`.

tony1tf, 2021-02-25:
Here's the lsusb -v entry for the camera:

`Bus 001 Device 006: ID 0471:0311 Philips (or NXP) PCVC740K ToUcam Pro [pwc]
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0471 Philips (or NXP)
  idProduct          0x0311 PCVC740K ToUcam Pro [pwc]
  bcdDevice            0.03
  iManufacturer           0 
  iProduct                0 
  iSerial                 1 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x01ea
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              32
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0000  1x 0 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
[... 520 lines cut ...]
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
        bRefresh                0
        bSynchAddress           0
        AudioStreaming Endpoint Descriptor:
          bLength                 7
          bDescriptorType        37
          bDescriptorSubtype      1 (EP_GENERAL)
          bmAttributes         0x00
          bLockDelayUnits         0 Undefined
          wLockDelay         0x0000
`
And the tail of dmesg:
[ 7197.987428] usb 1-1.1: new full-speed USB device number 6 using xhci_hcd
[ 7198.237734] usb 1-1.1: New USB device found, idVendor=0471, idProduct=0311, bcdDevice= 0.03
[ 7198.237743] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=1
[ 7198.237749] usb 1-1.1: SerialNumber: 016900006D1E7201
[ 7198.274025] pwc: Philips PCVC740K (ToUCam Pro)/PCVC840 (ToUCam II) USB webcam detected.
[ 7198.838947] pwc: Registered as video0.
[ 7198.839038] input: PWC snapshot button as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/input/input3
[ 7198.839254] usbcore: registered new interface driver Philips webcam
[ 7198.923691] usbcore: registered new interface driver snd-usb-audio
[ 7279.883021] usb 1-1.2: USB disconnect, device number 3

P33M, 2021-02-25:
And what is the output of dmesg once you have attempted to use the camera (and seen it fail)?

tony1tf, 2021-02-25:
Here:
[ 7957.755194] ------------[ cut here ]------------
[ 7957.755215] WARNING: CPU: 2 PID: 4451 at kernel/dma/mapping.c:149 dma_map_page_attrs+0x380/0x394
[ 7957.755220] Modules linked in: snd_usb_audio snd_hwdep snd_usbmidi_lib snd_rawmidi snd_seq_device pwc cmac rfcomm bnep hci_uart btbcm bluetooth ecdh_generic ecc fuse 8021q garp stp llc cdc_acm brcmfmac brcmutil rpivid_mem sha256_generic bcm2835_codec(C) v3d bcm2835_isp(C) v4l2_mem2mem gpu_sched videobuf2_dma_contig vc4 cec drm_kms_helper raspberrypi_hwmon bcm2835_v4l2(C) cfg80211 bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common rfkill videodev mc vc_sm_cma(C) snd_bcm2835(C) drm drm_panel_orientation_quirks snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[ 7957.755545] CPU: 2 PID: 4451 Comm: vlc Tainted: G         C        5.10.11-v7l+ #1399
[ 7957.755548] Hardware name: BCM2711
[ 7957.755552] Backtrace: 
[ 7957.755564] [<c0b58f24>] (dump_backtrace) from [<c0b592b8>] (show_stack+0x20/0x24)
[ 7957.755569]  r7:ffffffff r6:00000000 r5:60000013 r4:c12e69bc
[ 7957.755576] [<c0b59298>] (show_stack) from [<c0b5d644>] (dump_stack+0xcc/0xf8)
[ 7957.755583] [<c0b5d578>] (dump_stack) from [<c0220af4>] (__warn+0xfc/0x114)
[ 7957.755589]  r10:00000002 r9:00000009 r8:c029f66c r7:00000095 r6:00000009 r5:c029f66c
[ 7957.755592]  r4:c0e22b00 r3:c1205094
[ 7957.755598] [<c02209f8>] (__warn) from [<c0b59a28>] (warn_slowpath_fmt+0x70/0xd8)
[ 7957.755602]  r7:00000095 r6:c0e22b00 r5:c1205048 r4:00000000
[ 7957.755608] [<c0b599bc>] (warn_slowpath_fmt) from [<c029f66c>] (dma_map_page_attrs+0x380/0x394)
[ 7957.755613]  r9:00002580 r8:00000000 r7:dbdf1040 r6:00002580 r5:c1205048 r4:c5243478
[ 7957.755631] [<c029f2ec>] (dma_map_page_attrs) from [<bf4ff384>] (start_streaming+0x298/0x4ac [pwc])
[ 7957.755636]  r10:c1205048 r9:c79904f8 r8:00000000 r7:c7990570 r6:00002580 r5:c65d1400
[ 7957.755640]  r4:c7990000
[ 7957.755658] [<bf4ff0ec>] (start_streaming [pwc]) from [<bf30c150>] (vb2_start_streaming+0x6c/0x178 [videobuf2_common])
[ 7957.755663]  r10:bf5056b8 r9:bf257068 r8:c799050c r7:00000001 r6:c7990398 r5:c7990498
[ 7957.755666]  r4:c7990350
[ 7957.755684] [<bf30c0e4>] (vb2_start_streaming [videobuf2_common]) from [<bf30c8c8>] (vb2_core_streamon+0x90/0x180 [videobuf2_common])
[ 7957.755688]  r7:00000001 r6:c1205048 r5:c59ffe40 r4:c7990398
[ 7957.755705] [<bf30c838>] (vb2_core_streamon [videobuf2_common]) from [<bf322044>] (vb2_streamon+0x24/0x64 [videobuf2_v4l2])
[ 7957.755708]  r5:c59ffe40 r4:00000001
[ 7957.755720] [<bf322020>] (vb2_streamon [videobuf2_v4l2]) from [<bf3220d0>] (vb2_ioctl_streamon+0x4c/0x50 [videobuf2_v4l2])
[ 7957.755758] [<bf322084>] (vb2_ioctl_streamon [videobuf2_v4l2]) from [<bf257094>] (v4l_streamon+0x2c/0x30 [videodev])
[ 7957.755762]  r5:40045612 r4:bf322084
[ 7957.755814] [<bf257068>] (v4l_streamon [videodev]) from [<bf25c114>] (__video_do_ioctl+0x234/0x460 [videodev])
[ 7957.755818]  r5:40045612 r4:c7990000
[ 7957.755868] [<bf25bee0>] (__video_do_ioctl [videodev]) from [<bf25cb28>] (video_usercopy+0x128/0x5ac [videodev])
[ 7957.755873]  r10:c2a75e3c r9:998fe2a8 r8:00000000 r7:c59ffe40 r6:c1205048 r5:40045612
[ 7957.755876]  r4:40045612
[ 7957.755925] [<bf25ca00>] (video_usercopy [videodev]) from [<bf25cfcc>] (video_ioctl2+0x20/0x24 [videodev])
[ 7957.755929]  r10:00000018 r9:c2a74000 r8:00000000 r7:c59ffe40 r6:c59ffe41 r5:c1205048
[ 7957.755933]  r4:bf25cfac
[ 7957.755982] [<bf25cfac>] (video_ioctl2 [videodev]) from [<bf2551f4>] (v4l2_ioctl+0x4c/0x60 [videodev])
[ 7957.756013] [<bf2551a8>] (v4l2_ioctl [videodev]) from [<c044fc04>] (sys_ioctl+0x1d4/0x8ec)
[ 7957.756017]  r5:c1205048 r4:40045612
[ 7957.756023] [<c044fa30>] (sys_ioctl) from [<c0200040>] (ret_fast_syscall+0x0/0x28)
[ 7957.756027] Exception stack(0xc2a75fa8 to 0xc2a75ff0)
[ 7957.756032] 5fa0:                   a1866300 99c36004 00000018 40045612 998fe2a8 00000001
[ 7957.756036] 5fc0: a1866300 99c36004 40045612 00000036 994f8e10 99c36004 00000018 998ff8b0
[ 7957.756040] 5fe0: 998fe2a8 998fe298 99c1f808 b6e88af0
[ 7957.756045]  r10:00000036 r9:c2a74000 r8:c0200204 r7:00000036 r6:40045612 r5:99c36004
[ 7957.756049]  r4:a1866300
[ 7957.756053] ---[ end trace 7004563b997e857b ]---
[ 7957.756062] pwc: Failed to allocate urb buffer 0
[ 7957.932972] pwc: Failed to allocate urb buffer 0

P33M, 2021-02-25:
The pwc driver appears to do dma mapping itself in pwc_alloc_urb_buffer(), rather than using the USB HCD's automatic handling of mapping/unmapping on URB submission/hand-back or using usb_alloc_coherent()/usb_free_coherent(). It's also unique in doing it this way.

tony1tf, 2021-02-25:
Hi P33M
Thanks for investigating - is there a fix, apart from buying a different webcam ?
Tony

P33M, 2021-02-25:
I would suggest that the webcam belongs in a museum...

The driver breakage is a regression, possibly due to internal API changes. It would need bisecting to find the exact commit between 5.4 and 5.10 that broke the driver. Bisecting upstream regressions is a bit of a pain on a Pi, since there are many downstream patches. It's usually faster to reproduce the bug on x86, then test the unmodified upstream kernel that way.

My inclination would be to make the pwc driver do whatever the large pile of other webcam drivers do, but without hardware to test it on, chances of fixing it are small.

6by9, 2021-02-25:
If you can reproduce the issue on x86 then it'd be worth reporting it upstream via the linux-media mailing list (http://vger.kernel.org/vger-lists.html#linux-media).
They are the ones who maintain the drivers, and have the knowledge of how they work, although quirks due to weird hardware (often anything tht isn't x86) will sometimes be ignored.

tony1tf, 2021-02-25:
Thanks P33M - unfortunately you speak a different language from me as a mere engineer ;-) 
Gordon77 on the raspberry pi forum has written software to use this old webcam for telescope guiding with an R Pi, so this is now broken with the later release. How does a newbie to this advance linux stuff do regressions on the software?

6by9, 2021-02-25:
Forum thread (now locked) - https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=305091

P33M, 2021-02-25:
If reproducible on 5.10 x86 (e.g Ubuntu Linux 21.04) and you want to pin down the exact version yourself, then there's a pretty comprehensive guide - might want to skip to "bisecting upstream kernels" though.
https://wiki.ubuntu.com/Kernel/KernelBisection

If installing a dozen or so different kernels is daunting then I'd do what @6by9 suggests - report it upstream.

P33M, 2021-02-25:
As if by magic:
https://www.spinics.net/lists/linux-media/msg185882.html

Already reported last month. Presumably patch will make its way through in due course.

6by9, 2021-02-25:
Checking on patchwork.linuxtv.org, https://patchwork.linuxtv.org/project/linux-media/patch/20210104170007.20625-1-matwey@sai.msu.ru/ has been accepted.

The patch you linked to has been marked as obsoleted (https://patchwork.linuxtv.org/project/linux-media/patch/20210121202855.17400-1-tiwai@suse.de/). Although [comment 8](https://patchwork.linuxtv.org/project/linux-media/patch/20210121202855.17400-1-tiwai@suse.de/#125989) is a little amusing considering this discussion. That fix will be in 5.12.

tony1tf, 2021-02-25:
Hi all

Thanks for finding the patch. Re comment 8 - yes indeed, there are people out here using non-intel devices. I have a collection of USB webcams which are slowly going 'stale'. When you say the fix will be in 5.12, is there a patch for armhf now, or do I build the driver from source, or something similar - sorry that I know so little about these matters as a retired space scientist/engineer !

Tony

tony1tf, 2021-02-25:
I just looked at the link to bisecting upstream kernels - sorry - completely over my head! I'll wait for you experts to propagate the patch to us lesser mortals.

6by9, 2021-02-25:
It has made it to Linus' [master branch](https://github.com/torvalds/linux/commit/69c9e825e812ec6d663e64ebf14bd3bc7f37e2c7), and has a Fixes: tag, so it should make it's way back to the older stable kernels in due course (probably the next 2 weeks).

Unless it is a critical patch then we don't expedite pulling the patch back to our trees.

Gordon999, 2021-02-25:
Great there will be a fix. Maybe an old webcam but still very useful,  and was very popular for astrophotography .

FlareP1, 2021-03-10:
I reported this earlier in the Month see https://github.com/Hexxeh/rpi-firmware/issues/254
Just tried the latest pre-release branch today which has been updated to  5.10.20-v7+and can confirm that the PWC webcam is now working again.  In order to get this you would need to follow the instuction here
https://www.raspberrypi.org/documentation/raspbian/applications/rpi-update.md

tony1tf, 2021-03-10:
> I reported this earlier in the Month see Hexxeh/rpi-firmware#254

Hi FlareP1

Thanks for the heads up. I can also confirm that my PCVC740K ToUcam Pro is
now working. It's a useful webcam to use with the RPi400 which no longer
has the RPi camera connector. It also has a microphone in the camera - good
for Zoom meetings I hope.

Tony

P33M, 2021-03-15:
Closing as fixed upstream.

closed 2021-03-15 by P33M