raspberrypi/linux #4176: 8250.nr_uarts not working on 5.10.x and 5.4.x

TopgunZh, 2021-02-25:
### I test the raspberry pi Compute Module 4.
`
pi@raspberrypi:~ $ sudo uname -a
Linux raspberrypi 5.10.16-v7l #2 SMP Wed Feb 24 12:19:14 CST 2021 armv7l GNU/Linux
`
### I add "8250.nr_uarts=10" in cmdline.txt
`
pi@raspberrypi:~ $ cat /boot/cmdline.txt 
8250.nr_uarts=10 console=serial0,115200 console=tty1 root=PARTUUID=9d11c471-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles pci=nomsi
`
### But the boot change my value ，repacle"8250.nr_uarts=10" with "8250.nr_uarts=00"
```
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1010432
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60,margin_left=48,margin_right=48,margin_top=48,margin_bottom=48 smsc95xx.macaddr=DC:A6:32:DC:C2:D1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  8250.nr_uarts=00 console=ttyS0,115200 console=tty1 root=PARTUUID=9d11c471-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles pci=nomsi
[    0.000000] Kernel parameter elevator= does not have any effect anymore.
               Please use sysfs to set IO scheduler for individual devices.
```

pelwell, 2021-02-25:
Why do you want to set 8250.nr_uarts to 10? What did you think would happen?

TopgunZh, 2021-03-03:
I've solved the problem.”8250.nr_uarts=9“ is work well!

closed 2021-03-03 by TopgunZh