raspberrypi/linux #4178: raspberry pi zero -  reproducible - unable to handle kernel NULL pointer dereference

mhano, 2021-02-26:
**Is this the right place for my bug report?**

reliable/repeatable crash with most basic of steps to reproduce

**Describe the bug**

Raspberry Pi zero w crashes (upgraded or not) on downloading 1GB file with curl.

Have reproduced with the following installs:
 - tried two different wifi networks / different brands of router
 - full desktop raspbian offline install from noobs sd card
 - raspbian lite
 - raspbian lite - upgraded (apt update and upgrade after initial setup)
 - two different 16GB micro sd cards

**To reproduce**

install 
- either full desktop raspbian offline; or 
- install raspbian lite

configure basics -
- host name “raspi4z”
- wifi AU wifi region
- default locale - AU/Sydney time zone
- Boot to CLI instead of desktop (if installed desktop)
- Set password on pi account
- Enable ssh
- Install pivpn via standard curl | bash (not enabling auto upgrade)

Reboot login as pi / run

for i in {0..9}; do
curl -o ~/test-$i.dl http://speedcheck.cdn.on.net/1000meg.test
done

usually crashes before downloading first 1GB file

**Expected behaviour** doesn’t crash.
**Actual behaviour** crashes.

**System**
 
* Which model of Raspberry Pi? PiZeroW
* Which OS and version: Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068ba
d27d, stage5
* Which firmware version? version e90cba19a98a0d1f2ef086b9cafcbca00778f094 (clean) (release) (start)
* Which kernel version? 
Linux raspi4z 5.4.51+ #1333 Mon Aug 10 16:38:02 BST 2020 armv6l GNU/Linux

**Logs**
Screen shot / photo of crash attached (ssh connection dies on crash - but local hdmi console shows error).

[image]

**Additional context**
RaspberryPi completely unresponsive / frozen after crash (can’t interact with local keyboard or via ssh).

** raspinfo **

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       12437304 7013092   4769392  60% /
devtmpfs          188256       0    188256   0% /dev
[... 507 lines cut ...]
[   35.248907] 8021q: 802.1Q VLAN Support v1.8
[   36.032338] uart-pl011 20201000.serial: no DMA platform data
[   38.148074] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   39.349738] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   40.798921] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   40.893899] Bluetooth: Core ver 2.22
[   40.894056] NET: Registered protocol family 31
[   40.894071] Bluetooth: HCI device and connection manager initialized
[   40.894109] Bluetooth: HCI socket layer initialized
[   40.894130] Bluetooth: L2CAP socket layer initialized
[   40.894178] Bluetooth: SCO socket layer initialized
[   40.949264] Bluetooth: HCI UART driver ver 2.3
[   40.949290] Bluetooth: HCI UART protocol H4 registered
[   40.949429] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   40.949749] Bluetooth: HCI UART protocol Broadcom registered
[   41.881778] tun: Universal TUN/TAP device driver, 1.6
[   42.580672] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.wlan0.
retrans_time - use net.ipv6.neigh.wlan0.retrans_time_ms instead
[   43.307686] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   43.307706] Bluetooth: BNEP filters: protocol multicast
[   43.307741] Bluetooth: BNEP socket layer initialized
[   44.123017] Bluetooth: RFCOMM TTY layer initialized
[   44.123065] Bluetooth: RFCOMM socket layer initialized
[   44.123124] Bluetooth: RFCOMM ver 1.11
pi@raspi4z:~$

mhano, 2021-02-26:
syslog of crash caused by writing 1GB files from /dev/urandom...

Feb 26 19:00:39 raspi4z systemd[1]: Started Session c7 of user pi.
Feb 26 19:06:49 raspi4z systemd[1]: Started Session c8 of user pi.
Feb 26 19:12:19 raspi4z kernel: [ 5322.086977] 8<--- cut here ---
Feb 26 19:12:19 raspi4z kernel: [ 5322.091703] Unable to handle kernel paging request at virtual address fadcf041
Feb 26 19:12:19 raspi4z kernel: [ 5322.096425] pgd = 9dd85b50
Feb 26 19:12:19 raspi4z kernel: [ 5322.100795] [fadcf041] *pgd=00000000
Feb 26 19:12:19 raspi4z kernel: [ 5322.105406] Internal error: Oops: 5 [#1] ARM
Feb 26 19:12:19 raspi4z kernel: [ 5322.109990] Modules linked in: binfmt_misc aes_arm aes_generic cmac rfcomm bnep tun hci_uart btbcm bluetooth ecdh_generic ecc libaes 8021q garp stp llc nft_counter nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack nf_defrag_ipv4 xt_comment nft_compat nf_tables nfnetlink evdev joydev brcmfmac brcmutil sha256_generic libsha256 cfg80211 rfkill raspberrypi_hwmon bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig snd_bcm2835(C) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common snd_pcm snd_timer snd videodev mc vc_sm_cma(C) uio_pdrv_genirq uio fixed i2c_dev ip_tables x_tables ipv6 nf_defrag_ipv6
Feb 26 19:12:19 raspi4z kernel: [ 5322.126404] CPU: 0 PID: 34 Comm: kswapd0 Tainted: G         C        5.4.51+ #1333
Feb 26 19:12:19 raspi4z kernel: [ 5322.132041] Hardware name: BCM2835
Feb 26 19:12:19 raspi4z kernel: [ 5322.137834] PC is at shrink_page_list+0x688/0xfd4
Feb 26 19:12:19 raspi4z kernel: [ 5322.143721] LR is at 0x0
Feb 26 19:12:19 raspi4z kernel: [ 5322.149268] pc : [<c01562d0>]    lr : [<00000000>]    psr: 60000013
Feb 26 19:12:19 raspi4z kernel: [ 5322.155279] sp : d7649cb8  ip : c0a473fc  fp : d7649d64
Feb 26 19:12:19 raspi4z kernel: [ 5322.161245] r10: d7649f10  r9 : 00000000  r8 : d7649ce0
Feb 26 19:12:19 raspi4z kernel: [ 5322.166853] r7 : d7649d98  r6 : d7649d90  r5 : d78c7864  r4 : d78c7868
Feb 26 19:12:19 raspi4z kernel: [ 5322.172752] r3 : fadcf041  r2 : fadcf041  r1 : 00011eb0  r0 : 00000001
Feb 26 19:12:19 raspi4z kernel: [ 5322.178620] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Feb 26 19:12:19 raspi4z kernel: [ 5322.184202] Control: 00c5387d  Table: 16bac008  DAC: 00000055
Feb 26 19:12:19 raspi4z kernel: [ 5322.190060] Process kswapd0 (pid: 34, stack limit = 0x1b0e0147)
Feb 26 19:12:19 raspi4z kernel: [ 5322.195940] Stack: (0xd7649cb8 to 0xd764a000)
Feb 26 19:12:19 raspi4z kernel: [ 5322.201487] 9ca0:                                                       c0a47028 c0adee88
Feb 26 19:12:19 raspi4z kernel: [ 5322.207394] 9cc0: 00000000 00000007 00000000 d5705690 d5705690 00000001 c00148c0 0000355c
Feb 26 19:12:19 raspi4z kernel: [ 5322.213389] 9ce0: d7649ce0 d7649ce0 d78c7844 d78d5aac 00000000 d7649d00 c00135d0 c00148a8
Feb 26 19:12:19 raspi4z kernel: [ 5322.219158] 9d00: d7649d90 00000000 c0158074 d7649e84 d7649df0 c00d9878 c0b6a700 c015757c
Feb 26 19:12:19 raspi4z kernel: [ 5322.225258] 9d20: c0a60dfc c0158074 00000001 c0adf4b0 d7649d64 fadcf041 c00d9884 c0b6a700
Feb 26 19:12:19 raspi4z kernel: [ 5322.231298] 9d40: 00000020 d7649f10 c0adf518 00000001 c0adf4b0 d7649d90 d7649dec d7649d68
Feb 26 19:12:19 raspi4z kernel: [ 5322.237345] 9d60: c01575cc c0155c54 d7649d98 00000000 c059b8a8 00000000 c0768378 c0adf51c
Feb 26 19:12:19 raspi4z kernel: [ 5322.243629] 9d80: c0adf51c c0adee88 d7649dec 00000020 d795a208 d78c788c 00000000 00000000
Feb 26 19:12:19 raspi4z kernel: [ 5322.249688] 9da0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 fadcf041
Feb 26 19:12:19 raspi4z kernel: [ 5322.256100] 9dc0: d7649f10 00000002 d7649e1c 00000060 d7649f10 00000000 00000020 c0adf4b0
Feb 26 19:12:19 raspi4z kernel: [ 5322.262545] 9de0: d7649e84 d7649df0 c0158074 c01573ec ffffffff ffffffff 00000000 00000000
Feb 26 19:12:19 raspi4z kernel: [ 5322.268743] 9e00: d7649e40 00000003 0000330c 00001800 c0154b98 00000000 00000000 00000172
Feb 26 19:12:19 raspi4z kernel: [ 5322.275308] 9e20: 00000006 c0a65c64 00000000 00000000 000001f2 00000066 c0a65c64 00000000
Feb 26 19:12:19 raspi4z kernel: [ 5322.281759] 9e40: d7649e40 d7649e40 d7649e48 d7649e48 00000000 fadcf041 00000000 d7649f10
Feb 26 19:12:19 raspi4z kernel: [ 5322.288397] 9e60: 00000000 00000000 00000000 c0a47858 000001e7 c0adee88 d7649edc d7649e88
Feb 26 19:12:19 raspi4z kernel: [ 5322.295235] 9e80: c0158508 c0157ce0 d7649f04 d757e560 d7649eb4 d7649f2c d7649f48 00000000
Feb 26 19:12:19 raspi4z kernel: [ 5322.301893] 9ea0: 00000000 000001e7 00012c66 fadcf041 00000ab7 00001800 00000000 00000001
Feb 26 19:12:19 raspi4z kernel: [ 5322.308934] 9ec0: 00000001 00000000 c0adf4a8 c0adee88 d7649f7c d7649ee0 c01598d4 c015845c
Feb 26 19:12:19 raspi4z kernel: [ 5322.315784] 9ee0: d75bb900 00000000 00000000 d7649f1c 00000000 00000000 c0adf480 000001e7
Feb 26 19:12:19 raspi4z kernel: [ 5322.323008] 9f00: d7649f80 00000000 00000000 00000000 00001800 00000000 00000000 00070007
Feb 26 19:12:19 raspi4z kernel: [ 5322.330264] 9f20: 00000cc0 00000068 000001e7 00000000 00000000 00000000 00000000 00000000
Feb 26 19:12:19 raspi4z kernel: [ 5322.337229] 9f40: 00000060 00000060 00000001 fadcf041 c00433d0 d75fc980 d76083c0 00000000
Feb 26 19:12:19 raspi4z kernel: [ 5322.344536] 9f60: c0adee88 c015955c d74e1e18 d75fc998 d7649fac d7649f80 c0044134 c0159568
[... cut ...]
Feb 26 19:12:30 raspi4z kernel: [ 5334.276271] 1fa0:                   00000000 00001000 00000001 01788000 00001000 00000000
Feb 26 19:12:30 raspi4z kernel: [ 5334.285008] 1fc0: 00000000 00001000 0002f1e0 00000004 00000000 00000001 00000000 0002f150
Feb 26 19:12:30 raspi4z kernel: [ 5334.294012] 1fe0: 0000006c be94f950 000145a4 b6e69944
Feb 26 19:12:30 raspi4z kernel: [ 5334.302641] Code: e5943000 e3130001 12433001 01a03005 (e5933000) 
Feb 26 19:12:30 raspi4z kernel: [ 5334.311820] ---[ end trace 410ca2b57413bd42 ]---
Feb 26 19:15:48 raspi4z systemd[1]: Started Session c9 of user pi.

mhano, 2021-03-02:
I’ll purchase another pi zero w board from a different supplier and try to confirm whether it’s a hardware issue specific to this board.

mhano, 2021-03-02:
New board ordered - should arrive tomorrow.

almost certain this is a heat/hardware issue...

- writing large files causes it to crash.
- crashes during heavy workloads (big installs/upgrades)
- downloading data to it doesn’t cause crashes
- Uploading from it causes crashes (e.g. 
from another host run: ssh pi@raspi4z "timeout $timeout cat /dev/urandom | base64 -w 100" | pv | wc)
- Uploading from it whilst the board is out of its case draped over a desk fan has not crashed it yet...

After performing the upload from it test (in its case - the board was warm - but not too hot to touch).

popcornmix, 2021-03-02:
If it is a hardware issue then it may be worth trying either:
```
over_voltage=2
```
or
```
arm_freq=800
```
in config.txt. Maybe it is borderline reliable at 1000MHz and the extra heat pushes it over.

mhano, 2021-03-03:
Having clocked to 800 per suggestion above I haven’t managed to cause it to crash.

Im running the upload via ssh test (a few GBs in) in parallel with the writing large files from urand.

will be interesting to see how I go with replacement board I’ve ordered.

Is there a way I should report this hardware issue or just return to the retailer?

mhano, 2021-03-04:
Definitely hardware issue

No issue running the same tests on new board

closed 2021-03-04 by mhano

mhano, 2021-03-05:
Board appears physically fine.

[image]