raspberrypi/linux #4179: Please remove limitation to only 8 USB MIDI devices

mark-orion, 2021-02-26:
**Describe the bug**
I am unable to connect more than 8 USB MIDI devices to my PI. I am getting this error in the dmesg kernel log:
snd-usb-audio 1-1.2.3:1.0: cannot find the slot for index -1 (range 0-7), error: -19
I am experiencing the issue on Zynthian OS 

**Expected behaviour**
Modern Raspberry Pis (version 3 and up) should not have such a strict limit for USB MIDI.
I would suggest to either remove the limit completely or add an option to /boot/config.txt to change the limit without the need to recompile the kernel.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Pi4b
* Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage2
* Aug 19 2020 17:38:16 
Copyright (c) 2012 Broadcom
version e90cba19a98a0d1f2ef086b9cafcbca00778f094 (clean) (release) (start)
* Linux zynthian 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux

Koshdukai, 2023-10-04:
Please ? :)
I suspect the kernel wasn't (re)built yet with CONFIG_SND_MAX_CARDS=16 or better yet =32 ;)

https://forums.raspberrypi.com//viewtopic.php?t=177458

To make things worse, some MIDI controllers have multiple USB MIDI ports like Arturia KeyLab Essentials, KeyLab mkII or MiniLab mk3 which makes that kernel-level USB limitation even more limiting.

This helped a little bit but it's not a real solution for this solvable issue.
https://discourse.zynthian.org/t/using-more-than-4-midi-usb-controllers-partially-solved/5605/25?page=2

Thanks! :)

6by9, 2023-10-04:
It needs CONFIG_SND_DYNAMIC_MINORS to be set, and then CONFIG_SND_MAX_CARDS to be defined (default is 32 if CONFIG_SND_DYNAMIC_MINORS is set).

CONFIG_SND_DYNAMIC_MINORS changes the kernel ABI though. 
```
  If you say Y here, the minor numbers of ALSA device files in
	  /dev/snd/ are allocated dynamically.  This allows you to have
	  more than 8 sound cards, but requires a dynamic device file
	  system like udev.
```
Seeing as it was introduced in 2005, I'd expect all normal users of the ABI to understand the new version.

I don't see any significant arrays that are sized by that config, so change in memory usage should be minimal.

pelwell, 2023-10-06:
Pull Request #5637 enables "dynamic minors", raising the limit to 32 sounds cards as @6by9 said. If you wait an hour or so until the build checks have completed you'll be able to install a trial kernel using `sudo rpi-update pulls/5637`.

PR #5637 (2023-10-11): configs: Enable more sound devices
This should resolve #4179.
closed 2023-10-11 by pelwell