raspberrypi/linux #4183: MCP251x failes setup CAN-bus communication with error=110

timok90, 2021-03-01:
My project uses the mcp2515 chip to communicate via CAN-bus.
After kernel update following failure occurs:

```
~ $ dmesg | grep -i spi
[    7.567794] mcp251x spi0.0: MCP251x didn't enter in conf mode after reset 
[    7.567855] mcp251x spi0.0: Probe failed, err=110
[    7.567948] mcp251x: probe of spi0.0 failed with error -110 

```
My overlay configuration is:

```
dtparam=spi=on
dtoverlay=mcp2515-can0,oscillator=16000000,interrupt=25,spimaxfrequency=500000
```

My current kernel version is: 
`Linux 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux`

os: 
`PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"`

and additionally I use the can-utils package.

pelwell, 2021-03-04:
This sounds like a hardware problem - have you double-checked all of your wiring?

timok90, 2021-03-15:
I've checked it but with no changes. Rework all the wiring helped. So, it was a wiring issue.

closed 2021-03-15 by timok90

Elhkiati2000, 2024-02-16:
I encountered the same error while attempting to use the mcp2515 chip for CAN-bus communication over SPI 1-0/1/2 on the RPi 3B+. Communication over SPI 0-0/1cs is successful using the "mcp2515-can0.dtbo" and "mcp2515-can1.dtbo" overlays. I've also successfully tested SPI 0 with the newly introduced overlay "mcp2515.dtbo" found in the Raspberry OS, which functions correctly. However, attempting to use the chip with SPI1-0/1/2cs results in the same error. 

Below is the description of the "mcp2515.dtbo" overlay located in /boot/overlays/README:

```
Name:   mcp2515
Info:   Configures the MCP2515 CAN controller on spi0/1/2
        For devices on spi1 or spi2, the interfaces should be enabled
        with one of the spi1-1/2/3cs and/or spi2-1/2/3cs overlays.
Load:   dtoverlay=mcp2515,<param>=<val>
        Params: spi<n>-<m>              Configure device at spi<n>, cs<m> (boolean, required)
        oscillator                                  Clock frequency for the CAN controller (Hz)
        speed                                       Maximum SPI frequence (Hz)
        interrupt                                  GPIO for interrupt signal
```

- The connections are : 

INT -- 22 (GPIO 25)
SCK -- 40 (GPIO 21 , SPI1 SCLK)
SI -- 38 (GPIO 20 , SPI1 MOSI)
SO -- 35 (GPIO 19 , SPI1 MISO)
CS -- 12 (GPIO 18 , SPI1 CS0)
GND -- 6 (GND)
VCC - 1 (3V3)

- The content of /boot/config.txt file is as follows : 
`dtparam=spi=on
dtoverlay=spi1-3cs,cs0_spidev=disabled,cs1_spidev=disabled
dtoverlay=mcp2515,spi1-2=16,oscillator=16000000,speed=10000000,interrupt=25
`

- The decompiled content of the "mcp2515.dtbo" file located in /boot/overlays, resulting in "mcp2515.dts", is as detailed below:

`/dts-v1/;

/ {
	compatible = "brcm,bcm2835";

	fragment@0 {
		target = <0xffffffff>;

		__dormant__ {
			status = "disabled";
		};
	};

	fragment@1 {
		target = <0xffffffff>;

		__dormant__ {
			status = "disabled";
		};
	};

	fragment@2 {
		target-path = "spi1/spidev@0";

		__dormant__ {
			status = "disabled";
		};
	};

	fragment@3 {
		target-path = "spi1/spidev@1";

		__dormant__ {
			status = "disabled";
		};
	};
[... 114 lines cut ...]
			__overlay__ {
				mcp2515@0 {
					pinctrl-0 = <0x00>;
					clocks = <0x00>;
				};
			};
		};

		__overrides__ {
			spi0-0 = <0x07 0x19 0x25 0x42>;
			spi0-1 = <0x07 0x19 0x25 0x42>;
			spi1-0 = <0x07 0x19 0x25 0x42>;
			spi1-1 = <0x07 0x19 0x25 0x42>;
			spi1-2 = <0x07 0x19 0x25 0x42>;
			spi2-0 = <0x07 0x19 0x25 0x42>;
			spi2-1 = <0x07 0x19 0x25 0x42>;
			spi2-2 = <0x07 0x19 0x25 0x42>;
			oscillator = <0x00>;
			speed = <0x00>;
			interrupt = <0x00 0x10>;
		};
	};
};
`
Additionally, I've tested the "mcp2515-can2-overlay.dtbo" and "mcp2515-can0-1.dtbo" overlays mentioned in issue #1804 , which resulted in the same error. I've double-checked the wiring, and it appears to be correct. Can someone assist?