raspberrypi/linux #4184: eth0: hw csum failure

ralphrmartin, 2021-03-01:
**Is this the right place for my bug report?**
I believe so

**Describe the bug**
I am Running
Linux services 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l
on 2 different raspberry Pi 4s, my syslog is rapidly filling up with messages containing
```
Mar  1 11:23:01 services kernel: [ 1625.153516] [<c1000b30>] (start_kernel) from [<00000000>] (0x0)
Mar  1 11:23:01 services kernel: [ 1625.153541] eth0: hw csum failure
Mar  1 11:23:01 services kernel: [ 1625.153557] skb len=68 headroom=164 headlen=68 tailroom=1944
Mar  1 11:23:01 services kernel: [ 1625.153557] mac=(130,14) net=(144,20) trans=164
Mar  1 11:23:01 services kernel: [ 1625.153557] shinfo(txflags=0 nr_frags=0 gso(size=0 type=0 segs=0))
Mar  1 11:23:01 services kernel: [ 1625.153557] csum(0x20035181 ip_summed=2 complete_sw=0 valid=0 level=0)
Mar  1 11:23:01 services kernel: [ 1625.153557] hash(0x0 sw=0 l4=0) proto=0x0800 pkttype=0 iif=2
Mar  1 11:23:01 services kernel: [ 1625.153567] dev name=eth0 feat=0x0x0000010000004829
Mar  1 11:23:01 services kernel: [ 1625.153576] sk family=2 type=1 proto=6
```
As this is happening on 2 different machines, it seems unlikely this is a real hardware failure, and indeed, networking
seems to be working OK on these machines.
The logs are rapidly filling with many GB of such messages, which is hardly helpful, either.

**To reproduce*
This started within the last month since doing an apt full-upgrade. Otherwise the 2 machines are running a variety of services quite unrelated to each other, so it seems doing any networking causes the problem

Bothe machines are connected to a Unifi switch, powered by PoE with an official PoE hat.

**Expected behaviour**
Networking should not be generating spurious(?) hw error messages.

**Actual behaviour**
See above. Further details of syslog contents can be provided on request

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

command not found: raspinfo

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi 4 8GB

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage2

* Which firmware version (`vcgencmd version`)?
Jan 27 2021 22:19:57 
Copyright (c) 2012 Broadcom
version 99d9a48302e4553cff3688692bb7e9ac760a03fa (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux services 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux

**Logs**
see above. more can be provided if needed

**Additional context**
Seems to have arisen in the last month's updates. I noticed the enormous log file sizes doing a monthly backup, and the log file sizes were as expected last month

6by9, 2021-03-01:
As this hasn't been reported by anyone else (that I'm aware of), is there anything odd about your networking setup? Do you have any VLANs configured? Or a custom MTU defined on the interface? Unusual TCP options? IPv6 running? Multicast traffic?

The dump is kicked up off https://elixir.bootlin.com/linux/latest/source/net/core/dev.c#L3413

```sk family=2 type=1 proto=6```
sk family 2 is [```AF_INET``` (aka IPv4)](https://elixir.bootlin.com/linux/latest/source/include/linux/socket.h#L179)
type 1 is [```SOCK_STREAM```](https://elixir.bootlin.com/linux/latest/source/include/linux/net.h#L64)
proto 6 is [```IPPROTO_TCP```](https://elixir.bootlin.com/linux/latest/source/include/uapi/linux/in.h#L37)
TCP over IPv4 is used everywhere, so it can't be a generic bug with it.

The error message is rate limited, but what is the rate that these events are being logged at?

ralphrmartin, 2021-03-01:
No VLANS, No custom MTU, No special TCP options. IPV6 is in use, No multicast traffic. In short, nothing unusual at all.

Apart from setting a hostname, my dhcpcd.conf file only has
```
clientid
persistent
option rapid_commit
option domain_name_servers, domain_name, domain_search, host_name
option classless_static_routes
option ntp_servers
option interface_mtu
require dhcp_server_identifier
slaac hardware
interface eth0
static ip_address=192.168.1.100/24
static routers=192.168.1.254
static domain_name_servers=192.168.1.102 1.1.1.1 8.8.8.8
noarp
denyinterfaces wlan0
```

The errors are happening at a tremendous rate, many times every second. I've got 600MB of error messages in my syslog since resetting it perhaps 5 hours ago.

ralphrmartin, 2021-03-01:
Found the raspinfo command. Here's the output.

```
System Information
------------------

Raspberry Pi 4 Model B Rev 1.4
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2014-06-20 (armhf)
Generated using spindle, http://asbradbury.org/projects/spindle/, 8fe9fd4, stage4-lxde-edu.qed

Linux www 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
Revision	: d03114
Serial		: 100000006d1ed62e
Model		: Raspberry Pi 4 Model B Rev 1.4
Throttled flag  : throttled=0x0
Camera          : supported=1 detected=0

Videocore information
---------------------

Jan 27 2021 22:20:57 
Copyright (c) 2012 Broadcom
version 99d9a48302e4553cff3688692bb7e9ac760a03fa (clean) (release) (start_x)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks      Used Available Use% Mounted on
/dev/root      960128032 186785184 724501224  21% /
devtmpfs         4004100         0   4004100   0% /dev
tmpfs            4037892         0   4037892   0% /dev/shm
tmpfs            4037892    411112   3626780  11% /run
tmpfs               5120         0      5120   0% /run/lock
tmpfs            4037892         0   4037892   0% /sys/fs/cgroup
/dev/sda1         254700     49264    205436  20% /boot
tmpfs             807576         0    807576   0% /run/user/1000

Filename				Type		Size		Used		Priority
/var/swap                               file		2097148		0		-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: 1.20201210+nmu1
raspberrypi-sys-mods:
  Installed: 20210208
openbox:
  Installed: 3.6.1-8+rpt5
lxpanel:
  Installed: 0.10.0-2+rpt14
pcmanfm:
  Installed: 1.3.1-1+rpt26
rpd-plym-splash:
  Installed: 0.26

Networking Information
----------------------

/usr/bin/raspinfo: line 67: ifconfig: command not found

USB Information
---------------
[... 2421 lines cut ...]
[20996.186332]  r4:c120308c
[20996.186340] [<c0201370>] (__do_softirq) from [<c02272fc>] (irq_exit+0xd0/0xf8)
[20996.186346]  r10:c0e1d860 r9:c1200000 r8:c1810800 r7:00000001 r6:00000000 r5:00000000
[20996.186351]  r4:ffffe000
[20996.186361] [<c022722c>] (irq_exit) from [<c02870b4>] (__handle_domain_irq+0x70/0xc4)
[20996.186365]  r5:00000000 r4:c1094d10
[20996.186372] [<c0287044>] (__handle_domain_irq) from [<c020135c>] (gic_handle_irq+0x90/0xa4)
[20996.186378]  r9:c1200000 r8:c1094d1c r7:c1201ec8 r6:f081400c r5:f0814000 r4:c1205b3c
[20996.186385] [<c02012cc>] (gic_handle_irq) from [<c0200abc>] (__irq_svc+0x5c/0x7c)
[20996.186389] Exception stack(0xc1201ec8 to 0xc1201f10)
[20996.186395] 1ec0:                   00000000 0080cfc8 eff722c4 c021ab60 ffffe000 c120509c
[20996.186401] 1ee0: c12050e4 00000001 00000001 c133a97e c0e1d860 c1201f24 c1201f28 c1201f18
[20996.186406] 1f00: c0208878 c020887c 60000013 ffffffff
[20996.186412]  r9:c1200000 r8:00000001 r7:c1201efc r6:ffffffff r5:60000013 r4:c020887c
[20996.186421] [<c0208834>] (arch_cpu_idle) from [<c0b68da0>] (default_idle_call+0x4c/0x118)
[20996.186428] [<c0b68d54>] (default_idle_call) from [<c025837c>] (do_idle+0x118/0x168)
[20996.186435] [<c0258264>] (do_idle) from [<c02586a0>] (cpu_startup_entry+0x28/0x30)
[20996.186441]  r10:00000189 r9:c1053a60 r8:ffffffff r7:c1053a60 r6:c1205040 r5:c1205048
[20996.186445]  r4:000000d8 r3:c108a294
[20996.186451] [<c0258678>] (cpu_startup_entry) from [<c0b61cf8>] (rest_init+0xbc/0xc4)
[20996.186459] [<c0b61c3c>] (rest_init) from [<c1000ab4>] (arch_call_rest_init+0x18/0x1c)
[20996.186466]  r5:c1205048 r4:c1353068
[20996.186473] [<c1000a9c>] (arch_call_rest_init) from [<c1001098>] (start_kernel+0x568/0x59c)
[20996.186480] [<c1000b30>] (start_kernel) from [<00000000>] (0x0)
```

ralphrmartin, 2021-03-01:
And in case it helps, ip addr gives
```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether dc:a6:32:e7:6f:ad brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.100/24 brd 192.168.1.255 scope global noprefixroute eth0
       valid_lft forever preferred_lft forever
    inet6 2a00:23c6:7993:c001:dea6:32ff:fee7:6fad/64 scope global dynamic mngtmpaddr noprefixroute 
       valid_lft 2591979sec preferred_lft 604779sec
    inet6 fe80::dea6:32ff:fee7:6fad/64 scope link 
       valid_lft forever preferred_lft forever
```

6by9, 2021-03-01:
This is why we ask for the full logs.

Extracting a small part from one of those error splats
```
[20996.183043] eth0: hw csum failure
[20996.183069] skb len=68 headroom=144 headlen=68 tailroom=1964
               mac=(130,14) net=(144,20) trans=164
               shinfo(txflags=0 nr_frags=0 gso(size=0 type=0 segs=0))
               csum(0x4807 ip_summed=2 complete_sw=0 valid=0 level=0)
               hash(0x0 sw=0 l4=0) proto=0x0800 pkttype=2 iif=0
[20996.183084] dev name=eth0 feat=0x0x0000010000004829
[20996.183095] skb headroom: 00000000: 00 99 13 da 00 00 00 00 10 69 83 c2 18 ea 08 db
[20996.183104] skb headroom: 00000010: d8 d3 27 cd 18 ea c0 c0 00 00 00 00 00 00 00 00
[20996.183113] skb headroom: 00000020: 00 00 00 00 00 80 20 00 00 40 08 00 00 00 00 00
[20996.183123] skb headroom: 00000030: 00 00 00 00 00 00 00 00 b8 97 13 da b8 97 13 da
[20996.183132] skb headroom: 00000040: a0 7f 98 00 80 01 00 00 48 07 00 00 00 00 00 00
[20996.183141] skb headroom: 00000050: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[20996.183150] skb headroom: 00000060: 72 6f 70 22 20 3a 22 74 30 35 37 31 22 20 2c 30
[20996.183159] skb headroom: 00000070: 74 73 6f 68 74 6e 69 5f 33 20 3a 22 30 31 33 30
[20996.183168] skb headroom: 00000080: 00 00 01 00 5e 00 00 fb f0 18 98 24 52 4a 08 00

[20996.183177] skb linear:   00000000: 45 00 00 44 19 93 00 00 ff 11 ff 08 **c0 a8 01 69**
[20996.183186] skb linear:   00000010: **e0 00 00 fb** 14 e9 14 e9 00 30 e8 96 00 00 00 00

[20996.183195] skb linear:   00000020: 00 01 00 00 00 00 00 00 0b 5f 67 6f 6f 67 6c 65
[20996.183205] skb linear:   00000030: 63 61 73 74 04 5f 74 63 70 05 6c 6f 63 61 6c 00
[20996.183213] skb linear:   00000040: 00 0c 00 01
```
c0 a8 01 69 = 192 168 01 105
e0 00 00 fb = 224 0 0 251 (a multicast address), assigned to the mDNS service, as defined in https://tools.ietf.org/html/rfc6762

To manually dissect the header (https://en.wikipedia.org/wiki/IPv4_header#Header)
```[20996.183177] skb linear:   00000000:
45 - VERSION (4) / IHL (5)
00 - DSCP / ECN
00 44 - TOTAL LENGTH 
19 93 - IDENTIFICATION
00 00 - FLAGS/FRAGMENT OFFSET
ff - TIME TO LIVE (TTL)
11 - PROTOCOL (UDP)
ff 08 - HEADER CHECKSUM
c0 a8 01 69 - SRC ADDRESS
[20996.183186] skb linear:   00000010:
e0 00 00 fb - DEST ADDRESS
'''
To then continue dissecting the UDP headers (9https://en.wikipedia.org/wiki/User_Datagram_Protocol#UDP_datagram_structure)
```
14 e9 - SRC PORT (5353)
14 e9 - DEST PORT (5353)
00 30 - LENGTH
e8 96 - UDP CHECKSUM
```

mDNS is not a protocol I have dealt with previously, but looking at the UDP payload
```
[20996.183186] skb linear:   00000010:                                           00 00 00 00
[20996.183195] skb linear:   00000020: 00 01 00 00 00 00 00 00 0b 5f 67 6f 6f 67 6c 65
[20996.183205] skb linear:   00000030: 63 61 73 74 04 5f 74 63 70 05 6c 6f 63 61 6c 00
[20996.183213] skb linear:   00000040: 00 0c 00 01
```
0b is likely to be a length (11), and the values which are between 61 and 7a are all lower case ASCII values...
```_googlecast_tcplocal```

Ah, an online packet analyser - https://hpd.gasmi.net/?data=4500004419930000FF11FF08C0A80169E00000FB14E914E90030E8960000000000010000000000000B5F676F6F676C6563617374045F746370056C6F63616C00000C0001&force=ipv4

So whilst that does appear to be a valid packet, at least you now know the source of your woes, and it is multicast related.

ralphrmartin, 2021-03-01:
192.168.1.105 is a MacBook; it has Google Chrome as a browser, which can "cast". There's also a Sony Bravia TV on the network with an inbuilt service for receiving casts. (I was not "casting" anything at the time of the hw csum failures though).

I have not recently updated the software on the TV, although I have recently installed the latest system update and Chrome on the MacBook.

So, we can see what kind of packets are triggering the problem. Now what?

6by9, 2021-03-01:
> So, we can see what kind of packets are triggering the problem. Now what?

Now someone at least has a chance of recreating your problem. Doing so may be non-trivial though.

ralphrmartin, 2021-03-01:
Thank you. Let me know if I can do anything to assist, try out proposed changes, etc. (Sorry if my "now what" came across the wrong way - it was meant in the spirit of "what can I do to help?", but I can see now it hardly sounds like it).

ralphrmartin, 2021-03-01:
In case anyone else has this issue, a temporary way to prevent the flood of messages from swamping logs is to do
ethtool -K eth0 rx off 
(this does nothing to resolve the underlying cause of the problem).

ralphrmartin, 2021-03-04:
Problem still present in  5.10.17

pelwell, 2021-03-04:
And it still appears to be a problem only affecting one person. If you want some engineering effort put into this you'll have to make some attempts to work out what in your environment is triggering this bug by changing your configuration until the errors stop:
* Remove PoE - power the Pi using a conventional power supply.
* If you have another Ethernet switch, even a small one, try interposing that between the Pi and the Unifi.
* Close down Chrome. And if that makes no difference, turn off the MacBook.

Those three changes give 8 possible combinations. I would start by making as many changes as possible - if the problem still occurs then you've just ruled out several causes but now need to think of some more.

ralphrmartin, 2021-03-04:
(1) Problem persists when powered by official wall wart instead of PoE.
(2) Problem persists when powered by official wall wart, connected via IP Phone instead of via Unifi switch.
(3) Problem persists when instances of Chrome are shut down on Macs.
(4) Problem persists when all Macs are shut down.
(5) Problem persists when TV is turned off.

Note that Chrome may be running on Android phones on this network; it is not practical to try to stop it on them, nor is it practicable to turn these phones off.

This probably deoesn't help, but I have a Debian (testing) system on a Dell laptop on the same network, and there are no similar messages in its syslog. 

If you have further ideas of what else to try, need more logs, etc, please let me know.

pelwell, 2021-03-04:
> I have a Debian (testing) system on a Dell laptop on the same network, and there are no similar messages in its syslog.

Assuming this isn't some marginally faulty hardware, this is likely to be a bug in the GENET Ethernet controller, or more likely its driver.

How long does it typically take to see the problem? I have a Unifi Amplifi at home, and it might conceivably show the same issue.

ralphrmartin, 2021-03-04:
The chance of dodgy hardware is low as the 2 RPi 4's involved were bought quite a long time apart.
As soon as either RPi starts, the errors start coming in. 
Just in case it is relevant in any way, both boot from a USB attached SanDisk SSD.

ralphrmartin, 2021-03-04:
The switch is a Unifi US-8-150W, running 5.43.23.12533, in case that is useful.
--

--

ralphrmartin, 2021-03-04:
MTU is set to 1500 on this network.

pelwell, 2021-03-04:
If it is at all possible to briefly take your AP (and hence the mobile clients) out of the equation it would be a useful experiment. Note again that we're just trying to pinpoint the necessary elements to recreate the problem.

6by9, 2021-03-04:
Analysing a couple more of the skb dumps from the log (just the `skb linear` bits), they're all sorts of traffic.
- SSH and a TCP ACK between 192.168.1.100 and 192.168.1.105
- mDNS multicast query
- broadcast from 192.168.1.101 to UDP port 32412 (Plex).

There are also a couple which appear to be random fragments, as they don't start with IP headers (expect 45 00 at the start for a standard IPv4 packet). There is an IP header in the headroom, but it implies something odd is going on.

The SKB dumps only show the IP layer of the data. Grabbing a short wireshark dump may be of more use.

ralphrmartin, 2021-03-04:
It was not practical to turn the APs off (there are five off them, used for real time monitoring and control). However, I did manage to briefly turn off all Android and iOS devices that might be running Google and the problem persisted.

ralphrmartin, 2021-03-04:
If you can tell me the wireshark command line to use I'll collect some info. Does this need running on the RPi, or will any device on the net do to collect traffic? Where should I put the collected data (given that it may have private info in it)?

ralphrmartin, 2021-03-05:
Tried a 64-bit kernel to see if it made a difference. Still got errors, as was to be expected.

ralphrmartin, 2021-03-06:
An RPi 3B with the same kernel, connected to the same switch, does not show these errors.

6by9, 2021-03-06:
Pi3 uses a totally different ethernet chip so it's totally expected that it won't show issues. It would only do so if the problem was in the networking core code, and that sort of issue would be picked up fairly swiftly.

ralphrmartin, 2021-03-06:
As I say, I'll try to collect a wireshark capture if you give me some instructions.

ralphrmartin, 2021-03-18:
Finally tracked down the cause of this problem. Because of other ethernet issues I had earlier, I had
`genet.skip_umac_reset=y`
in my cmdline,txt. Changing it to
`genet.skip_umac_reset=n`
stopped the flood of hw csum failure errors.

Omoeba, 2021-03-23:
I'm facing the exact same issue on kernel version `5.10.17-v8+`. Adding `genet.skip_umac_reset=n` to my cmdline.txt didn't change anything.

Omoeba, 2021-03-23:
The only way to stop the dmesg flood is `ethtool -K eth0 rx off`.

noroutine, 2023-03-25:
just chiming in that same occured for me, no special networking, no special configuration for skip_umac_reset

```
eth0: hw csum failure
skb len=52 headroom=144 headlen=52 tailroom=1980
mac=(130,14) net=(144,20) trans=164
shinfo(txflags=0 nr_frags=0 gso(size=0 type=0 segs=0))
csum(0x8fda ip_summed=2 complete_sw=0 valid=0 level=0)
hash(0x0 sw=0 l4=0) proto=0x0800 pkttype=0 iif=0
dev name=eth0 feat=0x0000010000004829
skb headroom: 00000000: 32 f6 9e 38 77 0f 8d 89 db 4a ab e5 b9 92 0a 75
skb headroom: 00000010: 57 aa 9b 48 51 b3 a5 14 52 44 ad df 0d 49 cb a9
skb headroom: 00000020: 7c 92 06 28 d8 f5 a5 b2 15 d6 08 f6 ff 11 8a ee
skb headroom: 00000030: 0d cc e9 61 b6 80 62 c2 ac 8d bf 02 01 00 78 d2
skb headroom: 00000040: 80 7f 84 00 80 80 00 00 8f da 00 00 00 00 00 00
skb headroom: 00000050: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
skb headroom: 00000060: 80 7f 84 00 80 80 00 00 8f da 00 00 00 00 00 00
skb headroom: 00000070: a6 dc 00 00 d3 51 44 32 09 0f 09 00 00 08 1a 00
skb headroom: 00000080: 00 00 dc a6 32 44 51 d3 00 09 0f 09 00 1a 08 00
skb linear:   00000000: 45 00 00 34 a6 d8 40 00 3f 06 6f a5 0a 00 0a 3d
skb linear:   00000010: 0a 00 07 0a 03 6a 08 01 43 b7 57 69 e2 96 79 83
skb linear:   00000020: 80 10 4e e0 a9 d6 00 00 01 01 08 0a 6d e0 cd fa
skb linear:   00000030: 6d 13 ad 2b
skb tailroom: 00000000: 80 00 00 e4 39 0c fa a1 00 00 00 00 00 00 00 02
skb tailroom: 00000010: 00 01 86 a3 00 00 00 04 00 00 00 01 00 00 00 01
skb tailroom: 00000020: 00 00 00 40 00 00 00 00 00 00 00 25 62 6f 30 31
skb tailroom: 00000030: 2d 76 6d 2d 6e 6f 64 65 30 31 2e 6e 6f 64 65 2e
skb tailroom: 00000040: 62 6f 30 31 2e 6e 6f 72 6f 75 74 69 6e 65 2e 6d
skb tailroom: 00000050: 65 00 00 00 00 00 ff fe 00 00 ff fe 00 00 00 01
skb tailroom: 00000060: 00 00 ff fe 00 00 00 00 00 00 00 00 00 00 00 00
skb tailroom: 00000070: 00 00 00 02 00 00 00 03 00 00 00 35 ec 1c 0f 64
skb tailroom: 00000080: ac 6d 37 03 0d 00 00 00 00 00 00 00 00 00 71 76
skb tailroom: 00000090: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 16
skb tailroom: 000000a0: 00 00 00 24 01 00 07 01 01 00 24 00 00 00 00 00
skb tailroom: 000000b0: 53 b4 ba 23 0b 4b 4c ad a5 3f a1 7c 4d ee d0 61
skb tailroom: 000000c0: 44 7c 50 00 81 72 a6 ea 00 00 00 19 00 00 00 00
skb tailroom: 000000d0: ec 1c 0f 64 ac 6d 37 03 14 03 00 00 00 00 00 00
skb tailroom: 000000e0: 03 ba 00 00 00 02 00 00 3d dd 2e 82 18 01 26 1f
skb tailroom: 000000f0: 4e b6 df 39 15 36 c2 5a 96 e0 47 31 83 55 b7 7c
skb tailroom: 00000100: 29 d7 06 ef f5 21 e6 9b b0 8c c3 d4 c6 14 89 b7
skb tailroom: 00000110: 1b 7d b9 ab 6c 48 d5 58 22 21 eb 15 6c e8 93 34
skb tailroom: 00000120: 96 6b 70 e0 98 5a 31 0d f7 86 e9 ef 76 d6 b2 31
skb tailroom: 00000130: 13 f9 1d cc db 3f d6 2f 53 b7 49 f9 b9 0f ec 3e
skb tailroom: 00000140: c0 5c e4 23 6b 55 48 ad df bc 62 50 93 6b db d8
skb tailroom: 00000150: 79 ce b7 a2 54 80 ad 84 d8 4a ce b0 51 6e 91 63
skb tailroom: 00000160: 17 f4 9b 86 3b 04 30 be f1 c7 32 7c dd 7e e4 cf
skb tailroom: 00000170: ca da 69 03 22 97 14 52 48 ae e1 d0 83 a8 7b 65
skb tailroom: 00000180: de da 2e 91 50 b0 8c 60 46 d3 a9 04 a0 86 27 62
skb tailroom: 00000190: 19 f0 91 b9 b6 52 8c 5a f0 6c b0 0b 14 bf 17 04
skb tailroom: 000001a0: cb 1d 25 4a 13 0b fe 90 0d 0d dc f7 9c 2a cf 88
skb tailroom: 000001b0: 21 9d 5d 61 5d 8c 2f ec 9c cb 1a d4 db 96 0a af
skb tailroom: 000001c0: 75 61 5d 32 f6 9e 38 77 0f 8d 89 db 4a ab e5 b9
skb tailroom: 000001d0: 92 0a 75 57 aa 9b 48 51 b3 a5 14 52 44 ad df 0d
skb tailroom: 000001e0: 49 cb a9 7c 92 06 28 d8 f5 a5 b2 15 d6 08 f6 ff
skb tailroom: 000001f0: 11 8a ee 0d cc e9 61 b6 80 62 c2 ac 8d bf 02 01
skb tailroom: 00000200: 00 78 d2 f7 a4 bb e2 61 40 ee 74 40 00 00 00 00
skb tailroom: 00000210: bf 81 02 56 31 a8 09 98 7d 0e 82 2d 6d a1 0b 08
skb tailroom: 00000220: 9f 53 6e 4c 2a 80 e8 3e 1d c2 a0 54 65 b0 85 f5
skb tailroom: 00000230: 66 82 94 f4 5b 45 af e6 50 f3 c8 df d6 e9 3f a6
skb tailroom: 00000240: e7 d4 ee ec 21 06 51 0a a7 85 57 49 6a 05 de 05
skb tailroom: 00000250: fe 59 5a 68 30 e1 3b b2 9d 0e 05 f7 61 8a 37 04
skb tailroom: 00000260: 16 42 49 5d 0a 80 a7 67 61 46 72 e3 85 62 4e 93
skb tailroom: 00000270: 52 59 3d b7 19 05 8c 5d ed ca d5 46 cc a0 45 d6
skb tailroom: 00000280: ce 2d be bf 63 ef c3 73 69 9b 94 36 c4 ea b6 e9
skb tailroom: 00000290: 38 a1 9d d3 2c b0 e1 ab cc 42 3a 21 58 56 8e 2b
skb tailroom: 000002a0: 38 64 15 82 23 3d 38 86 a2 8a 55 0b 27 14 91 6b
skb tailroom: 000002b0: 0a 5f 84 bc f4 08 ed f9 60 8d 0b 38 88 1b 54 3b
skb tailroom: 000002c0: a2 e8 9b 97 7e c5 c9 26 ec 0f 2b d5 d2 45 4d c2
skb tailroom: 000002d0: 6e 62 5a 04 78 61 b8 a0 85 b5 22 ec 69 44 6e f2
skb tailroom: 000002e0: 0c 0f f7 79 69 87 54 4d ae 31 61 a1 1a 2f 51 7e
[... 1371 lines cut ...]
[986880.612935] [<c02290b4>] (run_ksoftirqd) from [<c024cce0>] (smpboot_thread_fn+0xfc/0x1ac)
[986880.612945] [<c024cbe4>] (smpboot_thread_fn) from [<c02488a4>] (kthread+0x178/0x194)
[986880.612955]  r9:c1511ddc r8:00000000 r7:c15060c0 r6:c024cbe4 r5:c1506100 r4:c1506140
[986880.612959] [<c024872c>] (kthread) from [<c02000d4>] (ret_from_fork+0x14/0x20)
[986880.612967] Exception stack(0xc1555fb0 to 0xc1555ff8)
[986880.612972] 5fa0:                                     00000000 00000000 00000000 00000000
[986880.612978] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[986880.612983] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[986880.612988]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c024872c
[986880.612992]  r4:c1506100

EEPROM
------

BOOTLOADER: up to date
   CURRENT: Wed 11 Jan 2023 05:40:52 PM UTC (1673458852)
    LATEST: Wed 11 Jan 2023 05:40:52 PM UTC (1673458852)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0
```

wulfy23, 2023-06-16:
Fingers crossed for 6.1 test this weekend. Issue persists on 5.10.176. MLD / Android triggers.

And while this commit is likely not directly related? Comment seems to sounds very on point with what is going on in the nic driver on 5.10 / 5.15...

https://github.com/raspberrypi/linux/commit/331615d837f4979eb91a336a223a5c7f7886ecd5

yutayu, 2023-11-01:
https://github.com/tohojo/sqm-scripts/issues/166#issue-1968667436

After I added ifb device ( for ingress shaping ) , This error occurs.