raspberrypi/linux #4185: miniuart-bt,krnbt=on broken on RPi4b

vianpl, 2021-03-02:
With this appended on top of the stock config.txt:
```
dtoverlay=miniuart-bt,krnbt=on
core_freq_min=500
core_freq=500
```
Kernel outputs the following:

```
[    6.771162] Bluetooth: Core ver 2.22
[    6.771303] Bluetooth: HCI device and connection manager initialized
[    6.771332] Bluetooth: HCI socket layer initialized
[    6.771364] Bluetooth: L2CAP socket layer initialized
[    6.771431] Bluetooth: SCO socket layer initialized
[    6.785501] Bluetooth: HCI UART driver ver 2.3
[    6.785524] Bluetooth: HCI UART protocol H4 registered
[    6.785631] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    6.786340] Bluetooth: HCI UART protocol Broadcom registered
[    9.113411] Bluetooth: hci0: command 0xfc18 tx timeout
[    9.736229] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    9.736250] Bluetooth: BNEP filters: protocol multicast
[    9.736279] Bluetooth: BNEP socket layer initialized
[   17.113329] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[   17.113347] Bluetooth: hci0: Failed to set baudrate
[   19.193319] Bluetooth: hci0: command 0x0c03 tx timeout
[   27.353338] Bluetooth: hci0: BCM: Reset failed (-110)
[   29.443304] Bluetooth: hci0: command 0xfc18 tx timeout
[   37.593349] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[   37.593367] Bluetooth: hci0: Failed to set baudrate
[   39.673334] Bluetooth: hci0: command 0x0c03 tx timeout
[   47.833333] Bluetooth: hci0: BCM: Reset failed (-110)
```
I've replicated this with the stock image kernel, with 5.10.17-v7l+ and RPi OS 64bit. Note that using the `hciattach` method works fine, but something I can't use as it's currently being deprecated by bluez.

Rings any bells?

pelwell, 2021-03-02:
It's not one I've seen. What's in your cmdline.txt?

vianpl, 2021-03-02:
```
console=serial0,115200 console=tty1 root=PARTUUID=7efb9ce6-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
```

vianpl, 2021-03-02:
```
pi@raspberrypi:~ $ ls -la /dev/serial*
lrwxrwxrwx 1 root root 7 Mar  2 11:58 /dev/serial0 -> ttyAMA0
```
Obviously, neither `ttyS0` nor  `serial1` are present

pelwell, 2021-03-02:
Yes, that's all fine. For context, is this a configuration that has worked for you before?

pelwell, 2021-03-02:
Note that the explicit core_freq and core_freq_min are redundant - the firmware should spot when UART1 is in use and lock the core clock.

vianpl, 2021-03-02:
> Yes, that's all fine. For context, is this a configuration that has worked for you before?

No, this is the first time I try such a setup. Could I be the first one to do it? 

> Note that the explicit core_freq and core_freq_min are redundant - the firmware should spot when UART1 is in use and lock the core clock.

Noted.

pelwell, 2021-03-02:
It's also failing here - leave it with me.

pelwell, 2021-03-02:
So far I can only make it work by actually _enabling_ the flow control pins on the mini-UART - something I failed to at the birth of Pi 3.

Try this overlay to force the pin mapping:
```
/dts-v1/;
/plugin/;

/{
	compatible = "brcm,bcm2835";

	fragment@0 {
		target = <&uart1>;
		__overlay__ {
			pinctrl-names = "default";
			pinctrl-0 = <&my_uart1_pins>;
			status = "okay";
		};
	};

	fragment@1 {
		target = <&gpio>;
		__overlay__ {
			my_uart1_pins: my_uart1_pins {
				brcm,pins = <30 31 32 33>;
				brcm,function = <2 2 2 2>; /* alt5 */
				brcm,pull = <2 0 0 2>;
			};
		};
	};

	fragment@2 {
		target-path = "/chosen";
		__overlay__ {
			bootargs = "8250.nr_uarts=1";
		};
	};

	fragment@3 {
		target = <&minibt>;
		__overlay__ {
			max-speed = <1000000>;
		};
	};
};
```
Let me know how you get on and I might have to reconsider the default miniuart-bt overlay.

vianpl, 2021-03-03:
Hi @pelwell, thanks for taking the time to look into this. Just so we're clear, you're applying this overlay below `dtoverlay=miniuart-bt,krnbt=on`

pelwell, 2021-03-03:
Yes - to avoid anything in miniuart-bt overwriting something it has also changed (although I didn't look to see if that was a possibility).

vianpl, 2021-03-03:
So I tested this on a fresh install, with 5.10.11-v7l+.  I added the following into config.txt: 
```
dtoverlay=miniuart-bt,krnbt=on
dtoverlay=miniuart-bt-fix
```
The issue comes and goes now. It'll boot alright for a couple of times and then fail. I'd say it's a 2/1 ratio of successes. I tried to lower the UART speed just for safety, but no luck. Makes any sense to you?

pelwell, 2021-03-03:
Not really - having the modem signals on hidden GPIOs makes it hard to debug.

idun-project, 2024-02-18:
Is there any more progress on this issue in the past 3 years?

This seems to be at the root of the problem I'm having with Bluetooth on an RPi Zero 2 with the latest kernel (6.6.16) and with the latest bluez release. Without this functionality, it appears impossible to use the latest kernel/bluez with Bluetooth utilizing the mini UART.

Thus, if one needs to use both the fast/reliable UART on the GPIO header AND the Bluetooth radio, they are just out of luck?

pelwell, 2024-02-19:
In what way is it not working for you? What are you attempting to do with Bluetooth, and with what kind of bandwidth? When using the mini-UART there is no flow control - it's a best-effort service.

idun-project, 2024-02-19:
@pelwell Hi and thanks for the response.

I'd like to pair HID devices with a Pi Zero 2 W. Using miniuart-bt, and no other settings, Bluetooth appears in the dmesg log and no errors. However, the bluez bluetooth.service fails while starting (see errors below), and commands like `scan on` and `power on` fail in bluetoothctl. I'm _assuming_ that's a conflict between the kernel driver and what is provided by the overlay.

dmesg:
```
[    9.904771] Bluetooth: Core ver 2.22
[    9.904900] NET: Registered PF_BLUETOOTH protocol family
[    9.904907] Bluetooth: HCI device and connection manager initialized
[    9.904927] Bluetooth: HCI socket layer initialized
[    9.904942] Bluetooth: L2CAP socket layer initialized
[    9.904959] Bluetooth: SCO socket layer initialized
[    9.931459] Bluetooth: HCI UART driver ver 2.3
[    9.931491] Bluetooth: HCI UART protocol H4 registered
[    9.931591] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    9.931925] hci_uart_bcm serial0-0: supply vbat not found, using dummy regulator
[    9.932104] hci_uart_bcm serial0-0: supply vddio not found, using dummy regulator
[    9.932239] Bluetooth: HCI UART protocol Broadcom registered
[   10.168446] brcmfmac_wcc: brcmf_wcc_attach: executing
[   10.177479] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   10.178094] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7
[   10.319242] Bluetooth: hci0: BCM: chip id 94
[   10.320337] Bluetooth: hci0: BCM: features 0x2e
[   10.331898] Bluetooth: hci0: BCM43430A1
[   10.331927] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0000
[   10.334473] Bluetooth: hci0: BCM43430A1 'brcm/BCM43430A1.hcd' Patch
[   11.835162] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   11.835194] Bluetooth: BNEP filters: protocol multicast
[   11.835220] Bluetooth: BNEP socket layer initialized
[   12.248728] Bluetooth: hci0: BCM: features 0x2e
[   12.260288] Bluetooth: hci0: BCM43438A1 37.4MHz Raspberry Pi 3-0141
[   12.260315] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0508
[   12.328983] Bluetooth: MGMT ver 1.22
```

systemctl status bluetooth:
```
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: disabled)
     Active: active (running) since Mon 2024-02-19 09:37:22 EST; 2min 4s ago
       Docs: man:bluetoothd(8)
   Main PID: 307 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 380)
        CPU: 194ms
     CGroup: /system.slice/bluetooth.service
             └─307 /usr/lib/bluetooth/bluetoothd

Feb 19 09:37:22 idunpi4 bluetoothd[307]: src/adapter.c:reset_adv_monitors_complete() Failed to reset Adv Monitors: Failed (0x03)
Feb 19 09:37:22 idunpi4 bluetoothd[307]: Battery Provider Manager created
Feb 19 09:37:22 idunpi4 bluetoothd[307]: Failed to clear UUIDs: Failed (0x03)
Feb 19 09:37:22 idunpi4 bluetoothd[307]: Failed to add UUID: Failed (0x03)
Feb 19 09:37:22 idunpi4 bluetoothd[307]: Failed to add UUID: Failed (0x03)
Feb 19 09:37:22 idunpi4 bluetoothd[307]: Failed to add UUID: Failed (0x03)
Feb 19 09:37:22 idunpi4 bluetoothd[307]: Failed to set mode: Failed (0x03)
Feb 19 09:37:22 idunpi4 bluetoothd[307]: Failed to add UUID: Failed (0x03)
Feb 19 09:37:22 idunpi4 bluetoothd[307]: Failed to add UUID: Failed (0x03)
Feb 19 09:37:22 idunpi4 bluetoothd[307]: Failed to add UUID: Failed (0x03)
```

bluetoothctl:
```
[idun@idunpi4 ~]$ bluetoothctl scan on
SetDiscoveryFilter failed: org.bluez.Error.NotReady
Failed to start discovery: org.bluez.Error.NotReady
[idun@idunpi4 ~]$ bluetoothctl power on
Failed to set power on: org.bluez.Error.Failed
```

pelwell, 2024-02-19:
When you say "latest Bluez release", am I right in thinking this isn't an RPiOS image? (Not that that's bad or unsupported, just trying to understanding what I'm dealing with).

pelwell, 2024-02-19:
Is that the mainline/upstream kernel or ours?

idun-project, 2024-02-19:
I think it's from you, based on the package descriptions.

```
[idun@idunpi4 ~]$ sudo pacman -Qs rpi
local/linux-rpi 6.6.16-2
    Linux kernel and modules (RPi Foundation fork)
[idun@idunpi4 ~]$ sudo pacman -Qs firmware
local/firmware-raspberrypi 20231022-1
    Additional firmware for Raspberry Pi
```