raspberrypi/linux #4186: tvservice not working with vc4-kms-v3d

audas, 2021-03-03:
I am using the latest 5.10 and the latest firmware and vc4-kms-v3d - it appears that tvservice is not returning things, or under-reporting.

For example 

`tvservice -l`

will return 

`0 attached device(s), display id's are :`

on the other hand things like 

`tvservice -m CEA
`

Will return nothing as well.

pi4B
Raspberry Pi reference 2020-02-05
January 27 2021 - version 99d9a48302....
5.10.11

pelwell, 2021-03-03:
tvservice and vc4-kms-v3d driver are incompatible - you either use tvservice and vc4-fkms-v3d (whereby the VPU controls the display) or vc4-kms-v3d (which gives full control to Linux).

pelwell, 2021-03-03:
Did you think of waiting at least half an hour for a response before spamming the forums as well?
* https://www.raspberrypi.org/forums/viewtopic.php?p=1829997#p1829997
* https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=305840

closed 2021-03-03 by pelwell

audas, 2021-03-03:
Sure- I posted those issues before I came here - I came here because with the new driver vc4-kms-v3d there is no way of setting the Screen Resolution - it does not show up with latest firmware, IOS and new driver running. You also can not set the resolution via raspi-config.

Would have thought that was an issue worth exploring.