raspberrypi/linux #4187: Pi4: KMS HDMI image off by 1 pixel

audetto, 2021-03-03:
**Describe the bug**
When I enable ``dtoverlay=vc4-kms-v3d`` the HDMI output is off by one pixel.
This means that he first pixel on the left should actually be the last on the right.
Easy to see when the mouse is moved to the right, one can see one pixel of the pointer on the left. 

**To reproduce**
Enable ``kms``. With ``fkms`` there are no issues.

System Information
------------------

Raspberry Pi 400 Rev 1.0
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage5

Linux raspberrypi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
Revision	: c03130
Serial		: 10000000dc6bd0c6
Model		: Raspberry Pi 400 Rev 1.0
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Jan 27 2021 22:19:57 
Copyright (c) 2012 Broadcom
version 99d9a48302e4553cff3688692bb7e9ac760a03fa (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root       14987608 10190628   4088728  72% /
devtmpfs         1827472        0   1827472   0% /dev
tmpfs            1959568    38820   1920748   2% /dev/shm
tmpfs            1959568     8776   1950792   1% /run
tmpfs               5120        4      5116   1% /run/lock
tmpfs            1959568        0   1959568   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095    48678    209418  19% /boot
tmpfs             391912        4    391908   1% /run/user/1000

Filename				Type		Size		Used	Priority
/var/swap                               file		102396		0	-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: 1.20201210+nmu1
raspberrypi-sys-mods:
  Installed: 20210208
openbox:
  Installed: 3.6.1-8+rpt5
lxpanel:
  Installed: 0.10.0-2+rpt14
pcmanfm:
  Installed: 1.3.1-1+rpt26
rpd-plym-splash:
  Installed: 0.26

Networking Information
----------------------
[... 544 lines cut ...]
[   12.835819] fuse: init (API version 7.32)
[   13.441310] Bluetooth: Core ver 2.22
[   13.441368] NET: Registered protocol family 31
[   13.441375] Bluetooth: HCI device and connection manager initialized
[   13.441389] Bluetooth: HCI socket layer initialized
[   13.441398] Bluetooth: L2CAP socket layer initialized
[   13.441413] Bluetooth: SCO socket layer initialized
[   13.477628] Bluetooth: HCI UART driver ver 2.3
[   13.477638] Bluetooth: HCI UART protocol H4 registered
[   13.477701] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   13.477858] Bluetooth: HCI UART protocol Broadcom registered
[   13.874880] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   13.874889] Bluetooth: BNEP filters: protocol multicast
[   13.874903] Bluetooth: BNEP socket layer initialized
[   14.089189] vc4_hdmi fef00700.hdmi: ASoC: error at snd_soc_dai_startup on fef00700.hdmi: -19
[   14.089736] vc4_hdmi fef00700.hdmi: ASoC: error at snd_soc_dai_startup on fef00700.hdmi: -19
[   14.090637] vc4_hdmi fef00700.hdmi: ASoC: error at snd_soc_dai_startup on fef00700.hdmi: -19
[   14.303969] vc4_hdmi fef00700.hdmi: ASoC: error at snd_soc_dai_startup on fef00700.hdmi: -19
[   14.304733] vc4_hdmi fef00700.hdmi: ASoC: error at snd_soc_dai_startup on fef00700.hdmi: -19
[   14.306475] vc4_hdmi fef00700.hdmi: ASoC: error at snd_soc_dai_startup on fef00700.hdmi: -19
[   14.306751] vc4_hdmi fef00700.hdmi: ASoC: error at snd_soc_dai_startup on fef00700.hdmi: -19
[   14.576126] Bluetooth: RFCOMM TTY layer initialized
[   14.576146] Bluetooth: RFCOMM socket layer initialized
[   14.576171] Bluetooth: RFCOMM ver 1.11
[   34.107450] v3d fec00000.v3d: MMU error from client L2T (0) at 0x7d21000, pte invalid

popcornmix, 2021-03-03:
Yes, well known issue (e.g. mentioned as known issue [here](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=288234).
No fix yet.

alanbork, 2021-05-11:
what's the full range of documented ways this shows up? I just noticed on this boot that the left-most line of pixels are shifted down by at least 1 pixel, which I'm guessing could be this bug as well? And there was the report I filed above (https://github.com/raspberrypi/linux/issues/4306) where it seems like the left-most line of pixels came from the past (or future) buffer, not a wrap around from the far right of the same buffer. Are these all the same problem?

popcornmix, 2021-06-25:
There's a [url=https://github.com/raspberrypi/linux/pull/4410]potential fix[/url] for the 1-pixel wrap issue when using kms in latest rpi-update kernel.
If you are currently seeing this issue, then we'd be interested in reports of behaviour after updating.