raspberrypi/linux #4191: kms: "no signal" when changing screen resolution through arandr

maxnet, 2021-03-05:
If using kms instead of fkms changing to another mode through arandr will result in "no signal" on monitor in most cases:

Tried a couple different ones like:

1920x1080i
1280x1024
1140x900
800x600

As an exception changing to 1680x1050 did work ONCE.
But not repeatedly.
And it did was able to boot with 1920x1080p, but when changing from the 1680x1050 that did work once back to 1920x1080p this also resulted in "no signal".

In none of the case arandr is switching back to a working mode either (which it is supposed to do if user does not confirm mode change within X seconds), requires a reboot to get visual again.

/sys/class/drm/card0/card0-HDMI-A-2/edid  : [edid.zip](https://github.com/raspberrypi/linux/files/6092034/edid.zip)

- Clean install of latest RPI OS offered through Imager, without any updates.

raspinfo

System Information
------------------

Raspberry Pi 400 Rev 1.0
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2021-01-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 21090519d85bdaa1615d5d5057d37b09368ea5d2, stage4

Linux raspberrypi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux
Revision        : c03130
Serial          : 10000000c35f69b0
Model           : Raspberry Pi 400 Rev 1.0
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Jan  8 2021 14:31:16 
Copyright (c) 2012 Broadcom
version 194a85abd768c7334bbadc3f1911c10a7d18ed14 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       61229808 3080020  55609840   6% /
devtmpfs         1827796       0   1827796   0% /dev
tmpfs            1959892       0   1959892   0% /dev/shm
tmpfs            1959892    8772   1951120   1% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs            1959892       0   1959892   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   46272    211824  18% /boot
tmpfs             391976       4    391972   1% /run/user/1000

Filename                                Type            Size    Used    Priority
/var/swap                               file            102396  0       -2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: 1.20201210+nmu1
[... 542 lines cut ...]
[   13.546573] Bluetooth: HCI device and connection manager initialized
[   13.546585] Bluetooth: HCI socket layer initialized
[   13.546593] Bluetooth: L2CAP socket layer initialized
[   13.546606] Bluetooth: SCO socket layer initialized
[   13.567067] Bluetooth: HCI UART driver ver 2.3
[   13.567076] Bluetooth: HCI UART protocol H4 registered
[   13.567110] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   13.567256] Bluetooth: HCI UART protocol Broadcom registered
[   13.793908] vc4_hdmi fef00700.hdmi: ASoC: can't open interface fef00700.hdmi: -19
[   13.794259] vc4_hdmi fef00700.hdmi: ASoC: can't open interface fef00700.hdmi: -19
[   13.794832] vc4_hdmi fef00700.hdmi: ASoC: can't open interface fef00700.hdmi: -19
[   13.887313] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   13.887321] Bluetooth: BNEP filters: protocol multicast
[   13.887334] Bluetooth: BNEP socket layer initialized
[   14.016793] Bluetooth: hci0: unexpected event for opcode 0x0c7a
[   14.357349] debugfs: File 'le_min_key_size' in directory 'hci0' already present!
[   14.357366] debugfs: File 'le_max_key_size' in directory 'hci0' already present!
[   14.481852] vc4_hdmi fef00700.hdmi: ASoC: can't open interface fef00700.hdmi: -19
[   14.482794] vc4_hdmi fef00700.hdmi: ASoC: can't open interface fef00700.hdmi: -19
[   14.484992] vc4_hdmi fef00700.hdmi: ASoC: can't open interface fef00700.hdmi: -19
[   14.485332] vc4_hdmi fef00700.hdmi: ASoC: can't open interface fef00700.hdmi: -19
[   14.794356] Bluetooth: RFCOMM TTY layer initialized
[   14.794370] Bluetooth: RFCOMM socket layer initialized
[   14.794394] Bluetooth: RFCOMM ver 1.11
[   32.923458] v3d fec00000.v3d: MMU error from client L2T (0) at 0x3aa1000, pte invalid

maxnet, 2021-03-05:
Issue no longer present in 5.10.20

closed 2021-03-05 by maxnet

pelwell, 2021-03-05:
Awesome - you can come again.

6by9, 2021-03-05:
My first response was going to be "you're running 5.4. Upgrade to 5.10 if you want to use KMS"