raspberrypi/linux #4192: 5Ghz WiFi bands are no longer visible after upgrade to Linux version 5.10.17-v8+

mgrouch, 2021-03-06:
5Ghz WiFi bands are no longer visible after upgrade to Linux version 5.10.17-v8+

Enjoy-Combi, 2021-03-08:
[same issue](https://github.com/raspberrypi/linux/issues/4188#issue-821867939)

pelwell, 2021-03-08:
They're working here. Do you want to say a bit more about your setup, or should I just close this now?

mgrouch, 2021-03-09:
They get found eventually. Takes way long. Might be it’s my enclosure issue (it’s not fully plastic).

mgrouch, 2021-03-13:
3 hours and still no 5G WiFi and all 2.4G wifi connections fail (with correct passwords).
5.10.17-v8+

Might be it's latest firmware issue. 

https://github.com/raspberrypi/linux/issues/4134

mgrouch, 2021-03-13:
I've noticed raspberry pi OS added this line:

dtoverlay=dwc2,dr_mode=host

at the end of my boot/config.txt

without it wifi works.

mgrouch, 2021-03-13:
WiFi still very bad. 

5.10.11-v8+ 
was finding 5Ghz band right away. 
I’ve boot it it from SSD and I can tell
Somewhere between 5.10.11 and 5.0.17 things got worse. 

I’ve upgraded with apt to 5.10.17 and started getting lots of issues with WiFi (basically not working) and one usb error on boot which I reported separately. 

Armhf version with 5.10.17 has same issues with WiFi and USB. 

so both 32 and 64 bit kernels are impacted.

mgrouch, 2021-03-26:
I’ve found the reason for issues. 
I didn’t set WiFi country. 
After setting WiFi country in
wpasupplicant it works just fine

closed 2021-03-26 by mgrouch