raspberrypi/linux #4193: HDMI audio noisy / distorted / clipped with vc4-kms-v3d on Pi3B+

ykazakov, 2021-03-06:
In my `/boot/config.txt` I disable vc4 hdmi audio and enable snd_bcm2835

```
dtparam=audio=on
dtoverlay=vc4-kms-v3d,audio=off
```
In raspi-config the hdmi audio is enabled and alsa can see the hdmi card of bcm2835:

```
$ aplay -lL
null
    Discard all samples (playback) or generate zero samples (capture)
default
sysdefault:CARD=b1
    bcm2835 HDMI 1, bcm2835 HDMI 1
    Default Audio Device
dmix:CARD=b1,DEV=0
    bcm2835 HDMI 1, bcm2835 HDMI 1
    Direct sample mixing device
dsnoop:CARD=b1,DEV=0
    bcm2835 HDMI 1, bcm2835 HDMI 1
    Direct sample snooping device
hw:CARD=b1,DEV=0
    bcm2835 HDMI 1, bcm2835 HDMI 1
    Direct hardware device without any conversions
plughw:CARD=b1,DEV=0
    bcm2835 HDMI 1, bcm2835 HDMI 1
    Hardware device with all software conversions
sysdefault:CARD=Headphones
    bcm2835 Headphones, bcm2835 Headphones
    Default Audio Device
dmix:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct sample mixing device
dsnoop:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct sample snooping device
hw:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct hardware device without any conversions
plughw:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Hardware device with all software conversions
**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
``` 

Playing a sample, however, produces a noisy output (as if the some frequencies are clipped):

```
$ aplay -D hw:0,0 /usr/share/sounds/alsa/Front_Center.wav
Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
```

Audio played over gstreamer (`gst-play-1.0`) appear to be similarly distorted/clipped.

The flag `audio=off` for `dtoverlay=vc4-kms-v3d` comes from here: #2489. Without disabling the vc4 audio the hdmi output doesn't show up (in `raspi-config` and in  `aplay -l`), but instead a (non-functional) vc4hdmi interface becomes available:

```
$ aplay -lL
null
    Discard all samples (playback) or generate zero samples (capture)
default
sysdefault:CARD=Headphones
    bcm2835 Headphones, bcm2835 Headphones
    Default Audio Device
dmix:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct sample mixing device
dsnoop:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct sample snooping device
hw:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct hardware device without any conversions
plughw:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Hardware device with all software conversions
sysdefault:CARD=vc4hdmi
    vc4-hdmi, MAI PCM vc4-hdmi-hifi-0
    Default Audio Device
front:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM vc4-hdmi-hifi-0
    Front speakers
iec958:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM vc4-hdmi-hifi-0
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM vc4-hdmi-hifi-0
    Direct sample mixing device
dsnoop:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM vc4-hdmi-hifi-0
    Direct sample snooping device
hw:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM vc4-hdmi-hifi-0
    Direct hardware device without any conversions
plughw:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM vc4-hdmi-hifi-0
    Hardware device with all software conversions
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: vc4hdmi [vc4-hdmi], device 0: MAI PCM vc4-hdmi-hifi-0 [MAI PCM vc4-hdmi-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

This has also been recently reported by [some](https://www.raspberrypi.org/forums/viewtopic.php?t=296737) [users](https://www.raspberrypi.org/forums/viewtopic.php?t=291263).

With `dtoverlay=vc4-fkms-v3d` the hdmi audio works fine. But it looks like `vc4hdmi` is not available in this configuration.

Interestingly, the the demo included in /opt/vc/ also appears to be working fine in all configurations over hdmi, so it may seem that the issue could be with alsa?

```
/opt/vc/src/hello_pi/hello_audio/hello_audio.bin 1
```

When disabling vc4 audio using `dtoverlay=vc4-kms-v3d,audio=off`, I noticed from `dmesg` that vc4hdmi is still loaded, so maybe the vc4 audio is not disabled completely and this causes a conflict?

```
[    8.203974] rc rc0: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0
[    8.205253] input: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0/input0
[    8.212851] vc4_hdmi 3f902000.hdmi: 'dmas' DT property is missing or empty, no HDMI audio
[    8.213035] vc4-drm soc:gpu: bound 3f902000.hdmi (ops vc4_hdmi_ops [vc4])
[    8.213342] vc4-drm soc:gpu: bound 3f806000.vec (ops vc4_vec_ops [vc4])
[    8.213630] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
[    8.213991] vc4-drm soc:gpu: bound 3f004000.txp (ops vc4_txp_ops [vc4])
[    8.214451] vc4-drm soc:gpu: bound 3f206000.pixelvalve (ops vc4_crtc_ops [vc4])
[    8.215043] vc4-drm soc:gpu: bound 3f207000.pixelvalve (ops vc4_crtc_ops [vc4])
[    8.215627] vc4-drm soc:gpu: bound 3f807000.pixelvalve (ops vc4_crtc_ops [vc4])
[    8.216003] vc4-drm soc:gpu: bound 3fc00000.v3d (ops vc4_v3d_ops [vc4])
```

I am on Pi3B+ with latest raspberry os (32 bit), firmware, and kernel:

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2021-01-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 21090519d85bdaa1615d5d5057d37b09368ea5d2, stage2
$ vcgencmd version
Feb 25 2021 12:12:09 
Copyright (c) 2012 Broadcom
version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start)
$ uname -a
Linux raspberry 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux
```

I tried to load the lates firmware using `rpi-update` but it did not notice any difference.

6by9, 2021-03-06:
Hdmi audio using bcm2835_audio with vc4-kms-v3d isn't going to work.
Hdmi audio parameters depend on the video mode, and with vc4-kms-v3d that has all been passed across to the kernel, so the firmware isn't going to get the correct config. There is no "fix". 

The vc4hdmi audio device needs the correct alsa config file to use the alsa plugin to do iec958 packing.

ykazakov, 2021-03-06:
Thanks @6by9, I see... Does it mean that vc4hdmi is not really disabled by the `audio=off` option? How to explain that `hello_audio.bin` is working fine?

Has anyone been able to make hdmi audio work on vc4-kms-v3d at all? I tried a few alsa configs for `/usr/share/alsa/cards/vc4-hdmi.conf` like [this](https://retropie.org.uk/forum/post/239124) and [this](https://github.com/LibreELEC/LibreELEC.tv/blob/master/projects/RPi/filesystem/usr/share/alsa/cards/vc4-hdmi.conf), but so far without much success.

6by9, 2021-03-06:
Vc4hdmi is disabled by adding audio=off to the vc4-kms-v3d line, but that doesn't mean it can fully work via the firmware at that point because it relies on the video config side. 

Analogue audio requires bcm2835-audio, which is why it is still there.

I don't immediately recall the full asound conf file for vc4hdmi, but Libreelec builds have been using it successfully for the past 5 months or so.

ykazakov, 2021-03-06:
Success! I was able to play through vc4-hdmi without modifying any config files as follows:
```
aplay -D iec958:1,0 Test.wav
```
Now just need to figure out how to change the volume as alsamixer does not see any control.

closed 2021-03-06 by ykazakov

popcornmix, 2021-03-09:
See [here](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=269769&p=1636828#p1636828)
That makes vc4-hdmi the default and provides a plug interface that allows conversion to iec958 and volume to work.

ykazakov, 2021-03-09:
@popcornmix thanks for the suggestion! That seems to be the same config as as I linked before. Unfortunately, it does not make any difference to me. The sound still plays from the headphone output. I guess, by default, alsa picks up the first hardware device. It is not possible to change the order using `raspi-config` since it does not detect vc4-hdmi. When I disable it in `/boot/config.txt`, nothing plays at all:
```
aplay: set_params:1343: Sample format non available
Available formats:
- IEC958_SUBFRAME_LE
```
In alsamixer I also do not see any difference. I still see `vc4-hdmi` without any volume controls. I see no new "alsa device" anywhere.

I solved the problem by replacing the content of `~/.asoundrc` with:
```
pcm.!default "iec958:1,0"
```
With this, I can play through hdmi and control volume in the application such as `gst-play-1.0` but not in alsamixer.

popcornmix, 2021-03-10:
You are probably seeing an artifact of moving to pulseaudio.
Check the updates here:
https://www.raspberrypi.org/blog/new-raspberry-pi-os-release-december-2020/

which include deleting the old `~/.asoundrc` which will stop `/usr/share/alsa/cards/vc4-hdmi.conf` from working.

ykazakov, 2021-03-10:
@popcornmix thanks for the hint! However, I am not using pulseaudio or a desktop. Just a "naked" raspberry os lite. (I install most of the stuff inside containers to keep the base os "clean"). Could it be that the desktop version has additional alsa configs that are missing in the lite version?
Also, if the original `vc-hdmi.conf` is broken, shouldn't it be reported upstream? ;-)

popcornmix, 2021-03-10:
Have you tried with  ~/.asoundrc deleted and `/usr/share/alsa/cards/vc4-hdmi.conf` as described in thread?

ykazakov, 2021-03-10:
@popcornmix success! Removing `~/. asoundrc` worked! I can now control the volume of in alsamixer! However, how do I now set `vc4-hdmi` to be the default playback device in alsa? I think I need an appropriate `~/.asoundrc` config for that.

ykazakov, 2021-03-10:
Sure, if I remove the headphone output then hdmi output will be used by default. I would, however, prefer to keep both and switch between the outputs in the software. Normally it enough to specify the default device in `~/.asoundrc` like I mentioned a few comments above, but I am not sure how to do that with the new config.

popcornmix, 2021-03-10:
@HiassofT do you know the right answer here?

HiassofT, 2021-03-10:
you can still use ~/.asoundrc to set the default device - just use set it to `default:CARD=vc4hdmi0`

ykazakov, 2021-03-11:
@HiassofT unfortunately this still does not work for me. The headphones are still used when playing:
```
$ cat ~/.asoundrc
default:CARD=vc4hdmi0
$ aplay /usr/share/sounds/alsa/Front_Center.wav -v
Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
Plug PCM: Hardware PCM card 0 'bcm2835 Headphones' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 1
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 24000
  period_size  : 6000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 6000
  period_event : 0
  start_threshold  : 24000
  stop_threshold   : 24000
  silence_threshold: 0
  silence_size : 0
  boundary     : 1572864000
  appl_ptr     : 0
  hw_ptr       : 0
```

Are you sure the syntax  `default:CARD=` is correct? I haven't seen that one can set the cards like that.

HiassofT, 2021-03-11:
@ykazakov sorry, that was a copy-and-paste error, it should have been `sysdefault:CARD=vc4hdmi0` - and you have to use that in the pcm.!default definition, eg like this:
```
pcm.!default "sysdefault:CARD=vc4hdmi0"
```

If want the default PCM to show up in `aplay -L` output (which can be handy when debugging things), you have to add a hint with description (note: you could also use `type plug` instead of `type empty` but that's redundant as vc4hdmi (sys)default already has plug in it's chain). Directing default ctl to vc4hdmi is also handy so you can just use "alsamixer" and don't have to switch cards.

```
pcm.!default {
  type emtpy
  slave.pcm "sysdefault:CARD=vc4hdmi0"
  hint {
    show on
    description "Default changed to vc4hdmi"
  }
}

ctl.!default {
  type hw
  card vc4hdmi0
}
```

Note: in general it's best to use explicit card names (eg `CARD=vc4hdmi0`) instead of the numeric index as these are stable - index numbers might change

ykazakov, 2021-03-11:
@HiassofT many thanks! That finally worked if I replace `vc4hdmi0` with `vc4hdmi` and `type empty` with `type plug`.
With `type empty` I get an error:
```
$ aplay /usr/share/sounds/alsa/Front_Center.wav -v
ALSA lib dlmisc.c:287:(snd1_dlobj_cache_get) Cannot open shared library /usr/lib/arm-linux-gnueabihf/alsa-lib/libasound_module_pcm_emtpy.so ((null): /usr/lib/arm-linux-gnueabihf/alsa-lib/libasound_module_pcm_emtpy.so: cannot open shared object file: No such file or directory)
aplay: main:828: audio open error: No such device or address
```

So a working config for `~/.asoundrc` is as follows:

```
pcm.!default {
  type plug
  slave.pcm "sysdefault:CARD=vc4hdmi"
  hint {
    show on
    description "Default changed to vc4hdmi"
  }
}

ctl.!default {
  type hw
  card vc4hdmi
}
```

HiassofT, 2021-03-11:
Sorry, this is not my day :(

I always seem to forget that on RPi0-3 the card name is `vc4hdmi` (not vc4hdmi0/1) and of course a typo sneaked into the asoundrc I posted (em**tp**y should have read em**pt**y)

Glad you got it working despite my errors!