raspberrypi/linux #4195: Micro SD card

Dell-Glitch, 2021-03-09:
when I use my rpi4 sometimes /dev/mmcblk0p1 and /dev/mmcblk0 dismount out of thin air and i got to restart the whole thing i can't do it by software shutdown i got to do it by cutting power to it

pelwell, 2021-03-09:
That sounds like a power supply or hardware problem. You might be able to get some logs out of it if you can `ssh pi@<ip-address> dmesg -w` to constantly capture the kernel log - after the initial boot it should be relatively quiet during normal use.

Dell-Glitch, 2021-03-09:
well it started when I got a 64gb micro sd card

Dell-Glitch, 2021-03-09:
may a bug with the huge disk capacity
it's a pny 64gb elite-X

pelwell, 2021-03-09:
> well it started when I got a 64gb micro sd card

SD cards count as hardware. 64GB is not huge, but there could be some other problem with it. Have you tried running an SD card checking tool - h2testw or similar?

Dell-Glitch, 2021-03-09:
> let me try

Buffer IO error

pelwell, 2021-03-09:
Is that from a card tester? Running on a Pi or a PC?

Dell-Glitch, 2021-03-09:
no logs

Dell-Glitch, 2021-03-09:
> Understood. If you have a Pi image on another SD card, and a USB card
> reader, and if you don't mind destroying the image on the 64GB card, you
> could install the f3 utilities:
> https://fight-flash-fraud.readthedocs.io/en/stable/introduction.html
> You just have to download and extract a zip file, then type make (and
> optionally sudo make install).

ok

pelwell, 2021-03-09:
I could also have said that you boot the other Pi image from the Pi's SD slot, install the f3 tools onto that Card, then put the 64GB card in the USB card reader and test it there - but you've probably worked that out already.

Dell-Glitch, 2021-03-09:
they're not fake and it was a 4.5 star on amazon too idk if the rating
counts

Dell-Glitch, 2021-03-09:
i used h2testw

pelwell, 2021-03-09:
You've not mentioned which power supply you are using and what else is attached to the Pi - all of which would be useful information.

pelwell, 2021-03-09:
I don't think we can rule it out completely, but at this point other causes seem more likely.

Dell-Glitch, 2021-03-09:
> Labists Rating for 5V 3 Amps

Overclocked:
CPU: 2ghz GPU: 600mhz Overvolt: 6

I think when I had mine to default settings it work fine it's prob because
of overclock

pelwell, 2021-03-09:
Ookaayyy - close when you're ready.

Dell-Glitch, 2021-03-09:
So default clock for me then

Dell-Glitch, 2021-03-09:
> And everytime my fans ramp up to full max speed but levels out after a min

Nvm you just gotta slap it to make it stop being so noisy

Dell-Glitch, 2021-03-09:
Like what the my teacher does to make the crying kid stop crying but she
doesn't slap she uses this purple rod and shoves it in they ass

Dell-Glitch, 2021-03-09:
FAILED AGAIN io error

Dell-Glitch, 2021-03-09:
im trying my 32gb

Dell-Glitch, 2021-03-09:
they are faulty

closed 2021-03-09 by Dell-Glitch