raspberrypi/linux #4196: Cannot find wlan0 device in RPi 400 when using branch rpi-4.19.y

woqidaideshi, 2021-03-09:
When using [branch rpi-4.19.y](https://github.com/raspberrypi/linux/tree/rpi-4.19.y)in RPi 400,  the wlan0 device is missing. 

This problem can be solved by applying this patch: <https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=e3062e05e1cfe378bb9b3fa0bef46711372bcf13>. So, shall we apply this patch to [rpi-4.19-y branch](https://github.com/raspberrypi/linux/tree/rpi-4.19.y)?

pelwell, 2021-03-09:
Applying this patch is straightforward enough, and it will do what you want, but it sets a precedent of supporting new hardware on defunct kernel branches. You must be building the kernel yourself because we no longer release 4.19 kernels, so can't you just apply it yourself?

woqidaideshi, 2021-03-10:
@pelwell Sure, thanks.