raspberrypi/linux #4197: TW6869 PCIe capture card DMA allocation trouble

walmis, 2021-03-09:
I'm trying to get TW6869 TV capture card running, but I'm seem to be stuck at a DMA allocation issue. I think this is related to DMA coherent allocations. I've increased coherent_pool=32M, but that did not help. Increasing CMA also does not help. Kernel config attached. I'd be grateful for any pointers on how to debug this.

```
[    1.550499] brcm-pcie 47d500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.550534] brcm-pcie 47d500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.550608] brcm-pcie 47d500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.550688] brcm-pcie 47d500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0400000000
[    1.584131] brcm-pcie 47d500000.pcie: link up, 2.5 GT/s PCIe x1 (SSC)
[    1.584319] brcm-pcie 47d500000.pcie: PCI host bridge to bus 0000:00
[    1.584342] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.584365] pci_bus 0000:00: root bus resource [mem 0x600000000-0x63fffffff] (bus address [0xc0000000-0xffffffff])
[    1.584449] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    1.584683] pci 0000:00:00.0: PME# supported from D0 D3hot
[    1.588387] pci 0000:00:00.0: bridge configuration invalid ([bus ff-ff]), reconfiguring
[    1.588634] pci 0000:01:00.0: TW686x PCI class overridden (0x000000 -> 0x048001)
[    1.588656] pci 0000:01:00.0: [1797:6869] type 00 class 0x048001
[    1.588713] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x00000fff pref]
[    1.588847] pci 0000:01:00.0: enabling Extended Tags
[    1.589049] pci 0000:01:00.0: PME# supported from D0 D3hot D3cold
[    1.601400] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.601444] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6000fffff]
[    1.601468] pci 0000:01:00.0: BAR 0: assigned [mem 0x600000000-0x600000fff pref]
[    1.601495] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.601520] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6000fffff]
```
modprobe tw686x
```
[   18.516474] ------------[ cut here ]------------
[   18.521163] Failed to get suitable pool for 0000:01:00.0
[   18.526585] WARNING: CPU: 2 PID: 490 at kernel/dma/pool.c:282 dma_alloc_from_pool+0x98/0x1c0
[   18.535139] Modules linked in: tw686x(+) videobuf2_dma_sg hci_uart btbcm bluetooth ecdh_generic ecc ipv6 vcan brcmfmac brcmutil vc4 cec v3d drm_kms_helper gpu_sched drm cfg80211 drm_panel_orientation_quirks raspberrypi_hwmon bcm2835_isp(C) snd_soc_core bcm2835_codec(C) rfkill bcm2835_v4l2(C) snd_compress snd_pcm_dmaengine snd_pcm i2c_brcmstb bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common snd_timer vc_sm_cma(C) videodev mc snd syscopyarea sysfillrect sysimgblt fb_sys_fops rpivid_mem backlight uio_pdrv_genirq uio
[   18.587864] CPU: 2 PID: 490 Comm: modprobe Tainted: G        WC        5.10.17-v8 #6
[   18.595713] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[   18.602241] pstate: 00000005 (nzcv daif -PAN -UAO -TCO BTYPE=--)
[   18.608329] pc : dma_alloc_from_pool+0x98/0x1c0
[   18.612917] lr : dma_alloc_from_pool+0x98/0x1c0
[   18.617504] sp : ffffffc01e8eb5e0
[   18.620857] x29: ffffffc01e8eb5e0 x28: ffffffc01e01a000 
[   18.626242] x27: ffffff8041056700 x26: ffffffc01e8eb6d0 
[   18.631626] x25: ffffff80410b30b0 x24: ffffffc0112e18f8 
[   18.637010] x23: fffffffeffe00000 x22: ffffffc0100bb880 
[   18.642393] x21: ffffffc010d148b0 x20: 0000000000001000 
[   18.647777] x19: 000000002ac01000 x18: 0000000000000010 
[   18.653160] x17: 000000000000016c x16: ffffffc015c59620 
[   18.658543] x15: ffffff804238df40 x14: ffffffffffffffff 
[   18.663926] x13: ffffffc09e8eb297 x12: ffffffc01e8eb29f 
[   18.669310] x11: 0000000000000001 x10: ffffffc011147950 
[   18.674694] x9 : ffffffc0100978e0 x8 : 0000000000000001 
[   18.680077] x7 : 0000000000017fe8 x6 : 0000000000000000 
[   18.685461] x5 : 0000000000000000 x4 : ffffff80ff79d818 
[   18.690844] x3 : ffffffc010e89000 x2 : 0000000000000001 
[   18.696228] x1 : d775422ddf87a300 x0 : 0000000000000000 
[   18.701612] Call trace:
[   18.704086]  dma_alloc_from_pool+0x98/0x1c0
[   18.708323]  dma_direct_alloc+0x278/0x330
[   18.712382]  dma_alloc_attrs+0x88/0x110
[   18.716273]  tw686x_sg_dma_alloc+0x144/0x1a0 [tw686x]
[   18.721392]  tw686x_set_format+0x258/0x2e0 [tw686x]
[   18.726335]  tw686x_video_init+0x134/0x640 [tw686x]
[   18.731278]  tw686x_probe+0x338/0x390 [tw686x]
[   18.735781]  pci_device_probe+0xc0/0x190
[   18.739753]  really_probe+0xec/0x3c0
[   18.743371]  driver_probe_device+0x60/0xc0
[   18.747519]  device_driver_attach+0x7c/0x90
[   18.751754]  __driver_attach+0x60/0xf0
[   18.755550]  bus_for_each_dev+0x78/0xd0
[   18.759433]  driver_attach+0x2c/0x40
[   18.763052]  bus_add_driver+0x158/0x200
[   18.766935]  driver_register+0x6c/0x130
[   18.770818]  __pci_register_driver+0x4c/0x60
[   18.775145]  tw686x_pci_driver_init+0x34/0x1000 [tw686x]
[   18.780528]  do_one_initcall+0x4c/0x2e0
[   18.784412]  do_init_module+0x60/0x260
[   18.788207]  load_module+0x20bc/0x2710
[   18.792002]  __do_sys_finit_module+0xb8/0x100
[   18.796414]  __arm64_sys_finit_module+0x28/0x40
[   18.801004]  el0_svc_common.constprop.0+0x84/0x1d0
[   18.805857]  do_el0_svc+0x2c/0xa0
[   18.809215]  el0_svc+0x20/0x30
[   18.812305]  el0_sync_handler+0xb0/0xc0
[   18.816188]  el0_sync+0x174/0x180
[   18.819543] ---[ end trace d8b5a8764c3af7c7 ]---
[   18.824254] tw686x 0000:01:00.0: dma0: unable to allocate P-buffer
[   18.830631] tw686x 0000:01:00.0: can't register video
[   18.835816] tw686x: probe of 0000:01:00.0 failed with error -12
```

trying loading module using dma_mode=contig: module loads, but when trying to launch gstreamer it fails with an allocation error.
```
[  416.499577] tw6869: PCI 0000:01:00.0, IRQ 70, MMIO 0x600000000 (contig mode)
[  416.515201] ALSA pcmC2D0c,0:tw686x PCM: cannot preallocate for size 65536
[  416.524472] ALSA pcmC2D0c,1:tw686x PCM: cannot preallocate for size 65536
[  416.531869] ALSA pcmC2D0c,2:tw686x PCM: cannot preallocate for size 65536
[  416.539066] ALSA pcmC2D0c,3:tw686x PCM: cannot preallocate for size 65536
[  431.177298] tw686x 0000:01:00.0: dma_alloc_coherent of size 692224 failed
```

cmdline
```
coherent_pool=32M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 video=HDMI-A-2:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:FE:BA:A4 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 ip=none elevator=deadline panic=3 consoleblank=0 
```
kernel config
[config.txt](https://github.com/raspberrypi/linux/files/6107959/config.txt)

6by9, 2021-03-09:
There seems to be a common assumption with some drivers that they can set up a 32bit DMA mask and guarantee memory allocations. 
https://elixir.bootlin.com/linux/latest/source/drivers/media/pci/tw686x/tw686x-core.c#L279
```
	err = pci_set_dma_mask(pci_dev, DMA_BIT_MASK(32));
```
The PCIe window on the Pi is defined as being at 0x0600000000 - 0x063fffffff, which is at 6-7GB, and therefore not addressable via a 32bit mask.

pelwell, 2021-03-09:
Note that with a recent firmware the PCIe inbound window (where system RAM appears in PCIe space) is at 0x4_00000000 (16GB).

vianpl, 2021-03-09:
I had a quick look at the driver and it seems that the physical address registers the DMA protocol uses are only 32bits*, tough luck. The current inbound window setup will not work for that driver. It maybe makes sense to create an overlay to move the inbound window into the the 32bit address space at the expense of having to bounce buffers, as I'm sure more drivers will show up with this limitation.

```
struct tw686x_sg_desc {
	/* 3 MSBits for flags, 13 LSBits for length */
	__le32 flags_length;
	__le32 phys;
};
```

walmis, 2021-03-09:
how about memcpy mode of the driver? the *sg* stuff is used when dma_mode=sg is selected when loading the module.

vianpl, 2021-03-09:
> how about memcpy mode of the driver? the _sg_ stuff is used when dma_mode=sg is selected when loading the module.

The same limitation applies :disappointed:.

I attached a modified device tree that should hopefully do the trick for you (it bypasses firmware's PCIe inbound memory config). It should replace your current bcm2711-rpi-cm4.dtb file, which you'll probably find it in the /boot/ folder (make sure to backup the current one).  [bcm2711-rpi-cm4.zip](https://github.com/raspberrypi/linux/files/6108832/bcm2711-rpi-cm4.zip)

walmis, 2021-03-09:
I tried the dtb. Not much changes

```
[    1.530627] brcm-pcie 47d500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.530662] brcm-pcie 47d500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.530736] brcm-pcie 47d500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.530817] brcm-pcie 47d500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0400000000
[    1.563743] brcm-pcie 47d500000.pcie: link up, 2.5 GT/s PCIe x1 (SSC)
[    1.563947] brcm-pcie 47d500000.pcie: PCI host bridge to bus 0000:00
[    1.563970] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.563992] pci_bus 0000:00: root bus resource [mem 0x600000000-0x63fffffff] (bus address [0xc0000000-0xffffffff])
[    1.564075] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    1.564313] pci 0000:00:00.0: PME# supported from D0 D3hot
[    1.567951] pci 0000:00:00.0: bridge configuration invalid ([bus ff-ff]), reconfiguring
[    1.568198] pci 0000:01:00.0: TW686x PCI class overridden (0x000000 -> 0x048001)
[    1.568220] pci 0000:01:00.0: [1797:6869] type 00 class 0x048001
[    1.568276] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x00000fff pref]
[    1.568409] pci 0000:01:00.0: enabling Extended Tags
[    1.568604] pci 0000:01:00.0: PME# supported from D0 D3hot D3cold
[    1.580929] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.580974] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6000fffff]
[    1.580997] pci 0000:01:00.0: BAR 0: assigned [mem 0x600000000-0x600000fff pref]
[    1.581024] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.581049] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6000fffff]
[    1.581467] pcieport 0000:00:00.0: enabling device (0000 -> 0002)
[    1.581707] pcieport 0000:00:00.0: PME: Signaling with IRQ 47
```
```
[   90.020305] tw6869: PCI 0000:01:00.0, IRQ 47, MMIO 0x600000000 (memcpy mode)
[   90.027527] tw686x 0000:01:00.0: enabling device (0000 -> 0002)
[   90.034611] ------------[ cut here ]------------
[   90.039358] Failed to get suitable pool for 0000:01:00.0
[   90.044788] WARNING: CPU: 1 PID: 505 at kernel/dma/pool.c:282 dma_alloc_from_pool+0x98/0x1c0
[   90.053342] Modules linked in: tw686x(+) videobuf2_dma_sg snd_pcm snd_timer snd hci_uart btbcm bluetooth ecdh_generic ecc ipv6 vcan brcmfmac brcmutil bcm2835_isp(C) bcm2835_v4l2(C) bcm2835_codec(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) cfg80211 videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common rfkill raspberrypi_hwmon videodev vc_sm_cma(C) mc rpivid_mem uio_pdrv_genirq uio
[   90.089998] CPU: 1 PID: 505 Comm: modprobe Tainted: G        WC        5.10.17-v8 #7
[   90.097847] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[   90.104376] pstate: 00000005 (nzcv daif -PAN -UAO -TCO BTYPE=--)
[   90.110464] pc : dma_alloc_from_pool+0x98/0x1c0
[   90.115052] lr : dma_alloc_from_pool+0x98/0x1c0
[   90.119638] sp : ffffffc017ccb5e0
[   90.122991] x29: ffffffc017ccb5e0 x28: ffffffc01776e000 
[   90.128376] x27: ffffff8041054300 x26: ffffffc017ccb6d0 
[   90.133760] x25: ffffff80410b30b8 x24: ffffffc0112f28f8 
[   90.139144] x23: fffffffeffe00000 x22: ffffffc0100bb880 
[   90.144527] x21: ffffffc010d26a30 x20: 00000000000a9000 
[   90.149910] x19: 0000000003900000 x18: 0000000000000010 
[   90.155293] x17: 0000000000000170 x16: ffffffc017ab9620 
[   90.160677] x15: ffffff804525c140 x14: ffffffffffffffff 
[   90.166061] x13: ffffffc097ccb297 x12: ffffffc017ccb2a0 
[   90.171444] x11: 00000000000000a9 x10: ffffffc011157950 
[   90.176828] x9 : ffffffc0100978e0 x8 : 0000000000000001 
[   90.182212] x7 : 0000000000017fe8 x6 : 0000000000000000 
[   90.187595] x5 : ffffffc0110e9000 x4 : ffffffc017ccb3d0 
[   90.192978] x3 : ffffffc0110e9190 x2 : 0000000000000001 
[   90.198362] x1 : 68542b5a4a094500 x0 : 0000000000000000 
[   90.203746] Call trace:
[   90.206219]  dma_alloc_from_pool+0x98/0x1c0
[   90.210456]  dma_direct_alloc+0x278/0x330
[   90.214516]  dma_alloc_attrs+0x88/0x110
[   90.218407]  tw686x_memcpy_dma_alloc+0xc8/0x1b0 [tw686x]
[   90.223791]  tw686x_set_format+0x258/0x2e0 [tw686x]
[   90.228734]  tw686x_video_init+0x134/0x640 [tw686x]
[   90.233677]  tw686x_probe+0x338/0x390 [tw686x]
[   90.238184]  pci_device_probe+0xc0/0x170
[   90.242159]  really_probe+0xec/0x3c0
[   90.245777]  driver_probe_device+0x60/0xc0
[   90.249925]  device_driver_attach+0x7c/0x90
[   90.254160]  __driver_attach+0x60/0xf0
[   90.257955]  bus_for_each_dev+0x78/0xd0
[   90.261838]  driver_attach+0x2c/0x40
[   90.265456]  bus_add_driver+0x158/0x200
[   90.269339]  driver_register+0x6c/0x130
[   90.273223]  __pci_register_driver+0x4c/0x60
[   90.277548]  tw686x_pci_driver_init+0x34/0x1000 [tw686x]
[   90.282931]  do_one_initcall+0x4c/0x2e0
[   90.286814]  do_init_module+0x60/0x260
[   90.290609]  load_module+0x20bc/0x2710
[   90.294403]  __do_sys_finit_module+0xb8/0x100
[   90.298815]  __arm64_sys_finit_module+0x28/0x40
[   90.303404]  el0_svc_common.constprop.0+0x84/0x1d0
[   90.308257]  do_el0_svc+0x2c/0xa0
[   90.311613]  el0_svc+0x20/0x30
[   90.314703]  el0_sync_handler+0xb0/0xc0
[   90.318586]  el0_sync+0x174/0x180
[   90.321940] ---[ end trace f52bbeddf4c4bfc3 ]---
[   90.326668] tw686x 0000:01:00.0: dma0: unable to allocate P-buffer
[   90.333017] tw686x 0000:01:00.0: can't register video
[   90.338193] tw686x: probe of 0000:01:00.0 failed with error -12
```
I ran the diff between dtb's, can't see any changed addressses.
Could you point out which changes you made?

BTW, I'm running
`Linux version 5.10.17-v8`

vianpl, 2021-03-09:
>[    1.530817] brcm-pcie 47d500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0400000000

Yes, my patch didn't had the expected effect. Don't know why.

> Could you point out which changes you made?

I got rid of the pcie0 DT alias, which is what firmware uses to edit into the pcie DT node. 

Let's try again, this time I've prepared an overlay. I'll be simpler to work with it. I verified it works locally on my CM4 running the latest available firmware, specific kernel versions shouldn't matter as this is purely a firmware thing:
```
/*
 * pcie-32bit-dma-overlay.dts
 */

/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2711";

	fragment@0 {
		target-path = "/aliases";
		__overlay__ {
			pcie0 = "";
		};
	};

};
```
Here's the compiled version:  [pcie-32bit-dma.zip](https://github.com/raspberrypi/linux/files/6109306/pcie-32bit-dma.zip), extract it and copy it into /boot/overlays and append the following in /boot/config.txt: `dtoverlay=pcie-32bit-dma`.

After rebooting you should see this log:
`[    1.233849] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00bfffffff -> 0x0000000000
`

walmis, 2021-03-09:
Success! tw686x got modprobe'd successfully. Thanks!

vianpl, 2021-03-09:
@pelwell My overlay is rather hacky, do you envision a cleaner way of fixing this?

vianpl, 2021-03-15:
@pelwell sorry for insisting, any comments?

pelwell, 2021-03-15:
There is currently no cleaner way of doing this. If you are asking whether such mechanism could be added then the answer is yes of course, but it would take months to roll out so doesn't really help you.

vianpl, 2021-03-24:
With the addition of the `pcie-32bit-dma` DT overlay I think we can close this. :slightly_smiling_face:

closed 2021-03-24 by pelwell