raspberrypi/linux #4202: v3d cuase garbled window with firefox-esr wayland

emojifreak, 2021-03-11:
**Describe the bug**
firefox-esr with gnome wayland session gives garbled windows as attached below. `module_blacklist=v3d` in `cmdline.txt` or selecting `Gnome on X.org` from  a display manager suppresses the symptom.

**To reproduce**

`dtoverlay=vc4-kms-v3d-pi4` or `dtoverlay=vc4-kms-v3d` or `dtoverlay=vc4-fkms-v3d` in `config.txt. Full config.txt looks like

```
disable_overscan=1
disable_fw_kms_setup=1
dtparam=audio=on,watchdog=on
hdmi_enable_4kp60=1
dtoverlay=vc4-fkms-v3d
arm_64bit=1
```

```
$ dpkg-query -W | egrep 'raspberry|firefox|gnome-shell'
chrome-gnome-shell	10.1-5
firefox-esr	78.8.0esr-1
firefox-esr-l10n-ja	78.8.0esr-1
gnome-shell	3.38.3-3
gnome-shell-common	3.38.3-3
gnome-shell-extension-prefs	3.38.3-3
gnome-shell-extensions	3.38.2-1
libraspberrypi-bin	2+git20201022~151804+e432bc3-1
libraspberrypi0	2+git20201022~151804+e432bc3-1
raspberrypi-archive-keyring	2016.10.31
raspberrypi-bootloader	1.20210303-1
raspberrypi-kernel	1.20210303-1
```

Log in to a gnome wayland session, and start firerfox-esr.

**Expected behaviour**
Show a sensible window by firefox.

**Actual behaviour**
The firerox window is garbled as attached below.

**System**
 Copy and paste the results of the raspinfo command in to this section. 

```
System Information
------------------

Raspberry Pi 4 Model B Rev 1.4
PRETTY_NAME="Debian GNU/Linux bullseye/sid"
NAME="Debian GNU/Linux"
ID=debian
HOME_URL="https://www.debian.org/"

Linux raspios-test 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux
Revision	: d03114
Serial		: 10000000488d2af3
Model		: Raspberry Pi 4 Model B Rev 1.4
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Feb 25 2021 12:10:40 
Copyright (c) 2012 Broadcom
version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0
[... 639 lines cut ...]
[   52.823136] vc4_hdmi fef05700.hdmi: ASoC: error at snd_soc_dai_startup on fef05700.hdmi: -19
[   52.829057] vc4_hdmi fef05700.hdmi: ASoC: error at snd_soc_dai_startup on fef05700.hdmi: -19
[   52.829793] vc4_hdmi fef05700.hdmi: ASoC: error at snd_soc_dai_startup on fef05700.hdmi: -19
[   52.830425] vc4_hdmi fef05700.hdmi: ASoC: error at snd_soc_dai_startup on fef05700.hdmi: -19
[   54.179435] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   60.044080] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[   84.870641] rfkill: input handler disabled
[  107.176951] rfkill: input handler enabled
[  108.458200] vc4_hdmi fef05700.hdmi: ASoC: error at snd_soc_dai_startup on fef05700.hdmi: -19
[  108.460599] vc4_hdmi fef05700.hdmi: ASoC: error at snd_soc_dai_startup on fef05700.hdmi: -19
[  108.554460] vc4_hdmi fef05700.hdmi: ASoC: error at snd_soc_dai_startup on fef05700.hdmi: -19
[  108.556372] vc4_hdmi fef05700.hdmi: ASoC: error at snd_soc_dai_startup on fef05700.hdmi: -19
[  108.562474] vc4_hdmi fef05700.hdmi: ASoC: error at snd_soc_dai_startup on fef05700.hdmi: -19
[  108.563535] vc4_hdmi fef05700.hdmi: ASoC: error at snd_soc_dai_startup on fef05700.hdmi: -19
[  108.564263] vc4_hdmi fef05700.hdmi: ASoC: error at snd_soc_dai_startup on fef05700.hdmi: -19
[  120.055632] rfkill: input handler disabled
```

**Logs**
The output of `raspinfo` includes dmesg output.

**Additional context**

All Debian packages are from Bullseye, not Buster.
[image]

pelwell, 2021-03-11:
Wayland is not supported in Raspberry Pi OS (or our kernel branches).

emojifreak, 2021-03-11:
> Wayland is not supported in Raspberry Pi OS (or our kernel branches).

Does that mean DRI/DRM is unsupported? This issue seems to uncover a bug in DRI/DRM part of v3d.ko.

pelwell, 2021-03-11:
Do you have another way to demonstrate the problem?

vanvugt, 2021-03-19:
AFAIK this is a mutter bug -> https://gitlab.gnome.org/GNOME/mutter/-/issues/1520

pelwell, 2021-03-19:
I don't think this is a bug in the Linux v3d driver.

closed 2021-03-19 by pelwell