raspberrypi/linux #4204: 24 bit audio output on Raspberry Pi 4 b model

CoppolaVH, 2021-03-11:
I am need some help solving a hi-res audio issue on the Raspberry Pi 4 b model over the HDMI. 

I am trying to stream 24 bit output over an HDMI plugged into the pi and then into my AVR, which can play 24 bit FLACs (als DSF files).  Currently the pi downsamples all 24 bit audio to 16 bit.   

How do we get the pi to output 24 bit via HDMI, please.

CoppolaVH, 2021-03-12:
Regarding 24 bit output:

Encoded at 24 bit, 192 kHz, Stereo FLAC
Decoded to 24 bit, 192 kHz, Stereo, 4.846 Mbps

DestinationLocal
Output rate 16 bit, 192 kHz, Stereo, 6.144 Mbps

I added **dtoverlay=vc4-fkms-v3d** to config.txt (below), but still the pi downsamples to 16 bit as you can see above.

disable_splash=1                                                                                                                         
disable_overscan=1                                                                                                                       
hdmi_drive=2                                                                                                                             
hdmi_blanking=0                                                                                                                          
hdmi_force_edid_audio=1                                                                                                                  
hdmi_force_hotplug=1                                                                                                                     
hdmi_group=1                                                                                                                             
hdmi_mode=4                                                                                                                              
**dtoverlay=vc4-fkms-v3d**                                                                                                                   
#config_hdmi_boost=5
dtparam=i2c_arm=on                                                                                                                       
dtparam=i2s=on                                                                                                                           
dtparam=audio=on                                                                                                                         
#dtoverlay=disable-wifi
#dtoverlay=disable-bt
enable_uart=1                                                                                                                            
hdmi_enable_4kp60=1                                                                                                                      
[all]

pelwell, 2021-03-12:
Have you tried with `dtoverlay=vc4-kms-v3d` instead? That will bypass the firmware completely and at least reduce the number of places that need to support 24-bit audio. Note that HDMI audio is not something I'm very familiar with, especially how ALSA manages any conversions from PCM to IEC958 subframes, but with more information people like @mripard, @popcornmix and @6by9 might have some suggestions.

popcornmix, 2021-03-12:
I would expect the vc4-hdmi to support 24-bit audio. I can't see anything restricting it. @HiassofT may know.
You want [this](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=269769&p=1636828#p1636828) alsa config file, and make sure you don't have an old `~/.asoundrc` file.

HiassofT, 2021-03-12:
IEC958 subframe formatting in alsa-lib will handle 24bit audio passed in via S24 and S32 formats fine.

I'm not sure if the two alsa-lib patches I sent upstream last year (which are in alsa-lib 1.2.4 / Debian Bullseye) have been backported to RPiOS.

The HBR audio patch https://git.alsa-project.org/?p=alsa-lib.git;a=commit;h=8defc5c2a65345a1aee613ebf0999d203c2e0433 isn't strictly necessary for PCM output, but the channel status bit patch https://git.alsa-project.org/?p=alsa-lib.git;a=commit;h=36e4b296d2f4ad0034f7028256c64b8af003068d is probably required so that 20/24bit audio data and sample rate are properly signalled in the channel status bits when using the default PCM - if you use the hdmi PCM you can (read: have to) set the AES bits manually.

If the patches are present (or if you use alsa-lib 1.2.4) it's best to use the vc4-hdmi.conf from here: https://github.com/LibreELEC/LibreELEC.tv/blob/master/projects/RPi/filesystem/usr/share/alsa/cards/vc4-hdmi.conf - the AES bit changes in the default pcm make sure alsa-lib will set channel status bits correctly, matching the audio stream parameters.

6by9, 2021-03-12:
Just for reference, duplicated question asked on the forums (now locked) - https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=306656&p=1835193

andrzej-gmt, 2021-03-12:
aplay reports version 1.1.8. Does it mean, that audio with vc4-kms-v3d won't work?
I have tried raspios and moode upgraded to current version of system. I have tried vc4-hdmi.cong from libreelec too. No success, but I don't know much about alsa configuration, so it could be my fault.

Other question: why bcm2835_audio is limited to 16 bits?

andrzej-gmt, 2021-03-13:
I have compiled and installed alsa-lib and alsa-utils 1.2.4 on Raspios Buster. Desktop configuration for sound shows audio out as IEC958 stereo or 5.1, but I can't get any sound form Chrome or VLC. I have tried vc4-hdmi.config from LibeElec, but then desktop shows no audio out.

popcornmix, 2021-03-17:
Have you switched to kms driver (see first post of forum thread I linked)?

pelwell, 2021-03-17:
> (see first post of forum thread I linked)?
And my reply.

andrzej-gmt, 2021-03-17:
From RPi0:

pi@moode:~ $ alsacap
*** Scanning for playback devices ***
Card 0, ID `vc4hdmi', name `vc4-hdmi'
  Device 0, ID `MAI PCM vc4-hdmi-hifi-0', name `MAI PCM vc4-hdmi-hifi-0', 1 subdevices (1 available)
    2..8 channels, sampling rate 32000..192000 Hz
    Sample formats: IEC958_SUBFRAME_LE
    Buffer size range from 16 to 65536
    Period size range from 8 to 32768

      Subdevice 0, name `subdevice #0'

pi@moode:~ $ aplay --version
aplay: version 1.1.8 by Jaroslav Kysela <perex@perex.cz>

pi@moode:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi [vc4-hdmi], device 0: MAI PCM vc4-hdmi-hifi-0 [MAI PCM vc4-hdmi-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

pi@moode:~ $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
btaplay_dmix
alsaequal
plug_alsaequal
btstream
camilladsp
crossfeed
plug_bs2b
eqfa12p
plug_eqfa12p
invpolarity
default:CARD=vc4hdmi
    vc4-hdmi, MAI PCM vc4-hdmi-hifi-0
    Default Audio Device
sysdefault:CARD=vc4hdmi
    vc4-hdmi, MAI PCM vc4-hdmi-hifi-0
    Default Audio Device
front:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM vc4-hdmi-hifi-0
    Front speakers
iec958:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM vc4-hdmi-hifi-0
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM vc4-hdmi-hifi-0
    Direct sample mixing device
dsnoop:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM vc4-hdmi-hifi-0
    Direct sample snooping device
hw:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM vc4-hdmi-hifi-0
    Direct hardware device without any conversions
plughw:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM vc4-hdmi-hifi-0
    Hardware device with all software conversions

jcucurull, 2021-09-24:
Any news on that? I have exactly the same issue with a Raspberry Pi 3b. I use the Moode player and it clearly shows me that the audio output via HDMI is downsampled to 16 bits (despite the sources I have are 24 bit FLAC files). Is there any solution for this?

CoppolaVH, 2021-09-24:
No new news from my side.

As a side note, I also bought a Justboom Digi Hat (https://shop.justboom.co/products/justboom-digi-hat?_pos=1&_sid=c3bb83bfd&_ss=r) and tested it wit both a coaxle and fibre cable to the AVR.  This resulted in 24 bit in and out at the AVR for 2 channel music with Moode audio which is promising, but it does not send 5.1 channel music, although they told me at Justboom it should.  It outputs as 2ch rather then 5.1, ugh!

So, I think we need an HDMI based solution at the RPI board as we started with, but still I have not solved it.

Let us know, anybody out there, can you help us solve this conundrum?

J.C.

jcucurull, 2021-09-24:
The problem is the one you just described, through SPDIF it is not possible to output uncompressed multichannel PCM audio, only stereo PCM audio. Thus, as a workaround the Justboom Digi Hat might help, but only for stereo audio (if we are talking about lossless audio).

CoppolaVH, 2021-09-24:
Indeed, that is fact regarding SPDIF.

I think given Moode Audio is a LINUX kernel 5.4 branch that we are stuck
with this issue, which is surprising to me because with a DAC and a USB
connection or the Digi Hat and SPDIF I think 2ch is all that is possible
with the kernel build of Moode Audio.

So, from what I have considered, the best solve for playback improvment for
me is likely upgrade to a DIRAC Live capable AVR as that should be a bigger
improvement then worrying about bit perfect pass through for 2ch or 6ch.

------------------------------
*From:* jcucurull ***@***.***>
*Sent:* Friday, September 24, 2021 11:00:16 PM
*To:* raspberrypi/linux ***@***.***>
*Cc:* CoppolaVH ***@***.***>; Author <
***@***.***>
*Subject:* Re: [raspberrypi/linux] 24 bit audio output on Raspberry Pi 4 b
model (#4204)

The problem is the one you just described, through SPDIF it is not possible
to output uncompressed multichannel PCM audio, only stereo PCM audio. Thus,
as a workaround the Justboom Digi Hat might help, but only for stereo audio
(if we are talking about lossless audio).

—
or Android

CoppolaVH, 2021-09-24:
Indeed, that is fact regarding SPDIF.

I think given Moode Audio is a LINUX kernel 5.4 branch that we are stuck with this issue, which is surprising to me because with a DAC and a USB connection or the Digi Hat and SPDIF I think 2ch is all that is possible with the kernel build of Moode Audio.

So, from what I have considered, the best solve for playback improvment for me is likely upgrade to a DIRAC Live capable AVR as that should be a bigger improvement then worrying about bit perfect pass through for 2ch or 6ch.

jcucurull, 2021-09-25:
But with newer kernel versions is this problem solved?

CoppolaVH, 2021-09-25:
From what I have read it seems in newer kernals of Linux there is a way to solve this issue, but frankly I did not go further as Moode Audio is built using 5.4, and I don't see that being updated.

Also, I am a sound enthusiast, so in my final analysis, it is more psychological gratification to see that Moode Audio showing 24 bit output, because I can not discern a difference in an a/b comparison between the 16 bit downsample 2ch audio versus the Digi Hat at bit perfect 24 bit 2ch, and this is with music I know very well for many years, in the best version of mastering I have tested for in those recordings.

So, it seemed like diminishing returns to pursue further, and this is why I gave up at the RPI out level, again,  recall I do get bit perfect 2ch with the Justboom Digi Hat.

Therefore I started to turn my attention to Dirac DSP options as implementing a good DSP such as DIRAC will yield far more audible improvement then wrestling down this bit perfection for "gratification" and only with 6ch music for me.  6ch is such a limited universe, and frankly the handful of recodings in 6ch I have sound really great at 16 bit output (I.e. crank up 6ch Metallica Black DVD or GNR Blu-ray version of Appetite for Destruction).

Let me know if you come up with any better ideas.

marrobHD, 2022-03-12:
I'm running moode 8pre.
`Linux moode 5.15.23-v7l+ #1525 SMP Wed Feb 16 14:35:14 GMT 2022 armv7l GNU/Linux`
Sadly it seems that there's still the same issue.

DragonQ, 2022-04-05:
I got this working a while back with buster (not sure if the same steps apply to bullseye). In short, you need to _remove_ `dtparam=audio=on` and `dtoverlay=vc4-fkms-v3d` from config.txt and add `dtoverlay=vc4-kms-v3d`. Then you need to replace the contents of `/usr/share/alsa/cards/vc4-hdmi.conf` with the code [here](https://forums.raspberrypi.com/viewtopic.php?f=29&t=269769&p=1636828#p1636828) and reboot.

Something further I needed to do to get channel mapping correct in mpd is mentioned at the bottom of [this post](https://forums.raspberrypi.com/viewtopic.php?p=1892612&hilit=24+bit+hdmi#p1892612).

marrobHD, 2022-04-05:
> I got this working a while back with buster (not sure if the same steps apply to bullseye). In short, you need to _remove_ `dtparam=audio=on` and `dtoverlay=vc4-fkms-v3d` from config.txt and add [...]

It's working now. Thanks!

jcucurull, 2022-05-29:
I have also tried the solution proposed by DragonQ, however without
success. In my case I have a RP3 with Moode 8.0.2.

If I update the config.txt, removing the dtparam=audio=on and
dtoverlay=vc4-fkms-v3d and adding the dtoverlay=vc4-kms-v3d, the UI shown
in the display connected via HDMI stops working and I cannot reproduce any
song. The error when I try to reproduce a song (using the web connection)
is the following: "Failed to open "ALSA Default" (alsa); Error opening ALSA
device "_audioout"; Failed to configure format 24: Invalid argument". This
error is with a 24 bit song, but with a 16 bit song it is the same but with
a 16 in the number shown in the error.

I have also updated the /usr/share/alsa/cards/vc4-hdmi.conf file, although
I have to say that the content was identical to the original one. Also, I
think the problem is not with ALSA, but with the HDMI itself because the
x.org server stops working too.

I have checked the dmesg to see if there are kernel errors, but I cannot
see any VC4 related error, just regular VC4 messages:

[    9.578370] fb0: switching to vc4 from simple
[    9.583367] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
[    9.602150] rc rc0: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0
[    9.603460] input: vc4 as
/devices/platform/soc/3f902000.hdmi/rc/rc0/input5
[    9.784458] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
[    9.787963] rc rc0: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0
[    9.788374] input: vc4 as
/devices/platform/soc/3f902000.hdmi/rc/rc0/input6
[   10.235446] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
[   10.249900] rc rc0: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0
[   10.287225] input: vc4 as
/devices/platform/soc/3f902000.hdmi/rc/rc0/input7
[   10.757793] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
[   10.775171] rc rc0: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0
[   10.775473] input: vc4 as
/devices/platform/soc/3f902000.hdmi/rc/rc0/input13
[   10.930166] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
[   10.970837] rc rc0: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0
[   10.990151] input: vc4 as
/devices/platform/soc/3f902000.hdmi/rc/rc0/input14
[   11.260694] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
[   11.270694] rc rc0: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0
[   11.290499] input: vc4 as
/devices/platform/soc/3f902000.hdmi/rc/rc0/input15
[   11.713178] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
[   11.716067] rc rc0: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0
[   11.754452] input: vc4 as
/devices/platform/soc/3f902000.hdmi/rc/rc0/input16
[   11.766108] vc4-drm soc:gpu: bound 3f902000.hdmi (ops vc4_hdmi_ops [vc4])
[   11.766746] vc4-drm soc:gpu: bound 3f004000.txp (ops vc4_txp_ops [vc4])
[   11.767167] vc4-drm soc:gpu: bound 3f206000.pixelvalve (ops vc4_crtc_ops
[vc4])
[   11.767577] vc4-drm soc:gpu: bound 3f207000.pixelvalve (ops vc4_crtc_ops
[vc4])
[   11.767992] vc4-drm soc:gpu: bound 3f807000.pixelvalve (ops vc4_crtc_ops
[vc4])
[   11.768345] vc4-drm soc:gpu: bound 3fc00000.v3d (ops vc4_v3d_ops [vc4])
[   11.903456] [drm] Initialized vc4 0.0.0 20140616 for soc:gpu on minor 0
[   12.123352] vc4-drm soc:gpu: [drm] fb0: vc4drmfb frame buffer device

From the Alsa-info script I get the following:

!!Aplay/Arecord output
!!--------------------

APLAY

**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi [vc4-hdmi], device 0: MAI PCM i2s-hifi-0 [MAI PCM
[... 128 lines cut ...]
vc_sm_cma
videobuf2_common
videobuf2_dma_contig
videobuf2_memops
videobuf2_v4l2
videobuf2_vmalloc
videodev
x_tables

!!ALSA/HDA dmesg
!!--------------

[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0
snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1
***@***.*** vc_mem.mem_base=0x3ec00000
vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1
root=PARTUUID=4c7e9f41-02 rootfstype=ext4 fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288
bytes, linear)

Anyone have any idea of why the new HDMI driver is not working in my case?

Missatge de TechHome ***@***.***> del dia dt., 5 d’abr. 2022
a les 11:51:

jcucurull, 2022-05-29:
I forgot to mention that the kernel included in Moodle 8.0.2 is 5.15.28-v7+.

Any idea of what might be happening is welcome.

Thank you!

Missatge de Jordi Cucurull ***@***.***> del dia dg., 29 de maig
2022 a les 15:40:

popcornmix, 2022-05-30:
@jcucurull it would be helpful if you could try a recent RPiOS image. 
That uses kms by default - does it give a picture on your TV? If it doesn't then we can try to reproduce.
If that does work, then we know it's something in your Moodle image and we can investigate the differences.