raspberrypi/linux #4205: bcm2711_defconfig: include CONFIG_RTC_HCTOSYS=y to automatically set system time from rtc on cm4io

kralo, 2021-03-11:
Considering f9b2a4d6a5f18e0aaf715206a056565c56889d9f

> rtc: class: support hctosys from modular RTC drivers
> 
> Due to distribution constraints it may not be possible to statically
> compile the required RTC driver into the kernel.
> 
> Expand RTC_HCTOSYS support to cover all RTC devices (statically compiled
> or not) by checking at the end of RTC device registration whether the
> time should be synced.
> 

I suggest to include

> CONFIG_RTC_HCTOSYS=y
> CONFIG_RTC_HCTOSYS_DEVICE="rtc0"

in the default kernel configuration. I have done so and compiled a kernel for CM4 (b03140 on a cm4io-board), which additionally has `dtoverlay=i2c-rtc,pcf85063a,i2c_csi_dsi` in `config.txt`

This works very well and spares of  `hwclock -r` - shenanigans.

I have verified it works well on cm4 (note my kernel is additionally built with `CONFIG_RTC_DEBUG=y` for more verbose messages)

```
# dmesg | grep -iP 'rtc|0311' 
[    0.000000] Linux version 5.10.20-v7l-0311+ (pi@raspberrypi) (gcc (Raspbian 8.3.0-6+rpi1) 8.3.0, GNU ld (GNU Binutils for Raspbian) 2.31.1) #1 SMP Thu Mar 11 12:33:33 CET 2021
[    5.442072] usb usb1: Manufacturer: Linux 5.10.20-v7l-0311+ dwc2_hsotg
[    6.635833] rtc-pcf85063 10-0051: pcf85063_probe
[    6.662836] rtc-pcf85063 10-0051: char device (252:0)
[    6.662873] rtc-pcf85063 10-0051: registered as rtc0
[    6.665400] rtc-pcf85063 10-0051: setting system clock to 2021-03-11T14:39:59 UTC (1615473599)
```

and have verified it has no adverse effects on rpi without rtc, in this case rpi4b. There the first timejump is from systemd-timesyncd (without internet connection, from saved timestamp).
```
Feb 14 11:12:02 raspberrypi systemd[1]: Received SIGRTMIN+20 from PID 157 (plymouthd).
Feb 14 11:12:02 raspberrypi systemd[1]: plymouth-read-write.service: Succeeded.
Feb 14 11:12:02 raspberrypi systemd[1]: Started Tell Plymouth To Write Out Runtime Data.
Feb 14 11:12:02 raspberrypi systemd-timesyncd[314]: System clock time unset or jumped backwards, restoring from recorded timestamp: Thu 2021-03-11 12:11:43 CET
Mär 11 12:11:43 raspberrypi systemd-udevd[152]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.

```

config diff for reference

```
$ diff config_*
3c3
< # Linux/arm 5.10.20 Kernel Configuration
---
> # Linux/arm 5.10.17 Kernel Configuration
5c5
< CONFIG_CC_VERSION_TEXT="gcc (Raspbian 8.3.0-6+rpi1) 8.3.0"
---
> CONFIG_CC_VERSION_TEXT="arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0"
7,8c7,8
< CONFIG_GCC_VERSION=80300
< CONFIG_LD_VERSION=231010000
---
> CONFIG_GCC_VERSION=80400
> CONFIG_LD_VERSION=234000000
23c23
< CONFIG_LOCALVERSION="-v7l-0311"
---
> CONFIG_LOCALVERSION="-v7l"
217d216
< CONFIG_KCMP=y
1963,1964c1962,1963
< CONFIG_NVME_CORE=y
< CONFIG_BLK_DEV_NVME=y
---
> CONFIG_NVME_CORE=m
> CONFIG_BLK_DEV_NVME=m
5924,5925c5923
< CONFIG_RTC_HCTOSYS=y
< CONFIG_RTC_HCTOSYS_DEVICE="rtc0"
---
> # CONFIG_RTC_HCTOSYS is not set
5928c5926
< CONFIG_RTC_DEBUG=y
---
> # CONFIG_RTC_DEBUG is not set
6870a6869
> # CONFIG_USB_LGM_PHY is not set
```

@pelwell

pelwell, 2021-03-11:
Thanks - https://github.com/raspberrypi/linux/commit/f9b2a4d6a5f18e0aaf715206a056565c56889d9f makes me wonder why it wasn't always implemented this way. See https://github.com/raspberrypi/linux/commit/4413b6e0a939a0713bb647571d1e537bd12294a2.

kralo, 2021-03-11:
confused about https://github.com/raspberrypi/linux/commit/4413b6e0a939a0713bb647571d1e537bd12294a2
Your Commit Message says `configs: Add CONFIG_RTS_HCTOSYS=y` but the referenced commit only removes comment lines? Is there some kind of magic? Don't you need to explicitly set it to =y ?

pelwell, 2021-03-11:
defconfig files only contain differences from the norm, which is helpful when new options are added (but does have a few pitfalls). In the case of `RTC_HCTOSYS` the default is `y`, so all we have to do is not unset it.

kralo, 2021-03-11:
Your handling of this issue has been a delight - fast and knowledgeable: Thank you very much!
Do you expect me to close this?

nullr0ute, 2021-03-15:
I only had a quick look but you likely also want to add the matching config pair to this which is the option to sync from from the system to the hardware clock (no, I've no idea why it's not implemented as part of the same config option):
CONFIG_RTC_SYSTOHC=y
CONFIG_RTC_SYSTOHC_DEVICE="rtc0"

pelwell, 2021-03-16:
Have a quick look at the .config file instead.